MichaIng/DietPi #5251: Failed to install nextcloud

p4cm4n, 2022-02-07:
#### Details:
- Date           | Tue Feb  8 00:40:10 CET 2022
- DietPi version | v8.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux pi.hole 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ncc maintenance:install --no-interaction --database mysql --database-name nextcloud --database-user tmp_root --database-pass XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX --admin-user admin --admin-pass XXXXXXXXXXXX --data-dir /mnt/dietpi_userdata/nextcloud_data`
- Exit code      | 255
- Software title | DietPi-Software
#### Steps to reproduce:

1. install nextcloud from dietpi-software
#### Expected behaviour:

- Software installed whitout having the not enough ram error
#### Actual behaviour:

- php gets with no ram and cannot install 
#### Extra details:

- ...
#### Additional logs:
```
PHP Fatal error:  Allowed memory size of 2097152 bytes exhausted (tried to allocate 438272 bytes) in /var/www/nextcloud/3rdparty/composer/autoload_real.php on line 37

MichaIng, 2022-02-08:
Many thanks for your report.

Interesting, the memory limit should have been raised to 512 MiB before that command is called.

Can you show the following command's outputs:
```sh
php -r 'print_r(ini_get_all());' | sed -n '/memory_limit/,/)/p'
cat /etc/php/7.4/cli/conf.d/98-dietpi-nextcloud.ini
```

p4cm4n, 2022-02-08:
Sure! It looks like 512M128 🤔 

`
❯ php -r 'print_r(ini_get_all());' | sed -n '/memory_limit/,/)/p'
cat /etc/php/7.4/cli/conf.d/98-dietpi-nextcloud.ini
    [memory_limit] => Array
        (
            [global_value] => 512M128
            [local_value] => 512M128
            [access] => 7
        )
; Nextcloud PHP settings
; priority=98
memory_limit=512M128
opcache.revalidate_freq=5
apc.enable_cli=1
`

MichaIng, 2022-02-08:
Ay, a dumb bug. Fixed with: https://github.com/MichaIng/DietPi/commit/70f2796

Please remove the trailing `128` from that line in `/etc/php/7.4/cli/conf.d/98-dietpi-nextcloud.ini`:
```sh
sed -i 's/128$//' /etc/php/7.4/cli/conf.d/98-dietpi-nextcloud.ini
```
I'll send a live patch as well.

MichaIng, 2022-02-08:
Okay, you can now run `dietpi-update` which should allow you to apply the live patch. Then rerun the Nextcloud installation:
```sh
dietpi-software install 114
```

p4cm4n, 2022-02-08:
It can happen to all of us :D

Thanks for your amazing work ^^

MichaIng, 2022-02-09:
Marking this as closed now. With the live patch online for >24h, there shouldn't be any more users which run into this and benefit from this issue visible 🙂.

closed 2022-02-09 by MichaIng

WaaromZoMoeilijk, 2022-02-10:
Hello, would this still be an issue on older images say (7.x) that update to 8.x and have nextcloud as install option in dietpi.txt for auto install?

Thanks for the quick fix.

Joulinar, 2022-02-10:
Usually DietPi should perform the update to 8.1 first (including the live patch). Means the fix should be included before running auto install option.

Joulinar, 2022-02-10:
there are a couple of logs files for the initial boot that might be interesting. Order of events should be done as follow:

```
/var/tmp/dietpi/logs/dietpi-firstboot.log
/var/tmp/dietpi/logs/dietpi-update.log
/var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```