MichaIng/DietPi #5256: update DietPi from v7.9.3 to v8.1.2 failed

beohna, 2022-02-09:
#### Details:
- Date           | Wed Feb  9 08:58:49 CET 2022
- DietPi version | v7.9.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux Server 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. start dietpi-update
2. ...
#### Expected behaviour:
normal update to dietpi v8.1.2
- ...
#### Actual behaviour:
updated failed
- ...
#### Extra details:
rasperberry pi 4 with nextcloud v23 and full text search, (openjdk v11 und elasticsearch)- ...
#### Additional logs:
```
Hit:1 https://artifacts.elastic.co/packages/7.x/apt stable InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Get:5 https://deb.debian.org/debian bullseye-backports InRelease [44.2 kB]
Get:6 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Reading package lists...
E: Release file for https://deb.debian.org/debian/dists/bullseye-updates/InRelease is not valid yet (invalid for another 24min 2s). Updates for this repository will not be applied.
E: Release file for https://deb.debian.org/debian/dists/bullseye-backports/InRelease is not valid yet (invalid for another 24min 2s). Updates for this repository will not be applied.
E: Release file for https://archive.raspberrypi.org/debian/dists/bullseye/InRelease is not valid yet (invalid for another 2min 37s). Updates for this repository will not be applied.
```

Joulinar, 2022-02-09:
Your system time seems to be out of sync. Pls try to update

```
/boot/dietpi/func/run_ntpd 1
```

beohna, 2022-02-09:
Thanks for the fast anwser.
Yes, now the dietpi-update works.

Now the problem is that elasticsearch isn't running anymore. ;-)

MichaIng, 2022-02-09:
Well, for this you need to check browser console, webserver, PHP and Nextcloud logs, probably in that order, depending on how exactly it does not work anymore.

Probably also it just needs to some to rebuild the search index.

Joulinar, 2022-02-09:
@MichaIng 
Should we probably add a time sync in front of a `dietpi-update`?

MichaIng, 2022-02-09:
We do, but not a forced one, i.e. when it did run successful a day ago, an hour ago or at boot (depending on chosen time sync mode), it is skipped. But indeed users reporting issues with system time off a few hours to days is way too often.

@beohna 
Which time sync mode do you use?
```sh
grep CONFIG_NTP_MODE /boot/dietpi.txt
```

MichaIng, 2022-02-09:
So it syncs forcefully on daily cron job. Can you check whether it succeeded this night?
```sh
journalctl -u systemd-timesyncd
```

beohna, 2022-02-09:
Sorry, i have no log of the  night anymore
 
`-- Journal begins at Wed 2022-02-09 13:58:30 CET, ends at Wed 2022-02-09 15:55:17 CET. --
Feb 09 13:58:31 Server systemd[1]: Starting Network Time Synchronization...
Feb 09 13:58:31 Server systemd[1]: Started Network Time Synchronization.
Feb 09 13:58:38 Server systemd[1]: Stopping Network Time Synchronization...
Feb 09 13:58:38 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 09 13:58:38 Server systemd[1]: Stopped Network Time Synchronization.
Feb 09 13:58:38 Server systemd[1]: Starting Network Time Synchronization...
Feb 09 13:58:38 Server systemd[1]: Started Network Time Synchronization.
Feb 09 14:00:57 Server systemd-timesyncd[468]: Initial synchronization to time server 85.214.83.151:123 (0.debian.pool.ntp.org).`

But I know that there was a problem tonight with  elasticsearch (memory problems) and after that there is a time problem also.

MichaIng, 2022-02-09:
I guess you did a reboot then? With this mode a time sync should be done at boot automatically, verified either by the DietPi update check (when not disabled) or run_ntpd from dietpi-postboot otherwise. But somehow there are multiple time sync daemon start/stop logs without any log about an actual synchronisation (before you ran it manually). run_ntpd should never stop systemd-timesyncd as long as no synchronisation flag was set, and it should restart it only when being forced on daily/nightly cron job 🤔.

Can you check:
```sh
systemctl is-enabled systemd-timesyncd
```
And probably have a look at
```sh
journalctl -u systemd-timesyncd
```
tomorrow to verify it synced successfully at night, seen via "Initial synchronization to time server" log message.

beohna, 2022-02-10:
Over night, the time changes , is now wrong - 1 hour left.

`root@Server:~# systemctl is-enabled systemd-timesyncd
enabled

root@Server:~# journalctl -u systemd-timesyncd
-- Journal begins at Wed 2022-02-09 16:09:49 CET, ends at Thu 2022-02-10 07:43:14 CET. --
Feb 09 16:09:50 Server systemd[1]: Starting Network Time Synchronization...
Feb 09 16:09:50 Server systemd[1]: Started Network Time Synchronization.
Feb 09 16:09:56 Server systemd[1]: Stopping Network Time Synchronization...
Feb 09 16:09:56 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 09 16:09:56 Server systemd[1]: Stopped Network Time Synchronization.
Feb 09 16:09:56 Server systemd[1]: Starting Network Time Synchronization...
Feb 09 16:09:57 Server systemd[1]: Started Network Time Synchronization.
Feb 09 16:10:09 Server systemd-timesyncd[468]: Initial synchronization to time server 176.9.1.211:123 (0.debian.pool.ntp.org).
Feb 10 01:35:31 Server systemd[1]: Stopping Network Time Synchronization...
Feb 10 01:35:31 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 01:35:31 Server systemd[1]: Stopped Network Time Synchronization.
Feb 10 01:35:31 Server systemd[1]: systemd-timesyncd.service: Consumed 53.057s CPU time.
Feb 10 01:35:31 Server systemd[1]: Starting Network Time Synchronization...
Feb 10 01:35:32 Server systemd[1]: Started Network Time Synchronization.
Feb 10 01:35:32 Server systemd-timesyncd[10024]: Initial synchronization to time server 85.220.190.246:123 (0.debian.pool.ntp.org).
Feb 10 01:35:33 Server systemd[1]: Stopping Network Time Synchronization...
Feb 10 01:35:33 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 01:35:33 Server systemd[1]: Stopped Network Time Synchronization.`

Joulinar, 2022-02-10:
Can you share content of `cat /boot/config.txt` pls

beohna, 2022-02-10:
`# Docs: https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/README.md
# Overlays: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

#-------Display---------
# Max allocated framebuffers: Set to "0" in headless mode to reduce memory usage
# - Defaults to "2" on RPi4 and "1" on earlier RPi models
#max_framebuffers=0

# If you get no picture, set the following to "1" to apply most compatible HDMI settings.
#hdmi_safe=1

# Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
# - Values from "0" to "11" are allowed, use values above "7" only if required, e.g. with very long
# - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
#config_hdmi_boost=5

# Uncomment if HDMI display is not detected and composite is being outputted.
#hdmi_force_hotplug=1

# Uncomment to disable HDMI even if plugged, e.g. to force composite output.
#hdmi_ignore_hotplug=1

# Uncomment to force a console size. By default it will be display's size minus overscan.
#framebuffer_width=1280
#framebuffer_height=720

# Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
#enable_tvout=1
# SDTV mode
#sdtv_mode=0

# Uncomment to force a specific HDMI mode (this will force VGA).
#hdmi_group=1
#hdmi_mode=1

# Uncomment to force an HDMI mode rather than DVI. This can make audio work in DMT (computer monito
#hdmi_drive=2

# Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
# With default value "0", the display shows a blank screen instead, but will not go into standby.
# NB: With "1" some applications (e.g. Kodi, OMXPlayer) cannot prevent display standby due to missi
#hdmi_blanking=1

# Set to "1" if your display has a black border of unused pixels visible.
disable_overscan=1

# Uncomment the following to adjust overscan. Use positive numbers if console goes off screen, and
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# Rotation
#display_hdmi_rotate=0
#lcd_rotate=0

#-------RPi camera module-------
#start_x=1
#disable_camera_led=1

#-------GPU memory splits-------
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16

#-------Boot splash screen------
disable_splash=1

#-------Onboard sound-----------
dtparam=audio=off

#-------I2C-------------
#dtparam=i2c_arm=off
#dtparam=i2c_arm_baudrate=100000

#-------SPI-------------
#dtparam=spi=off

#-------Serial/UART-----
# NB: Enabled for 1st run only. If you want to keep this setting, please set CONFIG_SERIAL_CONSOLE_
core_freq=500
enable_uart=1

#-------Overclock-------
temp_limit=70
initial_turbo=20

over_voltage=2
arm_freq=1800
#core_freq=500

#over_voltage_min=0
arm_freq_min=300
#core_freq_min=250
#sdram_freq_min=400
arm_64bit=1
dtoverlay=disable-wifi
dtparam=sd_poll_once`

beohna, 2022-02-10:
Elasticsearch stopped at night also

`root@Server:~# service elasticsearch status
● elasticsearch.service - Elasticsearch
     Loaded: loaded (/lib/systemd/system/elasticsearch.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Thu 2022-02-10 01:35:31 CET; 6h ago
       Docs: https://www.elastic.co
    Process: 495 ExecStart=/usr/share/elasticsearch/bin/systemd-entrypoint -p ${PID_DIR}/elasticsearch.pid --quiet (code=killed, signal=KILL)
   Main PID: 495 (code=killed, signal=KILL)
        CPU: 19min 10.931s

Feb 09 16:10:09 Server systemd[1]: Starting Elasticsearch...
Feb 09 16:11:12 Server systemd[1]: Started Elasticsearch.
Feb 10 01:35:31 Server systemd[1]: elasticsearch.service: Main process exited, code=killed, status=9/KILL
Feb 10 01:35:31 Server systemd[1]: elasticsearch.service: Failed with result 'signal'.
Feb 10 01:35:31 Server systemd[1]: elasticsearch.service: Unit process 1449 (controller) remains running after unit stopped.
Feb 10 01:35:31 Server systemd[1]: elasticsearch.service: Consumed 19min 10.920s CPU time.`

Maybe there is a correlation?

Joulinar, 2022-02-10:
You are running an overclocking profile. Old could you set it back to default?

beohna, 2022-02-10:
Yes. I use "medium ARM" - Current: 1800 MHz ARM | 500 MHz core | 2 overvolt.

Now, i set it back to default and reboot.

Joulinar, 2022-02-10:
OK pls verify time sync has been done on reboot.

`journalctl -u systemd-timesyncd`

As well, keep an eye on system time if it is still drifting.

beohna, 2022-02-10:
`-- Journal begins at Thu 2022-02-10 09:41:05 CET, ends at Thu 2022-02-10 09:50:13 CET. --
Feb 10 09:41:06 Server systemd[1]: Starting Network Time Synchronization...
Feb 10 09:41:06 Server systemd[1]: Started Network Time Synchronization.
Feb 10 09:41:16 Server systemd[1]: Stopping Network Time Synchronization...
Feb 10 09:41:16 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 09:41:16 Server systemd[1]: Stopped Network Time Synchronization.
Feb 10 09:41:16 Server systemd[1]: Starting Network Time Synchronization...
Feb 10 09:41:16 Server systemd[1]: Started Network Time Synchronization.
Feb 10 09:44:00 Server systemd-timesyncd[469]: Initial synchronization to time server 135.181.163.94:123 (0.debian.pool.ntp.org).`

Yes timesync has been done. 
I will keep an eye on system time over night

Joulinar, 2022-02-10:
@MichaIng 
not sure if I'm hunting a ghost but the number and order of events is quite strange. I guess it is not related to the initial issue but an observation 😃 . Time sync is started, stopped and started again. Once succeeded, it still seems to be `active/running` after boot. But usually it should be `inactive` after completion, isn't it?

I could replicate the behaviour on my test systems as well.

Some logs from my side. RPi4B right after boot

```
root@DietPi3:~# journalctl -u systemd-timesyncd
-- Journal begins at Thu 2022-02-10 01:46:23 CET, ends at Thu 2022-02-10 11:14:39 CET. --
Feb 10 01:46:25 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 01:46:25 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 01:46:26 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 10 01:46:26 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 01:46:26 DietPi3 systemd[1]: Stopped Network Time Synchronization.
Feb 10 01:46:26 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 01:46:26 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 11:14:14 DietPi3 systemd-timesyncd[427]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
root@DietPi3:~#
```

Service is active

```
root@DietPi3:~# systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-02-10 01:46:26 CET; 9h ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 427 (systemd-timesyn)
     Status: "Initial synchronization to time server 192.168.0.10:123 (diskstation.lan)."
      Tasks: 2 (limit: 2178)
        CPU: 196ms
     CGroup: /system.slice/systemd-timesyncd.service
             └─427 /lib/systemd/systemd-timesyncd

Feb 10 01:46:26 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 01:46:26 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 11:14:14 DietPi3 systemd-timesyncd[427]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
root@DietPi3:~#
```

forcing a time sync manually, that looks fine

```
root@DietPi3:~# /boot/dietpi/func/run_ntpd 1
[  OK  ] DietPi-TimeSync | rm /run/systemd/timesync/synchronized
[  OK  ] DietPi-TimeSync | systemctl restart systemd-timesyncd
[ INFO ] DietPi-TimeSync | Time sync completed
[  OK  ] DietPi-TimeSync | systemctl stop systemd-timesyncd
[  OK  ] DietPi-TimeSync | mkdir -p /run/systemd/timesync
[  OK  ] Network time sync | Completed
root@DietPi3:~#
```

Now the strange thing happen

```
root@DietPi3:~# journalctl -u systemd-timesyncd
-- Journal begins at Thu 2022-02-10 01:46:23 CET, ends at Thu 2022-02-10 11:16:09 CET. --
Feb 10 01:46:25 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 01:46:25 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 01:46:26 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 10 01:46:26 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 01:46:26 DietPi3 systemd[1]: Stopped Network Time Synchronization.
Feb 10 01:46:26 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 01:46:26 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 11:14:14 DietPi3 systemd-timesyncd[427]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
Feb 10 11:16:08 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 10 11:16:08 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 11:16:08 DietPi3 systemd[1]: Stopped Network Time Synchronization.
Feb 10 11:16:08 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 11:16:08 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 11:16:08 DietPi3 systemd-timesyncd[685]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
Feb 10 11:16:08 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 10 11:16:09 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 11:16:09 DietPi3 systemd[1]: Stopped Network Time Synchronization.
root@DietPi3:~#
```

- initial time sync happen at 11:14:14 during boot
- while forcing a time sync, first service is stopped at 11:16:08, yes as it was running before **still**
- 2nd step is now to start service again and do the time sync
- 3rd step is to stop service again
- and finally service is showing as `inactive`

```
root@DietPi3:~# systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Thu 2022-02-10 11:16:09 CET; 1min 40s ago
       Docs: man:systemd-timesyncd.service(8)
    Process: 685 ExecStart=/lib/systemd/systemd-timesyncd (code=exited, status=0/SUCCESS)
   Main PID: 685 (code=exited, status=0/SUCCESS)
     Status: "Idle."
        CPU: 200ms

Feb 10 11:16:08 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 11:16:08 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 11:16:08 DietPi3 systemd-timesyncd[685]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
Feb 10 11:16:08 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 10 11:16:09 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 11:16:09 DietPi3 systemd[1]: Stopped Network Time Synchronization.
root@DietPi3:~#
```

MichaIng, 2022-02-10:
Note that the order of logs can be distorted by the time sync itself. One downside of `journalctl` is that it does not list logs in order of occurrence but ordered by their timestamps. But indeed there is one service restart too much, which shouldn't be done by `run_ntpd` anymore, which instead only starts the service, a noop when the service runs already. Of course also that it is active afterwards shouldn't be. Does `htop` then also show the process running?

Here is what should happen:
1. `systemd-timesyncd` is enabled, hence it starts at relatively early boot stage.
2. On non-zero time sync modes, `dietpi-postboot` either triggers a `dietpi-update` check, or an `run_ntpd` call. The first implies the second. Both do not force a resync (like `run_ntpd 1` does), hence the service is not restarted but started only (as said, a noop as it is running already).
3. Once `systemd-timesyncd` created its flag file, or `run_ntpd` timed out waiting for it (60 seconds), for time sync modes 1-3 the service is stopped (regardless whether successful or not).

I'm just testing it on a VM here. First with DietPi update checks disabled:
```
# journalctl -u systemd-timesyncd
-- Journal begins at Thu 2022-02-10 14:33:37 CET, ends at Thu 2022-02-10 14:42:12 CET. --
Feb 10 14:33:37 VM-Bullseye systemd[1]: Starting Network Time Synchronization...
Feb 10 14:33:38 VM-Bullseye systemd[1]: Started Network Time Synchronization.
Feb 10 14:34:08 VM-Bullseye systemd-timesyncd[148]: Initial synchronization to time server [2a01:4f9:6b:46ce:1::40]:123 (2.debian.pool.ntp.org).
Feb 10 14:34:09 VM-Bullseye systemd[1]: Stopping Network Time Synchronization...
Feb 10 14:34:09 VM-Bullseye systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 14:34:09 VM-Bullseye systemd[1]: Stopped Network Time Synchronization.
```
With update checks enabled (so the time sync is triggered from within `dietpi-update`:
```
# journalctl -u systemd-timesyncd
-- Journal begins at Thu 2022-02-10 14:44:19 CET, ends at Thu 2022-02-10 14:45:14 CET. --
Feb 10 14:44:19 VM-Bullseye systemd[1]: Starting Network Time Synchronization...
Feb 10 14:44:20 VM-Bullseye systemd[1]: Started Network Time Synchronization.
Feb 10 14:44:49 VM-Bullseye systemd-timesyncd[149]: Initial synchronization to time server [2a01:4f8:c2c:477d::2]:123 (2.debian.pool.ntp.org).
Feb 10 14:44:50 VM-Bullseye systemd[1]: Stopping Network Time Synchronization...
Feb 10 14:44:50 VM-Bullseye systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 14:44:50 VM-Bullseye systemd[1]: Stopped Network Time Synchronization.
```
So it works as expected here 🤔.

In theory, when `run_ntpd` triggered during boot is still running/waiting, and you then manually run `run_ntpd 1`, it may cause this:
1. `systemd-timesyncd` starts at boot
2. `run_ntpd` starts, waiting for the flag, `systemd-timesyncd` is still active
3. `run_ntpd 1` restarts the service
4. The flag is created with some delay (due to service restart breaking sync process), which makes both `run_ntpd` processes end, the first one effectively stopping `systemd-timesyncd`.

Does this fit to your observations?

Joulinar, 2022-02-10:
@MichaIng 
Well I had it running on a test system and the status of service did not changed for 10 minutes after boot.

MichaIng, 2022-02-10:
In `/boot/dietpi/postboot`, could you change the line:
```sh
/boot/dietpi/dietpi-update 2 &> /dev/null &
```
Either removing the redirect when a screen is attached:
```sh
/boot/dietpi/dietpi-update 2
```
or forward it to a log file:
```sh
/boot/dietpi/dietpi-update 2 &> /var/log/dietpi-update_check.log
```
I assume DietPi update checks are enabled, else do the same for the `run_ntpd` line below.

Also the complete `journalctl` could be helpful to see whether there is any other service messing with this.

beohna, 2022-02-11:
Next day und the time ist wrong , more than 1 hour .

`-- Journal begins at Thu 2022-02-10 09:41:05 CET, ends at Fri 2022-02-11 08:20:03 CET. --
Feb 10 09:41:06 Server systemd[1]: Starting Network Time Synchronization...
Feb 10 09:41:06 Server systemd[1]: Started Network Time Synchronization.
Feb 10 09:41:16 Server systemd[1]: Stopping Network Time Synchronization...
Feb 10 09:41:16 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 09:41:16 Server systemd[1]: Stopped Network Time Synchronization.
Feb 10 09:41:16 Server systemd[1]: Starting Network Time Synchronization...
Feb 10 09:41:16 Server systemd[1]: Started Network Time Synchronization.
Feb 10 09:44:00 Server systemd-timesyncd[469]: Initial synchronization to time server 135.181.163.94:123 (0.debian.pool.ntp.org).
Feb 10 18:29:27 Server systemd-timesyncd[469]: Timed out waiting for reply from 135.181.163.94:123 (0.debian.pool.ntp.org).
Feb 10 18:29:41 Server systemd-timesyncd[469]: Initial synchronization to time server 213.209.109.45:123 (0.debian.pool.ntp.org).
Feb 11 02:28:29 Server systemd[1]: Stopping Network Time Synchronization...
Feb 11 02:28:29 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 11 02:28:29 Server systemd[1]: Stopped Network Time Synchronization.
Feb 11 02:28:29 Server systemd[1]: systemd-timesyncd.service: Consumed 1min 6.458s CPU time.
Feb 11 02:28:29 Server systemd[1]: Starting Network Time Synchronization...
Feb 11 02:28:29 Server systemd[1]: Started Network Time Synchronization.
Feb 11 02:28:29 Server systemd-timesyncd[71225]: Initial synchronization to time server 94.16.116.137:123 (0.debian.pool.ntp.org).
Feb 11 02:28:30 Server systemd[1]: Stopping Network Time Synchronization...
Feb 11 02:28:30 Server systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 11 02:28:30 Server systemd[1]: Stopped Network Time Synchronization.`

root@Server:~# date
Fri Feb 11 08:20:53 CET 2022

Joulinar, 2022-02-11:
I guess the issue for @beohna is still drifting system time. This should be unrelated to how the time sync is behaving on reboot.  Could you share again your `config.txt``?

@MichaIng 
to your question. Yes I have a PiHole overwrite.

```
root@DietPiProd:~# cat /etc/dnsmasq.d/42-pihole-ntp-dhcp.conf
###############################################################################
#  DHCP SERVER CONFIG FILE TO PROVIDE ADDITIONAL NTP FUNCTION.                #
#                                                                             #
###############################################################################
dhcp-option=option:ntp-server,192.168.0.10
root@DietPiProd:~#
```

But usually on DietPi I set it manually inside `dietpi-config`

MichaIng, 2022-02-11:
Ah yes right, what I was looking at for now was the unexpected timesyncd behaviour, which may require some changes our end. But indeed the time drift is a larger issue. Jep let's check overclocking, at least the CPU frequency on RPi is generally known to have potential effect on system time.

@Joulinar 
Btw, is systemd-timesyncd running also after daily cron on your case, or only after boot?

Joulinar, 2022-02-11:
@MichaIng 
default setting. `boot + daily`

```
root@DietPi3:~# journalctl -u systemd-timesyncd
-- Journal begins at Thu 2022-02-10 22:31:31 CET, ends at Fri 2022-02-11 12:43:45 CET. --
Feb 10 22:31:33 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 22:31:33 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 22:31:34 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 10 22:31:34 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 10 22:31:34 DietPi3 systemd[1]: Stopped Network Time Synchronization.
Feb 10 22:31:34 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 10 22:31:34 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 10 22:32:04 DietPi3 systemd-timesyncd[435]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
Feb 11 01:25:01 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 11 01:25:01 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 11 01:25:01 DietPi3 systemd[1]: Stopped Network Time Synchronization.
Feb 11 01:25:01 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 11 01:25:02 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 11 01:25:02 DietPi3 systemd-timesyncd[996]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
Feb 11 01:25:02 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 11 01:25:02 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 11 01:25:02 DietPi3 systemd[1]: Stopped Network Time Synchronization.
```

MichaIng, 2022-02-11:
Okay, so it is only at boot in your case. While the additional restart can be explained, not being a big issue, the question remains why the service is not stopped via `run_ntpd` after the successful sync.

When you find time, in `/boot/dietpi/postboot`, could you change the line:
```sh
/boot/dietpi/dietpi-update 2 &> /dev/null &
```
to:
```sh
/boot/dietpi/dietpi-update 2 2>&1 | logger -t 'dietpi-postboot' &
```
Can then be checked with:
```sh
journalctl -t dietpi-postboot
```
So we can compare output + timestamps with the service restarts and time sync. I'm now setting up a DHCP server as well.

Joulinar, 2022-02-11:
that's how it looks like

```
root@DietPi3:~# journalctl -t dietpi-postboot
-- Journal begins at Fri 2022-02-11 14:11:28 CET, ends at Fri 2022-02-11 14:12:22 CET. --
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]:
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]:  DietPi-Update
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: ─────────────────────────────────────────────────────
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]:  Mode: Checking for available DietPi update
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]:
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [ .... ] DietPi-Update | Checking network connectivity (1/2)
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [  OK  ] DietPi-Update | Checking network connectivity
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [ .... ] DietPi-Update | Checking DNS resolver (1/2)
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [  OK  ] DietPi-Update | Checking DNS resolver
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [  OK  ] Network time sync | Completed
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [  OK  ] DietPi-Update | Got valid latest version: 8.1.2
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [ INFO ] DietPi-Update | Current version : v8.1.2
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [ INFO ] DietPi-Update | Latest version  : v8.1.2
Feb 11 14:12:01 DietPi3 dietpi-postboot[464]: [ INFO ] DietPi-Update | Checking for new available live patches
Feb 11 14:12:02 DietPi3 dietpi-postboot[464]: [ INFO ] DietPi-Update | APT update, please wait...
Feb 11 14:12:02 DietPi3 dietpi-postboot[464]: Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Feb 11 14:12:02 DietPi3 dietpi-postboot[464]: Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Feb 11 14:12:07 DietPi3 dietpi-postboot[464]: Reading package lists...
Feb 11 14:12:07 DietPi3 dietpi-postboot[464]: [  OK  ] DietPi-Update | APT update
Feb 11 14:12:13 DietPi3 dietpi-postboot[464]: [ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
```

```
root@DietPi3:~# journalctl -u systemd-timesyncd.service
-- Journal begins at Fri 2022-02-11 14:11:28 CET, ends at Fri 2022-02-11 14:12:29 CET. --
Feb 11 14:11:30 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 11 14:11:30 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 11 14:11:31 DietPi3 systemd[1]: Stopping Network Time Synchronization...
Feb 11 14:11:31 DietPi3 systemd[1]: systemd-timesyncd.service: Succeeded.
Feb 11 14:11:31 DietPi3 systemd[1]: Stopped Network Time Synchronization.
Feb 11 14:11:31 DietPi3 systemd[1]: Starting Network Time Synchronization...
Feb 11 14:11:31 DietPi3 systemd[1]: Started Network Time Synchronization.
Feb 11 14:12:00 DietPi3 systemd-timesyncd[435]: Initial synchronization to time server 192.168.0.10:123 (diskstation.lan).
root@DietPi3:~#
```

MichaIng, 2022-02-11:
Found it. `run_ntpd` only stops the service if the flag file doesn't exist already. When time sync finished before `run_ntpd` starts at boot (common with a local NTP server at least), then it exits immediately, hence the service keeps running. Solved with: https://github.com/MichaIng/DietPi/commit/1699e77

MichaIng, 2022-02-11:
Okay, "solution" refers only to the still running systemd-timesyncd, which was not even related to the DHCP client (which triggers restarts only when the service is running). As mentioned above, to investigate the time drift:
```sh
cat /boot/config.txt
dmesg
```

MichaIng, 2022-02-25:
@beohna 
Do you still face the time drift over night?

MichaIng, 2022-02-25:
It looks like it is off for some hours at night as the time is jumping from 1:25 to 4:xx - 5:xx o'clock. Did you try already to commend `arm_freq_min` in your `dietpi.txt`? While 300 MHz is working fine on all other systems we tested or got report from, it is the only setting known (so far) to affect system time drift.

MichaIng, 2022-03-04:
Any news on the issue?

MichaIng, 2022-03-05:
Marking this as closed. Feel free to reopen if issue persists.

closed 2022-03-05 by MichaIng