MichaIng/DietPi #5258: [Fail] Update error when using chrony instead of timesyncd

twikedk, 2022-02-09:
#### Details:
- Date           | Wed Feb  9 20:22:18 CET 2022
- DietPi version | v8.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi B (armv6l) (ID=0)
- Kernel version | `Linux server 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync

#### Steps to reproduce:
run dietpi-update

#### Expected behaviour:
expecting it to run the update and finish nicely

```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```

It is fair it it can't find the service as I have replaced it with chrony and timesyncd will not coexist with chrony. I would though expect it to run smoothly anyway as this is just another time-daemon.
I need to run a local time-server for my home automation that is why I installed chrony instead.

Joulinar, 2022-02-09:
Did you disabled time sync option inside `dietpi-config`?

twikedk, 2022-02-09:
No I can say for sure I didn't.
I have now changed it to mode 0 - Custom and it works as expected.
Thanks for the quick reply.

closed 2022-02-09 by twikedk

MichaIng, 2022-02-10:
I wonder how we could make users more aware of this. I don't want to add a check for all known time sync daemons on every time sync attempt, but maybe there is another smart way via error handler or so 🤔.

I'll check whether, as long as the `systemd-timesyncd` package is installed, `systemctl start systemd-timesyncd` fails immediately as of its internally defined conditions. In this case, we could add an additional menu entry to this one error handler when starting the service, adding the hint and the option to switch to mode 0.