MichaIng/DietPi #5266: Can't boot up Pi after transferring rootfs

hellotinh03, 2022-02-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `v8.1.2`
- Distro version | probably `bullseye 0`
- Kernel version | 
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | _CY-0530 5V 3A 15W_
- SD card used | _USB Apacher AH354 16GB 3.0_ [original boot USB] and _SanDisk Ultra USB 3.0 64GB (SDCZ48-064G)_ [copied rootfs USB]

#### Steps to reproduce

1. Have DietPi running normally on the Apacher USB.
2. Plug in the SanDisk USB.
3. Use `dietpi-drive_manager`'s `Transfer rootfs` thingy to move rootfs to the SanDisk one.
4. Everything seems to operate normally (no error).
5. Screen says that I need to reboot after transferring rootfs.
6. Press OK.
7. No more VNC connection (as expected).
8. The Pi turn on with both LEDs bright, the Apacher USB flashes its LED for like 5 seconds and then completely dead (the USB normally flashes LED forever, indicating that it's being used).
9. Can't `ssh`, VNC, `ping`, or even anything.

#### Expected behaviour

- DietPi boots up normally with rootfs on the other USB for more storage.

#### Actual behaviour

- DietPi seems to not be able to boot, even with the SanDisk USB removed.

#### Extra details

- Since the Pi won't even let me `ssh` or anything, I don't know if I could provide a log file or not (the information provided on the _Required Information_ section was from an earlier screenshot and from a past issue). Both USBs still connect normally to my Mac, and they both have all the rootfs files on them.

Joulinar, 2022-02-12:
Hi,

do you have access to the `ext4` partition on the original USB Apacher AH354? Would it be possibile to share content of `/etc/fstab`. As well pls share `cmdline.txt` located on the `vfat` boot partition.

One more question out of interest. Why not moving your entire system to the SanDisk Ultra? Because keeping only 128MB `bootFS` on a 16GB Apacher AH354, is a little bit wast of space 😃

hellotinh03, 2022-02-12:
Hi, thanks for replying.

I do have access to both partition of the Apacher USB, but I'm currently at work right now, I'll provide any needed information once my shift is over.

> Why not moving your entire system to the SanDisk Ultra?
I'm actually trying to do so, but I've mistaken transferring rootfs as migrating my entire system. My bad 😬. Also, do you guys have any documentation about that procedure? It'll be really useful for me.

Thanks in advance!

Joulinar, 2022-02-12:
usually it should quite simple. Theoretically you could try to create a clone of the existing system and flash it again to your new device. On Windows I would use W32 disk imager. I guess there will be a similar tool on MacOS. 

A user on our forum created a small how-to https://dietpi.com/phpbb/viewtopic.php?t=8190

But it would be working on CLI as well. We just launched a blog post for moving an SD card to USB. But it should work similar from USB to USB https://dietpi.com/blog/?p=1236

hellotinh03, 2022-02-12:
> `/etc/fstab`

```
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------

#----------------------------------------------------------------
# TMPFS
#----------------------------------------------------------------
tmpfs /tmp tmpfs size=1922M,noatime,lazytime,nodev,nosuid,mode=1777
tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777

#----------------------------------------------------------------
# MISC: ecryptfs, vboxsf, glusterfs, bind, Btrfs subvolume mounts
#----------------------------------------------------------------

#----------------------------------------------------------------
# SWAP SPACE
#----------------------------------------------------------------

#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=9bcc366d-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=9bcc366d-01 /boot vfat noatime,lazytime,rw 0 2
UUID=26f7532b-a9d0-4427-972f-e3b49a110c90 /mnt/26f7532b-a9d0-4427-972f-e3b49a110c90 ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount
```

> `cmdline.txt`

`root=PARTUUID= rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=tty1`

Joulinar, 2022-02-12:
the following setting is empty in `cmdline.txt`

```
root=PARTUUID=
```

This should not be the case. Can you do the change on the original USB stick as follow and boot from it 

```
root=PARTUUID=9bcc366d-02
```

MichaIng, 2022-02-12:
@hellotinh03 
The `/etc/fstab` you pasted is from the SD card, while it would be interesting to see the one from the target drive, where you moved the rootfs to. I tried to find a path how it is possible that the PARTUUID is empty, but couldn't find any 🤔. It is also not possible to continue with moving the rootfs if the drive was not formatted before, and it is automatically mounted as well, and every formatted drive is assured to have a PARTUUID.

However, maybe I overlooked something. With the step by @Joulinar you will be able to boot with rootfs on SD card. Please mount the external drive, or mount it on an external system (with ext4 support) and paste the content of `/etc/fstab` from that drive. Also interesting would be what `findmnt -Ufnro PARTUUID -M /mnt/external` (with `/mnt/external` being the mount point of the drive) returns.

Joulinar, 2022-02-12:
just to avoid a misunderstanding. It's 2 USB sticks, while the original one was Apacher USB one.  U guess @hellotinh03 is trying to migrate from a 16GB USB stick to the 64GB SanDisk Ultra.

hellotinh03, 2022-02-13:
Thanks for the helps! I've successfully migrated my system to the SanDisk Ultra.
Another day, another great work from you guys! 👏

closed 2022-02-13 by hellotinh03

Joulinar, 2022-02-13:
Just out of interest. How did you migrate to the new USB stick finally?

hellotinh03, 2022-02-13:
I used BalenaEtcher clone option and dietpi-drive_manager to resize the SanDisk to its maximum capacity.

StephanStS, 2022-02-13:
> I used BalenaEtcher clone option and dietpi-drive_manager to resize the SanDisk to its maximum capacity.

So you had the luck, that the target had a higher capacity than the original. This simplifies it a bit.

MichaIng, 2022-02-13:
> I used BalenaEtcher clone option

Perfect 🙂.