MichaIng/DietPi #5267: When updating, get error: cannot remove microsoft.gpg; operation not permitted

linxiaopi, 2022-02-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
buster 1

- Kernel version | `uname -a`
5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 2 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 1A
- SD card used | (EG: SanDisk ultra)
SanDisk Ultra 8GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...run command: dietpi-update and allow the update to run
2. ...update runs then exits back to update window with the error: cannot remove microsoft.gpg: operation not permitted
3. try to manually rm microsoft.gpg by logging in as root, but get same error

#### Expected behaviour

- ...update should run smoothly with no errors

#### Actual behaviour

- ...get error: cannot remove microsoft.gpg: operation not permitted

#### Extra details

- ...I had tried to remove the microsoft repository in the past, but I don't remember the commands nor if it was successful

linxiaopi, 2022-02-12:
Sure. Here it is:
`drwxr-xr-x 2 root root 4096 Mar  5  2021 .
drwxr-xr-x 7 root root 4096 Feb  9 17:00 ..
-rw-r--r-- 1 root root    0 Mar  5  2021 microsoft.gpg`

linxiaopi, 2022-02-12:
Sorry, that didn't format correctly. Here it is again:
```
drwxr-xr-x 2 root root 4096 Mar  5  2021 .
drwxr-xr-x 7 root root 4096 Feb  9 17:00 ..
-rw-r--r-- 1 root root    0 Mar  5  2021 microsoft.gpg
```

Joulinar, 2022-02-12:
no need to post the information twice. You could simply use the edit function on GitHub. I will go to remove the first post 😃 

`microsoft.gpg` seems to be an empty file. Could you try following

`rm /etc/apt/trusted.gpg.d/microsoft.gpg`

MichaIng, 2022-02-12:
Added a pre-patch: https://github.com/MichaIng/DietPi/commit/9e39014

linxiaopi, 2022-02-12:
> Looks like you followed an overly sensitive guide stating to make that `microsoft.gpg` immutable to prevent RPi packages from re-adding the MS VSCode repository 😄. It was always the best and easiest to simply remove it and have a little trust in the RPi Foundation that they didn't meant this sneaky or harmfully but simply wanted to make it easier for their users to install VSCode. Us as well as the related `raspberrypi-sys-mods` package both attempt to remove the repo and key, and the immutable flag prevents that.
> 
> ```shell
> chattr -i /etc/apt/trusted.gpg.d/microsoft.gpg
> dietpi-update 1
> ```
> 
> Probably I should update our pre-patch to remove the flag automatically, when present.

That did the job. Thanks!

MichaIng, 2022-02-16:
I'll mark this as closed then. Feel free to reopen if required.

closed 2022-02-16 by MichaIng