MichaIng/DietPi #5268: Kernel | Manually cross-compiled RPI v5.15 results in no boot?

xRiot, 2022-02-12:
#### Required Information
- DietPi version | `8.1.2`
- Distro version | `Bullseye`
- Kernel version | `Cross-compiled 5.15.21-v8+`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | `15.3W 5.1V 3A Raspberry Pi official power supply`
- SD card used | `SanDisk Extreme - 64GB - 160MB/s`

#### Additional Information (if applicable)
- Software title | https://github.com/raspberrypi/linux/tree/rpi-5.15.y
- Was the software title installed freshly or updated/migrated? `Cross-compiled from a x86_64 Debian Bullseye computer`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

#### Steps to reproduce
Build the kernel according to official documentation from RPI for **cross-compiling 64bit ARM RPI4 kernel**: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel
```shell
# Mount a existing DietPi installation from a SD card (see lsblk) onto a separate Debian Bullseye system
mkdir -p mnt/fat32 mnt/ext4
sudo mount /dev/sda1 mnt/fat32
sudo mount /dev/sda2 mnt/ext4

# Compile official 5.15 RPI kernel
sudo apt install git bc bison flex libssl-dev make libc6-dev libncurses5-dev crossbuild-essential-arm64 kmod
git clone --depth=1 --branch rpi-5.15.y https://github.com/raspberrypi/linux
cd linux
make -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
make -j$(nproc) ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs

# Install onto SD card
sudo env PATH=$PATH make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- INSTALL_MOD_PATH=mnt/ext4 modules_install
sudo cp arch/arm64/boot/Image mnt/fat32/rpi.img
sudo cp arch/arm64/boot/dts/broadcom/*.dtb mnt/fat32/
sudo cp arch/arm64/boot/dts/overlays/*.dtb* mnt/fat32/overlays/
sudo cp arch/arm64/boot/dts/overlays/README mnt/fat32/overlays/

# Update config.txt to use manually compiled kernel
echo "kernel=rpi.img" >> mnt/fat32/config.txt

# Finalize
sudo umount mnt/fat32
sudo umount mnt/ext4
sudo eject sda
```

#### Expected behaviour
The system should boot successfully.

#### Actual behaviour
- System doesn't boot (the green status lamp blinks a few times and then stops while the red lamp remains static on the RPI. There is no picture from the HDMI port even though the signal is detected. The device isn't connected to the network either but the ethernet port's lights are lit).
- No journalctl/dmesg log entries with full logging configured in `dietpi-software`.

#### Extra details
**However, when using the v5.15 pre-compiled kernel from https://github.com/raspberrypi/firmware/tree/master/boot (`kernel8.img`) it works successfully?**
```bash
$ uname -r
5.15.21-v8+
```

Am I doing something wrong or is anything missing in the official RPI kernel compilation steps?

MichaIng, 2022-02-12:
Since we ship with the official RPi kernel packages, I never tried to compile one for RPi myself. I suggest to ask this on the [RPi forum](https://forums.raspberrypi.com/) or on the [RPi kernel GitHub repo](https://github.com/raspberrypi/linux/issues).

Generally the steps look fine and complete. Easiest to debug such early boot issues is to attach a serial adapter and check via UART/serial console.

First idea I have is that the new bootloader is required as well as of the minor version switch from v5.10 to v5.15, but when the pre-compiled 5.15 kernel works, obviously this is not the case.

About the size, I'm not sure whether the 64-bit kernel is compressed automatically, though since the 32-bit kernel definitely is "**z**Image", the 64-bit kernel without different build flags should as well. That it is called "Image" without the "z", is common for modern Linux images. Probably the official kernel build doesn't build into the kernel as much modules, though here I also suspect that it uses the same [bcm2711_defconfig](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm64/configs/bcm2711_defconfig) you used 🤔.

xRiot, 2022-02-12:
Thank you @MichaIng for your insights :+1: 

I forgot to mention that I'm actually looking into compiling the Xanmod v5.15 Real Time kernel once I can successfully run the cross-compiled official kernel.

I hope we can figure out what's going wrong here since I haven't found any other report for this particular issue.

---

**Update 1**
To keep you guys in loop. I just tried using the latest `/boot` files in my DietPi installation with a cross-compiled v5.15 kernel:
```shell
git clone --depth=1 https://github.com/raspberrypi/firmware && cd firmware
find mnt/fat32/* ! -name 'dietpi' ! -name '*.txt' -print -delete
cp -rvn ./boot/* mnt/fat32
```
But unfortunately it doesn't boot. The pre-compiled v5.15 still works though..

---

**Update 2**
I just tried the Raspberry Pi OS (64-bit) Lite edition (2022-01-28-raspios-bullseye-arm64-lite.img) which came with the v5.10.92-v8+ kernel. Then I cross-compiled the official RPI v5.15 kernel and it works!?

This is really confusing :thinking: 
| OS                            | Version       | Bootloader              | Kernel                       | Bootable? |
| ----------------------------- | ------------- | ----------------------- | ---------------------------- | --------- | 
| Raspberry Pi OS (64-bit) Lite | 2022-01-28    | 2022-01-25 (1643121041) | 5.10.92-v8+ (stock)          | Yes       |
| Raspberry Pi OS (64-bit) Lite | 2022-01-28    | 2022-01-25 (1643121041) | 5.15.21-v8+ (pre-compiled)   | Yes       |
| Raspberry Pi OS (64-bit) Lite | 2022-01-28    | 2022-01-25 (1643121041) | 5.15.21-v8+ (cross-compiled) | Yes       |
| DietPi RPi ARMv8 64-bit       | 8.1.2         | 2022-01-25 (1643121041) | 5.10.92-v8+ (stock)          | Yes       |
| DietPi RPi ARMv8 64-bit       | 8.1.2         | 2022-01-25 (1643121041) | 5.15.21-v8+ (pre-compiled)   | Yes       |
| DietPi RPi ARMv8 64-bit       | 8.1.2         | 2022-01-25 (1643121041) | 5.15.21-v8+ (cross-compiled) | **No**    |

Just to recap the status: I tried to remove `quiet` option from `cmdline.txt` but I still get no display. Also no journalctl/dmesg logs and unfortunately I don't have a serial adapter at hand.

I'm confused as to how the pre-compiled v5.15 kernel boots in DietPi while my cross-compiled official v5.15 RPI kernel doesn't :cry:

MichaIng, 2022-02-12:
I'm confused how any kernel which boots on RPi OS does not boot on DietPi, since hardware, bootloader, firmware are all the same. However, we found some cases where (generally officially supported) config.txt entries trigger bugs in kernel, firmware or so, e.g. `initial_turbo` breaking CPU frequency scaling on RPi 3B+ with current stable kernel, while it worked with one version before and works with Linux 5.15: https://github.com/raspberrypi/linux/issues/4875

Hence please go through the config.txt settings, starting with `initial_turbo` (which e.g. causes higher power consumption at boot). Here the one from RPi OS 64-bit for comparison: https://github.com/RPi-Distro/pi-gen/blob/arm64/stage1/00-boot-files/files/config.txt

xRiot, 2022-02-13:
### SOLVED
Thank you @MichaIng. You were right, the issue originated from `config.txt`.
I compared and tested the **enabled** options in [DietPi](https://github.com/MichaIng/DietPi/blob/master/config.txt) versus the ones shipped with [Raspberry Pi OS](https://github.com/RPi-Distro/pi-gen/blob/arm64/stage1/00-boot-files/files/config.txt):

| Option                   | In Raspberry Pi OS | In DietPi | Required to boot? |
| ------------------------ | ------------------ | --------- | ----------------- |
| `hdmi_blanking=1`        | No                 | Yes       | No                |
| `gpu_mem_256=16`         | No                 | Yes       | No                |
| `gpu_mem_512=16`         | No                 | Yes       | No                |
| `gpu_mem_1024=16`        | No                 | Yes       | No                |
| `disable_splash=1`       | No                 | Yes       | No                |
| `enable_uart=0`          | No                 | Yes       | No                |
| `dtparam=sd_poll_once`   | No                 | Yes       | No                |
| `temp_limit=75`          | No                 | Yes       | No                |
| `initial_turbo=20`       | No                 | Yes       | No                |
| `dtoverlay=disable-wifi` | No                 | Yes       | No                |
| `dtoverlay=vc4-kms-v3d`  | Yes                | No        | No                |
| `dtparam=audio=off`      | No                 | Yes       | No                |
| `dtparam=audio=on`       | Yes                | No        | No                |
| `disable_overscan=1`     | Yes                | Yes       | No                |
| `camera_auto_detect=1`   | Yes                | No        | No                |
| `display_auto_detect=1`  | Yes                | No        | No                |
| `max_framebuffers=2`     | Yes                | No        | No                |
| `arm_64bit=1`            | Yes                | No        | **Yes**           |
| `[cm4]`<br>`otg_mode=1`  | Yes                | No        | No                |
| `[pi4]`<br>`arm_boost=1` | Yes                | No        | No                |

In summary, I had to add `arm_64bit=1` to my `/boot/config.txt` to boot the cross-compiled v5.15 kernel on DietPi

xRiot, 2022-02-13:
For future reference, I've combined the DietPi and Raspberry Pi OS configs and this is what I'm running now:

<details>
  <summary>config.txt</summary>

   ```ini
   # Docs: https://www.raspberrypi.com/documentation/computers/config_txt.html
   # Overlays: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README
   
   #-------Kernel----------
   kernel=rpi.img
   arm_64bit=1
   
   #-------Display---------
   # Enable DRM VC4 V3D driver
   #dtoverlay=vc4-kms-v3d
   
   # Max allocated framebuffers: Set to "0" in headless mode to reduce memory usage
   # - Defaults to "2" on RPi4 and "1" on earlier RPi models
   max_framebuffers=2
   
   # If you get no picture, set the following to "1" to apply most compatible HDMI settings.
   #hdmi_safe=1
   
   # Uncomment to adjust the HDMI signal strength if you have interferences, blanking, or no display.
   # - Ranges from "0" to "11", use values above "7" only if required, e.g. with very long HDMI cable.
   # - Default on first RPi1 A/B is "2", else "5", on RPi4 this setting is ignored.
   #config_hdmi_boost=5
   
   # Uncomment if HDMI display is not detected and composite is being outputted.
   #hdmi_force_hotplug=1
   
   # Uncomment to disable HDMI even if plugged, e.g. to force composite output.
   #hdmi_ignore_hotplug=1
   
   # Uncomment to force a console size. By default it will be display's size minus overscan.
   #framebuffer_width=1280
   #framebuffer_height=720
   
   # Uncomment to enable SDTV/composite output on RPi4. This has no effect on previous RPi models.
   #enable_tvout=1
   # SDTV mode
   #sdtv_mode=0
   
   # Uncomment to force a specific HDMI mode (this will force VGA).
   #hdmi_group=1
   #hdmi_mode=1
   
   # Uncomment to force an HDMI mode rather than DVI. This enables HDMI audio in DMT modes.
   #hdmi_drive=2
   
   # Set "hdmi_blanking=1" to allow the display going into standby after 10 minutes without input.
   # With default value "0", the display shows a blank screen instead, but will not go into standby.
   # NB: Some legacy OpenMAX applications (OMXPlayer) cannot wake screens from real standby.
   hdmi_blanking=0
   
   # Set to "1" if your display has a black border of unused pixels visible.
   disable_overscan=1
   
   # Uncomment the following to adjust overscan.
   # Use positive numbers if console goes off screen, and negative if there is too much border.
   #overscan_left=16
   #overscan_right=16
   #overscan_top=16
   #overscan_bottom=16
   
   # Rotation
   #display_hdmi_rotate=0
   #lcd_rotate=0
   
   # Automatically load overlays for detected DSI displays
   display_auto_detect=1
   
   #-------RPi camera module-------
   #start_x=1
   disable_camera_led=1
   camera_auto_detect=0
   
   #-------GPU memory splits-------
   gpu_mem_256=16
   gpu_mem_512=16
   gpu_mem_1024=16
   
   #-------Boot splash screen------
   disable_splash=1
   
   #-------Onboard sound-----------
   dtparam=audio=off
   
   #-------I2C-------------
   #dtparam=i2c_arm=off
   #dtparam=i2c_arm_baudrate=100000
   
   #-------I2S-------------
   #dtparam=i2s=off
   
   #-------SPI-------------
   #dtparam=spi=off
   
   #-------GPIO-------------
   #dtoverlay=gpio-ir,gpio_pin=17
   #dtoverlay=gpio-ir-tx,gpio_pin=18
   
   #-------Serial/UART-----
   # NB: "enable_uart=1" will enforce "core_freq=250" on RPi models with onboard WiFi.
   enable_uart=0
   
   #-------SD card HPD-----
   # Comment to enable SD card hot-plug detection, while booting via USB or network.
   # NB: This causes constant CPU load and kernel errors when no SD card is inserted.
   dtparam=sd_poll_once
   
   #-------Overclock-------
   temp_limit=75
   initial_turbo=20
   
   over_voltage=4
   arm_freq=1900
   core_freq=600
   
   #over_voltage_min=0
   #arm_freq_min=300
   #core_freq_min=250
   #sdram_freq_min=400
   
   # Run as fast as firmware / board allows
   [pi4]
   arm_boost=1
   
   #-------USB-------------
   [cm4]
   # Enable host mode on the 2711 built-in XHCI USB controller.
   # This line should be removed if the legacy DWC2 controller is required
   # (e.g. for USB device mode) or if USB support is not required.
   otg_mode=1
   ```
  
</details>

MichaIng, 2022-02-13:
Wasn't it our 64-bit/ARMv8 image you started with (which ships this option, of course)? It doesn't make any sense, only causes trouble, when you run a 64-bit on a 32-bit userland image. Use the ARMv8/64-bit RPi image we provide or compile the 32-bit kernel.

xRiot, 2022-02-13:
> Wasn't it our 64-bit/ARMv8 image you started with

@MichaIng Yeah that's correct, I've used the ARMv8 64-bit edition for RPi (DietPi_RPi-ARMv8-Bullseye.7z).

> It doesn't make any sense, only causes trouble, when you run a 64-bit kernel on a 32-bit userland image

It feels like something is definitely missing in the official steps for compiling a 64-bit RPi kernel.. I'm only compiling a 64-bit version of the kernel with `ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-` YET it only boots if `arm_64bit=1` is explicitly specified in the `config.txt` WHILE that's not the case with the [pre-compiled v5.15](https://github.com/raspberrypi/firmware/blob/577aef2bc2f8e9dcd6955a85cf0621cc73b29db7/boot/kernel8.img) kernel (it doesn't matter if `arm_64bit=1` is used or not).

Joulinar, 2022-02-13:
why are you going to cross compile a 64bit kernel on a 64bit system? This is not required isn't it?

MichaIng, 2022-02-13:
The kernel was cross-compiled on an x86 system.

Of course any 64-bit kernel requires `arm_64bit=1`. If you use the pre-compiled kernel and do not set the `kernel=` setting, it simply boots the 32-bit kernel, with `arm_64bit=1` the 64-bit kernel. The 32-bit kernel is however not present on our 64-bit image, the userland is 64-bit only, so it wouldn't boot if your really took the ARMv8 64-bit image, and the `arm_64bit=1` would be definitely present. So this cannot be true and I'm pretty sure that you took the ARMv6 or ARMv7 image accidentally 😉.

xRiot, 2022-02-13:
> why are you going to cross compile a 64bit kernel on a 64bit system? This is not required isn't it?

I'm personally cross-compiling in Docker on a beefy workstation just to save time and keep things clean :slightly_smiling_face: 

Doing it like this if anyone is interested:
```shell
# Environment variables
RPI_MOUNT_BOOT_FS=/dev/sda1
RPI_MOUNT_ROOT_FS=/dev/sda2

# Docker container
sudo mkdir -p /mnt/rpi_boot_fs /mnt/rpi_root_fs
sudo mount $RPI_MOUNT_BOOT_FS /mnt/rpi_boot_fs && sudo mount $RPI_MOUNT_ROOT_FS /mnt/rpi_root_fs
sudo docker run -it --rm -v /mnt/rpi_boot_fs:/mnt/fat32 -v /mnt/rpi_root_fs:/mnt/ext4 debian:bullseye-slim
```

xRiot, 2022-02-13:
> Of course any 64-bit kernel requires arm_64bit=1.

I see @MichaIng. In that case it sounds like everything works as intended. RPi's official documentation could have mentioned it :facepalm:  

> The 32-bit kernel is however not present on our 64-bit image and the userland is 64-bit only, so it wouldn't boot if your really took the ARMv8 64-bit image, and the arm_64bit=1 setting would be definitely present. So this cannot be true and I'm pretty sure that you took the ARMv6 or ARMv7 image accidentally wink.

I've been struggling with this issue for a couple of days now and I've triple checked all the steps as well as which image I've downloaded (only have the ARM8 edition in my downloads). `arm_64bit=1` isn't unfortunately a part of the .img in DietPi_RPi-ARMv8-Bullseye.7z, maybe this is a deployment issue then :thinking: ?

MichaIng, 2022-02-13:
Hah, you are right, I wonder how it boots in the first place, it seems to pick `kernel8.img` and switches to 64-bit mode automatically when nothing else is present while probably with `kernel=` set it doesn't expect it to be a 64-bit kernel without `arm_64bit=1`. Checking why this is missing and patching the image.

MichaIng, 2022-02-13:
Image rebuilt, `arm_64bit=1` was added as expected, not sure how it was missing before. Probably I messed it up when fixing the turbo_mode issue in `config.txt` or so.

xRiot, 2022-02-14:
Awesome! All is now good on my end too :slightly_smiling_face:  
```shell
# uname -r
5.15.21-rt30-xanmod1-v8
```