MichaIng/DietPi #5269: Error installing Portainer, breaks installation on fresh install Odroid HC4

echtfrank, 2022-02-12:
### Creating a bug report/issue

#### Required Information
- DietPi version |
      G_DIETPI_VERSION_CORE=8
      G_DIETPI_VERSION_SUB=1
      G_DIETPI_VERSION_RC=2
      G_GITBRANCH='master'
      G_GITOWNER='MichaIng'
      G_LIVE_PATCH_STATUS[0]='applied'
      G_LIVE_PATCH_STATUS[1]='applied'
      G_LIVE_PATCH_STATUS[2]='not applicable'
- Distro version | bullseye
- Kernel version | Linux biblio 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux
- SBC model | Odroid C4/HC4 (aarch64)
- Power supply used | 15V 4A from Hardkernel
- SD card used | SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | Portainer
- Was the software title installed freshly or updated/migrated? Freshly. Error 
- Can this issue be replicated on a fresh installation of DietPi? The issue was in fact on a fresh install.
- Bug report ID | c4a9d489-3f0b-4bd1-9437-d754bb569dd2

#### Steps to reproduce

1.  Perform base-install of DietPi, using latest Bullseye image from DietPi.com for Odroid HC4
2.  Note that I have NOT removed Petitboot from SPI; this never lead to any problem.
3.  Base configuration steps (passwords, hostname, and verified that system could succesfully rebooted (via SSH)
4.  Run Dietpi-software and select a pretty good list: LEMP, Docker (Compose), Portainer, LetsEncrypt, Nextcloud, SambaServer, Jellyfin

#### Expected behaviour

- all selected software packages are succesfully installed, without errors

#### Actual behaviour

- installation of Portainer generates an error:

> docker: Error response from daemon: Conflict. The container name "/portainer" is already in use by container "3995e9c306f063f98209157c8833c27fe3603fcf56a65602c89e3fa943206c23". You have to remove (or rename) that container to be able to reuse that name.

I have choosen to "Exit", then following message appears:

> [FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

If I enter "sudo dietpi-software list", it becomes clear that none of the selected software is installed, not just Portainer is missing...

#### Extra details

I will install my software in smaller logical 'chunks'.

Joulinar, 2022-02-12:
would it be possible to share following

```
docker container ls
```

MichaIng, 2022-02-12:
> The container name "/portainer" is already in use by container

I guess you hit "Retry" from the error handler when it failed first, the retry in case of a Docker install is doomed to fail. Do you still have the full install log so that we can review the initial error message?

If it was the first boot dietpi-software instance (so you didn't manually ran it from console), the log is here:
```sh
cat /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```

> Note that I have NOT removed Petitboot from SPI; this never lead to any problem.

Whether this work well or limited or not at all depends on which bootloader is installed on the SPI. Out of interest, could you show the output of:
```sh
cat /proc/cmdline /boot/boot.ini
```
Probably the `boot.ini` is not parsed by the SPI bootloader, which breaks all attempts to configure the device that way, one of the reason to prefer using the intended bootloader shipped with the image instead of one where we have no idea what works and what doesn't 😉. Generally it can be cleared like that:
```sh
apt install mtd-utils
flash_eraseall /dev/mtd0
flash_eraseall /dev/mtd1
flash_eraseall /dev/mtd2
flash_eraseall /dev/mtd3
```

echtfrank, 2022-02-12:
@Joulinar , of I can provide you more info. I will do my best:

> CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES

@MichaIng , you are totally right about hitting the Retry-button first. Good to know.
I think I will manage to resolve the Portainer/Docker issue, but I will focus first on configuring Nextcloud to have that up and running. Almost finished I think.

**By the way, thanks al lot for all good work resulting in DietPi!**

Attached is the firstrun-setup-log that was in the folder you metioned.
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/8054550/dietpi-firstrun-setup.log)
 

I know little to nothing about bootloaders, U-boot, Petitboot, etc. I saw there is a lot to do about it. I should take some time to learn about it. I'm a bit scared to remove it, since HC4 is delivered with it and I have lack of knowledge what is does.... For  example does emptying the SPI break the abbility to Wake-On-RTC, my OLED-display or switch HC4 on by IR-remote? I could test and play, but that takes time you know ;-).

However, here is the output  of `cat /proc/cmdline /boot/boot.ini`

```
> root=/dev/mmcblk1p2 rootwait ro console=ttyS0,115200n8  no_console_suspend fsck.repair=yes net.ifnames=0 elevator=noop hdmimode=1080p60hz cvbsmode=576cvbs max_freq_a55= maxcpus=4 voutmode=hdmi  disablehpd=false cvbscable=0 overscan=100  monitor_onoff=false logo=osd0,loaded  sdrmode=auto enable_wol=0
ODROIDC4-UBOOT-CONFIG

setenv board "odroidc4"

# Default Console Device Setting
setenv condev "console=ttyS0,115200n8"   # on both

# Auto Detection of Monitor settings based on your Screen information
setenv display_autodetect "true"

# HDMI Mode
# Resolution Configuration
#    Symbol             | Resolution
# ----------------------+-------------
#    "480x272p60hz"     | 480x272 Progressive 60Hz
#    "480x320p60hz"     | 480x320 Progressive 60Hz
#    "480p60hz"         | 720x480 Progressive 60Hz
#    "576p50hz"         | 720x576 Progressive 50Hz
#    "720p60hz"         | 1280x720 Progressive 60Hz
#    "720p50hz"         | 1280x720 Progressive 50Hz
#    "1080p60hz"        | 1920x1080 Progressive 60Hz
#    "1080p50hz"        | 1920x1080 Progressive 50Hz
#    "1080p30hz"        | 1920x1080 Progressive 30Hz
#    "1080p24hz"        | 1920x1080 Progressive 24Hz
#    "1080i60hz"        | 1920x1080 Interlaced 60Hz
#    "1080i50hz"        | 1920x1080 Interlaced 50Hz
#    "2160p60hz"        | 3840x2160 Progressive 60Hz
#    "2160p50hz"        | 3840x2160 Progressive 50Hz
#    "2160p30hz"        | 3840x2160 Progressive 30Hz
#    "2160p25hz"        | 3840x2160 Progressive 25Hz
#    "2160p24hz"        | 3840x2160 Progressive 24Hz
#    "smpte24hz"        | 3840x2160 Progressive 24Hz SMPTE
#    "2160p60hz420"     | 3840x2160 Progressive 60Hz YCbCr 4:2:0
#    "2160p50hz420"     | 3840x2160 Progressive 50Hz YCbCr 4:2:0
#    "640x480p60hz"     | 640x480 Progressive 60Hz
#    "800x480p60hz"     | 800x480 Progressive 60Hz
#    "800x600p60hz"     | 800x600 Progressive 60Hz
#    "1024x600p60hz"    | 1024x600 Progressive 60Hz
#    "1024x768p60hz"    | 1024x768 Progressive 60Hz
#    "1280x800p60hz"    | 1280x800 Progressive 60Hz
#    "1280x1024p60hz"   | 1280x1024 Progressive 60Hz
#    "1360x768p60hz"    | 1360x768 Progressive 60Hz
#    "1440x900p60hz"    | 1440x900 Progressive 60Hz
#    "1600x900p60hz"    | 1600x900 Progressive 60Hz
#    "1600x1200p60hz"   | 1600x1200 Progressive 60Hz
#    "1680x1050p60hz"   | 1680x1050 Progressive 60Hz
#    "1920x1200p60hz"   | 1920x1200 Progressive 60Hz
#    "2560x1080p60hz"   | 2560x1080 Progressive 60Hz
#    "2560x1440p60hz"   | 2560x1440 Progressive 60Hz
#    "2560x1600p60hz"   | 2560x1600 Progressive 60Hz
#    "3440x1440p60hz"   | 3440x1440 Progressive 60Hz
[... 82 lines cut ...]
# Load kernel, dtb and initrd
fatload mmc ${devno}:1 ${k_addr} Image.gz
if test "x${variant}" != "x"; then
        fatload mmc ${devno}:1 ${dtb_loadaddr} meson64_odroid${variant}.dtb
else
        fatload mmc ${devno}:1 ${dtb_loadaddr} meson64_odroidc4.dtb
fi
fatload mmc ${devno}:1 ${initrd_loadaddr} uInitrd
fdt addr ${dtb_loadaddr}

if test "x{overlays}" != "x"; then
        fdt resize ${overlay_resize}
        for overlay in ${overlays}; do
                fatload mmc ${devno}:1 ${dtbo_addr_r} overlays/${board}/${overlay}.dtbo && fdt apply ${dtbo_addr_r}
        done
fi

# unzip the kernel
unzip ${k_addr} ${loadaddr}

# boot
booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}
dietpi@biblio:~$ ^C
dietpi@biblio:~$
```

MichaIng, 2022-02-12:
> Good to know.

Generally, a retry does not hurt, but in case of Docker, when a container was installed already, probably partly, it fails to install again. Sadly there is no "install or reinstall" command that we could use.

Ah Portainer install was not during first run setup, so it's not part of the log 😢. Strange that `docker container ls` does not list something, though, actually the command we use is `docker ps -a`, not `docker container ls`. Can you please show:
```sh
docker ps -a
docker images -a
```

The SPI bootloader loads `boot.ini`, that is good. To test the SD card bootloader, there is a button at the bottom of the board which you can hold, then it will skip the SPI/petitboot. But indeed there is no other reason I can currently think of to not leave it as it is, when it boots and successfully reads and applies `boot.ini`.

echtfrank, 2022-02-12:
Do you mean with the _first-run-setup_ the possibility to 'slipstream' software during OS-install? I didn't use that; I installed the software directly after booting into the freshly installed system. 

Here is the info you asked for:
```
CONTAINER ID   IMAGE                    COMMAND        CREATED       STATUS    PORTS                                                           NAMES
3995e9c306f0   portainer/portainer-ce   "/portainer"   3 hours ago   Created   8000/tcp, 9443/tcp, 0.0.0.0:9002->9000/tcp, :::9002->9000/tcp   portainer
REPOSITORY               TAG       IMAGE ID       CREATED      SIZE
portainer/portainer-ce   latest    8000bafa387b   3 days ago   267MB
```

Anything else you want to examine?

Joulinar, 2022-02-12:
ah my fault, it should have been `docker container ls -a`, as it shows running container only without the `-a` option

https://docs.docker.com/engine/reference/commandline/container_ls/

MichaIng, 2022-02-12:
> I installed the software directly after booting into the freshly installed system.

When you login to the system the very first time, dietpi-software prompts automatically, where you chose e.g. OpenSSH, but not yet Portainer. This very initial automatic dietpi-software run is what I meant. It is logged since it performs some additional setup steps, compared to a regular `dietpi-software` call from console.

> Here is the info you asked for:

Ah here the Portainer container and image are listed. That is good. When done with everything else, please try to install Portainer again. A so listed container and image are removed automatically (no data or configs, only the static container+image) by dietpi-software before the new one is installed. If it fails again, it could be great when you could paste the log.

@Joulinar 
I wonder what the difference to [docker ps -a](https://docs.docker.com/engine/reference/commandline/ps/) is. Both have exactly the same description 😄. Probably the second is a short form of the first, only that such are commonly shown together, like `-v, --verbose` like option short forms.

Joulinar, 2022-02-12:
@MichaIng 
there you go https://stackoverflow.com/questions/45254677/is-there-a-difference-between-docker-ps-and-docker-container-ls

MichaIng, 2022-02-12:
Ah, so `docker container` is new. Indeed it seems better structured and better to remember, I mean what shall "ps" stand for 😄. Means we should earlier or later switch to the new one. Only strange that it didn't give any result [above](https://github.com/MichaIng/DietPi/issues/5269#issuecomment-1037502959) 🤔.

Joulinar, 2022-02-12:
Container was not active, therefore it was not shown. I missed the `-a` to show all 🙄

MichaIng, 2022-02-12:
Ah, makes sense.

echtfrank, 2022-02-13:
OK, I didn't expect this error again, but unfortunately it did.
After a reboot and making a dietpi-backup, I installed Portainer and Docker-compose again. See attached files with CLI-output and the dietpi-software screenshot. This time I didn't go for a Retry but Exit/abborted the installation.

I can manually remove al remains of docker, etc, but maybe you would want me to look into something else first?

[CLI-logging-error_instaling_Portainer_and_Docker-Compose_20220213.log](https://github.com/MichaIng/DietPi/files/8055445/CLI-logging-error_instaling_Portainer_and_Docker-Compose_20220213.log)

Regards
[Screendump_error_instaling_Portainer_and_Docker-Compose_20220213.txt](https://github.com/MichaIng/DietPi/files/8055446/Screendump_error_instaling_Portainer_and_Docker-Compose_20220213.txt)
,

Joulinar, 2022-02-13:
Well Docker is not able to start correctly. It is related to the old kernel 4.9 missing kernel feature `BPF_CGROUP_DEVICE`

```
setting cgroup config for procHooks process caused: bpf_prog_query(BPF_CGROUP_DEVICE) failed: invalid argument: unknown.
```

We had this a couple of times with these old kernel version. As far as I can see a possible workaround could be following to get Docker working. Add `systemd.unified_cgroup_hierarchy=0` into `boot.ini`.

Joulinar, 2022-02-13:
The challenge is with the Docker version as it require this kernel feature now.

MichaIng, 2022-02-13:
Ah of course, please try to add `systemd.unified_cgroup_hierarchy=0` to the end of the line stat starts with `setenv bootargs` (separated by a space) and reboot. That should allow Docker to start:
```sh
journalctl -u docker
```

Actually I though it is present on the current image already (we add it [here](https://github.com/MichaIng/DietPi/blob/6cf4a0b/PREP_SYSTEM_FOR_DIETPI.sh#L1840-L1847)), but seems that is not the case. I'll update the image. I'll work on a new image from scratch based on Armbian mainline kernel, like in staging for Odroid N2 already. Will ping you once is there for testing. Sadly the existing image cannot be upgraded since a different partitioning with a single ext4 filesystem is required only.

MichaIng, 2022-02-13:
Image updated.

echtfrank, 2022-02-13:
I added the line to boot.ini, rebooted. The outrput of `journalctl -u docker` is:

> -- Journal begins at Sun 2022-02-13 22:43:17 CET, ends at Sun 2022-02-13 22:43:34 CET. --
Feb 13 22:43:24 biblio systemd[1]: Started Docker Application Container Engine.
Feb 13 22:43:26 biblio dockerd[2466]: time="2022-02-13T22:43:26.063212962+01:00" level=warning msg="Your kernel does not support swap memory limit"
Feb 13 22:43:26 biblio dockerd[2466]: time="2022-02-13T22:43:26.063430295+01:00" level=warning msg="Your kernel does not support CPU CFS scheduler"
Feb 13 22:43:26 biblio dockerd[2466]: time="2022-02-13T22:43:26.063465212+01:00" level=warning msg="Unable to find blkio cgroup in mounts"

Despite these warnings, the installation of Portainer was succesful. Thanks a lot!

MichaIng, 2022-02-13:
To check why this hasn't been applied with the v7.7 update already, could you run these commands:
```sh
uname -r
uname -r | mawk -F'.' '{print $1,$2}'
(( $G_DISTRO > 5 && ( $(uname -r | mawk -F'.' '{print $1}') < 4 || ( $(uname -r | mawk -F'.' '{print $1}') == 4 && $(uname -r | mawk -F'.' '{print $2}') < 15 ) ) )) && echo yes
[[ $G_HW_MODEL -gt 9 && $G_HW_MODEL -le 16 && -f '/boot/boot.ini' ]] && echo yes
```

MichaIng, 2022-02-13:
Did that: https://github.com/MichaIng/DietPi/commit/c3f6726
The parameter is not understood by systemd < 233, hence not on Buster (which ships v232) unless systemd is upgraded from backports. But it doesn't hurt either, remains just as dummy but becomes effective once a distro upgrade is done.

echtfrank, 2022-02-13:
>$ uname -r
4.9.241-arm64

>$ uname -r | mawk -F'.' '{print $1,$2}'
(( $G_DISTRO > 5 && ( $(uname -r | mawk -F'.' '{print $1}') < 4 || ( $(uname -r | mawk -F'.' '{print $1}') == 4 && $(uname -r | mawk -F'.' '{print $2}') < 15 ) ) )) && echo yes
[[ $G_HW_MODEL -gt 9 && $G_HW_MODEL -le 16 && -f '/boot/boot.ini' ]] && echo yes
4 9
yes
yes

Sorry, I made myself not clear. I MOVED from Buster to Bullseye, I haven't used a (automated) migration script.
Just if you're interested...

- My first attempt was to use the latest Debian 11 image (5.15-kernel) from Hardkernel/tobetter and use the Prep_System_for_Dietpi to convert it to DietPi (I chose odroid c4 board) --> no succes; the converted image wouldn't boot
- Second attempt was like the first, but then I chose "general board" (may-be that would prevent altering the boot-partition or so something like that --> no succes, this image wouldn't boot either
- Third attempt was to follow the steps in [HowTo Upgrade to Bullseye-blog](https://dietpi.com/blog/?p=811) on production-copy of the system (SD-card). This worked perfectly, except that I forgot to remove my HDD that contained dietpi-userdata including my Nextcloud database and I didn't make a DB-dump...

After I put my original production-SD-card in the board, the system was wrecked. The upgraded database didn't work on Buster, so I decided to take benefit from it and start from scratch, using a ready made Bullseye-image from the DietPi-site (the fourth attempt). This way I can move from Lighttpd to Nginx too!

MichaIng, 2022-02-13:
Yes, this is what I meant: `systemd.unified_cgroup_hierarchy=0` was added with DietPi v7.7, but only on Bullseye systems (on Buster it is not required). So when you did a distro upgrade following our HowTo, after being on DietPi v7.7 or later already, the then required argument is missing.

DietPi-PREP even went through when you used tobetter's image and selected Odroid C4 target? It tries to install `linux-image-arm64-odroid-c4 meveric-keyring u-boot` packages, which are only available on Meveric's repository, so this should have failed. Whether "Generic Device" works or not depends e.g. on whether the essential kernel, device tree and U-Boot packages have common package names. But another thing, tobetter's images are Ubuntu and not Debian, isn't it?

echtfrank, 2022-02-13:
I'll read your first paragaph with more attention later ;-) 

About the Prep-script, I didn't look quite well. I thought it would 'only' 

Regarding the Odroid OS images. Look at [Please use the [official ODROID Wiki page]https://wiki.odroid.com/ if you are using Hardkernel's stock OS images, Ubuntu 18.04/20.04 server or Mate version.](http://docs.linuxfactory.or.kr/#quick-start)

It appears that there is a difference between 'stock' (Ubuntu only) and 'tobetter' 's images (Ubuntu and Debian also).

MichaIng, 2022-02-13:
> and didn't install a bootloader

Indeed it does not flash the bootloader in most cases (unless the package postinst script does), but a kernel upgrade from the same repository should of course still be compatible with the existing bootloader, especially when it is not flashed by any postinst script.

While tobetter's repository seems to be generally Debian compatible (makes sense, the kernel doesn't care at all), I do not see any Debian image for download on the forum, only the three Ubuntu images. Also your link lists Debian 10 and 11 but actual download links only for Ubuntu.

echtfrank, 2022-02-13:
Just click a little bit further and you will reach http://docs.linuxfactory.or.kr/install/odroidhc4/image.html or http://docs.linuxfactory.or.kr/install/odroidn2/image.html or a few others.

These pages contains download links and the files are named _debian_bullseye-server*.img.xz_ As a novice I have to rely on the filename only.... Are you saying these so called debian images are in facrt ubuntu?

MichaIng, 2022-02-14:
Ah, thanks. I checked the image, not sure why "Generic device" wouldn't work, probably `flash-kernel` + `bootscript-odroidc4` packages are required for updating the boot configuration on kernel upgrades so that the bootloader is not looking for a non-existent kernel image.

MichaIng, 2022-02-16:
A new Armbian mainline kernel based image is ready for testing: https://github.com/MichaIng/DietPi/issues/5278#issuecomment-1039611089

This renders dealing with `systemd.unified_cgroup_hierarchy=0` obsolete, since it fully supports cgroupv2 (unified cgroup hierarchy). Sadly it is not possible to upgrade an existing legacy image since a single ext4 partition is required while the legacy one has a dedicated boot FAT partition.

echtfrank, 2022-02-17:
Thanks for letting me know, that's quick!

echtfrank, 2022-02-17:
I cannot reach  https://dietpi.com/downloads/images/testing/DietPi_OdroidC4-ARMv8-Bullseye.7z . Was it removed already?
Not that I have time to test, I was just curious.

What I was wondering: there is no kernel (name or version) in the filename of the image; the images of Hardkernel or Armbian do mention that info. Is that on purpose? Why? And how about the Dietpi-version (like v8.1)?

Joulinar, 2022-02-17:
Yes looks like the image has been removed. I guess @MichaIng tried to move it to stable? Maybe it got lost in translation. 

Basically our images do not contain a version flag inside the name. The v8 did not state it is DietPi v8. Its is `ARMv8`, means 64bit version. On first boot DietPi will update automatically to latest version (always).

MichaIng, 2022-02-17:
I moved it to stable already since the old one had this strange filesystem expansion issue that we observed on the legacy C2 image already. Before wasting time to investigate and fix this for the legacy C4 image, the new one has been moved into place: https://dietpi.com/downloads/images/

Joulinar, 2022-02-17:
https://dietpi.com/downloads/images/DietPi_OdroidC4-ARMv8-Bullseye.7z

echtfrank, 2022-02-17:
Thanks, running already ;-) (without HDD)...

echtfrank, 2022-02-17:
As aspectected - based on various posts on forums (DietPi, Odroid, Armbian), it is neccesary for the newest image to boot correctly to bypass Petitboot (keep button pressed on powe-up). This is also the case for reboots. So I'll see myself forced to get rid of Petitboot. But for now I'll keep it... for production I'll stay on the old 4.9 kernel.

So far not a single problem with docker, docker-compose and portainer in the NEW image!!! I haven't tested in relation with Jellyfin, nginx en nextcloud, but I am happy. It looks certainly promising!!!

MichaIng, 2022-02-17:
The commands for clearing the SPI was given above:
```sh
apt update
apt install mtd-utils
flash_eraseall /dev/mtd0
flash_eraseall /dev/mtd1
flash_eraseall /dev/mtd2
flash_eraseall /dev/mtd3
```

echtfrank, 2022-02-22:
When migrating to the new kernel , I understand that I have to start from scratch. Given the facts that a) my current system (old 4.9 kernel) is running well and b) I have installed Nextcloud and c) I have moved the dietpi_userdata to my external HDD, am I correct that I can follow the steps mentioned in this blog [migrate to bullseye / fresh install](https://dietpi.com/blog/?p=811#2.1-fresh-install)?

I prefer to disconnect my external SATA-disk during initial install for security reasons. Is that possible? At what moment should I remount the drive? Should I choose install or re-install the software (Nextcloud and Jellyfin)?

echtfrank, 2022-02-22:
By the way, this issue could be closed, since I had not any error using the new kernel. Thanks again!

Joulinar, 2022-02-22:
Best to my knowledge, using the upgrade guide will not update the kernel. If I'm not mistaken, it has to do with how the old image was built in the past.

MichaIng, 2022-02-22:
Yes, that guide covers the major info about migration to a fresh image. For migrating Nextcloud (and ownCloud) explicitly, there is a script you can use: https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-cloud-migration
- Execute this on the old system, it will store the old Nextcloud instance and database within `/mnt/dietpi_userdata/nextcloud_data`.
- On the new system, after you attached the data drive and setup dietpi-drive_manager to point `/mnt/dietpi_userdata` to that drive, you can install Nextcloud, and it will use the backup automatically instead of doing a fresh install it.

Generally you can attach the data drive after first run setup, but install all software (which stores something in `/mnt/dietpi_userdata`) after you attached it and changed the user data dir accordingly, so your existing data will be migrated.

Also all software needs to be "install"ed instead of "reinstall"ed. Both does basically do the same anyway, existing data and configs will be preserved in any case. It is more a formal separation which respects the install marks in `/boot/dietpi/.installed` so that we can run dietpi-software reinstall 1 2 3 on DietPi update and have it skipping non-installed software IDs automatically.

echtfrank, 2022-02-23:
> Whether SATA works would be indeed an important check, since there is a dedicated HC4 bootloader. However, reading through it and the recent commits seems to indicate that the correct device tree is used automatically, while I'm then not sure what the dedicated U-Boot is about at all, probably booting from SATA. For flashing to SPI the C4 bootloader is required, which points into the direction I thought, since then one obviously boots from SPI, not from SATA, while to have an HC4 usable, SATA as data drive must of course work: https://github.com/armbian/build/blob/master/config/boards/odroidhc4.conf

I can confirm that SATA - disk can be mounted!

echtfrank, 2022-02-23:
@MichaIng , regarding flashing the SPI, only `flash_eraseall /dev/mtd0` was working; mtd1 to mtd3 didn't exist.

Since you metioned it above (" For flashing to SPI the C4 bootloader is required"), is there a command to copy he bootloader to the SPI?

As far as I can see now, I have erased the SPI **after**  the inital install of DietPi. Is there a process during that DietPi-installation that installs the bootloader to a certain location (SD or SPI)?

MichaIng, 2022-02-23:
> regarding flashing the SPI, only `flash_eraseall /dev/mtd0` was working

Makes sense, all four are more a failsafe step.

> Since you metioned it above (" For flashing to SPI the C4 bootloader is required"), is there a command to copy/install he bootloader to the SPI?

What I meant is the bootloader on SD card, where a C4 and an HC4 variant exists. I'm not sure whether these can be used at all from SPI, that its config is not located on the same drive then should already break it. Also I do not really see a benefit to use the SPI at all, since bootloader and OS are tied to each other anyway, at least when not using the legacy Hardkernel kernel and bootloader builds.

> Is there a process during that DietPi-installation that installs the bootloader to a certain location (SD or SPI)?

The bootloader is and remains on the SD card.

Btw, just in case you want to revert to a legacy resp. petitboot compatible OS, here is how to recover petitboot with a recovery SD card image: https://forum.odroid.com/viewtopic.php?p=311828#p311828

echtfrank, 2022-02-23:
OK, thanks for your reaction. You are to kind for showing the restoration of petitboot - which I knew.

I'll keep it the way it is now (without Petitboot) because I see too many conflicts regarding it in combination with debian, armbian and dietpi. Hopefully here will be some day that I don't need to power-down my server just to reboot.

On the other hand, I don't frequently feel the need to reboot when everthing is working; only when changing images ;-).

closed 2022-02-23 by echtfrank

MichaIng, 2022-02-23:
> Hopefully here will be some day that I don't need to power-down my server just to reboot.

What do you mean by that? `reboot` does not work as intended?

echtfrank, 2022-02-23:
About not being able  to reboot is the following.

Normally - using the previous Dietpi-c4-image (kernel 4.9) or the ancient debian 10 OS that I installed via Net-boot via Petitboot from Hardkernel), rebooting worked as intended.

That means that when I gave the command `sudo reboot now`, the system was completely shut down; I could hear de HDD spin down, the OLED-screen was turned off and finally the LED's were turned off, immediately followed by a (re)boot.
everything comes to life again and after a while you SSH into it, web-services are running etc.

At the moment, rebooting looks like the device is just turned off. I can't see any differrence between `sudo reboot now` and `sudo shutdown now`. 

As part of a reboot-process I now have to manually turn the board on. Since the HC4 is not equiped with a phisical power button, the only way to do that is to unplug it, wait a few seconds and then plug it in again (unplug and replug power adapter). Then the system starts normally. It is a bit annoying but it is far better than not a working Portainer ;-).

By the way I have an IR remotecontrol which I could use to power it on (or off) but that is not working either. I don't think that is related to this, but that would be further steps for me to try to get working. The same applies for the built-in OLED, and probably for the built-in heat fan too - I haven't tried that. I suspected these kind of issues when moving away from the old kernel. If I can help to investigate thing, feel free.

MichaIng, 2022-02-23:
Found it, indeed a bug with current Armbian kernel, which seems to be solved with next LTS release: https://forum.armbian.com/topic/19340-odroid-c4-will-not-reboot-after-any-sort-of-kernel-update-have-tried-running-nand-sata-install/?do=findComment&comment=134650

You could test it by switching from "current" to "edge" bootloader, dtb and kernel packages, but I hope the next LTS is moved to "current" soon anyway.

echtfrank, 2022-02-23:
OK, great! I will try 'edge' ;-) I'll let you know.

MichaIng, 2022-02-23:
You can upgrade to the new kernel. But at best do an SD card backup/clone first, just in case:
```sh
apt install linux-{image,dtb}-edge-meson64 linux-u-boot-odroidc4-edge
# To flash the new bootloader the SD card (optional)
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
```

echtfrank, 2022-02-23:
Updating the kernel went well.

I can't tell if the process of flashing the bootloader to my SD-card went well. Both lines give no output/result.
Are these two separate commands or just one line (I tried both). I tried them as user dietpi, prepend 'sudo' and as user root.
I assume the bootloader is not flashed to the SD.

Updating the kernel only seems no solution to the reboot-issue.

MichaIng, 2022-02-23:
> I can't tell if the process of flashing the bootloader to my SD-card went well. Both lines give no output/result.

You could do the following to get output:
```sh
sed -i 's| > /dev/null 2>&1||' /usr/lib/u-boot/platform_install.sh
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
```
It's 3 lines, at best copy & paste them directly into our SSH console.

> Updating the kernel only seems no solution to the reboot-issue.

😞. The old kernel was removed, was it? Could you verify that the new initramfs, dtb and kernel is loaded:
```sh
ls -l /boot
```

echtfrank, 2022-02-23:
Thanks a lot, I could have figured that out (remove the redirecting to nul)...
Flashing seems okay:

```
442+0 records in
442+0 records out
442 bytes copied, 0.0886331 s, 5.0 kB/s
2172+1 records in
2172+1 records out
1112432 bytes (1.1 MB, 1.1 MiB) copied, 0.191658 s, 5.8 MB/s
```

And, the output of  `ls -l /boot` is:
```
total 136076
lrwxrwxrwx 1 root root       23 Feb 23 20:35 Image -> vmlinuz-5.15.18-meson64
-rw-r--r-- 1 root root  5289511 Nov 22 11:21 System.map-5.10.81-meson64
-rw-r--r-- 1 root root  5215392 Jan 29 16:01 System.map-5.15.18-meson64
-rw-r--r-- 1 root root     3007 Feb 13 23:08 boot.cmd
-rw-r--r-- 1 root root     3079 Feb 15 12:03 boot.scr
-rw-r--r-- 1 root root   229071 Nov 22 11:21 config-5.10.81-meson64
-rw-r--r-- 1 root root   238480 Jan 29 16:01 config-5.15.18-meson64
drwxr-xr-x 4 root root     4096 Feb 23 15:19 dietpi
-rw-r--r-- 1 root root    18092 Feb 13 23:08 dietpi-LICENSE.txt
-rw-r--r-- 1 root root    14544 Feb 13 23:08 dietpi-README.md
-rw-r--r-- 1 root root    15052 Feb 23 15:20 dietpi.txt
-rw-r--r-- 1 root root      342 Feb 17 15:41 dietpiEnv.txt
lrwxrwxrwx 1 root root       19 Feb 23 20:34 dtb -> dtb-5.15.18-meson64
drwxr-xr-x 4 root root     4096 Feb 15 12:03 dtb-5.10.81-meson64
drwxr-xr-x 4 root root     4096 Feb 23 20:33 dtb-5.15.18-meson64
-rw-r--r-- 1 root root 12656846 Feb 15 12:03 initrd.img-5.10.81-meson64
-rw-r--r-- 1 root root 12340498 Feb 23 20:35 initrd.img-5.15.18-meson64
-rw-r--r-- 1 root root 26001984 Feb 23 20:35 uImage
lrwxrwxrwx 1 root root       23 Feb 23 20:35 uInitrd -> uInitrd-5.15.18-meson64
-rw-r--r-- 1 root root 12656910 Feb 15 12:03 uInitrd-5.10.81-meson64
-rw-r--r-- 1 root root 12340562 Feb 23 20:35 uInitrd-5.15.18-meson64
-rw-r--r-- 1 root root 26264064 Nov 22 11:21 vmlinuz-5.10.81-meson64
-rw-r--r-- 1 root root 26001920 Jan 29 16:01 vmlinuz-5.15.18-meson64
```

MichaIng, 2022-02-23:
Okay, so all is correctly pointing to the new kernel. The CI/CD reboot tests of Armbian for this kernel succeed to reboot: https://github.com/armbian/scripts/runs/5018211208?check_suite_focus=true

There is not dedicated test for the HC4 bootloader, we however use the C4 bootloader anyway. I hope this is not causing the reboot issue on HC4 while it works on C4 🤔.

echtfrank, 2022-02-23:
Yes, I stumbled upon that link. And I also thought about the difference between C4 and HC4.

Is it possible to do this `apt install linux-{image,dtb}-edge-meson64 linux-u-boot-odroid**_h_**c4-edge` instead?

Does DietPi link to the Armbian kernels or use a copy (subset) of these kernels and bootloaders?

echtfrank, 2022-02-23:
FYI, `sudo shutdown -r 1` gives me:
```
Failed to set wall message, ignoring: Unit dbus-org.freedesktop.login1.service is masked.
Failed to call ScheduleShutdown in logind, no action will be taken: Unit dbus-org.freedesktop.login1.service is masked.
```
I'll stop with it for now.

MichaIng, 2022-02-23:
Use `reboot` to reboot. The delayed reboot requires `systemd-logind` which is disabled on DietPi by default. It can be enabled via:
```sh
apt install dbus
systemctl unmask systemd-logind
systemctl start systemd-logind
```
There is also a `dietpi.txt` option for this, which is applied on first boot or on next `dietpi-software` install + of course when any software install requires it, like desktops or Kodi.

> Is it possible to do this?

Ah yes, of course, simple to test 👍:
```sh
G_AGP 'linux-u-boot-odroidc4-*'
apt install linux-u-boot-odroidhc4-edge
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
reboot
```

echtfrank, 2022-02-23:
Thanks a lot of your time. Unfortunately, it made no difference at all.
May be another day,

MichaIng, 2022-02-23:
You do not have a serial/UART adapter, do you?

I'll report this back to Armbian.

echtfrank, 2022-02-24:
You're correct, I don't have that adapter.

Thanks to report it to Armbian. There is a minor difference in the kernel version that I use (5.15.18) and the one that is not facing the reboot-issue according [to Armbian](https://forum.armbian.com/topic/19340-odroid-c4-will-not-reboot-after-any-sort-of-kernel-update-have-tried-running-nand-sata-install/#comment-134650) (5.15.15). I'll put Armbian on my watch-list!

echtfrank, 2022-02-24:
I'll be happy to help and test. That will be no sooner than this weekend, I'm afraid.

echtfrank, 2022-02-25:
Hi, I was  too curious and did some testing. On the [Armbian-site](https://www.armbian.com/odroid-hc4/), I read more closely about clearing the SPI. Two different methods are mentioned:
`flash_eraseall /dev/mtd0`
and 
```
ls -ltr /dev/mtd*  
flash_erase /dev/mtd0 0 0
```
Earlier you suggested the first one, but if I understand it right from armbian-site that command should be done from the Petitboot-console, which I didn't. So I tried the second one too. `ls -ltr /dev/mtd*` gave a correct listing like on armbian-site; the flash-command also worked accordingly.  After all, still no reboot possible.

And now the happy news: booting the Arbian-HC4-image you asked, went totally well! See attached upload of the total SSH-session:
[SSH_session_install_Armbian_HC4_Bullseye_20220225.log](https://github.com/MichaIng/DietPi/files/8139782/SSH_session_install_Armbian_HC4_Bullseye_20220225.log)

Next step: shutting down Armbian and remove SD-card; insert my production DietPi-SD and start system --> OK
Reboot still does not work. My conclusion is that the boot-issue has nothing to do with with erasing Petitboot or clearing the SPI. I thought it might be that Armbian could do the SPI flashing better than DietPi, because Armbian might support more hardware components (DTB, etc). But now I think it really has something to do with installing the bootloader itself for some reason.

Anyway, the reboot-issue does NOT occur on Armbian HC4 kernel 5.10.81-meson64... so no need to update kernel to bleeding edge  5.15.18... There could be a difference between the images of the HC4 and the C4. I could test that to if you think that is useful.

echtfrank, 2022-02-25:
I have tested one more thing on the Armbian-system: verify if rebooting also worked after mounting the SATA-disk. This also worked well.

[2 more replies]