MichaIng/DietPi #5274: Pydio fail to install correctly, error on initial screen

holocronology, 2022-02-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.1.2
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.0-9-amd64 #1 SMP Debian 5.10.70-1 (2021-09-30) x86_64 GNU/Linux
- SBC model | Virtual Machine (x86_64)
- Power supply used | na
- SD card used | na

#### Additional Information (if applicable)
- Software title | Pydio
- Was the software title installed freshly or updated/migrated? Fresh and updated
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Install Pydio from dietpi-software
2. Go to initial page http://IPADDR/pydio

#### Expected behaviour
Launch startup

#### Actual behaviour
Error page display: "Impossible write into the AJXP_DATA_PATH folder: Make sure to grant write access to this folder for your webserver!"

Joulinar, 2022-02-14:
Hi,

indeed a bug we created on last release. Pls do following as workaround.

```
chown -R www-data:www-data /var/www/pydio /mnt/dietpi_userdata/pydio_data/
```

MichaIng, 2022-02-14:
Solved with: #5277

closed 2022-02-14 by MichaIng