MichaIng/DietPi #5278: {First deployment failure} Odroid C4 Bullseye DietPi V7

MDAR, 2022-02-14:
Failure to complete first run

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
Contents of `.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

- Kernel version | `uname -a` - Unable to obtain a command prompt
- SBC model | Odorid C4
- Power supply used | 12V 2.5Amp
- SD card used | eMMC 16GB

#### Additional Information (if applicable)
- Software title | Chromium
- Was the software title installed freshly or updated/migrated? - Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
<a93df9cf-11a6-49a4-b57d-9816ae857944>
<f752bac5-5acd-4feb-a58b-43d27fb58608>
- Bug report ID | `echo $G_HW_UUID` - unable to get a command prompt

#### Steps to reproduce

1.  Download fresh image
2.  Flash to eMMC
3. Boot to new image
4. Choose defaults, with the only exception of adding Chromium as a software package

#### Expected behaviour

- Achieve a simple system with Chromium loading after boot as a kiosk

#### Actual behaviour

- Fails to complete the initial setup with these error messages

```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] DietPi-Software | APT install for: xserver-xorg-core xserver-xorg-input-libinput xinit dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
 - Command: apt-get -qq --allow-change-held-packages install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code:	f752bac5-5acd-4feb-a58b-43d27fb58608
```

This command failed
```
apt -qq --allow-held-packages install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra	

```

#### Extra details

- This also happened when I tried to update a previous working system.

MichaIng, 2022-02-14:
Many thanks for your report.

I checked your bug report:
```
dpkg: error processing archive /tmp/apt-dpkg-install-mgmNpt/041-libllvm11_1%3a11.0.1-2_arm64.deb (--unpack):
 cannot copy extracted data for './usr/lib/aarch64-linux-gnu/libLLVM-11.so.1' to '/usr/lib/aarch64-linux-gnu/libLLVM-11.so.1.dpkg-new': failed to write (No space left on device)
...
dpkg: error processing archive /tmp/apt-dpkg-install-mgmNpt/045-libgl1-mesa-dri_20.3.5-1_arm64.deb (--unpack):
 cannot copy extracted data for './usr/lib/aarch64-linux-gnu/dri/armada-drm_dri.so' to '/usr/lib/aarch64-linux-gnu/dri/armada-drm_dri.so.dpkg-new': failed to write (No space left on device)
...
```
Can you check the free filesystem space, please:
```sh
df -h /
```

MichaIng, 2022-02-14:
But please verify that the used space really is 16 GiB already. It would be also possible that the partition/filesystem expansion failed for some reason. `df -h /` would reveal this.

MDAR, 2022-02-14:
Yep

There's the problem

```
root@DietPi:~# df -h /
Filesystem               Size        Used      Avail      Use%      Mounted on
/dev/mmcblk0p2     740M    577M     149M     80%        /
````

MDAR, 2022-02-14:
Wow, that was a fast reply.

Thanks.

``root@DietPi:~# cat /var/tmp/dietpi/logs/fs_partition_resize.log
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/mmcblk0: 14.56 GiB, 15634268160 bytes, 30535680 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x03823826

Old situation:

Device         Boot  Start     End Sectors   Size Id Type
/dev/mmcblk0p1        2048  526335  524288   256M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      526336 2039567 1513232 738.9M 83 Linux

/dev/mmcblk0p2: 
New situation:
Disklabel type: dos
Disk identifier: 0x03823826

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        2048   526335   524288  256M  e W95 FAT16 (LBA)
/dev/mmcblk0p2      526336 30535679 30009344 14.3G 83 Linux

The partition table has been altered.
resize2fs 1.46.2 (28-Feb-2021)
resize2fs: No space left on device While checking for on-line resizing support
Filesystem at /dev/mmcblk0p2 is mounted on /; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
``

MichaIng, 2022-02-14:
```
resize2fs: No space left on device While checking for on-line resizing support
```
I remember the exact same happened on the legacy Odroid C2 image: #5008
No idea where this is coming from, probably it has to do with the journal recreation we do, but it never was an issue with any other image and the modern Odroid images. I repacked the Odroid C4 image yesterday to add some kernel command-line parameter required for Docker to work, and other applications which require some cgroupv2 features that this old kernel does not support: https://github.com/MichaIng/DietPi/issues/5269#issuecomment-1038150039
I remember it become a little smaller. While 20% and more free space look pretty sufficient for any action, probably some specific part became too small for this kernel to perform a filesystem resize somehow.

I'll now work on the new C4 image. Working on images anyway currently. Let's see how fast I can set this up based on the Odroid N2 work that has been done already.

MDAR, 2022-02-14:
If there is anything I can do to help, please just ask.

I'm happy to give you remote access to a machine if that help you.

MDAR, 2022-02-14:
FYI

`fdisk /dev/mmcblk0p2` followed by `p`returns this

```
Welcome to fdisk (util-linux 2.36.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

The device contains 'ext4' signature and it will be removed by a write command. See fdisk(8) man page and --wipe option for more details.

Device does not contain a recognized partition table.
Created a new DOS disklabel with disk identifier 0x04224d6c.

Command (m for help): p
Disk /dev/mmcblk0p2: 14.31 GiB, 15364784128 bytes, 30009344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x04224d6c
```

and for mmcblk0p1 returns this

```
Disk /dev/mmcblk0p1: 256 MiB, 268435456 bytes, 524288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x01b14692
```

MDAR, 2022-02-14:
FYI

Just thought I'd see what happens now if I tried to add Chromium, Apache and openSSH

It still failed at the xfont-base x11-server-utils x11-util etc stage

```
root@DietPi:/opt# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Setting in /boot/dietpi/.installed adjusted: INDEX_WEBSERVER=0
[  OK  ] DietPi-Software | Setting in /boot/dietpi/.installed adjusted: INDEX_BROWSER=-2
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | OpenSSH Client will be installed
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask 
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://dietpi.com/meveric bullseye InRelease
Hit:6 https://dietpi.com/meveric all InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing OpenSSH Client: Feature-rich SSH, SFTP and SCP client

[ INFO ] DietPi-Software | APT install for: openssh-client, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libedit2:arm64.
(Reading database ... 15668 files and directories currently installed.)
Preparing to unpack .../libedit2_3.1-20191231-2+b1_arm64.deb ...
Unpacking libedit2:arm64 (3.1-20191231-2+b1) ...
Selecting previously unselected package libcbor0:arm64.
Preparing to unpack .../libcbor0_0.5.0+dfsg-2_arm64.deb ...
Unpacking libcbor0:arm64 (0.5.0+dfsg-2) ...
Selecting previously unselected package libfido2-1:arm64.
Preparing to unpack .../libfido2-1_1.6.0-2_arm64.deb ...
Unpacking libfido2-1:arm64 (1.6.0-2) ...
Selecting previously unselected package openssh-client.
Preparing to unpack .../openssh-client_1%3a8.4p1-5_arm64.deb ...
Unpacking openssh-client (1:8.4p1-5) ...
Setting up libedit2:arm64 (3.1-20191231-2+b1) ...
Setting up libcbor0:arm64 (0.5.0+dfsg-2) ...
Setting up libfido2-1:arm64 (1.6.0-2) ...
Setting up openssh-client (1:8.4p1-5) ...
Processing triggers for libc-bin (2.31-13+deb11u2) ...
[  OK  ] DietPi-Software | APT install for: openssh-client

 DietPi-Software
[... 226 lines cut ...]
                                       │ Unpacking libmpfr6:arm64 (4.1.0-3) ...^M                                                                               
                                       │ Selecting previously unselected package libmpc3:arm64.^M                                                               
                                       │ Preparing to unpack .../004-libmpc3_1.2.0-1_arm64.deb ...^M                                                            
                                       │ Unpacking libmpc3:arm64 (1.2.0-1) ...^M                                                                                
                                       │ Selecting previously unselected package cpp-10.^M                                                                      
                                       │ Preparing to unpack .../005-cpp-10_10.2.1-6_arm64.deb ...^M                                                            
                                       │ Unpacking cpp-10 (10.2.1-6) ...^M                                                                                      
                                       │ Selecting previously unselected package cpp.^M                                                                         
                                       │ Preparing to unpack .../006-cpp_4%3a10.2.1-1_arm64.deb ...^M                                                           
                                       │ Unpacking cpp (4:10.2.1-1) ...^M                                                                                       
                                       │ Selecting previously unselected package libxau6:arm64.^M                                                               
                                       │ Preparing to unpack .../007-libxau6_1%3a1.0.9-1_arm64.deb ...^M                                                        
[... cut ...]
                                       │                        Retry          : Re-run the last command that failed                 ↑                          
                                       │                        DietPi-Config  : Edit network, APT/NTP mirror settings etc           ▮    
```
Is this the key line ?

```
dpkg: error processing archive /tmp/apt-dpkg-install-eHAT9a/071-libxmu6_2%3a1.1.2-2+b3_arm64.deb (--unpack):
 cannot copy extracted data for './usr/lib/aarch64-linux-gnu/libXmu.so.6.2.0' to '/usr/lib/aarch64-linux-gnu/libXmu.so.6.2.0.dpkg-new': failed to write (No space left on device)
```

MichaIng, 2022-02-14:
No need to retry as long as the filesystem has not been expanded 😉. What `fdisk` shows you is not the filesystem size but the partition size, which has been expended successfully. It is only the contained ext4 partition which fails to be expanded.

MDAR, 2022-02-14:
> It is only the contained ext4 filesystem which fails to be expanded.

I'm not entirely sure I follow, but I get the general idea.

I'll go back to using a previous image I have and leave you to do your magic in your own time. (I'll just not update past V7)

There's no pressure from me.

Thanks a million Michal

Joulinar, 2022-02-14:
The issue is not with updating DietPi. It's the first run setup that did not expand the partition correctly. Question is why? This is something to find out.

MichaIng, 2022-02-14:
I'm not gonna invest further time into this resizing issue, since it only affects the legacy Odroid images and we already tried everything on Odroid C2, without success. Better have the new image ready. Good thing is that boot configuration is exactly the same as for Odroid N2. But Armbian offers two different bootloaders (and images) for C4 and HC4. I hope their are identical, especially that they load the correct device tree OOTB. We'll see, else we need to split the images as well 😢.

MichaIng, 2022-02-14:
Image ready for testing: https://dietpi.com/downloads/images/testing/DietPi_OdroidC4-ARMv8-Bullseye.7z

MDAR, 2022-02-15:
> Image ready for testing: https://dietpi.com/downloads/images/testing/DietPi_OdroidC4-ARMv8-Bullseye.7z

Trying it right now......

MichaIng, 2022-02-15:
> UUID=3ccf3790-5e80-44b9-83be-17e3b21083d9 does not exist. Dropping to a shell!

Let me see.

MichaIng, 2022-02-15:
Ah, I missed to edit the part in PREP where the UUID and root filesystem type are adjusted in `/boot/dietpiEnv.txt`. I just updated the image accordingly (same link).

If you have a possibility to mount that ext4 filesystem and check its UUID on another system, you can edit it yourself. It's the line starting with `rootdev=UUID=` and the value should be the output of what on Linux this gives (with the device path adjusted, in case):
```sh
lsblk -no UUID /dev/mmcblk0p1
```

MichaIng, 2022-02-15:
Hmm, so X11 with xterm works but not with Chromium. What does `cat /var/log/Xorg.0.log` show after trying to start `xinit /usr/bin/chromium`?

I wonder whether its the same as here: https://dietpi.com/phpbb/viewtopic.php?p=43184#p43184
Through other/additional error messages are thrown, not sure whether those are visible on primary console only (kernel logs), which should then be seen via `dmesg` elsewhere.

MDAR, 2022-02-15:
> Hmm, so X11 with xterm works but not with Chromium. What does `cat /var/log/Xorg.0.log` show after trying to start `xinit /usr/bin/chromium`?

brace yourself... this is a long log

```
root@DietPi:~# xinit /usr/bin/chromium

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon Nov 22 11:21:51 UTC 2021 aarch64
Kernel command line: root=UUID=28046172-8741-4334-96cd-b24d158eec46 rootfstype=ext4 rootwait console=tty1 loglevel=4                                consoleblank=0 coherent_pool=2M ubootpart=2c2a3e4f-01 libata.force=noncq usb-storage.quirks=0x2537:0x1066:u,0x2537:0x                               1068:u net.ifnames=0
Build Date: 16 December 2021  05:08:23PM
xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Feb 15 19:09:41 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

root@DietPi:~# cat /var/log/Xorg.0.log
[   141.048]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[   141.049] Build Operating System: linux Debian
[   141.049] Current Operating System: Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon Nov 22 11:21:51 UTC 2021                                aarch64
[   141.049] Kernel command line: root=UUID=28046172-8741-4334-96cd-b24d158eec46 rootfstype=ext4 rootwait console=tty                               1 loglevel=4 consoleblank=0 coherent_pool=2M ubootpart=2c2a3e4f-01 libata.force=noncq usb-storage.quirks=0x2537:0x106                               6:u,0x2537:0x1068:u net.ifnames=0
[   141.049] Build Date: 16 December 2021  05:08:23PM
[   141.049] xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support)
[   141.049] Current version of pixman: 0.40.0
[   141.049]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   141.049] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   141.049] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Feb 15 19:09:41 2022
[   141.050] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   141.050] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   141.050] (==) No Layout section.  Using the first Screen section.
[   141.050] (==) No screen section available. Using defaults.
[   141.050] (**) |-->Screen "Default Screen Section" (0)
[   141.050] (**) |   |-->Monitor "<default monitor>"
[   141.051] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[   141.051] (**) Option "BlankTime" "0"
[   141.051] (==) Automatically adding devices
[   141.051] (==) Automatically enabling devices
[   141.051] (==) Automatically adding GPU devices
[   141.051] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   141.051] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   141.051]    Entry deleted from font path.
[   141.051] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   141.051]    Entry deleted from font path.
[   141.051] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   141.051]    Entry deleted from font path.
[   141.051] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   141.051]    Entry deleted from font path.
[   141.051] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   141.051]    Entry deleted from font path.
[   141.051] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   141.051]    Entry deleted from font path.
[   141.051] (==) FontPath set to:
[... 200 lines cut ...]
[   142.678] (II) libinput: meson-ir: needs a virtual subdevice
[   142.678] (**) Option "config_info" "udev:/sys/devices/platform/soc/ff800000.bus/ff808000.ir/rc/rc0/input0/event0"
[   142.678] (II) XINPUT: Adding extended input device "meson-ir" (type: MOUSE, id 6)
[   142.680] (**) Option "AccelerationScheme" "none"
[   142.680] (**) meson-ir: (accel) selected scheme none/0
[   142.680] (**) meson-ir: (accel) acceleration factor: 2.000
[   142.680] (**) meson-ir: (accel) acceleration threshold: 4
[   142.689] (II) event0  - meson-ir: is tagged by udev as: Keyboard Pointingstick
[   142.690] (II) event0  - meson-ir: device is a pointer
[   142.690] (II) event0  - meson-ir: device is a keyboard
[   142.717] (**) meson-ir: Applying InputClass "libinput keyboard catchall"
[   142.717] (II) Using input driver 'libinput' for 'meson-ir'
[   142.717] (**) meson-ir: always reports core events
[   142.717] (**) Option "Device" "/dev/input/event0"
[   142.717] (**) Option "_source" "_driver/libinput"
[   142.717] (II) libinput: meson-ir: is a virtual subdevice
[   142.717] (**) Option "config_info" "udev:/sys/devices/platform/soc/ff800000.bus/ff808000.ir/rc/rc0/input0/event0"
[   142.717] (II) XINPUT: Adding extended input device "meson-ir" (type: KEYBOARD, id 7)
[   142.718] (**) Option "xkb_model" "pc105"
[   142.718] (**) Option "xkb_layout" "gb"
[   143.299] (II) event0  - meson-ir: device removed
[   143.377] (II) UnloadModule: "libinput"
[   143.380] (II) UnloadModule: "libinput"
[   143.521] (II) Server terminated successfully (0). Closing log file.
```

MichaIng, 2022-02-15:
It is like Chromium is simply stopping immediately after being started. No error messages from the X server, just a regular shutdown after startup finished 🤔. The other error messages on the forum thread may not even be related.

Does it work when you install a desktop and start Chromium from there?

MDAR, 2022-02-16:
> Does it work when you install a desktop and start Chromium from there?

Thanks Michal

I'm away for a few days, as soon as I get back I'll give it a try.

MDAR, 2022-02-18:
> Does it work when you install a desktop and start Chromium from there?

Hi

I'm back...

Here's what I have tried so far.

I ran `dietpi-software` and added XFDE
But I DIDN'T change the autoboot option from Chromium.

When rebooting I get a screen full of

```
login as: root
root@192.168.178.75's password:
Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon Nov 22 11:21:51 UTC 2021 a                                                                                                               arch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Feb 18 11:07:04 2022
 ─────────────────────────────────────────────────────
 DietPi v8.1.2 : 11:14 - Fri 02/18/22
 ─────────────────────────────────────────────────────
 - Device model : Odroid C4/HC4 (aarch64)
 - CPU temp : 33 °C / 91 °F : Cool runnings
 - LAN IP : 192.168.178.75 (eth0)
 - MOTD : DietPi v8.1 is here, check out the release notes:
          https://dietpi.com/docs/releases/v8_1/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Image by        : DietPi Core Team (pre-image: from scratch)
 Patreon Legends : Camry2731, Chris Gelatt
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# dietpi-config
[  OK  ] DietPi-AutoStart | Setting in /boot/dietpi.txt adjusted: SOFTWARE_CHROM                                                                                                               IUM_AUTOSTART_URL=https://www.dietpi.com/
[  OK  ] DietPi-AutoStart | Desired setting in /boot/dietpi.txt was already set:                                                                                                                AUTO_SETUP_AUTOSTART_LOGIN_USER=root
[  OK  ] DietPi-AutoStart | mkdir -p /etc/systemd/system/getty@tty1.service.d /e                                                                                                               tc/systemd/system/console-getty.service.d
[  OK  ] DietPi-AutoStart | systemctl daemon-reload
root@DietPi:~# find / -iname chromium*
/etc/chromium
/etc/chromium.d
/usr/bin/chromium
/usr/share/doc/chromium-common
/usr/share/doc/chromium
/usr/share/bug/chromium
/usr/share/man/man1/chromium.1.gz
/usr/share/bash-completion/completions/chromium-browser
/usr/share/bash-completion/completions/chromium
/usr/share/chromium
/usr/share/gnome-control-center/default-apps/chromium.xml
/usr/share/lintian/overrides/chromium
/usr/share/icons/hicolor/24x24/apps/chromium.png
/usr/share/icons/hicolor/256x256/apps/chromium.png
/usr/share/icons/hicolor/48x48/apps/chromium.png
/usr/share/icons/hicolor/16x16/apps/chromium.png
[... 49 lines cut ...]
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.113204:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.128790:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.144491:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.165640:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
[2681:2681:0218/111605.165751:FATAL:gpu_data_manager_impl_private.cc(448)] GPU process isn't usable. Goodbye.
[2719:2736:0218/111605.262777:ERROR:ssl_client_socket_impl.cc(995)] handshake failed; returned -1, SSL error code 1,                                                                           net_error -3
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

root@DietPi:~#
```

What I'm seeing on the device's monitor is the white start screen of Chromium, but only for a few ms.

I'll start the desktop and see if Chromium works

MDAR, 2022-02-18:
Trying to run each stage manually, 

Command prompt login
run xinit
start chromium from Xinit

I see this kind of error in Xinit

```
[2681:2712:0218/111604.891078:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknow                                                                          n address type (examples of valid types are "tcp" and on UNIX "unix")
[2681:2712:0218/111604.891201:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknow                                                                          n address type (examples of valid types are "tcp" and on UNIX "unix")
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.014512:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.095049:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.113204:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.128790:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.144491:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2681:2681:0218/111605.165640:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
FATAL:gpu_data_manager_impl_private.cc(448) GPU process isn't usable. Goodbye
handshake failed; returned -1, SSL error code 1, net_error -3
```

Is it.... This last line and the GPU error that needs attention?

MichaIng, 2022-02-18:
The missing `libpci3` dependency has already been reported to Debian: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=993308
Though no reaction so far, I'll try to give this a push.

Definitely an ARMv8 related bug in Chromium, I'll try to replicate and in case report at the Debian bug tracker as well.

MichaIng, 2022-02-19:
Firefox warning appears on all distro versions from Buster to Bookworm. I added it to `dietpi-software`, might help with Firefox internal render decisions: https://github.com/MichaIng/DietPi/commit/5e2dda0

Gave the bug report a bump, but at least our end it is solved: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=993308

MichaIng, 2022-02-19:
According to Chromium, I see `*** stack smashing detected ***: terminated` on an x86_64 VM as well, together with attempts to access the CPUFreq API, which isn't available on VMs. Probably another things to report to Debian, and in case forward upstream, so that it is checked first whether CPUFreq is available at all before it is tried to read it.

MichaIng, 2022-02-20:
I'll mark this as closed. Let's concentrate on Chromium ARMv8 in a dedicated issue: https://github.com/MichaIng/DietPi/issues/5300

closed 2022-02-20 by MichaIng