MichaIng/DietPi #5279: Error installing Jellyfin on Odroid HC4 Bullseye

echtfrank, 2022-02-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

- Distro version | bullseye
- Kernel version | Linux biblio 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux
- SBC model | Odroid C4/HC4 (aarch64)
- Power supply used | 12V 4A Hardkernel
- SD card used | SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | Jellyfin and Samba Server
- Was the software title installed freshly or updated/migrated? I thought freshly
- Can this issue be replicated on a fresh installation of DietPi? Unknown at this moment.
- Bug report ID | N/A

#### Steps to reproduce
1. sudo dietpi-software
2. select jellyfin AND Samba Server
3. choose "install"

#### Expected behaviour
- Software is installed correctly
- Jellyfin web interface available on http://<ip>:8097

#### Actual behaviour
- An error is visible on the screen (see attached file)
- It looks like Jellyfin is running but there is no webinterface visible

#### Extra details
- attached logging of installation:
- attached output of journalctl -u jellyfin
- yesterday I had a [interruption of installing software](https://github.com/MichaIng/DietPi/issues/5269#issuecomment-1038488656); may that corrupted this installation?

I'm sorry but I have little time to assist or try other steps tomorrow.
[journalctl-u-Jellyfin_20220214.log](https://github.com/MichaIng/DietPi/files/8064476/journalctl-u-Jellyfin_20220214.log)
[screenshot_of_error-installing_Jellyfin_20220214.txt](https://github.com/MichaIng/DietPi/files/8064432/screenshot_of_error-installing_Jellyfin_20220214.txt)
[install_Jellyfin_20220214.log](https://github.com/MichaIng/DietPi/files/8064452/install_Jellyfin_20220214.log)

MichaIng, 2022-02-15:
Many thanks for reporting.

This config file is not installed by the package but created by the service on startup, and we stop it too quickly 😄. I added another step, a function which starts a service until a specific configuration file exists and then stops it again. Works fine here.

If you want to adjust the network port to what was intended:
```sh
G_CONFIG_INJECT '<PublicPort>' '<PublicPort>8097</PublicPort>' /etc/jellyfin/network.xml
G_CONFIG_INJECT '<HttpServerPortNumber>' '  <HttpServerPortNumber>8097</HttpServerPortNumber>' /etc/jellyfin/network.xml
systemctl restart jellyfin-web
```
Otherwise the port is `8096`. We only change it since InfluxDB uses `8086` as well, to avoid a conflict when one installs both, but if you do not plan to install InfluxDB, there is not really a reason for changing it 😉.

echtfrank, 2022-02-15:
More than many thanks back for fixing this issue, @MichaIng!

I like to follow 'the standards' - you never know what future brings - so I'll switch to port 8097.
Rstarting jellyfin-web generates another error.

> Failed to restart jellyfin-web.service: Unit jellyfin-web.service not found.

Joulinar, 2022-02-15:
no need to reinstall. The name of the service is incorrect. It's `jellyfin ` only instead of `jellyfin-web` 😉 

I guess @MichaIng mixed it with Deluge. There we have `deluge-web`.

This should do it for you

```
systemctl restart jellyfin
```

echtfrank, 2022-02-15:
Haha, the latter worked! I forgive you :-))))))))

MichaIng, 2022-02-15:
Whoops, I mixed it up because there is indeed a dedicated `jellyfin-web` APT package, but it is invoked by the same `jellyfin` service 😄. Great that it worked. The commit has it luckily right, where I stumbled above exactly this question.

closed 2022-02-15 by MichaIng