MichaIng/DietPi #5280: [FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd

g0rge0us, 2022-02-15:
#### Details:
- Date           | Tue Feb 15 12:17:43 EET 2022
- DietPi version | v8.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux dietrpi4b 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd
[FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd
[ INFO ] DietPi-TimeSync | Executing alternative command: systemctl start systemd-timesyncd
Job for systemd-timesyncd.service failed.
See "systemctl status systemd-timesyncd.service" and "journalctl -xe" for details.
[FAILED] Alternative command execution | Exited with error
Press any key to return to error handler menu... 

2. I am posting here the request to open a ticket, which I got from the OS. Plese find the desired information below,captured with copy+paste like you have advised.

#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for systemd-timesyncd.service failed because the control process exited with error code.
See "systemctl status systemd-timesyncd.service" and "journalctl -xe" for details.
```

Joulinar, 2022-02-15:
Can you share following

`journalctl -u systemd-timesyncd.service`

g0rge0us, 2022-02-16:
dietpi@dell-9200:~$ journalctl -u systemd-timesyncd.service
Hint: You are currently not seeing messages from other users and the system.
      Users in groups 'adm', 'systemd-journal' can see all messages.
      Pass -q to turn off this notice.
No journal files were opened due to insufficient permissions.
dietpi@dell-9200:~$ sudo !!
sudo journalctl -u systemd-timesyncd.service
-- Journal begins at Mon 2022-02-14 06:08:36 EET, ends at Wed 2022-02-16 16:04:08 EET. --
фев 15 01:25:02 dell-9200 systemd[1]: Starting Network Time Synchronization...
фев 15 01:25:02 dell-9200 systemd[1]: Started Network Time Synchronization.
фев 15 01:25:01 dell-9200 systemd-timesyncd[166516]: Initial synchronization to time server 84.54.128.102:123 (0.debian.pool.ntp.org).
фев 15 01:25:02 dell-9200 systemd[1]: Stopping Network Time Synchronization...
фев 15 01:25:02 dell-9200 systemd[1]: systemd-timesyncd.service: Succeeded.
фев 15 01:25:02 dell-9200 systemd[1]: Stopped Network Time Synchronization.
фев 16 01:25:01 dell-9200 systemd[1]: Starting Network Time Synchronization...
фев 16 01:25:01 dell-9200 systemd[1]: Started Network Time Synchronization.
фев 16 01:25:00 dell-9200 systemd-timesyncd[179663]: Initial synchronization to time server 93.123.39.67:123 (0.debian.pool.ntp.org).
фев 16 01:25:01 dell-9200 systemd[1]: Stopping Network Time Synchronization...
фев 16 01:25:01 dell-9200 systemd[1]: systemd-timesyncd.service: Succeeded.
фев 16 01:25:01 dell-9200 systemd[1]: Stopped Network Time Synchronization.

g0rge0us, 2022-02-16:
фев = Februar (in Bulgarian)

Joulinar, 2022-02-16:
Daily time sync is working. Question is if your system time stays in sync over the day or is it still drifting?

g0rge0us, 2022-02-16:
Oh... sorry, that was the wrong server... I'll paste the requested info in 1 min.

g0rge0us, 2022-02-16:
dietpi@dietrpi4b:~$ sudo journalctl -u systemd-timesyncd.service
-- Journal begins at Sun 2022-01-30 19:02:48 EET, ends at Wed 2022-02-16 16:11:40 EET. --
яну 30 19:02:49 dietrpi4b systemd[1]: Starting Network Time Synchronization...
яну 30 19:02:49 dietrpi4b systemd[1]: Started Network Time Synchronization.
яну 30 19:02:53 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
яну 30 19:02:53 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
яну 30 19:02:53 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
яну 30 19:02:53 dietrpi4b systemd[1]: Starting Network Time Synchronization...
яну 30 19:02:54 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 06 16:45:28 dietrpi4b systemd-timesyncd[580]: Initial synchronization to time server 77.236.182.128:123 (0.debian.pool.ntp.org).
фев 07 01:25:01 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
фев 07 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
фев 07 01:25:01 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 07 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 07 01:25:01 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 07 01:25:01 dietrpi4b systemd-timesyncd[6670]: Initial synchronization to time server 91.210.88.37:123 (0.debian.pool.ntp.org).
фев 07 01:25:02 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
фев 07 01:25:02 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
фев 07 01:25:02 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 08 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 08 01:25:01 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 08 01:25:00 dietrpi4b systemd-timesyncd[135834]: Initial synchronization to time server 195.238.84.84:123 (0.debian.pool.ntp.org).
фев 08 01:25:01 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
фев 08 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
фев 08 01:25:01 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 09 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 09 01:25:01 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 09 01:25:00 dietrpi4b systemd-timesyncd[323110]: Initial synchronization to time server 213.91.190.229:123 (0.debian.pool.ntp.org).
фев 09 01:25:01 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
фев 09 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
фев 09 01:25:01 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 10 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 10 01:25:01 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 10 01:25:00 dietrpi4b systemd-timesyncd[500699]: Initial synchronization to time server 195.238.84.23:123 (0.debian.pool.ntp.org).
фев 10 01:25:01 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
фев 10 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
фев 10 01:25:01 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 11 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 11 01:25:01 dietrpi4b systemd[784530]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Stale file handle
фев 11 01:25:01 dietrpi4b systemd[784530]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Stale file handle
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
фев 11 01:25:01 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 1.
фев 11 01:25:01 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 11 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 11 01:25:01 dietrpi4b systemd[784536]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Stale file handle
фев 11 01:25:01 dietrpi4b systemd[784536]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Stale file handle
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
фев 11 01:25:01 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 2.
фев 11 01:25:01 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 11 01:25:01 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 11 01:25:01 dietrpi4b systemd[784555]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Stale file handle
фев 11 01:25:01 dietrpi4b systemd[784555]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Stale file handle
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
фев 11 01:25:01 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
[... cut ...]
фев 16 16:10:45 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 5.
фев 16 16:10:45 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Start request repeated too quickly.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
фев 16 16:10:45 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.

MichaIng, 2022-02-16:
> Main process exited, code=exited, status=226/NAMESPACE

Did you link `/var/tmp` to `/tmp`?
```sh
ls -dl /var/tmp
```

MichaIng, 2022-02-16:
Okay that is not a symlink, also modes are correct. Can you check the other directories which are part of the systemd-timesyncd sandboxing:
```sh
ls -dl /home /proc /sys /tmp /lib/modules
```
There is more, but I'm not sure which files or directories those are related to. One of these however seems to be missing or a symlink so that systemd cannot do the namespace sandboxing, something like a private mount onto these to make the content inaccessible to the service.

g0rge0us, 2022-02-16:
dietpi@dietrpi4b:~$ ls -dl /home /proc /sys /tmp /lib/modules
drwxr-xr-x   3 root root 4096 15 дек 17:38 /home
drwxr-xr-x   3 root root 4096 30 яну 18:06 /lib/modules
dr-xr-xr-x 325 root root    0  1 яну 1970 /proc
dr-xr-xr-x  12 root root    0  1 яну 1970 /sys
drwxrwxrwt   9 root root  180 16 фев 18:08 /tmp

MichaIng, 2022-02-16:
Looks all right. Here is a similar case with another systemd service (not active on DietPi by default): https://github.com/systemd/systemd/issues/16156
There is a way to get debug log to see where exactly it fails:
```sh
G_CONFIG_INJECT 'LogLevel=' 'LogLevel=debug' /etc/systemd/system.conf
systemctl daemon-reload
systemctl restart systemd-timesyncd
sleep 5
journalctl -u systemd-timesyncd | tail -20
```

g0rge0us, 2022-02-16:
So, what exactly do you want me to do?

g0rge0us, 2022-02-16:
Oh, sorry! I thought it should be 1st saved in some file... Anyway here it is:
```
sudo journalctl -u systemd-timesyncd | tail -20
фев 16 16:10:45 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 16 16:10:45 dietrpi4b systemd[2229815]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Stale file handle
фев 16 16:10:45 dietrpi4b systemd[2229815]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Stale file handle
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
фев 16 16:10:45 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 4.
фев 16 16:10:45 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 16 16:10:45 dietrpi4b systemd[2229818]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/: Stale file handle
фев 16 16:10:45 dietrpi4b systemd[2229818]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Stale file handle
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
фев 16 16:10:45 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 5.
фев 16 16:10:45 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Start request repeated too quickly.
фев 16 16:10:45 dietrpi4b systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
фев 16 16:10:45 dietrpi4b systemd[1]: Failed to start Network Time Synchronization.
```

MichaIng, 2022-02-16:
Here it is:
```sh
ls -dl /run/systemd/unit-root
```

g0rge0us, 2022-02-16:
```
drwxr-xr-x 18 root root 4096 15 дек 17:49 /
drwxr-xr-x 20 root root  600 10 фев 02:42 /run
drwxr-xr-x 20 root root  480 16 фев 16:10 /run/systemd
```

MichaIng, 2022-02-16:
All exactly like it should be. Okay, reading about what a "stable file handle" is: https://www.baeldung.com/linux/stale-file-handles#stale-file-handles

Looks like `/run/systemd/unit-root` is created with an inode and then somehow removed again, so that the file handle systemd was about to use for namespacing becomes stale. To me this looks like a bug or broken session.

Did you already try a `reboot`? If so, and error persists, please try to reinstall both, systemd and systemd-timesyncd:
```sh
apt update
apt install --reinstall systemd systemd-timesyncd
```

g0rge0us, 2022-02-17:
Hm... after reboot, I was able to normally execute dietpi-update without any errors. Should I reinstall both packages: systemd & systemd-timesyncd?

Joulinar, 2022-02-17:
I guess system time sync has been working on reboot. Can you have a look?

`journalctl -u systemd-timesyncd`

g0rge0us, 2022-02-17:
`sudo journalctl -u systemd-timesyncd
-- Journal begins at Thu 2022-02-17 21:29:25 EET, ends at Thu 2022-02-17 21:43:43 EET. --
фев 17 21:29:26 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 17 21:29:26 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 17 21:29:33 dietrpi4b systemd[1]: Stopping Network Time Synchronization...
фев 17 21:29:33 dietrpi4b systemd[1]: systemd-timesyncd.service: Succeeded.
фев 17 21:29:33 dietrpi4b systemd[1]: Stopped Network Time Synchronization.
фев 17 21:29:33 dietrpi4b systemd[1]: Starting Network Time Synchronization...
фев 17 21:29:33 dietrpi4b systemd[1]: Started Network Time Synchronization.
фев 17 21:30:00 dietrpi4b systemd-timesyncd[579]: Initial synchronization to time server 195.238.84.84:123 (0.debian.pool.ntp.org).
`

Joulinar, 2022-02-17:
yes has been working during boot as expected. Strange that it is failing after some time.

g0rge0us, 2022-02-17:
Many thanks for your prompt support and kudos for the good work so far!!!

MichaIng, 2022-02-17:
> Should I reinstall both packages: systemd & systemd-timesyncd?

It doesn't hurt, but probably just keep this step in mind when the issue reappears.