MichaIng/DietPi #5285: DietPi-Software | Deluge: Configure web UI to use local backend OOTB

MichaIng, 2022-02-15:
https://dietpi.com/phpbb/viewtopic.php?t=10026

Joulinar, 2022-02-15:
not sure if this is the best way, but assuming content of `/mnt/dietpi_userdata/deluge/.config/deluge/hostlist.conf` is as follow

```
{
    "file": 3,
    "format": 1
}{
    "hosts": [
        [
            "5de6b23c343r34_DEMO_5c19f0ea16b83",
            "127.0.0.1",
            58811,
            "localclient",
            "38affe20a92bc_DEMO_9c5468a2a9c8d56131"
        ]
    ]
}
```

This might gonna print what we are looking for

```
root@DietPi4:~# grep -m1 '            ' /mnt/dietpi_userdata/deluge/.config/deluge/hostlist.conf | awk '{print $1}'
"5de6b23c343r34_DEMO_5c19f0ea16b83",
root@DietPi4:~#
```

MichaIng, 2022-02-15:
We could also use the fact that its the first string value of hexadecimal characters.
```sh
mawk -F\" '/"[0-9a-f]+",/{print $2;exit}' /mnt/dietpi_userdata/deluge/.config/deluge/hostlist.conf
```
Is it also 32 characters long in our case?

Joulinar, 2022-02-15:
ah way better than my version of the function 🤣 
Yes working as well in my case.

MichaIng, 2022-02-26:
Done with: #5316

closed 2022-02-26 by MichaIng