MichaIng/DietPi #5286: Nextcloud+Jellyfin (and other) with nginx

kd9352, 2022-02-15:
Hi all dietpi users, now I need some help securing my servers.  I have a server computer with proxmox, so I run vitrual machines.
1-I want nextcloud + jellyfin on a Virtual machine (already installed).  
2- I created a new virtual machine for my other servers (running dietpi also belongs).  
3- I want to create a new Virtual machine with ubuntu server and nginx proxy manager installed.  

How can I use nginx to run my servers with SSL and my domain name? Any help would be good. 
Thx.

Joulinar, 2022-02-15:
Why do you need an additional VM running Ubuntu + NPM? NPM could be installed on one of the already existing VM's. Anyway, just configure the hosts inside NPM pointing to your apps. In Jellyfin you would need to set a base URL. This way you would use sub path `https://your.dns.org/nextcloud` and `https://your.dns.org/jellyfin`

kd9352, 2022-02-15:
> Why do you need an additional VM running Ubuntu + NPM? NPM could be installed on one of the already existing VM's. Anyway, just configure the hosts inside NPM pointing to your apps. In Jellyfin you would need to set a base URL. This way you could use sub path `https://your.dns.org/nextcloud` and `https://your.dns.org/jellyfin`

Ahh good question, I know it's a little weird but the only reason is I want it to look good, structured. 😅☺️
So I do not need to configure dietpi-letsencrypt and dietpi-ddns on dietpi hosts at all?

Joulinar, 2022-02-15:
Certificates are managed by NPM. No need to use `dietpi-letsencrypt`. For the DDNS its depends Some IPS router are able to manage DDNS as well. If your, you are free to use `dietpi-ddns` instead.

kd9352, 2022-02-15:
> Certificates are managed by NPM. No need to use `dietpi-letsencrypt`. For the DDNS its depends Some IPS router are able to manage DDNS as well. If your, you are free to use `dietpi-ddns` instead.

I have tried to use dietpi-ddns and dietpi-lets encrypt without any nginx installation, and i could access to my nextcloud server with https://my.domain.net/nextcloud but jellyfin didnt work, but i will try that you say tomorrow and update here. 😀
What is good with NPM is that lets ecrypt updates dona automaticly.

Joulinar, 2022-02-15:
inside Jellyfin GUI you can specify a base URL `/jellyfin`. This way you can use the sub path. 

Just to avoid a misunderstanding. If you use NPM, it's managing certificates independently from `dietpi-letsencrypt` It's basically 2 tools having their own certificate data. 

> What is good with NPM is that lets ecrypt updates dona automaticly.

This is done by `dietpi-letsencrypt` as well.

---

If you already configured NextCloud, you would not need to use NPM. Because you already configured the web server on the DietPi VM directly. And on this you could simply add a Jellyfin configuration.

https://jellyfin.org/docs/general/networking/nginx.html#nginx-with-subpath-exampleorgjellyfin

MichaIng, 2022-02-15:
Jep, running two VMs sounds a bid overkill since setting up a path-based vhost on Nginx as proxy for Jellyfin is not so difficult, so you do not run multiple webservers with multiple certificates. Basically:
```sh
cat << '_EOF_' > /etc/nginx/sites-dietpi/jellyfin.conf
location /jellyfin {
    return 302 $scheme://$host/jellyfin/;
}

location /jellyfin/ {
    # Proxy main Jellyfin traffic

    # The / at the end is significant.
    # https://www.acunetix.com/blog/articles/a-fresh-look-on-reverse-proxy-related-attacks/

    proxy_pass http://127.0.0.1:8097/jellyfin/;

    proxy_pass_request_headers on;

    proxy_set_header Host $host;

    proxy_set_header X-Real-IP $remote_addr;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header X-Forwarded-Proto $scheme;
    proxy_set_header X-Forwarded-Host $http_host;

    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $http_connection;

    # Disable buffering when the nginx proxy gets very resource heavy upon streaming
    proxy_buffering off;
}
_EOF_
systemctl restart nginx
```

MichaIng, 2022-02-16:
> if i have all dietpi servers on one vm, what would be the best way to have ssl for nextcloud

`dietpi-letsencrypt` would then be an option. Since Nextcloud and proxy both are no own vhosts but accessed location/path based, both will have HTTPS enabled with the default vhosts HTTPS configuration applied by Certbot (`dietpi-letsencrypt`).

kd9352, 2022-02-16:
> `dietpi-letsencrypt` would then be an option. Since Nextcloud and proxy both are no own vhosts but accessed location/path based, both will have HTTPS enabled with the default vhost's HTTPS [...]

I have done with nextcloud by configuring the dietpi-ddns and lets-encrypt but i cant access to my jellyfin server with domain name. Should i have nginx for jellyfin? I have only found nginx conf on their website.

Joulinar, 2022-02-16:
For Jellyfin you just need to add this configuration to Nginx https://github.com/MichaIng/DietPi/issues/5286#issuecomment-1040900513

In addition set a base URL inside Jellyfin

kd9352, 2022-02-16:
> For Jellyfin you just need to add this configuration to Nginx [#5286 (comment)](https://github.com/MichaIng/DietPi/issues/5286#issuecomment-1040900513)
> 
> In addition set a base URL inside Jellyfin

Ahh yeah sorry my bad, i am now understanding the concept of nginx. Both nextcloud and jellyfin working perfectly. Now i am tryng to do same configs for filebrowser and mineos 😅👍 Just one more question is there any webui of nginx on dietpi?

MichaIng, 2022-02-16:
Great that it works 👍. That proxy config is pretty complete regarding headers, so should work for the others as well. Probably some headers could even be skipped, but they shouldn't hurt.

No, plain Nginx has no (meta) web interface. Probably there are 3rd party ones, but it would need to run as root user and hence on an own webserver, to be able to adjust Nginx configuration files, which sounds not so great. All guides I ever read about Nginx provide steps to configure its config files from console, so at least there is no widely trusted admin interface, and I wouldn't recommend anything else.

MichaIng, 2022-02-16:
Quick duck search result: https://pythonrepo.com/repo/schenkd-nginx-ui-python-administrative-interfaces
But hmm, it just allows to edit the plain config files with an indeed dedicated web interface (running on an own webserver), not a nice interface with menus, input boxes and toggles, like what you likely (at least me) had in mind. Basically useful only when there is no direct console or SSH access 😄.

Lol:
> Containerization is now state of the art and therefore the application is delivered in a container.

Whether reasonable in this particular case of not seems to be not even though about, running a trivial, probably 10 MiB RAM application (based on what it does) with a 100 MiB RAM consuming container engine and wrapped into a [140 MiB](https://hub.docker.com/r/schenkd/nginx-ui/tags) container download size (300 MiB or more installed?), what a waste of resources. Containers, a major driver of the [rebound effect](https://en.wikipedia.org/wiki/Rebound_effect_(conservation)) in IT, no because it is not an awesome concept for sensitive environments or complex systems with multiple modules, but because it is so often used for tiny trivial applications, multiplying RAM and disk usage and adding complexity layers. However I'm drifting off topic 😄.

kd9352, 2022-02-17:
> Quick duck search result: https://pythonrepo.com/repo/schenkd-nginx-ui-python-administrative-interfaces But hmm, it just allows to edit the plain config files with an indeed dedicated web interface (running on an own webserver), not a nice interface with menus, input boxes and toggles, like what you likely (at least me) had in mind. Basically useful only when there is no direct console or SSH access 😄.
> 
> Lol:
> 
> > Containerization is now state of the art and therefore the application is delivered in a container.
> 
> Whether reasonable in this particular case or not seems to be not even though about, running a trivial, probably 10 MiB RAM application (based on what it does) with a 100 MiB RAM consuming container engine and wrapped into a [140 MiB](https://hub.docker.com/r/schenkd/nginx-ui/tags) container download size (300 MiB or more installed?), what a waste of resources. Containers, a major driver of the [rebound effect](https://en.wikipedia.org/wiki/Rebound_effect_(conservation)) in IT, not because it is not an awesome concept for sensitive environments or complex systems with multiple modules, but because it is so often used for tiny trivial applications, multiplying RAM and disk usage and adding complexity layers. However I'm drifting off topic 😄.

Hahahhaha it was very conceivable information but I agree 😂.

Yeah I thought more nice website but but so far I think I can manage via the terminal dietpi.  And then I just came up with one thing, because I run proxmox and I have my domain connected to dietpi, what about proxmox, should I continue to use public ip to access my server outside the network?
Sorry if i am asking to much 😅

MichaIng, 2022-02-17:
You mean to access the host server which runs the DietPi Proxmox VM? Which kind of web application/protocol you run on it, which runs on DietPi, too? SSH? Usually a subdomain could do, not sure whether the NAT/router can forward directly based on hostname, in case of another webserver, Nginx could get another proxy vhost (a real one, based on subdomain now). Also SSH has some forwarding feature, at least it can be disabled on OpenSSH, but I never had a closer look into this.

Joulinar, 2022-02-17:
It's again overkill in terms of resource consumption (due to Docker usage). But if you are looking for a nice WebUi to manage proxy configuration + multiple certificates, you can look into Nginx Proxy Manager.

MichaIng, 2022-02-17:
At least the NPM container ships not only the proxy interface but Nginx together, and it is exactly what I had in my biased mind, being not only a web based file editor. Such would be great as standalone application to configure a dedicated plain Nginx instance.

Joulinar, 2022-02-17:
found a way to get NPM running without docker https://github.com/ej52/proxmox-scripts/blob/main/lxc/nginx-proxy-manager/install/debian.sh

Taking time to install, as stuff needs to be compiled. But it is working. Yeah. Downside, it is not following DietPi standards 😉 . Some thinks are located at `/data/`. And of course it did not except other app using 80/443 without adjusting configs. Next to this it is expecting `amd64` for `openresty`. Some tweaks are needed on the install script to get `openresty` on `arm64`.

```
root@DietPi4:/tmp# ss -tulpn | grep LISTEN
tcp   LISTEN 0      511          0.0.0.0:80        0.0.0.0:*    users:(("nginx",pid=13311,fd=6),("nginx",pid=13310,fd=6),("nginx",pid=13309,fd=6),("nginx",pid=13308,fd=6),("nginx",pid=9935,fd=6))
tcp   LISTEN 0      511          0.0.0.0:81        0.0.0.0:*    users:(("nginx",pid=13311,fd=14),("nginx",pid=13310,fd=14),("nginx",pid=13309,fd=14),("nginx",pid=13308,fd=14),("nginx",pid=9935,fd=14))
tcp   LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=369,fd=4))
tcp   LISTEN 0      511          0.0.0.0:443       0.0.0.0:*    users:(("nginx",pid=13311,fd=12),("nginx",pid=13310,fd=12),("nginx",pid=13309,fd=12),("nginx",pid=13308,fd=12),("nginx",pid=9935,fd=12))
tcp   LISTEN 0      511             [::]:80           [::]:*    users:(("nginx",pid=13311,fd=11),("nginx",pid=13310,fd=11),("nginx",pid=13309,fd=11),("nginx",pid=13308,fd=11),("nginx",pid=9935,fd=11))
tcp   LISTEN 0      511             [::]:81           [::]:*    users:(("nginx",pid=13311,fd=15),("nginx",pid=13310,fd=15),("nginx",pid=13309,fd=15),("nginx",pid=13308,fd=15),("nginx",pid=9935,fd=15))
tcp   LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=369,fd=5))
tcp   LISTEN 0      511                *:3000            *:*    users:(("node",pid=13187,fd=21))
tcp   LISTEN 0      511             [::]:443          [::]:*    users:(("nginx",pid=13311,fd=13),("nginx",pid=13310,fd=13),("nginx",pid=13309,fd=13),("nginx",pid=13308,fd=13),("nginx",pid=9935,fd=13))
```

Node require around 80MB of memory

```
  0[                              0.0%] Mem[|||                     108M/7.68G]
  1[                              0.0%] Swp[                             0K/0K]
  2[||                            1.3%] Tasks: 22; 1 running
  3[                              0.0%] Load average: 0.19 0.13 0.05
Avg[|                             0.2%] Uptime: 00:01:46
    PID USER        RES▽ NI CPU%   TIME+  Command
    526 root      81780   0  0.0  0:04.75 /usr/bin/node index.js --abort_on_unca
    534 root       9944   0  0.0  0:00.07 nginx: master process /usr/local/openr
      1 root       9284   0  0.0  0:01.30 /sbin/init
```

MichaIng, 2022-02-17:
Yeah, not sure, if NPM is from ground up designed to run within a Docker container, I think it is not a good idea to run it outside.

kd9352, 2022-03-06:
Hi guys, long time :D
I have a problem with nextcloud now, i would like get a little bit help :D
I have install openvpn on dietpi and using my domin to access it and updated with letsencrypt, after that when i am trying to access nextcloud with http://my_domain/nextcloud, its redirecting me to https and my 443 port already in use by openvpn. How can i configure nextcloud to just use only http?
My jellyfin with http://my_domain/jellyfin working fine.

MichaIng, 2022-03-06:
As long as you do not want want to access the webserver (Nextcloud/Nginx) on a non-default port, which means you'd need to define it explicitly on every client and browser access), you need to change the OpenVPN port from 443 to something different. In theory, it should be even possible to bind it to UDP only since the webserver uses TCP. It does not depend on port 443, you can use the default port 1194 instead.

Of course you could disable forced HTTPS redirect for the webserver, but I strongly recommend to not do so, especially for accessing your private data on Nextcloud, otherwise what do you want to use HTTPS for? With cert auto-updates it can only work on port 443 and OpenVPN does not use it (entirely different protocol).

kd9352, 2022-03-06:
> As long as you do not want want to access the webserver (Nextcloud/Nginx) on a non-default port, which means you'd need to define it explicitly on every client and browser access), you need to change the OpenVPN port from 443 to something different. In theory, it should be even possible to bind it to UDP only since the webserver uses TCP. OpenVPN does not depend on port 443, you can use the default port 1194 instead.
> 
> Of course you could disable forced HTTPS redirect for the webserver, but I strongly recommend to not do so, especially for accessing your private data on Nextcloud, otherwise what do you want to use HTTPS for? With cert auto-updates it can only work on port 443 and OpenVPN does not use it (entirely different protocol).

I have tried exactly with that, changed openvpn tcp port to 8555 and web server port still 943 than when i am trying to access web ui to openvpn with https://my_domain/admin i am getting 404 not found nginx error, tried with use only udp 1194 but still same error. :/

MichaIng, 2022-03-06:
> web ui to openvpn

You use the OpenVPN Access Server (which is not installed by default)? When it listens on port 943 then this shouldn't be affected at all by the webserver, HTTPS etc. But I'm not sure how this Access Server works, probably you need to make it aware (check config files) about the changed OpenVPN tunnel port.

Otherwise it should be possible to bind it to the loopback IP `127.0.0.1` and create a proxy config for it in Nginx. Surprisingly I couldn't find instructions about how to do that, e.g. whether/which headers need to be rewritten. It seems to not work with a simple proxy config: https://forums.openvpn.net/viewtopic.php?t=30316

kd9352, 2022-03-06:
> 

Yeah, when i am changing tcp port to 8555 and udp 1194 (default) and web server port is still 943 (default) i can access web ui with my local ip but not with my domain. Very strange...

MichaIng, 2022-03-06:
You need to setup port forwarding for port 943 then. Also, if not yet done, you should configure the OpenVPN Access Server to use the Let's Encrypt certificate and key for TLS.

kd9352, 2022-03-06:
> You need to setup port forwarding for port 943 then. Also, if not yet done, you should configure the OpenVPN Access Server to use the Let's Encrypt certificate and key for TLS.

I have already opened 943, TLS settings is default by openvpn and ssl certificates is uploaded. I have now tried to access openvpn web ui with my phone using mobile internet with https://example.ddns.net:943/admin aaaand i can access it, realy strange  i have to write the port number to.

MichaIng, 2022-03-06:
> realy strange i have to write the port number to.

Why strange? HTTP default port is 80 and HTTPS default port is 443, so this is what browser and HTTP(S) clients assume. If a server/service listens on any other port, it must be defined explicitly.

kd9352, 2022-03-06:
> > realy strange i have to write the port number to.
> 
> Why strange? HTTP default port is 80 and HTTPS default port is 443, so this is what browser and HTTP(S) clients assume when no port is given. If a server/service listens on any other port, it must be defined explicitly.

Okay, I did not know. I always thought if you have opened for example port 943 and you have a domain name then you can connect your server by typing just the domain name. But it was not so hahah, the reason I have domain names is that my public ip changes most often and now I have a domain from no ip and have installed their client, the domain name points to my public ip and their client updates as soon as public ipn changes. I use Openvpn to be able to access proxmox if something happens when I'm not home. I could not put the domain on proxmox via acme therefore openvpn would be suitable for it. I wanted to avoid using public ip as much as possible. Thank you so much again for helping me.

MichaIng, 2022-03-06:
> I always thought if you have opened for example port 943 and you have a domain name then you can connect your server by typing just the domain name.

Hehe, TCP/UDP/IP traffic is based on ports, this is why you explicitly need to open and forward ports on the router, so on every request, a client needs to define a port it wants to access on in the first place. Specific clients (like web browsers or SSH clients) assume the protocols default port when none is given, but otherwise it must be defined. The domain/DNS hostname on the other hand has nothing to do with the port, you can use it to access any service on any port, like SSH on port 22 or the DietPi-Dashboard on port 5252, as long as it is open and forwarded by the router. The domain It is not somehow bound to HTTPS or the ports you use to validate and renew your HTTPS certificate 😉.

kd9352, 2022-03-07:
> Hehe, TCP/UDP/IP traffic is based on ports, this is why you explicitly need to open and forward ports on the router, so on every request, a client needs to define a port it wants to access on in the [...]

Yes, a lot of things i have to learn 😂 thank you so much again for help 😊

MichaIng, 2022-04-14:
I'll mark this as closed. However, there is a lot of useful information in this issue. I'll give this a new tag for issues were we may find interesting information when required later.

closed 2022-04-14 by MichaIng