MichaIng/DietPi #5287: Mosquitto loses permission to access letsencrypt certificates (on cert update?)

DarrenSchwartz, 2022-02-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
>G_DIETPI_VERSION_CORE=8
> G_DIETPI_VERSION_SUB=1
> G_DIETPI_VERSION_RC=2
> G_GITBRANCH='master'
> G_GITOWNER='MichaIng'
> G_LIVE_PATCH_STATUS[0]='not applied'
> G_LIVE_PATCH_STATUS[1]='not applied'
> G_LIVE_PATCH_STATUS[2]='not applicable'
> G_LIVE_PATCH_STATUS[3]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
> bullseye 0
- Kernel version | `uname -a`
> Linux Rpi4 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
> RPi 4 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)
> CanaKit 3.5A Raspberry Pi 4 Power Supply (USB-C)
- SD card used | (EG: SanDisk ultra)
> sandisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
> mosquitto and letsencrypt
- Was the software title installed freshly or updated/migrated?

> Worked fine when freshly installed over a year ago. Lately, every few months clients are unable to connect to mosquitto. Re-assigning mosquitto permissions with: 
> ```
> chown -R mosquitto /etc/letsencrypt
> chgrp -R mosquitto /etc/letsencrypt
> ```
> to the letsencrypt directory fixes the problem for a few months. I can't be sure on timing, but seems like it could be coinciding with letsencrypt certificate renewal. `
- Can this issue be replicated on a fresh installation of DietPi?
> Haven't tried yet, but will if necessary.

- Bug report ID | `echo $G_HW_UUID`
> no
#### Steps to reproduce

1. install letsencrypt as usual
2. install mosquitto as usual
3. Follow these instructions to give mosquitto permission to the letsencrypt certificates. [https://github.com/nilathj/mosquitto-tls#setup-mosquitto-mqtt-broker-on-the-raspberry-pi](https://github.com/nilathj/mosquitto-tls#setup-mosquitto-mqtt-broker-on-the-raspberry-pi)

#### Expected behaviour

- Mosquitto should not lose permissions to the letsencrypt certificates

#### Actual behaviour

- Everything works and clients connect for a few months. One day clients cannot connect until the permissions are manually reset. Seems to coincide with certificate renewal (but not positive).

#### Extra details

- I had previously run mosquitto (with owntracks) for years without this issue. I can't be sure of when the problem started, but I think it was shortly after the switch to Bullseye (fresh install).

MichaIng, 2022-02-17:
Many thanks for your report.

This is basically expected since the renewal job runs as root user and creates new cert/key files. Generally to grant unprivileged users TLS certificate/key access, it is advised to create own copies for them, created via deploy hooks. Check out how we do it for MinIO: https://github.com/MichaIng/DietPi/blob/bca964e/dietpi/dietpi-letsencrypt#L309-L325

DarrenSchwartz, 2022-02-17:
Ah, understood. Thanks for that info. I was about to write my own cron for this, but should a more permanent fix be included with the mosquitto install? I'd offer to give it a try, but I'm just a hacker with shell scripting.
btw, DietPi is so awesome.

Joulinar, 2022-02-17:
I guess better to have it inside our online docs. Instead of creating the hook automatically.

MichaIng, 2022-02-17:
Agreed, for now a nice docs HowTo should be fine. If we think about implementing further applications into DietPi-LetsEncrypt, then IMHO we need a toggles to enable/disable for each application, as HTTPS may not be needed or wanted everywhere, e.g. for purely LAN or localhost applications.

closed 2022-04-14 by MichaIng