MichaIng/DietPi #5289: Cannot update DietPi from v8.0.2 to v8.1.2

oCtodur, 2022-02-17:
#### Details:
- Date           | Thu Feb 17 09:06:38 GMT 2022
- Bug report     | 502c43b0-b1f5-4ed2-bc66-e10eaa55720b
- DietPi version | v8.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Run dietpi-update
2. Wait, and error occurs
#### Expected behaviour:

- Update should be able to complete
#### Actual behaviour:

- It just borkes at this step. Been trying this for some time now, but always getting this error
#### Extra details:

root@DietPi:~# uname -a
Linux DietPi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux

root@DietPi:~# dietpi-update 

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.1.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.0.2
[ INFO ] DietPi-Update | Latest version  : v8.1.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=54276 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Get:2 https://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:3 https://deb.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:4 https://deb.debian.org/debian bullseye-backports InRelease [44.2 kB]
Get:5 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]
Get:6 https://deb.debian.org/debian-security bullseye-security/main arm64 Packages [116 kB]
Get:7 https://deb.debian.org/debian bullseye-backports/non-free arm64 Packages.diff/Index [7239 B]
Get:8 https://deb.debian.org/debian bullseye-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:9 https://deb.debian.org/debian bullseye-backports/contrib arm64 Packages.diff/Index [14.0 kB]
Get:10 https://deb.debian.org/debian bullseye-backports/non-free arm64 Packages T-2022-02-09-2007.33-F-2022-02-09-2007.33.pdiff [3755 B]
Get:10 https://deb.debian.org/debian bullseye-backports/non-free arm64 Packages T-2022-02-09-2007.33-F-2022-02-09-2007.33.pdiff [3755 B]
Get:11 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-02-17-0201.53-F-2022-02-08-0803.45.pdiff [34.4 kB]
Get:12 https://deb.debian.org/debian bullseye-backports/contrib arm64 Packages T-2022-02-16-2006.49-F-2022-02-16-2006.49.pdiff [477 B]
Get:11 https://deb.debian.org/debian bullseye-backports/main arm64 Packages T-2022-02-17-0201.53-F-2022-02-08-0803.45.pdiff [34.4 kB]
Get:12 https://deb.debian.org/debian bullseye-backports/contrib arm64 Packages T-2022-02-16-2006.49-F-2022-02-16-2006.49.pdiff [477 B]
Get:13 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages [251 kB]
Fetched 641 kB in 4s (166 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 17536 files and directories currently installed.)
Preparing to unpack .../bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb ...
Unpacking bsdutils (1:2.36.1-8+deb11u1) over (1:2.36.1-8) ...
dpkg: error processing archive /var/cache/apt/archives/bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb (--unpack):
 unable to stat './usr/share/doc/bsdutils/changelog.Debian.gz' (which was about to be installed): Bad message
Errors were encountered while processing:
 /var/cache/apt/archives/bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 502c43b0-b1f5-4ed2-bc66-e10eaa55720b
#### Additional logs:
```
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 17536 files and directories currently installed.)
Preparing to unpack .../bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb ...
Unpacking bsdutils (1:2.36.1-8+deb11u1) over (1:2.36.1-8) ...
dpkg: error processing archive /var/cache/apt/archives/bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb (--unpack):
 unable to stat './usr/share/doc/bsdutils/changelog.Debian.gz' (which was about to be installed): Bad message
Errors were encountered while processing:
 /var/cache/apt/archives/bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2022-02-17:
Hi,

could you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update && apt upgrade -y
```

oCtodur, 2022-02-17:
Heya! Thanks for the fast answer :) Here are the results, it still fails =/

root@DietPi:~# /boot/dietpi/func/dietpi-set_software apt-cache clean
apt update && apt upgrade -y

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
Get:1 https://deb.debian.org/debian bullseye InRelease [116 kB]
Get:2 https://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]                
Get:3 https://deb.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:4 https://deb.debian.org/debian bullseye-backports InRelease [44.2 kB]
Get:5 https://archive.raspberrypi.org/debian bullseye InRelease [23.5 kB]             
Get:6 https://deb.debian.org/debian bullseye/contrib arm64 Packages [40.8 kB]
Get:7 https://deb.debian.org/debian bullseye/main arm64 Packages [8070 kB]
Get:8 https://deb.debian.org/debian bullseye/non-free arm64 Packages [69.6 kB]
Get:9 https://deb.debian.org/debian bullseye-updates/main arm64 Packages [2596 B]
Get:10 https://deb.debian.org/debian-security bullseye-security/main arm64 Packages [116 kB]
Get:11 https://deb.debian.org/debian bullseye-backports/contrib arm64 Packages [4240 B]
Get:12 https://deb.debian.org/debian bullseye-backports/non-free arm64 Packages [6564 B]
Get:13 https://deb.debian.org/debian bullseye-backports/main arm64 Packages [203 kB]
Get:14 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages [251 kB]
Fetched 9030 kB in 4s (2340 kB/s)                          
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
17 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bsdutils fdisk firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek libblkid1 libcryptsetup12 libexpat1 libfdisk1 libmount1 libsmartcols1 libuuid1 mount rpi-eeprom util-linux
17 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 43.4 MB of archives.
After this operation, 2636 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian-security bullseye-security/main arm64 bsdutils arm64 1:2.36.1-8+deb11u1 [145 kB]
Get:2 https://deb.debian.org/debian-security bullseye-security/main arm64 libblkid1 arm64 2.36.1-8+deb11u1 [187 kB]
Get:3 https://deb.debian.org/debian-security bullseye-security/main arm64 libuuid1 arm64 2.36.1-8+deb11u1 [83.4 kB]
Get:4 https://deb.debian.org/debian-security bullseye-security/main arm64 libfdisk1 arm64 2.36.1-8+deb11u1 [224 kB]
Get:5 https://deb.debian.org/debian-security bullseye-security/main arm64 libmount1 arm64 2.36.1-8+deb11u1 [203 kB]
Get:6 https://deb.debian.org/debian-security bullseye-security/main arm64 libsmartcols1 arm64 2.36.1-8+deb11u1 [151 kB]
Get:7 https://deb.debian.org/debian-security bullseye-security/main arm64 fdisk arm64 2.36.1-8+deb11u1 [186 kB]
Get:8 https://deb.debian.org/debian-security bullseye-security/main arm64 util-linux arm64 2.36.1-8+deb11u1 [1097 kB]
Get:9 https://deb.debian.org/debian-security bullseye-security/main arm64 mount arm64 2.36.1-8+deb11u1 [182 kB]
Get:10 https://deb.debian.org/debian-security bullseye-security/main arm64 libcryptsetup12 arm64 2:2.3.7-1+deb11u1 [230 kB]
Get:11 https://deb.debian.org/debian-security bullseye-security/main arm64 libexpat1 arm64 2.2.10-2+deb11u1 [84.8 kB]
Get:12 https://archive.raspberrypi.org/debian bullseye/main arm64 firmware-atheros all 1:20210315-3+rpt5 [10.0 MB]
Get:13 https://archive.raspberrypi.org/debian bullseye/main arm64 firmware-brcm80211 all 1:20210315-3+rpt5 [5787 kB]
Get:14 https://archive.raspberrypi.org/debian bullseye/main arm64 firmware-iwlwifi all 1:20210315-3+rpt5 [13.4 MB]
Get:15 https://archive.raspberrypi.org/debian bullseye/main arm64 firmware-misc-nonfree all 1:20210315-3+rpt5 [8953 kB]
Get:16 https://archive.raspberrypi.org/debian bullseye/main arm64 firmware-realtek all 1:20210315-3+rpt5 [922 kB]
Get:17 https://archive.raspberrypi.org/debian bullseye/main arm64 rpi-eeprom arm64 13.8-1 [1529 kB]
Fetched 43.4 MB in 3s (16.6 MB/s)     
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 17536 files and directories currently installed.)
Preparing to unpack .../bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb ...
Unpacking bsdutils (1:2.36.1-8+deb11u1) over (1:2.36.1-8) ...
dpkg: error processing archive /var/cache/apt/archives/bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb (--unpack):
 unable to stat './usr/share/doc/bsdutils/changelog.Debian.gz' (which was about to be installed): Bad message
Errors were encountered while processing:
 /var/cache/apt/archives/bsdutils_1%3a2.36.1-8+deb11u1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~#

oCtodur, 2022-02-17:
ooops, there are a few there, hmmm

root@DietPi:~# dmesg -l emerg,alert,crit,err
[    0.580864] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    6.979710] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.377243] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.378469] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.396199] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[1379392.369041] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9364: comm rsync: iget: bad extra_isize 288 (inode size 256)
[1379392.386233] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9366: comm rsync: iget: checksum invalid
[1379392.401658] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9362: comm rsync: iget: checksum invalid
[1379392.417901] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9376: comm rsync: iget: checksum invalid
[1379392.433447] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9372: comm rsync: iget: checksum invalid
[1379392.592227] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9370: comm rsync: iget: checksum invalid
[1379392.607733] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9368: comm rsync: iget: checksum invalid
[1379392.623131] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9374: comm rsync: iget: checksum invalid
[1379392.643449] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9422: comm rsync: iget: checksum invalid
[1379392.659006] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9420: comm rsync: iget: checksum invalid
[1379402.330858] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9364: comm rsync: iget: bad extra_isize 288 (inode size 256)
[1379402.348502] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9366: comm rsync: iget: checksum invalid
[1379402.491533] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9362: comm rsync: iget: checksum invalid
[1379402.509835] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9376: comm rsync: iget: checksum invalid
[1379402.525348] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9372: comm rsync: iget: checksum invalid
[1379402.540805] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9370: comm rsync: iget: checksum invalid
[1379402.556329] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9368: comm rsync: iget: checksum invalid
[1379402.571740] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9374: comm rsync: iget: checksum invalid
[1379402.723912] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9422: comm rsync: iget: checksum invalid
[1379402.739536] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9420: comm rsync: iget: checksum invalid
[1379534.246344] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9365: comm rsync: iget: checksum invalid
[1379534.270179] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9413: comm rsync: iget: checksum invalid
[1379534.289674] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9457: comm rsync: iget: checksum invalid
[1379534.309264] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9459: comm rsync: iget: checksum invalid
[1379534.455167] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9375: comm rsync: iget: checksum invalid
[1379534.474668] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9423: comm rsync: iget: checksum invalid
[1379534.493817] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9463: comm rsync: iget: bad extra_isize 8224 (inode size 256)
[1379534.515732] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9419: comm rsync: iget: checksum invalid
[1379534.534987] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9409: comm rsync: iget: checksum invalid
[1379534.554401] EXT4-fs error (device mmcblk0p2): ext4_lookup:1706: inode #9471: comm rsync: iget: checksum invalid
[1379544.157936] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #6451: comm rsync: Directory block failed checksum
[1379545.329445] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #6463: comm rsync: Directory block failed checksum
[1379720.166244] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #4858: comm rsync: Directory block failed checksum
[1379720.752221] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #4859: comm rsync: Directory block failed checksum
[1379720.768616] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #4863: comm rsync: Directory block failed checksum
[1379720.837868] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #4941: comm rsync: Directory block failed checksum
[1379721.798231] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #5674: comm rsync: Directory block failed checksum
[1379721.820167] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #5677: comm rsync: Directory block failed checksum
[1379721.842089] EXT4-fs error (device mmcblk0p2): htree_dirblock_to_tree:1003: inode #5678: comm rsync: Directory block failed checksum
[... cut ...]
[1395616.016158] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1547: inode #6451: comm dpkg: checksumming directory block 0
[1413216.883703] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1547: inode #6451: comm dpkg: checksumming directory block 0
[2224378.208580] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1547: inode #6451: comm dpkg: checksumming directory block 0
[2225494.866416] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1547: inode #6451: comm dpkg: checksumming directory block 0

MichaIng, 2022-02-17:
That will be the reason. Try to fix them, e.g. on reboot:
```sh
> /forcefsck
reboot
```
Or if you have another Linux system you can attach the SD card to, checking and repairing it interactively there would be the best option.

Joulinar, 2022-02-17:
could you share

```
df -h
ls -la /run/dietpi
```

MichaIng, 2022-02-17:
Some more seems to be gone:
```sh
journalctl -u systemd-tmpfiles-setup
ls -l /run/dietpi
curl -V
```

oCtodur, 2022-02-17:
root@DietPi:~# df -h
ls -la /run/dietpi
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        57G  1.6G   54G   3% /
devtmpfs        3.9G     0  3.9G   0% /dev
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           1.6G  8.5M  1.6G   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.9G     0  3.9G   0% /tmp
tmpfs            50M  8.0K   50M   1% /var/log
/dev/mmcblk0p1  127M   32M   95M  25% /boot
ls: cannot access '/run/dietpi': No such file or directory
root@DietPi:~# 

root@DietPi:~# journalctl -u systemd-tmpfiles-setup
ls -l /run/dietpi
curl -V
-- Journal begins at Thu 2022-02-17 15:49:07 GMT, ends at Thu 2022-02-17 15:57:14 GMT. --
Feb 17 15:49:09 DietPi systemd[1]: Starting Create Volatile Files and Directories...
Feb 17 15:49:09 DietPi systemd[1]: Finished Create Volatile Files and Directories.
ls: cannot access '/run/dietpi': No such file or directory
-bash: /usr/bin/curl: cannot execute binary file: Exec format error
root@DietPi:~# 

Got a bad feeling about this one =/

oCtodur, 2022-02-17:
root@DietPi:~# cat /etc/tmpfiles.d/dietpi.conf
d /run/dietpk 0777

root@DietPi:~# apt update
apt install --reinstall curl
curl -V
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done                             
Building dependency tree... Done
Reading state information... Done
17 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 17 not upgraded.
Need to get 261 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 curl arm64 7.74.0-1.3+deb11u1 [261 kB]
Fetched 261 kB in 0s (2991 kB/s)
Illegal instruction
(Reading database ... 17536 files and directories currently installed.)
Preparing to unpack .../curl_7.74.0-1.3+deb11u1_arm64.deb ...
Unpacking curl (7.74.0-1.3+deb11u1) over (7.74.0-1.3+deb11u1) ...
Setting up curl (7.74.0-1.3+deb11u1) ...
curl: error while loading shared libraries: /lib/aarch64-linux-gnu/libsasl2.so.2: invalid ELF header
root@DietPi:~# 

no go =(

MichaIng, 2022-02-17:
```sh
apt install --reinstall libsasl2-2
```

oCtodur, 2022-02-17:
Mastermind! So this got curl re-installed again when I ran it and now I get a prompt to update to 8.1.2.

MichaIng, 2022-02-17:
Above your paste was:
```
root@DietPi:~# cat /etc/tmpfiles.d/dietpi.conf
d /run/dietpk 0777
```
Is it really `dietpk` with an `k` at the end? Would be quite interesting result of a filesystem corruption or write abortion or something like that 😄. To fix:
```sh
echo 'd /run/dietpi 0777' > /etc/tmpfiles.d/dietpi.conf
mkdir -m 0777 /run/dietpi
```

oCtodur, 2022-02-17:
Hmm! I did a few restarts of the Pi and ran it again and got this:
```
root@DietPi:~# cat /etc/tmpfiles.d/dietpi.conf
d /run/dietpi 0777
root@DietPi:~# 
```
but it must have been with an K in the end as I was just copy pasting directly in here :)

I'm running the update now to see if that works, then I'll check why the dns blocker isn't up and running after that :) Again many thanks for this! The same thing happened when I upgraded from a previous version, but I just ended up reinstalling the OS again. All in all, it was the SD card that had gone loco or something else? I haven't really done much to this new install, just installed Ad Guard and that was it :)

While running the update, I got this for the first time
[image]
I do have a wide screen, but it's the first time I see this :)

oCtodur, 2022-02-17:
Naw I can't go past it, I mean I can hit the button no worries, but it just pops back up again. Had to end the terminal and reconnect.

root@DietPi:~# sed -h
Segmentation fault

MichaIng, 2022-02-17:
```sh
apt install --reinstall sed
sed -h
```

oCtodur, 2022-02-17:
never heard of this package :) Reinstalled and trying to update again :)

oCtodur, 2022-02-17:
Awesome! Updated, saw that ad guard wasn't starting and giving some errors, rebooted and we're back online! 

Thank you MichaIng for the help on this and Joulinar!

A few things learned here so big ups for that too :) I'll close this out with a happy ending :)

closed 2022-02-17 by oCtodur

Joulinar, 2022-02-17:
keep in mind that you had quite some data corruption. Means no grantee on your system and there might be more that is not working 😉

MichaIng, 2022-02-17:
Yes, feel free to ask here when you face more errors, since APT and dpkg work fine, it should be possible to fix everything by reinstalling related packages.

oCtodur, 2022-02-18:
Thanks both of you! Yeah there was some corruption indeed, not sure why because not much has been done on this install since it was reinstalled 1-2 months ago. I see some weird things in apt though so I'll monitor if anything else comes up.

`root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  rpi-eeprom
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 1593 kB of archives.
After this operation, 522 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://archive.raspberrypi.org/debian bullseye/main arm64 rpi-eeprom arm64 13.9-1 [1593 kB]
Fetched 1593 kB in 1s (2979 kB/s)  
**Illegal instruction** <----------------- ???
(Reading database ... 17541 files and directories currently installed.)
Preparing to unpack .../rpi-eeprom_13.9-1_arm64.deb ...
Unpacking rpi-eeprom (13.9-1) over (13.8-1) ...
Setting up rpi-eeprom (13.9-1) ...
root@DietPi:~# 
`

MichaIng, 2022-02-18:
Hmm not sure which command/step this "Illegal instruction" is related to 🤔.

oCtodur, 2022-02-20:
yeah its very weird, I saw it happen on all apt things when I upgraded. I'll monitor and see if I can figure out where it's coming from. Thanks again for the support on this!

Joulinar, 2022-02-20:
Maybe apt to be reinstalled?

`apt install --reinstall apt`

oCtodur, 2022-02-20:
Hmm, yeah good idea! I'll give that a try now, thanks!

oCtodur, 2022-02-20:
Funny:

```
root@DietPi:~# apt install --reinstall apt

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 1433 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 apt arm64 2.2.4 [1433 kB]
Fetched 1433 kB in 0s (9826 kB/s)
Illegal instruction
(Reading database ... 17542 files and directories currently installed.)
Preparing to unpack .../archives/apt_2.2.4_arm64.deb ...
Unpacking apt (2.2.4) over (2.2.4) ...
Setting up apt (2.2.4) ...
Illegal instruction <-------------------------------------------------------------------------- happened again
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
root@DietPi:~# 
```

After reinstall I installed lynx and still got the message:

```
root@DietPi:~# apt install lynx
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libidn11 lynx-common
Recommended packages:
  mime-support
The following NEW packages will be installed:
  libidn11 lynx lynx-common
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 1901 kB of archives.
After this operation, 6113 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://deb.debian.org/debian bullseye/main arm64 libidn11 arm64 1.33-3 [115 kB]
Get:2 https://deb.debian.org/debian bullseye/main arm64 lynx-common all 2.9.0dev.6-3~deb11u1 [1192 kB]
Get:3 https://deb.debian.org/debian bullseye/main arm64 lynx arm64 2.9.0dev.6-3~deb11u1 [594 kB]
Fetched 1901 kB in 0s (10.4 MB/s)
Illegal instruction <-------------------------------------------------------------------------- here
Selecting previously unselected package libidn11:arm64.
(Reading database ... 17542 files and directories currently installed.)
Preparing to unpack .../libidn11_1.33-3_arm64.deb ...
Unpacking libidn11:arm64 (1.33-3) ...
Selecting previously unselected package lynx-common.
Preparing to unpack .../lynx-common_2.9.0dev.6-3~deb11u1_all.deb ...
Unpacking lynx-common (2.9.0dev.6-3~deb11u1) ...
Selecting previously unselected package lynx.
Preparing to unpack .../lynx_2.9.0dev.6-3~deb11u1_arm64.deb ...
Unpacking lynx (2.9.0dev.6-3~deb11u1) ...
Setting up libidn11:arm64 (1.33-3) ...
Setting up lynx-common (2.9.0dev.6-3~deb11u1) ...
Setting up lynx (2.9.0dev.6-3~deb11u1) ...
update-alternatives: using /usr/bin/lynx to provide /usr/bin/www-browser (www-browser) in auto mode
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
root@DietPi:~# 
```

MichaIng, 2022-02-20:
Might be an issue with an underlying library or dpkg:
```sh
apt install --reinstall libapt-pkg6.0 dpkg
```

oCtodur, 2022-02-20:
Hmm tried it, removed lynx and reinstalled it and still getting: Illegal instruction

MichaIng, 2022-02-20:
Hmm some more to try:
```sh
apt install --reinstall gpgv libgpg-error0 libc6 zlib1g libgcc-s1 libgnutls30 libseccomp2 'libstdc++6' libsystemd0 liblzma5 libgcrypt20
```

oCtodur, 2022-02-20:
Hmmm, something here:

```
root@DietPi:~# apt install --reinstall gpgv libgpg-error0 libc6 zlib1g libgcc-s1 libgnutls30 libseccomp2 'libstdc++6' libsystemd0 liblzma5 libgcrypt20
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 11 reinstalled, 0 to remove and 0 not upgraded.
Need to get 6039 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 libsystemd0 arm64 247.3-6 [354 kB]
Get:2 https://deb.debian.org/debian bullseye/main arm64 gpgv arm64 2.2.27-2 [610 kB]   
Get:3 https://deb.debian.org/debian bullseye/main arm64 libgcc-s1 arm64 10.2.1-6 [34.6 kB]
Get:4 https://deb.debian.org/debian bullseye/main arm64 libgcrypt20 arm64 1.8.7-6 [494 kB]
Get:5 https://deb.debian.org/debian bullseye/main arm64 libgnutls30 arm64 3.7.1-5 [1262 kB]
Get:6 https://deb.debian.org/debian bullseye/main arm64 libgpg-error0 arm64 1.38-2 [74.6 kB]
Get:7 https://deb.debian.org/debian bullseye/main arm64 liblzma5 arm64 5.2.5-2 [165 kB]
Get:8 https://deb.debian.org/debian bullseye/main arm64 libseccomp2 arm64 2.5.1-1+deb11u1 [47.4 kB]
Get:9 https://deb.debian.org/debian bullseye/main arm64 libstdc++6 arm64 10.2.1-6 [451 kB]
Get:10 https://deb.debian.org/debian bullseye/main arm64 zlib1g arm64 1:1.2.11.dfsg-2 [87.9 kB]
Get:11 https://archive.raspberrypi.org/debian bullseye/main arm64 libc6 arm64 2.31-13+rpt2+rpi1+deb11u2 [2459 kB]
Fetched 6039 kB in 1s (7941 kB/s)
Illegal instruction
(Reading database ... 17659 files and directories currently installed.)
Preparing to unpack .../libc6_2.31-13+rpt2+rpi1+deb11u2_arm64.deb ...
Illegal instruction
dpkg: error processing archive /var/cache/apt/archives/libc6_2.31-13+rpt2+rpi1+deb11u2_arm64.deb (--unpack):
 new libc6:arm64 package pre-installation script subprocess was killed by signal (Illegal instruction)
Preparing to unpack .../libgcc-s1_10.2.1-6_arm64.deb ...
Unpacking libgcc-s1:arm64 (10.2.1-6) over (10.2.1-6) ...
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.31-13+rpt2+rpi1+deb11u2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2022-02-20:
Please try:
```sh
export LC_ALL=C
export LC_ALL=C.UTF-8
sed 's/test/success/' <<< 'test'
dpkg --compare-versions 2.3 lt 2.2
/usr/share/debconf/confmodule
```

oCtodur, 2022-02-20:
Like this?

```
root@DietPi:~# export LC_ALL=C
root@DietPi:~# export LC_ALL=C.UTF-8
root@DietPi:~# sed 's/test/success/' <<< 'test'
success
root@DietPi:~# dpkg --compare-versions 2.3 lt 2.2
/usr/share/debconf/confmoduleroot@DietPi:~# /usr/share/debconf/confmodule
-bash: /usr/share/debconf/confmodule: Permission denied
root@DietPi:~# 
```

MichaIng, 2022-02-20:
```sh
bash /usr/share/debconf/confmodule
```

oCtodur, 2022-02-20:
Haha wow! Looks like you found it!

```
root@DietPi:~# bash /usr/share/debconf/confmodule
Illegal instruction
root@DietPi:~# 
```

MichaIng, 2022-02-20:
Note yet, this is only a shell script 😄.
```sh
echo test | debconf-escape -u
```
If this does not trigger the error:
```sh
apt install --reinstall debconf
bash /usr/share/debconf/confmodule
```

oCtodur, 2022-02-21:
Took the win out in advance =)

So the first command runs well, it echoed test in the terminal so I ran

```
root@DietPi:~# apt install --reinstall debconf
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for debconf:arm64
root@DietPi:~# 
```
And we got error

MichaIng, 2022-02-21:
Weird. Let's try to clean everything:
```sh
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt install --reinstall debconf
# if it still fails
dpkg -i /var/cache/apt/archives/*.deb
dpkg --configure -a
```

oCtodur, 2022-02-21:
So the first part failed on the --reinstall part so I ran:

```
root@DietPi:~# dpkg -i /var/cache/apt/archives/*.deb
dpkg: error: cannot access archive '/var/cache/apt/archives/*.deb': No such file or directory
root@DietPi:~# 
```

MichaIng, 2022-02-21:
`lock` is expected, empty `partial` as well when all APT downloads have finished. What is unexpected is that there is no `debconf` package, since APT should only delete packages when their installation was successful. Is there a hint that it as even downloaded in the `apt install --reinstall debconf` output? Or course we can download and install it manually:
```sh
cd /tmp
curl -LO 'https://deb.debian.org/debian/pool/main/d/debconf/debconf_1.5.77_all.deb'
dpkg -i debconf_1.5.77_all.deb
rm debconf_1.5.77_all.deb
```

MichaIng, 2022-02-21:
Does this produce the illegal instruction error?
```sh
echo test | debconf-escape -u
```
There are no other external commands called by `debconf.postinst` 🤔.

We could try to reinstall a bunch of further packages, but at least I have no specific idea which one could be related, apart of those we tried already. So probably its indeed better to restart with a fresh image. With the current one you would be never sure whether anything succeeded and did what you want or not.

oCtodur, 2022-02-21:
That doesn't proceed an Illegal Instruction, gives test as the result.

Yeah I think I might just start with a new image in this case. On this last install I don't really have much sacred on the disk so probably the best way to go about it. Big thanks to you both for assisting me on this! Love the support from DietPi which is and will be the one and only distro I'll ever have on my Pi :) <3