MichaIng/DietPi #5292: armbianEnv.txt - quotes in extraargs cause kernel to ignore parameters

belveder79, 2022-02-18:
- DietPi version 
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'
````
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux
`
- SBC model | `ROCKPro64 (aarch64)`
- Power supply used | `12V 5A`
- SD card used | `SanDisk ultra 32GB`

#### Additional Information (if applicable)
- `armbianEnv.txt` file - probably this is an Armbian bug itself, however, stock "extraargs" in (at least) latest image come with quotes, but those are ignored in the kernel arguments list - there should be no quotes, then it works.
```
root@DietPi:~# cat /boot/armbianEnv.txt 
verbosity=4
bootlogo=false
overlay_prefix=rockchip
rootdev=UUID=16fd56b1-1af3-4832-b584-e15c71d16b10
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
extraargs="net.ifnames=0"
docker_optimizations=off
```

#### Steps to reproduce

1. change armbianEnv.txt to
```
verbosity=4
bootlogo=false
overlay_prefix=rockchip
rootdev=UUID=16fd56b1-1af3-4832-b584-e15c71d16b10
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
extraargs="net.ifnames=0 memtest=1"
docker_optimizations=off
```

#### Expected behaviour
- Should run a memtest, however the argument in quotes is ignored, same for e.g. mem=512M that would limit the available RAM to 512MB for testing purposes;

#### Actual behaviour
- ignores parameters

#### Extra details
- it's likely not even a DietPi bug, but rather an Armbian issue...
- removing quote, eg.
```
verbosity=4
bootlogo=false
overlay_prefix=rockchip
rootdev=UUID=16fd56b1-1af3-4832-b584-e15c71d16b10
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
extraargs=net.ifnames=0 mem=512M memtest=1
docker_optimizations=off
```
works just fine...

MichaIng, 2022-02-18:
Many thanks for your report. Issue is solved already with next release: https://github.com/MichaIng/DietPi/blob/6c71f5c/.update/patches#L626-L627

MichaIng, 2022-02-19:
Marking this as closed. Feel free to reopen when required.

closed 2022-02-19 by MichaIng