MichaIng/DietPi #5293: Youtube-dl not downloading

spacedude609, 2022-02-18:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
- G_DIETPI_VERSION_CORE=8
- G_DIETPI_VERSION_SUB=1
- G_DIETPI_VERSION_RC=2
- G_GITBRANCH=MASTER
- G_GITOWNER=MichaIng
- G_LIVE_PATCH_STATUS[0]='applied'
- G_LIVE_PATCH_STATUS[1]='applied'
- (other patches not applicable)
- Distro version | `0 (based off of bullseye)`
- Kernel version | `5.10.63-v7l+`
- SBC model | `Raspberry Pi 4 Model B 2 Gb armv7l`
- Power supply used | I don't know (It works tho)
- SD card used | `Generic Multi-Card (etcher calls it that)`

#### Additional Information (if applicable)
- Software title | youtube-dl
- Was the software title installed freshly or updated/migrated? `installed freshly`
- Can this issue be replicated on a fresh installation of DietPi? `yes`

- Bug report ID | `9ebb38f3-62ec-430a-b649-6fc0e162a388`

#### Steps to reproduce

1. Select youtube-dl when installing softeare for the first time
2. Wait for it to give an error while checking the download url

#### Expected behaviour

- youtube-dl should install

#### Actual behaviour

- It cant check the url

#### Extra details

- I am doing this over ssh, wifi card definitly works

Joulinar, 2022-02-18:
Probably you like to share the error message as well!

spacedude609, 2022-02-18:
[eror.txt](https://github.com/MichaIng/DietPi/files/8098359/eror.txt)

spacedude609, 2022-02-18:
its really long

spacedude609, 2022-02-18:
did that help?

spacedude609, 2022-02-18:
Ok but I have to finish setup first

Joulinar, 2022-02-18:
you could open another session via SSH

spacedude609, 2022-02-18:
yeah but it says setup is running

spacedude609, 2022-02-18:
Im setting it up over ssh

spacedude609, 2022-02-18:
oh, ok but it's almost done

spacedude609, 2022-02-18:
and setup is done btw

Joulinar, 2022-02-18:
more than 5 sec to check a URL is extremely long. Usually this check should be done below 1 sec. Probably a timeout value on the `yt-dl.org` as we set 10 sec by default.

spacedude609, 2022-02-18:
output:

```
HTTP/2 302
date: Fri, 18 Feb 2022 16:02:09 GMT
content-type: text/html; charset=iso-8859-1
server: nginx
location: https://yt-dl.org/downloads/2021.12.17/youtube-dl
x-xss-protection: 1; mode=block
x-frame-options: SAMEORIGIN
strict-transport-security: max-age=31536000
x-content-type-options: nosniff
referrer-policy: strict-origin-when-cross-origin

HTTP/2 302
date: Fri, 18 Feb 2022 16:02:09 GMT
content-type: text/html; charset=iso-8859-1
server: nginx
location: https://github.com/ytdl-org/youtube-dl/releases/download/2021.12.17/youtube-dl
x-xss-protection: 1; mode=block
x-frame-options: SAMEORIGIN
strict-transport-security: max-age=31536000
x-content-type-options: nosniff
referrer-policy: strict-origin-when-cross-origin

HTTP/2 302
server: GitHub.com
date: Fri, 18 Feb 2022 16:02:15 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/1039520/65476747-919e-4c56-a57f-ded64bed461f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220218%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220218T160215Z&X-Amz-Expires=300&X-Amz-Signature=b164335865645b0c4b1f3514143b88c79528a3c08c0e3079d242cf06e9d4d8d4&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=1039520&response-content-disposition=attachment%3B%20filename%3Dyoutube-dl&response-content-type=application%2Foctet-stream
cache-control: no-cache
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com collector.github.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations raw.githubusercontent.com github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com collector.github.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
x-github-request-id: D3AA:353C:BEFF7C:1720172:620FC306

HTTP/2 200
content-type: application/octet-stream
content-md5: Lc8h+5dJEGrd4TTxGUFAbw==
last-modified: Thu, 16 Dec 2021 19:01:11 GMT
etag: "0x8D9C0C66D350675"
server: Windows-Azure-Blob/1.0 Microsoft-HTTPAPI/2.0
x-ms-request-id: ceb1f4f8-001e-0056-1ae0-24d38a000000
x-ms-version: 2020-04-08
x-ms-creation-time: Thu, 16 Dec 2021 19:01:11 GMT
x-ms-lease-status: unlocked
x-ms-lease-state: available
x-ms-blob-type: BlockBlob
content-disposition: attachment; filename=youtube-dl
x-ms-server-encrypted: true
fastly-restarts: 1
accept-ranges: bytes
date: Fri, 18 Feb 2022 16:02:20 GMT
via: 1.1 varnish
age: 138
x-served-by: cache-yul12822-YUL
x-cache: HIT
x-cache-hits: 2
x-timer: S1645200140.226692,VS0,VE26
content-length: 1837828

real    0m16.152s
user    0m0.271s
sys     0m0.031s
```

Joulinar, 2022-02-18:
puhh 16 sec. This is somehow worse. Is this working

```
time curl -ILfvm 20 https://yt-dl.org/downloads/latest/youtube-dl
```

MichaIng, 2022-02-18:
The reason is, without `-L` the redirects were not followed, so 5.x seconds was the time only to connect to `https://yt-dl.org/downloads/latest/youtube-dl` without any redirection or download. That is a very long time 🤔.

spacedude609, 2022-02-18:
output:
```
*   Trying 95.143.172.170:443...
* Connected to yt-dl.org (95.143.172.170) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: CN=yt-dl.org
*  start date: Feb 16 21:38:23 2022 GMT
*  expire date: May 17 21:38:22 2022 GMT
*  subjectAltName: host "yt-dl.org" matched cert's "yt-dl.org"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0xe05d20)
> HEAD /downloads/latest/youtube-dl HTTP/2
> Host: yt-dl.org
> user-agent: curl/7.74.0
> accept: */*
>
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 128)!
< HTTP/2 302
HTTP/2 302
< date: Fri, 18 Feb 2022 16:05:02 GMT
date: Fri, 18 Feb 2022 16:05:02 GMT
< content-type: text/html; charset=iso-8859-1
content-type: text/html; charset=iso-8859-1
< server: nginx
server: nginx
< location: https://yt-dl.org/downloads/2021.12.17/youtube-dl
location: https://yt-dl.org/downloads/2021.12.17/youtube-dl
< x-xss-protection: 1; mode=block
x-xss-protection: 1; mode=block
< x-frame-options: SAMEORIGIN
x-frame-options: SAMEORIGIN
< strict-transport-security: max-age=31536000
strict-transport-security: max-age=31536000
< x-content-type-options: nosniff
x-content-type-options: nosniff
< referrer-policy: strict-origin-when-cross-origin
referrer-policy: strict-origin-when-cross-origin

<
* Connection #0 to host yt-dl.org left intact
* Issue another request to this URL: 'https://yt-dl.org/downloads/2021.12.17/youtube-dl'
* Found bundle for host yt-dl.org: 0xe03550 [can multiplex]
* Re-using existing connection! (#0) with host yt-dl.org
* Connected to yt-dl.org (95.143.172.170) port 443 (#0)
* Using Stream ID: 3 (easy handle 0xe05d20)
> HEAD /downloads/2021.12.17/youtube-dl HTTP/2
[... 170 lines cut ...]
accept-ranges: bytes
< date: Fri, 18 Feb 2022 16:05:07 GMT
date: Fri, 18 Feb 2022 16:05:07 GMT
< via: 1.1 varnish
via: 1.1 varnish
< age: 0
age: 0
< x-served-by: cache-yul12828-YUL
x-served-by: cache-yul12828-YUL
< x-cache: MISS
x-cache: MISS
< x-cache-hits: 0
x-cache-hits: 0
< x-timer: S1645200308.921842,VS0,VE69
x-timer: S1645200308.921842,VS0,VE69
< content-length: 1837828
content-length: 1837828

<
* Connection #2 to host objects.githubusercontent.com left intact

real    0m11.244s
user    0m0.276s
sys     0m0.047s
```

spacedude609, 2022-02-18:
restarting install...

spacedude609, 2022-02-18:
weird

spacedude609, 2022-02-18:
```
 Step: Checking for prerequisite software

[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : vsftpd
[  OK  ] DietPi-Services | unmask : redis-server
[  OK  ] DietPi-Services | unmask : mariadb
[  OK  ] DietPi-Services | unmask : php7.4-fpm
[  OK  ] DietPi-Services | unmask : lighttpd
[  OK  ] DietPi-Services | unmask : qbittorrent
[  OK  ] DietPi-Services | unmask : gogs
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : gogs
[  OK  ] DietPi-Services | stop : qbittorrent
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm
[  OK  ] DietPi-Services | stop : mariadb
[  OK  ] DietPi-Services | stop : redis-server
[  OK  ] DietPi-Services | stop : vsftpd
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists...
[  OK  ] DietPi-Software | APT update
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing youtube-dl: Download videos from YouTube and other video sites

[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | Checking URL: https://yt-dl.org/downloads/latest/youtube-dl
[ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://yt-dl.org/downloads/latest/youtube-dl -o youtube-dl
[ INFO ] DietPi-Software | APT install for: python3, please wait...
[  OK  ] DietPi-Software | APT install for: python3
[  OK  ] DietPi-Software | G_THREAD: All threads finished
[  OK  ] DietPi-Software | mv youtube-dl /usr/local/bin/youtube-dl
[  OK  ] DietPi-Software | chmod +x /usr/local/bin/youtube-dl
[  OK  ] DietPi-Software | sed -i 1s/python$/python3/ /usr/local/bin/youtube-dl

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Finalising install

[  OK  ] DietPi-Software | systemctl daemon-reload
2022-02-18 11:09:29 [ INFO ] DietPi-RAMlog | Storing /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store...
2022-02-18 11:09:29 [  OK  ] DietPi-RAMlog | Stored /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store.
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : vsftpd
[  OK  ] DietPi-Services | dietpi_controlled : redis-server
[  OK  ] DietPi-Services | dietpi_controlled : mariadb
[  OK  ] DietPi-Services | dietpi_controlled : php7.4-fpm
[  OK  ] DietPi-Services | dietpi_controlled : lighttpd
[  OK  ] DietPi-Services | dietpi_controlled : qbittorrent
[  OK  ] DietPi-Services | dietpi_controlled : gogs
[  OK  ] DietPi-Services | dietpi_controlled : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : vsftpd
[  OK  ] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : php7.4-fpm
[  OK  ] DietPi-Services | restart : lighttpd
[  OK  ] DietPi-Services | restart : qbittorrent
[  OK  ] DietPi-Services | restart : gogs
[  OK  ] DietPi-Services | restart : cron
```

MichaIng, 2022-02-18:
I guess it doesn't take that long when you download https://yt-dl.org/downloads/latest/youtube-dl via browser, right? Joulinar gave the workaround, but it could make sense to play a bit with network settings, e.g. try to change the DNS nameserver (when you use a static IP). But probably it is only a temporary network issue unrelated to your settings or LAN, and it would resolve itself over night.

> that's odd... it's installing python3 after checking the url. The install finished

youtube-dl is a Python 3 script, so of course Python 3 is installed 😉. It is done in parallel, so the youtube-dl runs in background and the python3 package install happens in foreground.

spacedude609, 2022-02-18:
Oh, i know

spacedude609, 2022-02-18:
It was probably a misconfigured DNS I set up

closed 2022-02-18 by spacedude609

MichaIng, 2022-02-18:
Generally, I wouldn't use Pi-hole as DNS for the host system itself. I mean you do not browse websites form the server where ads pop up and trackers are triggered. As long as you do not want to use Pi-hole for monitoring reasons only, it is unnecessary overhead for a server system 😉.

But Pi-hole as well as Unbound both have a DNS cache, so after a possibly initial first connection time, repeating it would be much faster when really DNS was the culprit 🤔.

Joulinar, 2022-02-18:
On my PiHole/Unbound setup, DNS resolution for this topic is done quite fast and the whole request finished in less than 1 sec. It looks more than a slow connection.

spacedude609, 2022-02-18:
10/10 tech support, better than big companies

spacedude609, 2022-02-18:
also my wifi is 6 MB/s when downloading files (max speed)

spacedude609, 2022-02-18:
Ok, btw, started fresh, wont install pi-hole or unbound, I just wanted a nextcloud server