MichaIng/DietPi #5294: re-install of owncloud fails

MichaelStiglmayr, 2022-02-18:
[### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

- Kernel version | Linux raspberry 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

#### Additional Information (if applicable)
- Software title | owncloud
- Was the software title installed freshly or updated/migrated? install script of owncloud
- Bug report ID | 62008e84-5197-4964-87e1-276748921a3f

#### Steps to reproduce

1. re-install owncloud
2. ...

#### Expected behaviour

- ...

#### Actual behaviour
stops with error
- ...find: error while loading shared libraries: ld-linux-aroh&.so.3: cannot open shared object file: No such file or directory
[FAILED] DietPi-Software | phpenmod
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
[FAILED] DietPi-Services | ● mariadb.service - MariaDB 10.3.31 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2022-02-18 19:58:40 CET; 11min ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
 Main PID: 574 (code=exited, status=1/FAILURE)
   Status: "Starting final batch to recover 13 pages from redo log"

#### Extra details

- ...
](https://github.com/MichaIng/DietPi/issues)

Joulinar, 2022-02-18:
Looks like database crashed and was try to do some recovery. Can you post following

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

MichaelStiglmayr, 2022-02-18:
In fact it's true the database is lost, I want to start from scratch

MichaelStiglmayr, 2022-02-19:
My goal is not to recover the old installation / configuration but to start with a new database. Sorry maybe I wasn’t clear about that. 
How can I get rid of the old settings and start from scratch?
Thank you!

MichaIng, 2022-02-19:
> In fact it's true the database is lost, I want to start from scratch

Did you manually remove all database files? Well that breaks the database server itself 😄. Try the following if there is nothing else using MariaDB:
```sh
dietpi-software uninstall 88 114
rm -R /mnt/dietpi_userdata/{nextcloud_data,mariadb-database-backup.sql}
dietpi-software install 114
```

Joulinar, 2022-02-19:
It is Owncloud not Nextcloud

MichaIng, 2022-02-19:
Ah sorry, commands adjusted.

MichaelStiglmayr, 2022-02-20:
`dietpi-software uninstall 88`
results in an error, since MariaDB is not able to start in order to create the backup file (which is not necessary in my case)
would it be possible to remove mariadb with 
`apt-get remove mariadb-*`

Joulinar, 2022-02-20:
You should be able to skip the step. If the error occurs,  select `change command` from error handle screen. Replace the command given with `true` and try to continue.

MichaelStiglmayr, 2022-02-20:
while installing mariadb I now get the error message 
`
debconf: delaying package configuration, since apt-utils is not installed                            
Setting up mariadb-common (1:10.3.31-0+deb10u1) ...                                                
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist                  
dpkg: error processing package mariadb-common (--configure):                                       
 installed mariadb-common package post-installation script subprocess returned error exit status 2 
Errors were encountered while processing:                                                          
 mariadb-common                                                                                    
E: Sub-process /usr/bin/dpkg returned an error code (1)  `
It seems to me that the system is still not clean

MichaIng, 2022-02-20:
The database backup is only attempted if `command -v mysqldump` is present, hence the database server installed. Ah but makes sense when you only manually removed all database table, which broke the server without removing the commands.

Jep, I underline Joulinar's solution for this.

MichaIng, 2022-02-20:
Can you please manually purge all related packages, then retry the package install manually:
```sh
G_AGP '*mariadb*'
rm -Rf /mnt/dietpi_userdata/mysql /var/lib/mysql /etc/mysql
apt install mariadb-server
```

MichaelStiglmayr, 2022-02-20:
also during the manual installation I receive the error
```
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2022-02-20:
Did it show the same line above? At best page the whole command output.
```
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
```
Does it really not exist?
```
ls -l /etc/mysql
dpkg -S /etc/mysql
```

MichaelStiglmayr, 2022-02-20:
Yes. This is the complete output: 
```
apt install mariadb-server
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  galera-3 libaio1 libconfig-inifiles-perl libdbi-perl libreadline5 libsnappy1v5 lsof mariadb-client-10.3 mariadb-client-core-10.3 mariadb-common
  mariadb-server-10.3 mariadb-server-core-10.3 mysql-common rsync socat
Suggested packages:
  libclone-perl libmldbm-perl libnet-daemon-perl libsql-statement-perl mailx mariadb-test netcat-openbsd tinyca openssh-server
Recommended packages:
  libdbd-mysql-perl libterm-readkey-perl libhtml-template-perl
The following NEW packages will be installed:
  galera-3 libaio1 libconfig-inifiles-perl libdbi-perl libreadline5 libsnappy1v5 lsof mariadb-client-10.3 mariadb-client-core-10.3 mariadb-common
  mariadb-server mariadb-server-10.3 mariadb-server-core-10.3 mysql-common rsync socat
0 upgraded, 16 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 13.7 MB/17.3 MB of archives.
After this operation, 147 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf mysql-common all 5.8+1.0.5 [7324 B]
Get:2 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf mariadb-common all 1:10.3.31-0+deb10u1 [32.7 kB]
Get:3 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf galera-3 armhf 25.3.25-2 [811 kB]
Get:4 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libdbi-perl armhf 1.642-1+deb10u2 [767 kB]
Get:5 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf lsof armhf 4.91+dfsg-1 [307 kB] 
Get:6 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian buster/main armhf libconfig-inifiles-perl all 3.000001-1 [51.9 kB]
Get:7 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libaio1 armhf 0.3.112-3 [10.8 kB]                    
Get:8 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf libreadline5 armhf 5.2+dfsg-3 [103 kB]     
Get:9 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf libsnappy1v5 armhf 1.1.7-1 [15.4 kB]
Get:10 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf mariadb-client-core-10.3 armhf 1:10.3.31-0+deb10u1 [4731 kB]
Get:11 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf mariadb-client-10.3 armhf 1:10.3.31-0+deb10u1 [989 kB]
Get:12 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf mariadb-server-core-10.3 armhf 1:10.3.31-0+deb10u1 [5230 kB]
Get:13 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf rsync armhf 3.1.3-6 [356 kB]                                                     
Get:14 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf socat armhf 1.7.3.2-2 [324 kB]
Fetched 13.7 MB in 27s (513 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 26375 files and directories currently installed.)
Preparing to unpack .../00-mysql-common_5.8+1.0.5_all.deb ...
Unpacking mysql-common (5.8+1.0.5) ...
Preparing to unpack .../01-mariadb-common_1%3a10.3.31-0+deb10u1_all.deb ...
Unpacking mariadb-common (1:10.3.31-0+deb10u1) ...
Selecting previously unselected package galera-3.
Preparing to unpack .../02-galera-3_25.3.25-2_armhf.deb ...
Unpacking galera-3 (25.3.25-2) ...
Selecting previously unselected package libdbi-perl:armhf.
Preparing to unpack .../03-libdbi-perl_1.642-1+deb10u2_armhf.deb ...
Unpacking libdbi-perl:armhf (1.642-1+deb10u2) ...
Selecting previously unselected package lsof.
Preparing to unpack .../04-lsof_4.91+dfsg-1_armhf.deb ...
Unpacking lsof (4.91+dfsg-1) ...
Selecting previously unselected package libconfig-inifiles-perl.
Preparing to unpack .../05-libconfig-inifiles-perl_3.000001-1_all.deb ...
Unpacking libconfig-inifiles-perl (3.000001-1) ...
Selecting previously unselected package libaio1:armhf.
Preparing to unpack .../06-libaio1_0.3.112-3_armhf.deb ...
Unpacking libaio1:armhf (0.3.112-3) ...
Preparing to unpack .../07-libreadline5_5.2+dfsg-3_armhf.deb ...
Unpacking libreadline5:armhf (5.2+dfsg-3) ...
Selecting previously unselected package libsnappy1v5:armhf.
Preparing to unpack .../08-libsnappy1v5_1.1.7-1_armhf.deb ...
Unpacking libsnappy1v5:armhf (1.1.7-1) ...
Selecting previously unselected package mariadb-client-core-10.3.
Preparing to unpack .../09-mariadb-client-core-10.3_1%3a10.3.31-0+deb10u1_armhf.deb ...
Unpacking mariadb-client-core-10.3 (1:10.3.31-0+deb10u1) ...
Preparing to unpack .../10-mariadb-client-10.3_1%3a10.3.31-0+deb10u1_armhf.deb ...
Unpacking mariadb-client-10.3 (1:10.3.31-0+deb10u1) ...
Selecting previously unselected package mariadb-server-core-10.3.
Preparing to unpack .../11-mariadb-server-core-10.3_1%3a10.3.31-0+deb10u1_armhf.deb ...
Unpacking mariadb-server-core-10.3 (1:10.3.31-0+deb10u1) ...
Preparing to unpack .../12-rsync_3.1.3-6_armhf.deb ...
Unpacking rsync (3.1.3-6) ...
Selecting previously unselected package socat.
Preparing to unpack .../13-socat_1.7.3.2-2_armhf.deb ...
Unpacking socat (1.7.3.2-2) ...
Setting up mysql-common (5.8+1.0.5) ...
Setting up mariadb-common (1:10.3.31-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--configure):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```
> ls -l /etc/mysql
total 8
drwxr-xr-x 2 root root 4096 Feb 20 17:14 conf.d
drwxr-xr-x 2 root root 4096 Feb 20 17:13 mariadb.conf.d
> dpkg -S /etc/mysql
mariadb-client-10.3, mariadb-common, mysql-common: /etc/mysql
```

MichaIng, 2022-02-20:
```sh
dpkg -S /etc/mysql/mariadb.cnf
for i in /etc/apt/preferences{,.d/*}; do [[ -f $i ]] &&  echo -e "${i}:\n$(<$i)"; done
```

MichaelStiglmayr, 2022-02-20:
```
> dpkg -S /etc/mysql/mariadb.cnf
mariadb-common: /etc/mysql/mariadb.cnf
> for i in /etc/apt/preferences{,.d/*}; do [[ -f $i ]] &&  echo -e "${i}:\n$(<$i)"; done

```
no output of the second command

MichaIng, 2022-02-20:
Strange, the package ships this file, nothing is preventing it from being deployed, but it is missing regardless 🤔. Can you try to install the package manually:
```sh
cd /tmp
curl -LO 'https://archive.raspbian.org/raspbian/pool/main/m/mariadb-10.3/mariadb-common_10.3.31-0%2Bdeb10u1_all.deb'
dpkg -i mariadb-common_*
rm mariadb-common_*
```

MichaelStiglmayr, 2022-02-20:
That failed also. I'm starting to think that reinstalling dietpi might be easier. 
```
> dpkg -i mariadb-common_*
(Reading database ... 26827 files and directories currently installed.)
Preparing to unpack mariadb-common_10.3.31-0%2Bdeb10u1_all.deb ...
Unpacking mariadb-common (1:10.3.31-0+deb10u1) over (1:10.3.31-0+deb10u1) ...
Setting up mariadb-common (1:10.3.31-0+deb10u1) ...
update-alternatives: error: alternative path /etc/mysql/mariadb.cnf doesn't exist
dpkg: error processing package mariadb-common (--install):
 installed mariadb-common package post-installation script subprocess returned error exit status 2
Errors were encountered while processing:
 mariadb-common

```

MichaelStiglmayr, 2022-02-20:
Nevertheless it would be great if it wouldn't be necessary. Thanks for your help!

MichaIng, 2022-02-20:
Please try the following:
```sh
cd /tmp
curl -LO 'https://archive.raspbian.org/raspbian/pool/main/m/mariadb-10.3/mariadb-common_10.3.31-0%2Bdeb10u1_all.deb'
mkdir mariadb-common
dpkg-deb -R mariadb-common_10.3.31-0%2Bdeb10u1_all.deb mariadb-common
cp -R mariadb-common/. /
rm -R mariadb-common*
dpkg --configure -a
```

MichaelStiglmayr, 2022-02-20:
I've got a new error message, while setting up redis server. 
```
> dpkg --configure -a
Setting up libconfig-inifiles-perl (3.000001-1) ...
Setting up libreadline5:armhf (5.2+dfsg-3) ...
Setting up redis-server (5:5.0.14-1+deb10u2) ...
Segmentation fault
dpkg: error processing package redis-server (--configure):
 installed redis-server package post-installation script subprocess returned error exit status 139
Setting up libsnappy1v5:armhf (1.1.7-1) ...
Setting up mariadb-common (1:10.3.31-0+deb10u1) ...
update-alternatives: using /etc/mysql/mariadb.cnf to provide /etc/mysql/my.cnf (my.cnf) in auto mode
Setting up socat (1.7.3.2-2) ...
Setting up lsof (4.91+dfsg-1) ...
Setting up galera-3 (25.3.25-2) ...
Setting up libaio1:armhf (0.3.112-3) ...
Setting up libdbi-perl:armhf (1.642-1+deb10u2) ...
Setting up rsync (3.1.3-6) ...
Setting up mariadb-server-core-10.3 (1:10.3.31-0+deb10u1) ...
Setting up mariadb-client-core-10.3 (1:10.3.31-0+deb10u1) ...
Setting up mariadb-client-10.3 (1:10.3.31-0+deb10u1) ...
Processing triggers for systemd (241-7~deb10u8+rpi1) ...
Processing triggers for libc-bin (2.28-10+rpt2+rpi1) ...
Errors were encountered while processing:
 redis-server
```

MichaIng, 2022-02-20:
Looks like all this is the result of a corrupted filesystem. Can you check:
```sh
dmesg -l emerg,alert,crit,err
```

MichaelStiglmayr, 2022-02-21:
This is the result
```
[    0.544726] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
```

MichaIng, 2022-02-21:
Hmm, no filesystem I/O errors 🤔. Let's try to reinstall the packages which provide the commands from the Redis postinst script:
```sh
apt install --reinstall dpkg init-system-helpers
```

MichaelStiglmayr, 2022-02-21:
```
> apt install --reinstall dpkg init-system-helpers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 2 reinstalled, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 2188 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf dpkg armhf 1.19.7 [2141 kB]
Get:2 http://ftp.halifax.rwth-aachen.de/raspbian/raspbian buster/main armhf init-system-helpers all 1.56+nmu1 [46.8 kB]
Fetched 2188 kB in 1s (2928 kB/s)             
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 26375 files and directories currently installed.)
Preparing to unpack .../init-system-helpers_1.56+nmu1_all.deb ...
Unpacking init-system-helpers (1.56+nmu1) over (1.56+nmu1) ...
Setting up init-system-helpers (1.56+nmu1) ...
(Reading database ... 26375 files and directories currently installed.)
Preparing to unpack .../archives/dpkg_1.19.7_armhf.deb ...
Unpacking dpkg (1.19.7) over (1.19.7) ...
Setting up dpkg (1.19.7) ...
Setting up redis-server (5:5.0.14-1+deb10u2) ...
find: error while loading shared libraries: ld-linux-aroh&.so.1: cannot open shared object file: No such file or directory
dpkg: error processing package redis-server (--configure):
 installed redis-server package post-installation script subprocess returned error exit status 127
Errors were encountered while processing:
 redis-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2022-02-21:
The `find` command. Can you run, whether this produces the same error:
```sh
find /etc/redis -maxdepth 1 -type d -name 'redis-server.*.d' -empty -delete
```
`ld-linux-aroh&.so.1` contains false characters, is it really printed like that to console? It should be `ld-linux-armhf.so.3`. Probably reinstalling `find` solves it:
```sh
apt install --reinstall findutils
```

MichaelStiglmayr, 2022-02-21:
Yes it's really like this: `ld-linux-aroh&.so.1` and the find command yields the same result. After the install of findutils the error was gone. 
And afterwards also `dietpi-software install 47`worked fine, until the start of mariadb. So I ran

```
systemctl status mariadb
● mariadb.service - MariaDB 10.3.31 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-02-21 17:25:57 CET; 17min ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
  Process: 474 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
  Process: 475 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 477 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`cd /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemc
tl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 485 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 576 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 580 ExecStartPost=/etc/mysql/debian-start (code=exited, status=203/EXEC)
 Main PID: 485 (code=exited, status=0/SUCCESS)
   Status: "MariaDB server is down"

Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: FTS optimize thread exiting.
Feb 21 17:25:55 raspberry systemd[1]: mariadb.service: Control process exited, code=exited, status=203/EXEC
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Starting shutdown...
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Dumping buffer pool(s) to /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Buffer pool(s) dump completed at 220221 17:25:55
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Shutdown completed; log sequence number 1625479; transaction id 21
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] /usr/sbin/mysqld: Shutdown complete
Feb 21 17:25:57 raspberry systemd[1]: mariadb.service: Failed with result 'exit-code'.
Feb 21 17:25:57 raspberry systemd[1]: Failed to start MariaDB 10.3.31 database server.
```

MichaIng, 2022-02-21:
Please show the full log:
```sh
journalctl -u mariadb
```

MichaelStiglmayr, 2022-02-21:
```
> journalctl -u mariadb
-- Logs begin at Thu 2019-02-14 11:11:59 CET, end at Mon 2022-02-21 18:17:02 CET. --
Feb 21 17:24:20 raspberry systemd[1]: Starting MariaDB 10.3.31 database server...
Feb 21 17:25:28 raspberry mysqld[485]: 2022-02-21 17:25:28 0 [Note] /usr/sbin/mysqld (mysqld 10.3.31-MariaDB-0+deb10u1) starting as process 485 ...
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Using Linux native AIO
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Uses event mutexes
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Number of pools: 1
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Using generic crc32 instructions
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Completed initialization of buffer pool
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be cha
nged. See the man page of setpriority().
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Creating shared tablespace for temporary tables
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wai
t ...
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: 10.3.31 started; log sequence number 1625470; transaction id 20
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] Plugin 'FEEDBACK' is disabled.
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Buffer pool(s) load completed at 220221 17:25:54
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] Server socket created on IP: '::'.
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] Reading of all Master_info entries succeeded
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] Added new Master_info '' to hash table
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] /usr/sbin/mysqld: ready for connections.
Feb 21 17:25:55 raspberry mysqld[485]: Version: '10.3.31-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
Feb 21 17:25:55 raspberry systemd[580]: mariadb.service: Failed to execute command: No such file or directory
Feb 21 17:25:55 raspberry systemd[580]: mariadb.service: Failed at step EXEC spawning /etc/mysql/debian-start: No such file or director
y
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] Event Scheduler: Purging the queue. 0 events
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: FTS optimize thread exiting.
Feb 21 17:25:55 raspberry systemd[1]: mariadb.service: Control process exited, code=exited, status=203/EXEC
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Starting shutdown...
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Dumping buffer pool(s) to /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Buffer pool(s) dump completed at 220221 17:25:55
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Shutdown completed; log sequence number 1625479; transaction id 21
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] /usr/sbin/mysqld: Shutdown complete
Feb 21 17:25:57 raspberry systemd[1]: mariadb.service: Failed with result 'exit-code'.
Feb 21 17:25:57 raspberry systemd[1]: Failed to start MariaDB 10.3.31 database server.

```

MichaIng, 2022-02-21:
```
Feb 21 17:25:55 raspberry systemd[580]: mariadb.service: Failed to execute command: No such file or directory
Feb 21 17:25:55 raspberry systemd[580]: mariadb.service: Failed at step EXEC spawning /etc/mysql/debian-start: No such file or directory
```
🤔. `/etc/mysql/debian-start` is missing?
```sh
ls -l /etc/mysql
```

MichaelStiglmayr, 2022-02-21:
```
> ls -l /etc/mysql
total 16
drwxr-xr-x 2 root root 4096 Feb 21 16:20 conf.d
-rw------- 1 root root  277 Feb 21 16:25 debian.cnf
-rw-r--r-- 1 root root  869 Feb 21 08:27 mariadb.cnf
drwxr-xr-x 2 root root 4096 Feb 21 16:25 mariadb.conf.d
lrwxrwxrwx 1 root root   24 Feb 21 16:20 my.cnf -> /etc/alternatives/my.cnf

```

MichaIng, 2022-02-21:
Strange, this is already the second case where a MariaDB file is missing which is actually shipped by the package. Please try to reinstall it:
```sh
apt install --reinstall mariadb-server-10.3
```

MichaelStiglmayr, 2022-02-21:
```
> apt install --reinstall mariadb-server-10.3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 3567 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf mariadb-server-10.3 armhf 1:10.3.31-0+deb10u1 [3567 kB]
Fetched 3567 kB in 1s (4623 kB/s)              
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 26959 files and directories currently installed.)
Preparing to unpack .../mariadb-server-10.3_1%3a10.3.31-0+deb10u1_armhf.deb ...
/var/lib/mysql: found previous version 10.3
Unpacking mariadb-server-10.3 (1:10.3.31-0+deb10u1) over (1:10.3.31-0+deb10u1) ...
Setting up mariadb-server-10.3 (1:10.3.31-0+deb10u1) ...
mariadb.service is a disabled or a static unit, not starting it.
Processing triggers for systemd (241-7~deb10u8+rpi1) ...
> systemctl status mariadb
● mariadb.service - MariaDB 10.3.31 database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2022-02-21 17:25:57 CET; 1h 33min ago
     Docs: man:mysqld(8)
           https://mariadb.com/kb/en/library/systemd/
 Main PID: 485 (code=exited, status=0/SUCCESS)
   Status: "MariaDB server is down"

Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: FTS optimize thread exiting.
Feb 21 17:25:55 raspberry systemd[1]: mariadb.service: Control process exited, code=exited, status=203/EXEC
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Starting shutdown...
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Dumping buffer pool(s) to /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Buffer pool(s) dump completed at 220221 17:25:55
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Shutdown completed; log sequence number 1625479; transaction id 21
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] /usr/sbin/mysqld: Shutdown complete
Feb 21 17:25:57 raspberry systemd[1]: mariadb.service: Failed with result 'exit-code'.
Feb 21 17:25:57 raspberry systemd[1]: Failed to start MariaDB 10.3.31 database server.
```

MichaIng, 2022-02-21:
```sh
systemctl restart mariadb
sleep 5
journalctl -u mariadb
```

MichaelStiglmayr, 2022-02-21:
```
>> systemctl restart mariadb
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
>> sleep 5
>> journalctl -u mariadb
-- Logs begin at Thu 2019-02-14 11:11:59 CET, end at Mon 2022-02-21 21:03:42 CET. --
Feb 21 17:24:20 raspberry systemd[1]: Starting MariaDB 10.3.31 database server...
Feb 21 17:25:28 raspberry mysqld[485]: 2022-02-21 17:25:28 0 [Note] /usr/sbin/mysqld (mysqld 10.3.31-MariaDB-0+deb10u1) starting as process 485 ...
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Using Linux native AIO
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Uses event mutexes
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
Feb 21 17:25:53 raspberry mysqld[485]: 2022-02-21 17:25:53 0 [Note] InnoDB: Number of pools: 1
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Using generic crc32 instructions
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Completed initialization of buffer pool
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be cha
nged. See the man page of setpriority().
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Creating shared tablespace for temporary tables
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wai
t ...
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: 10.3.31 started; log sequence number 1625470; transaction id 20
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] Plugin 'FEEDBACK' is disabled.
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] InnoDB: Buffer pool(s) load completed at 220221 17:25:54
Feb 21 17:25:54 raspberry mysqld[485]: 2022-02-21 17:25:54 0 [Note] Server socket created on IP: '::'.
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] Reading of all Master_info entries succeeded
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] Added new Master_info '' to hash table
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] /usr/sbin/mysqld: ready for connections.
Feb 21 17:25:55 raspberry mysqld[485]: Version: '10.3.31-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
Feb 21 17:25:55 raspberry systemd[580]: mariadb.service: Failed to execute command: No such file or directory
Feb 21 17:25:55 raspberry systemd[580]: mariadb.service: Failed at step EXEC spawning /etc/mysql/debian-start: No such file or director
y
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] Event Scheduler: Purging the queue. 0 events
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: FTS optimize thread exiting.
Feb 21 17:25:55 raspberry systemd[1]: mariadb.service: Control process exited, code=exited, status=203/EXEC
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Starting shutdown...
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Dumping buffer pool(s) to /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 17:25:55 raspberry mysqld[485]: 2022-02-21 17:25:55 0 [Note] InnoDB: Buffer pool(s) dump completed at 220221 17:25:55
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] InnoDB: Shutdown completed; log sequence number 1625479; transaction id 21
Feb 21 17:25:57 raspberry mysqld[485]: 2022-02-21 17:25:57 0 [Note] /usr/sbin/mysqld: Shutdown complete
Feb 21 17:25:57 raspberry systemd[1]: mariadb.service: Failed with result 'exit-code'.
Feb 21 17:25:57 raspberry systemd[1]: Failed to start MariaDB 10.3.31 database server.
Feb 21 18:59:09 raspberry systemd[1]: Starting MariaDB 10.3.31 database server...
Feb 21 18:59:10 raspberry mysqld[1273]: 2022-02-21 18:59:10 0 [Note] /usr/sbin/mysqld (mysqld 10.3.31-MariaDB-0+deb10u1) starting as process 1273 ...
Feb 21 18:59:10 raspberry mysqld[1273]: 2022-02-21 18:59:10 0 [Note] InnoDB: Using Linux native AIO
Feb 21 18:59:10 raspberry mysqld[1273]: 2022-02-21 18:59:10 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
>>
```
```

MichaelStiglmayr, 2022-02-21:
Sorry the last page of output is missing
```
Feb 21 21:03:39 raspberry systemd[1]: Starting MariaDB 10.3.31 database server...
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] /usr/sbin/mysqld (mysqld 10.3.31-MariaDB-0+deb10u1) starting as process 1734 ...
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Using Linux native AIO
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Uses event mutexes
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Compressed tables use zlib 1.2.11
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Number of pools: 1
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Using generic crc32 instructions
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Initializing buffer pool, total size = 128M, instances = 1, chunk size = 128M
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Completed initialization of buffer pool
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be ch
anged. See the man page of setpriority().
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Creating shared tablespace for temporary tables
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wa
it ...
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: 10.3.31 started; log sequence number 1625497; transaction id 20
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] InnoDB: Loading buffer pool(s) from /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 21:03:39 raspberry mysqld[1734]: 2022-02-21 21:03:39 0 [Note] Plugin 'FEEDBACK' is disabled.
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] Server socket created on IP: '::'.
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] InnoDB: Buffer pool(s) load completed at 220221 21:03:40
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] Reading of all Master_info entries succeeded
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] Added new Master_info '' to hash table
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] /usr/sbin/mysqld: ready for connections.
Feb 21 21:03:40 raspberry mysqld[1734]: Version: '10.3.31-MariaDB-0+deb10u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10
Feb 21 21:03:40 raspberry systemd[1771]: mariadb.service: Failed to execute command: No such file or directory
Feb 21 21:03:40 raspberry systemd[1771]: mariadb.service: Failed at step EXEC spawning /etc/mysql/debian-start: No such file or directo
ry
Feb 21 21:03:40 raspberry systemd[1]: mariadb.service: Control process exited, code=exited, status=203/EXEC
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] /usr/sbin/mysqld (initiated by: unknown): Normal shutdown
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] Event Scheduler: Purging the queue. 0 events
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] InnoDB: FTS optimize thread exiting.
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] InnoDB: Starting shutdown...
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] InnoDB: Dumping buffer pool(s) to /mnt/dietpi_userdata/mysql/ib_buffer_pool
Feb 21 21:03:40 raspberry mysqld[1734]: 2022-02-21 21:03:40 0 [Note] InnoDB: Buffer pool(s) dump completed at 220221 21:03:40
Feb 21 21:03:42 raspberry mysqld[1734]: 2022-02-21 21:03:42 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
Feb 21 21:03:42 raspberry mysqld[1734]: 2022-02-21 21:03:42 0 [Note] InnoDB: Shutdown completed; log sequence number 1625506; transaction id 21
Feb 21 21:03:42 raspberry mysqld[1734]: 2022-02-21 21:03:42 0 [Note] /usr/sbin/mysqld: Shutdown complete
Feb 21 21:03:42 raspberry systemd[1]: mariadb.service: Failed with result 'exit-code'.
Feb 21 21:03:42 raspberry systemd[1]: Failed to start MariaDB 10.3.31 database server.
```

MichaIng, 2022-02-21:
Can you try
```sh
cd /tmp
curl -L 'https://archive.raspbian.org/raspbian/pool/main/m/mariadb-10.3/mariadb-server-10.3_10.3.31-0%2Bdeb10u1_armhf.deb' -o mariadb.deb
dpkg -i --force-confnew --force-confmiss mariadb.deb
ls -l /etc/mysql/debian-start
```

MichaelStiglmayr, 2022-02-21:
```
> cd /tmp
> /tmp# curl -L 'https://archive.raspbian.org/raspbian/pool/main/m/mariadb-10.3/mariadb-server-10.3_10.3.31-0%2Bdeb10u1_armhf.deb' -o mariadb.deb
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 3482k  100 3482k    0     0  4933k      0 --:--:-- --:--:-- --:--:-- 4933k
root@raspberry:/tmp# dpkg -i --force-confnew --force-confmiss mariadb.deb
(Reading database ... 26959 files and directories currently installed.)
Preparing to unpack mariadb.deb ...
/var/lib/mysql: found previous version 10.3
Unpacking mariadb-server-10.3 (1:10.3.31-0+deb10u1) over (1:10.3.31-0+deb10u1) ...
Setting up mariadb-server-10.3 (1:10.3.31-0+deb10u1) ...

Configuration file '/etc/default/mysql', does not exist on system.
Installing new config file as you requested.

Configuration file '/etc/init.d/mysql', does not exist on system.
Installing new config file as you requested.

Configuration file '/etc/mysql/debian-start', does not exist on system.
Installing new config file as you requested.

Configuration file '/etc/mysql/mariadb.conf.d/50-mysqld_safe.cnf', does not exist on system.
Installing new config file as you requested.

Configuration file '/etc/mysql/mariadb.conf.d/50-server.cnf', does not exist on system.
Installing new config file as you requested.
mariadb.service is a disabled or a static unit, not starting it.
Processing triggers for systemd (241-7~deb10u8+rpi1) ...
root@raspberry:/tmp# ls -l /etc/mysql/debian-start
-rwxr-xr-x 1 root root 1620 May 17  2021 /etc/mysql/debian-start
> /tmp# rm mariadb.deb
> /tmp# systemctl restart mariadb
```
That's good, isn't it?

MichaIng, 2022-02-21:
Yes, that look good. I have no idea why all those files were missing in the first place, but all files in `/etc` are automatically treated as "conffiles" by official Debian packages, which means that new configs do neither overwrite old configs, nor are they installed when the config was "manually" removed. Since it is unlikely that you manually removed these files, I thought they were somehow mussing as of a transient package install stage and that in this situation the package would install the missing ones automatically. Obviously this is not the case.

Okay, is everything up and running now?

MichaelStiglmayr, 2022-02-21:
Seems so. The owncloud install script ran without error. Thank you so much for your help!

MichaIng, 2022-02-21:
Okay great. I'll mark this issue as closed then. Feel free to reopen if required.

closed 2022-02-21 by MichaIng