MichaIng/DietPi #5297: Dietpi-config menu?

spacedude609, 2022-02-19:
### New feature request:
(It's me again)
#### Is your feature request related to a problem? Please describe:

- I find it can be a bit difficult to do manual config sometimes, or when trying to find the command to reconfigure a package.
- Ex. learning how to run the occ command included with nextcloud, changing config in php and lighttpd.

#### Describe the solution you'd like:

- Mabye some kind of dietpi-config menu? Allows user to reconfigure packages. Can run re-cnfiguration, mabye allows user to set variables of their own, which the program writes into a config file just like it does during install.

#### Describe alternatives you've considered:

- Looking online for a packages config and how to edit it. Looking fr the command to launch the reconfigurator.

#### Additional context

- menu design like dietpi-software. Still theoretical.

MichaIng, 2022-02-19:
Many thanks for your request.

The thing is that we offer close to 200 software titles. Adding individual configuration options for all of them is impossible to achieve and maintain. We thought about it for individual selected software titles, but basically since there is much more on the ToDo which does not bend the scope of what DietPi currently does, it has very low priority.

Generally, most software has very good own docs which explain how it can be configured. We link those from our documentation pages as well. E.g. for Nextcloud: https://dietpi.com/docs/software/cloud/#nextcloud

As also explained in our docs, for Nextcloud and ownCloud we created shortcuts for the occ command to make it easier. Simply run `ncc` (in case of ownCloud `occ`), and it will call Nextcloud's occ command with correct user etc.

spacedude609, 2022-02-19:
Oh thx, I was just talking about common packages, php, apache, lighttpd, etc.

spacedude609, 2022-02-19:
also ur docs didnt appear at the top of my results so I wasn't able to find them

Joulinar, 2022-02-19:
dieitpi.com/docs

spacedude609, 2022-02-19:
But thanks for that I guess. It was a thought. I guess you could just make commands that, for some commands, open the right config file in an editor.

spacedude609, 2022-02-19:
> dietpi.com/docs

Okie dokie

closed 2022-02-19 by spacedude609

MichaIng, 2022-02-19:
The docs are at least linked from `dietpi-software` > `Help` menu and when running `dietpi-software list`, but yeah we could make them more prominent, e.g. print them also for installed software titles after the install has finished.

For PHP, Apache and Lighttpd it is basically the same as for Nextcloud/ownCloud/...: check their official docs. A UI for configuring them, again, would add a high maintenance efforts, not worth it compared to other tasks, given that we could only add individual options step by step, which would create conflicts with custom options etc. Even "the right config file" is not as trivial as you might think, each of PHP, Apache and Lighttpd have several config files, with redundant possible config ways, it highly depends on what and how you want to achieve, whether even an existing config file is to be adjusted or whether it should better be a new one, which then also needs to be enabled etc. In fact we just recently talked about how nice an Nginx (webserver) web UI would be, but the only one I found was basically a web based text editor, without any menus, check boxes or so to enable/configure features, only reasonable when one has no console and SSH access. The fact that none exists out there shows how difficult (efforts vs result) it is for this case to create such a configuration interface 😉.

Finally it depends on whether a volunteer has time and motivation to create such a script for a software title (s)he uses, as then we would be happy to implement it 🙂.