MichaIng/DietPi #5300: chromium (kiosk mode) crush

almirus, 2022-02-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.1.2 
- Distro version | bullseye
- Kernel version | Linux DietPi 5.10.69-7-amlogic-gfd159ba07d5c #gfd159ba07d5c SMP PREEMPT Thu Nov 4 14:38:15 CST 2021 aarch64 GNU/Linux
- SBC model | Radxa Zero (aarch64)
- Power supply used | 5V 1A
- SD card used |SanDisk ultra

#### Additional Information (if applicable)
- Software title | chromium 98.0.4758.102
- Bug report ID | 16a389ef-4f5f-4795-ad55-81e728bafdde

#### Steps to reproduce
1. autostart chromium (kiosk mode)
2. or xinit /usr/bin/chromium ---kiosk

#### Extra details
```
root@DietPi:~# xinit /usr/bin/chromium ---kiosk

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux DietPi 5.10.69-7-amlogic-gfd159ba07d5c #gfd159ba07d5c SMP PREEMPT Thu Nov 4 14:38:15 CST 2021 aarch64
Kernel command line: initrd=initrd.img-5.10.69-7-amlogic-gfd159ba07d5c root=UUID=30422831-c223-492f-959c-0fc18139c228 rootwait rw rootfstype=ext4 console=tty1 console=ttyAML0,115200 panic=10 consoleblank=0 loglevel=7   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
Build Date: 16 December 2021  05:08:23PM
xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sun Feb 20 18:40:30 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
[2380:2410:0220/184033.652183:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2380:2410:0220/184033.652751:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2380:2410:0220/184033.702222:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2380:2410:0220/184033.702466:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2380:2410:0220/184033.842968:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[2380:2410:0220/184033.843115:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
*** stack smashing detected ***: terminated
[2380:2380:0220/184034.039483:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2380:2380:0220/184034.122230:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
[2380:2444:0220/184034.123897:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[2380:2444:0220/184034.124941:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[2380:2444:0220/184034.126014:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2380:2380:0220/184034.167681:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2380:2380:0220/184034.198424:ERROR:system_web_app_manager.cc(320)] Exceeded SWA install retry attempts.  Skipping installation, will retry on next OS update or when locale changes.
[2380:2380:0220/184034.200305:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2380:2380:0220/184034.225990:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2380:2380:0220/184034.240862:ERROR:gpu_process_host.cc(978)] GPU process exited unexpectedly: exit_code=6
[2380:2380:0220/184034.240968:FATAL:gpu_data_manager_impl_private.cc(448)] GPU process isn't usable. Goodbye.
[2418:2436:0220/184034.338564:ERROR:ssl_client_socket_impl.cc(995)] handshake failed; returned -1, SSL error code 1, net_error -3
[2418:2436:0220/184034.339536:ERROR:ssl_client_socket_impl.cc(995)] handshake failed; returned -1, SSL error code 1, net_error -3
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

```

MichaIng, 2022-02-20:
Many thanks for your report. It seems to be a general bug in Chromium affecting many or all ARMv8 systems:
- https://github.com/MichaIng/DietPi/issues/5278#issuecomment-1040592980
- https://dietpi.com/phpbb/viewtopic.php?t=9957

MDAR, 2022-02-21:
Please let me know if there is anything you want me to test on an Odroid C4.

MichaIng, 2022-02-25:
I found a workaround:
```sh
apt install chromium/bullseye chromium-common/bullseye
```
This will downgrade Chromium from v98 (from Debian security repo) to v90 (from regular Debian repo). That one starts up without issues.

I'll report this to the Debian bug tracker.

MichaIng, 2022-02-26:
This works for you as well guys? I'm not sure what to do about it. I reported it to Debian: https://bugs.debian.org/1006457
But I'm not sure whether we should really pin an old Chromium version on those systems via dietpi-software which obviously has security vulnerabilities solved with the one shipped by the security repo. As a personal workaround this is fine, but knowingly shipping vulnerable software in "stable" DietPi is not so awesome 🤔.

I think I'll wait with the decision until the final release, probably we get some helpful response on the Debian bug report until then.

MDAR, 2022-02-26:
> This works for you as well guys?

I'm away for a few weeks working on a client's site, as soon as I get back I'll fire up a machine and let you know.

> As a personal workaround this is fine, but knowingly shipping vulnerable software in "stable" DietPi is not so awesome 🤔.
> 
> I think I'll wait with the decision until the final release, probably we get some helpful response on the Debian bug report until then.

I agree completely

orlandozeta, 2022-03-05:
in my case, the chromium is opened, but not load the --homepage, but, if i press ctrl t (new tab) then load the page:

test:
dietpi version: 8.1.2
Chromium 98.0.4758.106 Built on Debian , running on Debian 11

`xinit /usr/bin/chromium-browser --autoplay-policy=no-user-gesture-required --no-sandbox --homepage https://www.google.com --app --start-fullscreen --check-for-update-interval=604800 --window-size=, --app-window-size=, --window-position=0,0 --noerrdialogs --disable-infobars --kiosk`

**OUTPUT:**
```
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux DietPi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=854 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  root=PARTUUID=9bcc366d-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=ttyS0,115200 console=tty1
Build Date: 04 January 2022  10:24:42AM
xorg-server 2:1.20.11-1+rpt1+deb11u1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  4 21:07:26 2022
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
/etc/chromium.d/README: line 1: Any: command not found
dpkg-query: no packages found matching bluealsa
[3623:3678:0304/210729.478320:ERROR:bus.cc(397)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[3623:3678:0304/210729.480972:ERROR:bus.cc(397)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[3623:3678:0304/210729.544250:ERROR:bus.cc(397)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[3623:3678:0304/210729.546522:ERROR:bus.cc(397)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
libEGL warning: DRI2: failed to authenticate
[3623:3678:0304/210729.790953:ERROR:bus.cc(397)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[3623:3678:0304/210729.792626:ERROR:bus.cc(397)] Failed to connect to the bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[3623:3725:0304/210730.017362:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[3623:3725:0304/210730.019597:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[3623:3725:0304/210730.021784:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[3686:3686:0304/210730.257685:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is
[3686:3686:0304/210730.318619:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[3745:3745:0304/210730.388484:ERROR:egl_util.cc(74)] Failed to load GLES library: /usr/lib/chromium-browser/libGLESv2.so: /usr/lib/chromium-browser/libGLESv2.so: cannot open shared object file: No such file or directory
[3745:3745:0304/210730.404488:ERROR:viz_main_impl.cc(188)] Exiting GPU process due to errors during initialization
[3757:3757:0304/210730.466733:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is disabled, ANGLE is
[3623:3661:0304/210735.098604:ERROR:chrome_browser_main_extra_parts_metrics.cc(227)] START: ReportBluetoothAvailability(). If you don't see the END: message, this is crbug.com/1216328.
[3623:3661:0304/210735.101069:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()

```

MichaIng, 2022-03-06:
@orlandozeta 
You are using an RPi, with `chromium-browser` from RPi repo instead of `chromium` from Debian repo, I guess (this is what `dietpi-software` installs)? This is known to not be affected, only the Chromium build from the Debian APT repository.

orlandozeta, 2022-03-06:
what could be the problem then?

with Chromium 95.0.4638.78 Built on Debian , running on Debian 11 It works well, 32 rpi running ok, 2 with updated chromium to 98, not

MichaIng, 2022-03-06:
I have no idea, which is why I reported it to Debian.

MDAR, 2022-03-11:
Hello everyone

I've just updated an Odroid C4 to DietPi 8.2.2 and the full screen Chromium kiosk in Xinit is working again.

Thanks to everyone that did all the hard work in the background.

FYI

I did play around with Firefox in the same situation, but it never quite did what Chromium did so easily.

MichaIng, 2022-03-11:
Strange, since there was not change regarding this, no Chromium package upgrade since this issue appeared the first time. I'll test on Odroid N2 as well, probably others here can test as well. If it is really resolved, then it was probably something with the X server or a dependency of Chromium which fixed it. I'll carefully watch all APT package upgrades on Odroid N2 as well 😄.

MDAR, 2022-03-11:
> no Chromium package upgrade

Curious, the machine in question here updated from DietPi 7 to 8.2.2 and I'm sure I saw a Chromium package update at some point.

But I can confirm that I did nothing else, other than update to 8.2.2

Good luck with the N2.

Do you have a C4 machine? Would you like one?

MichaIng, 2022-03-11:
> Curious, the machine in question here updated from DietPi 7 to 8.2.2 and I'm sure I saw a Chromium package update at some point.

Okay, DietPi v7 is old, so of course there were many Chromium package updates since then. I hope before reporting here you tried first to upgrade to the latest Chromium package 😉. I tested it with the (still) latest package on Odroid N2 and faced the same issue there.

MDAR, 2022-03-11:
> hope before reporting here

Yeah, fear not.

I reported it after I had updated the machine to 8.1 (I think, whatever it was a few weeks ago)

Today's machine was a "working / known to be good" image

MichaIng, 2022-03-11:
Okay, another package upgrade (not the `chromium` one) solved it most likely then.

orlandozeta, 2022-03-28:
today, after  a  sudo apt upgrade, the chromium-browser is working again

orlandozeta, 2022-03-28:
`root@DietPi:~# sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  base-files bind9-dnsutils bind9-host bind9-libs dirmngr dnsutils gnupg gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm gpgv libflac8 libpam-systemd libssl1.1 libsystemd0 libtiff5 libudev1 libxml2 libzzip-0-13 openssl
  systemd systemd-sysv systemd-timesyncd sysvinit-utils tzdata udev xterm
32 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 21.2 MB of archives.
After this operation, 58.4 kB of additional disk space will be used.
Do you want to continue? [Y/n]
`

orlandozeta, 2022-03-28:
tried on another device, not lucky, maybe in the first device, i install something did fix the bug

MichaIng, 2022-03-29:
The device where it is working is a Raspberry Pi? This issue is not affecting the Raspberry Pi's `chromium-browser` package, only Debian's `chromium` package.

MichaIng, 2022-03-29:
Then your issue is not related to this one, I guess you do not see the error messages reported above either, do you? Does this better fit your symptoms: #5342
You need to enable KMS on RPi to have it showing websites. See the first three commands provided here: https://github.com/MichaIng/DietPi/issues/5342#issuecomment-1061789323

MichaIng, 2022-04-01:
On my Odroid N2 it doesn't work yet either.

MDAR, 2022-04-01:
> On my Odroid N2 it doesn't work yet either.

That's disappointing.
Is there anything I can try that might help you?

(I'm tempted to try a N2, but can't justify it right now)

MichaIng, 2022-04-02:
> Is there anything I can try that might help you?

Many thanks for the offer, but I have really no idea how to further debug, and it is the very same issue on all boards, so I can do every debug step quickly by myself. I updated the Debian bug report, sadly I missed a reply, and sadly the suggested Chromium flags didn't help either, but at least there is some activity on the bug report.

Nasty that now where the new Chromium version got merged into the main Debian repo as well (previously it was shipped only by the security repo), the simply workaround above doesn't work anymore, but now one needs to downgrade Chromium by manually downloading and installing the older packages.

MDAR, 2022-05-31:
Hi

I'm sure this isn't news to you, but I thought I'd share the Xorg.log file from an Odoird C4 with DietPi V8.5

```
[   175.730] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[   175.730] Build Operating System: linux Debian
[   175.730] Current Operating System: Linux openHAB-C4 5.10.110-meson64 #22.05.1 SMP PREEMPT Sat May 28 07:50:27 UTC 2022 aarch64
[   175.730] Kernel command line: root=UUID=28046172-8741-4334-96cd-b24d158eec46 rootfstype=ext4 rootwait console=tty1 loglevel=4 consoleblank=0 coherent_pool=2M ubootpart=2c2a3e4f-01 libata.force=noncq usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u net.ifnames=0 
[   175.731] Build Date: 16 December 2021  05:08:23PM
[   175.731] xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support) 
[   175.731] Current version of pixman: 0.40.0
[   175.731] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   175.731] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   175.732] (==) Log file: "/home/dietpi/.local/share/xorg/Xorg.0.log", Time: Tue May 31 10:32:46 2022
[   175.737] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   175.737] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   175.738] (==) No Layout section.  Using the first Screen section.
[   175.738] (==) No screen section available. Using defaults.
[   175.738] (**) |-->Screen "Default Screen Section" (0)
[   175.738] (**) |   |-->Monitor "<default monitor>"
[   175.739] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   175.739] (**) Option "BlankTime" "0"
[   175.739] (==) Automatically adding devices
[   175.739] (==) Automatically enabling devices
[   175.739] (==) Automatically adding GPU devices
[   175.741] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   175.747] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   175.747] 	Entry deleted from font path.
[   175.747] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   175.747] 	Entry deleted from font path.
[   175.747] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   175.747] 	Entry deleted from font path.
[   175.747] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   175.747] 	Entry deleted from font path.
[   175.747] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   175.747] 	Entry deleted from font path.
[   175.747] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   175.747] 	Entry deleted from font path.
[   175.747] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[   175.747] (==) ModulePath set to "/usr/lib/xorg/modules"
[   175.747] (**) Extension "DPMS" is disabled
[   175.747] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   175.747] (II) Loader magic: 0xaaaab6791e28
[   175.747] (II) Module ABI versions:
[   175.747] 	X.Org ANSI C Emulation: 0.4
[   175.747] 	X.Org Video Driver: 24.1
[   175.747] 	X.Org XInput driver : 24.1
[   175.747] 	X.Org Server Extension : 10.0
[   175.753] (EE) systemd-logind: failed to get session: Unit dbus-org.freedesktop.login1.service is masked.
[   175.755] (II) xfree86: Adding drm device (/dev/dri/card0)
[   175.756] (II) xfree86: Adding drm device (/dev/dri/card1)
[   175.756] (II) no primary bus or device found
[   175.756] (II) LoadModule: "glx"
[   175.757] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   175.779] (II) Module glx: vendor="X.Org Foundation"
[   175.779] 	compiled for 1.20.11, module version = 1.0.0
[   175.779] 	ABI class: X.Org Server Extension, version 10.0
[   175.779] (==) Matched modesetting as autoconfigured driver 0
[   175.779] (==) Matched fbdev as autoconfigured driver 1
[   175.779] (==) Assigned the driver to the xf86ConfigLayout
[   175.779] (II) LoadModule: "modesetting"
[   175.780] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[   175.783] (II) Module modesetting: vendor="X.Org Foundation"
[   175.783] 	compiled for 1.20.11, module version = 1.20.11
[   175.783] 	Module class: X.Org Video Driver
[   175.783] 	ABI class: X.Org Video Driver, version 24.1
[   175.783] (II) LoadModule: "fbdev"
[   175.784] (WW) Warning, couldn't open module fbdev
[   175.784] (EE) Failed to load module "fbdev" (module does not exist, 0)
[   175.784] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   175.784] (EE) 
Fatal server error:
[   175.784] (EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
[   175.787] (EE) 
[   175.789] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   175.797] (EE) Please also check the log file at "/home/dietpi/.local/share/xorg/Xorg.0.log" for additional information.
[   175.799] (EE) 
[   175.802] (WW) xf86CloseConsole: KDSETMODE failed: Bad file descriptor
[   175.802] (WW) xf86CloseConsole: VT_GETMODE failed: Bad file descriptor
[   175.802] (EE) Server terminated with error (1). Closing log file.
```

Joulinar, 2022-05-31:
What was the user used to start Chromium?

`parse_vt_settings: Cannot open /dev/tty0 (Permission denied)`

MDAR, 2022-05-31:
> What was the user used to start Chromium?
> 
> `parse_vt_settings: Cannot open /dev/tty0 (Permission denied)`

On this occasion, it was set as ```dietpi```

I'll try ```root``` and see what happens

MDAR, 2022-07-05:
Hi

Just an update.

I've installed a fresh / vanilla DietPi V8.6.1 onto a new eMMC card for an Odroid C4 today and tried the whole AutoStart with Chromium thing. (yes, I installed Chromium)

When the machine rebooted, I got the whole "Stack Smashing Detected" warning again.

Is there any value in sharing any of the logs with you to hunt down this issue?

Cheers,
Stuart

MichaIng, 2022-07-05:
Sadly not. Bug has been reported to Debian bug tracker with other users verifying it, I guess the logs would just repeat what has been posted already: https://bugs.debian.org/1006457

kt1024, 2022-07-06:
can we get old and working version of chromium somewhere, then manually install it?

Joulinar, 2022-07-06:
well we install Chromium via apt package manger from official Debian repository. DietPi themselves is not providing a version.

ohdearaugustin, 2022-07-06:
You can get older debian packages directly from the repository at http://ftp.debian.org/debian/pool/main/c/chromium/. Maybe someone will find this helpful.

Joulinar, 2022-07-07:
be careful, These are not different versions you can install as they are dedicated to a Debian version. Have a look to the ending where you see deb9 (Stretch), deb10 (Buster) and deb11 (Bullseye). These are the same versions you will get via `apt install`
https://packages.debian.org/search?suite=all&searchon=names&keywords=chromium

kt1024, 2022-07-08:
install old version did not work.
what I did: 

- remove chromium with dietpi-software, and verify no chromium package remained.
- `dpkg -i chromium_89.0.4389.114-1_deb10u1_arm64.deb`  failed. 
- then run `apt install --fix-missing` & `apt --fix-broken install`
- reboot, kiosk screen not work.
- finds out chromium package version is `103` not `89`. (did not run dietpi-update or apt update/upgrade in above tests)

MichaIng, 2022-07-08:
Install the package like this:
```bash
apt install --allow-downgrades ./chromium_89.0.4389.114-1_deb10u1_arm64.deb
```
If that fails, show the full error log.

Joulinar, 2022-07-08:
This is the buster version. Does it gonna work on Bullseye?

MichaIng, 2022-07-08:
Probably not, maybe a missing (old) library is the reason for the install failure. There is a much newer version which worked: https://github.com/MichaIng/DietPi/issues/5300#issuecomment-1060045619
But not sure whether it's still available on repo. In Debian snapshots at least one should be able to find it.

kt1024, 2022-07-08:
> Install the package like this:
> 
> ```shell
> apt install --allow-downgrades ./chromium_89.0.4389.114-1_deb10u1_arm64.deb
> ```
> 
> If that fails, show the full error log.

<pre>
ken@tv1:~$ sudo apt install --allow-downgrades ./chromium_89.0.4389.114-1_deb10u1_arm64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'chromium' instead of './chromium_89.0.4389.114-1_deb10u1_arm64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 chromium : Depends: libevent-2.1-6 (>= 2.1.8-stable) but it is not installable
            Depends: libicu63 (>= 63.1-1~) but it is not installable
            Depends: libjsoncpp1 (>= 1.7.4) but it is not installable
            Depends: libre2-5 (>= 20160901) but it is not installable
            Depends: libvpx5 (>= 1.6.0) but it is not installable
            Depends: chromium-common (= 89.0.4389.114-1~deb10u1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
</pre>

kt1024, 2022-07-08:
just installed `chromium-browser_95.0.4638.78-rpt6_arm64.deb` from here `https://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/`

<pre>
ken@tv1:~$ sudo dpkg --list | grep chrom
rc  chromium-browser                95.0.4638.78-rpt6              arm64        Chromium web browser, open-source version of Chrome
</pre>

[image]

MichaIng, 2022-07-08:
Because you uninstalled Chromium via `dietpi-software` 😉. Does it start with `startx /usr/bin/chromium-browser`? If so, install it again via `dietpi-software`, then downgrade and `apt-mark hold chromium-browser` to prevent it from being upgraded.

kt1024, 2022-07-11:
tried v92 & 95, both failed. copied `chromium-autostart.sh` to `/var/lib/dietpi/dietpi-software/installed/`
[image]
<pre>
ken@tv1:/var/log$ cat Xorg.0.log 
[     7.554] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[     7.554] Build Operating System: linux Debian
[     7.554] Current Operating System: Linux tv1 5.10.123-meson64 #22.05.3 SMP PREEMPT Wed Jun 22 07:23:04 UTC 2022 aarch64
[     7.554] Kernel command line: root=UUID=dffc3253-1d83-4c04-9cf7-65ede07d866f rootfstype=ext4 rootwait console=tty1 loglevel=4 consoleblank=0 coherent_pool=2M ubootpart=4d4ec9c8-01 libata.force=noncq usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u net.ifnames=0
[     7.554] Build Date: 16 December 2021  05:08:23PM
[     7.555] xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support) 
[     7.555] Current version of pixman: 0.40.0
[     7.555] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[     7.555] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     7.555] (==) Log file: "/var/log/Xorg.0.log", Time: Mon Jul 11 14:51:22 2022
[     7.562] (==) Using config directory: "/etc/X11/xorg.conf.d"
[     7.562] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     7.563] (==) No Layout section.  Using the first Screen section.
[     7.563] (==) No screen section available. Using defaults.
[     7.563] (**) |-->Screen "Default Screen Section" (0)
[     7.563] (**) |   |-->Monitor "<default monitor>"
[     7.564] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[     7.564] (**) Option "BlankTime" "0"
[     7.564] (==) Automatically adding devices
[     7.565] (==) Automatically enabling devices
[     7.565] (==) Automatically adding GPU devices
[     7.567] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     7.577] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     7.577] 	Entry deleted from font path.
[     7.577] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     7.577] 	Entry deleted from font path.
[     7.577] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     7.577] 	Entry deleted from font path.
[     7.577] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[     7.577] 	Entry deleted from font path.
[     7.577] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     7.577] 	Entry deleted from font path.
[     7.577] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     7.577] 	Entry deleted from font path.
[     7.577] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[     7.577] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.577] (**) Extension "DPMS" is disabled
[     7.577] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[     7.577] (II) Loader magic: 0xaaaab5af1e28
[     7.577] (II) Module ABI versions:
[     7.577] 	X.Org ANSI C Emulation: 0.4
[     7.577] 	X.Org Video Driver: 24.1
[     7.577] 	X.Org XInput driver : 24.1
[     7.577] 	X.Org Server Extension : 10.0
[     7.579] (--) using VT number 2

[     7.579] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     7.580] (II) xfree86: Adding drm device (/dev/dri/card0)
[     7.588] (II) xfree86: Adding drm device (/dev/dri/card1)
[     7.593] (II) no primary bus or device found
[     7.593] 	falling back to /sys/devices/platform/soc/ff900000.vpu/drm/card0
[     7.593] (II) LoadModule: "glx"
[     7.595] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     7.621] (II) Module glx: vendor="X.Org Foundation"
[     7.621] 	compiled for 1.20.11, module version = 1.0.0
[     7.621] 	ABI class: X.Org Server Extension, version 10.0
[     7.621] (==) Matched modesetting as autoconfigured driver 0
[... 148 lines cut ...]
[     9.429] (II) Initializing extension RENDER
[     9.430] (II) Initializing extension RANDR
[     9.431] (II) Initializing extension COMPOSITE
[     9.431] (II) Initializing extension DAMAGE
[     9.431] (II) Initializing extension MIT-SCREEN-SAVER
[     9.432] (II) Initializing extension DOUBLE-BUFFER
[     9.432] (II) Initializing extension RECORD
[     9.433] (II) Initializing extension Present
[     9.433] (II) Initializing extension DRI3
[     9.434] (II) Initializing extension X-Resource
[     9.434] (II) Initializing extension XVideo
[     9.435] (II) Initializing extension XVideo-MotionCompensation
[     9.435] (II) Initializing extension SELinux
[     9.435] (II) SELinux: Disabled on system
[     9.435] (II) Initializing extension GLX
[     9.709] (II) AIGLX: Loaded and initialized meson
[     9.709] (II) GLX: Initialized DRI2 GL provider for screen 0
[     9.709] (II) Initializing extension XFree86-VidModeExtension
[     9.710] (II) Initializing extension XFree86-DGA
[     9.710] (II) Initializing extension XFree86-DRI
[     9.711] (II) Initializing extension DRI2
[     9.722] (II) modeset(0): Damage tracking initialized
[     9.722] (II) modeset(0): Setting screen physical size to 1016 x 571
[    10.418] (II) Server terminated successfully (0). Closing log file.
</pre>

MDAR, 2022-07-11:
Here's a thing

after posting that message above, I happened to catch the first line

```[82287:82287:0711/100846.508963:ERROR:process_singleton_posix.cc(353)] The profile appears to be in use by another Chromium process (2357) on another computer (DietPi). Chromium has locked the profile so that it doesn't get corrupted. If you are sure no other processes are using this profile, you can unlock the profile and relaunch Chromium.
```

So I cleared the profile with this command

`rm -rf ~/.config/chromium/Singleton*`

While it didn't fix much, I now see a much bigger white box with the mouse in, moments before the X session closes.

And the last line of the error on screen reads

`Closing log file.d; returned -1, SSL error code 1, net_error -3xinit: connection o S server lost`

MDAR, 2022-07-12:
> Did you verify that there is no other Chromium process running already? Of course it should then show up on screen, but it's possible it runs on a different TTY. The first error like "fd already in use" indicates that something is using the framebuffer already (while it should use DRM/KMS actually), but I might interpret it wrongly.

Well spotted, I hadn't seen that message.

From a fresh boot, I'm only trying to run Chromium, so I'd like to think nothing has got there before us.

However, anything is possible.

Does it mean the X Session it's meant to be running within?

kt1024, 2022-07-21:
test with raspberry pi 4 today (dietpi 8.1 upgrade to 8.6.1 also) . it is a working kiosk with chromium v98, after upgrade to chromium v101. kiosk mode still works.
if upstream will not fix the issue, mind consider use raspberry repo for chromium installation? @MichaIng

MichaIng, 2022-07-21:
On RPi this was never an issue, on any other the RPi `chromium-browser` package cannot be used.

MDAR, 2022-07-21:
> I was thinking whether the `dietpi-autostart` option is probably on, but on the other hand that would be on TTY1 and should be visible on screen 🤔. Any X session should however show up in `htop`.

I'm going to start again with a blank eMMC card and see what happens.

(I'm sure I haven't done anything, but happy to start over)

MDAR, 2022-07-27:
I'm not sure what to do with the following information.

Your guidance would be most welcome.

***

A brand new eMMC with nothing but a fresh DietPI image on a C4, trying to get Chromium working, just fails with the same stack smashing issue.

However.

I have got Vivaldi working full screen, it just complains that it doesn't have HTML5 H.264 support loaded.

********

I downloaded Debian Vivaldi from here
https://vivaldi.com/download/

and installed with it this ```apt install ./vivaldi-stable_5.3.2679.70-1_arm64.deb```

then I edited the Chromium autostart file (because I wanted a template to work from)

so that it looks like this

```
#!/bin/dash
# Autostart script for kiosk mode, based on @AYapejian: https://github.com/MichaIng/DietPi/issues/1737#issue-318697621

# Resolution to use for kiosk mode, should ideally match current system resolution
RES_X=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_RES_X=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
RES_Y=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_RES_Y=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)

# Command line switches: https://peter.sh/experiments/chromium-command-line-switches/
# - Review and add custom flags in: /etc/chromium.d
CHROMIUM_OPTS="--kiosk --window-size=${RES_X:-1280},${RES_Y:-720} --window-position=0,0 --no-sandbox"

# If you want tablet mode, uncomment the next line.
#CHROMIUM_OPTS+=' --force-tablet-mode --tablet-ui'

# Home page
URL=$(sed -n '/^[[:blank:]]*SOFTWARE_CHROMIUM_AUTOSTART_URL=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)

CHROMIUM_OPTS=$CHROMIUM_OPTS" --url="$URL

# RPi or Debian Chromium package
# FP_CHROMIUM=$(command -v chromium-browser)

 FP_CHROMIUM=$(command -v vivaldi)

[ "$FP_CHROMIUM" ] || FP_CHROMIUM=$(command -v chromium)

# Use "startx" as non-root user to get required permissions via systemd-logind
STARTX='xinit'
[ "$USER" = 'root' ] || STARTX='startx'

echo trying $FP_CHROMIUM with these options
echo  $CHROMIUM_OPTS

 exec "$STARTX" "$FP_CHROMIUM" $CHROMIUM_OPTS
# xinit  "$FP_CHROMIUM" $CHROMIUM_OPTS "${URL:-https://dietpi.com/}"

exit 0

```

****************

I get Vivaldi at full screen (after editing dietpi.txt and setting the X and Y res correctly

```
root@DietPi:/var/lib/dietpi/dietpi-software/installed# sh chromium-autostart.sh
[... 23 lines cut ...]
[6427:6456:0727/183354.588127:ERROR:bus.cc(398)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6427:6456:0727/183354.588255:ERROR:bus.cc(398)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6427:6427:0727/183354.718163:ERROR:image_generic.cc(77)] Unable to decode PNG for 1.
[6427:6517:0727/183354.821591:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6427:6517:0727/183354.823188:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6427:6517:0727/183354.824720:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6427:6427:0727/183356.121245:ERROR:gpu_process_host.cc(975)] GPU process exited unexpectedly: exit_code=11
[6427:6427:0727/183357.508576:ERROR:gpu_process_host.cc(975)] GPU process exited unexpectedly: exit_code=11
[6427:6427:0727/183358.845877:ERROR:gpu_process_host.cc(975)] GPU process exited unexpectedly: exit_code=11
[6493:6506:0727/183400.379673:ERROR:command_buffer_proxy_impl.cc(126)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
[6427:6427:0727/183401.874966:ERROR:CONSOLE(1)] "syncDetachedTabInformation: The message port closed before a response was received.", source: chrome-extension://mpognobbkildjkofajifpdfhcoklimli/bundle.js (1)
[6427:6427:0727/183401.876212:ERROR:CONSOLE(1)] "syncDetachedTabInformation: The message port closed before a response was received.", source: chrome-extension://mpognobbkildjkofajifpdfhcoklimli/bundle.js (1)
[6427:6427:0727/183401.877261:ERROR:CONSOLE(1)] "syncDetachedTabInformation: The message port closed before a response was received.", source: chrome-extension://mpognobbkildjkofajifpdfhcoklimli/bundle.js (1)

```

Anything you can do to help or resolve the H.264 playback would be amazing.

I'm just glad that I've got some kind of full screen browser again, I'll try it with the rest of my setup tomorrow.

Thanks to everyone in advance.

@kt1024 

Does this help you in any way?

MDAR, 2022-07-28:
Good morning

Well this didn't take long at all.

I just dropped Vivaldi onto my development machine
Installed it

Swapped any reference to Chromium to Vivaldi and it works for everything other than the video playback.
(Which isn't used by many, but I do need to get it working for when they do ask for it)

The command line that is being used to start Vivaldi is this

```sudo xinit /usr/bin/vivaldi ---kiosk --test-type --window-size=1920,1080 --no-sandbox --start-fullscreen --start-maximized --window-position=0,0 --no-user-gesture-required --app=http://127.0.0.1:1880/content/index.html -- -nocursor``` 

**Note the `--no-sandbox` when running as root**
and that I have hidden the cursor in Xinit

Which isn't an issue in the closed environment that I'm working in, but I guess it is a big issue for those that pull content from the web.

I would be delighted to hear about other people's results.

One small detail that does confuse me.... I think Vivaldi is based on Chromium..... _so why is this working?_

kt1024, 2022-07-29:
@MDAR does it have `chromium-codecs-ffmpeg-extra` package installed?

I may try this later sometime, coz I've made a rpi4 as kiosk.

MDAR, 2022-07-29:
> @MDAR does it have `chromium-codecs-ffmpeg-extra` package installed?
> 
> I may try this later sometime, coz I've made a rpi4 as kiosk.

@kt1024 
Hi

By the looks of it, that package isn't available

```
root@LiveStream-C4:~# apt install chromium-codecs-ffmpeg-extr* --fix-missing
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package chromium-codecs-ffmpeg-extr*
E: Couldn't find any package by glob 'chromium-codecs-ffmpeg-extr*'
```

However, the good people at Vivaldi suggest that the application should "download what it needs" on first boot.

I'm waiting for a reply on their forum (as per link above)

If you don't need video playback, Vivaldi certainly doing everything else I need it to, in full unattended kiosk mode, with WebSocket remote control.

kt1024, 2022-07-29:
my bad, should be 2 packages, `chromium-codecs-ffmpeg` and `chromium-codecs-ffmpeg-extra`

MichaIng, 2022-07-29:
These are RPi-only packages from the RPi repository, responsible only for video decoding, not for startup, and on RPi the issue does not exist.

MDAR, 2022-08-02:
@ruario
 posted a comment on the Vivaldi issue, that gets me H.264 playback in Vivaldi

https://gist.github.com/ruario/bec42d156d30affef655#gistcomment-4251813

>
>
>I know why H.264 is failing for you and hope to get out a fix soon. In the mean time I can give you a work around.
>
>If it is armhf
>
```
wget -O- https://launchpadlibrarian.net/614082413/chromium-codecs-ffmpeg-extra_103.0.5060.134-0ubuntu0.18.04.1_armhf.deb | tail -c+1157 | tar xOJ ./usr/lib/chromium-browser/libffmpeg.so > libffmpeg.so.5.3
sudo install -m755 libffmpeg.so.5.3 /opt/vivaldi/libffmpeg.so.5.3
```
>*********************
>If it is arm64
>
```
wget -O- https://launchpadlibrarian.net/614041844/chromium-codecs-ffmpeg-extra_103.0.5060.134-0ubuntu0.18.04.1_arm64.deb | tail -c+1157 | tar xOJ ./usr/lib/chromium-browser/libffmpeg.so > libffmpeg.so.5.3
sudo install -m755 libffmpeg.so.5.3 /opt/vivaldi/libffmpeg.so.5.3
```

It certainly enables h.264 playback, with user interaction, so for me that is >70% the way to a solution.

I need to send F11 to the browser *twice* (even if it is full screen / kiosk) before the videos will autoplay.

After that, everything works perfectly.
So only the AutoPlay without any user interaction to resolve

MDAR, 2022-10-07:
Hi

I think there has been a breakthough

I'm having an issue with a machine that is forcing me to update it to the very latest version of Debian.

I didn't want to do this due to the Chromium issue.
However, I've tried an update on a machine that doesn't use Chromium in Kiosk and it now seems to be working.

```
dpkg -s chromium

Installed-Size: 172351
Maintainer: Debian Chromium Team <chromium@packages.debian.org>
Architecture: arm64
Version: 89.0.4389.114-1~deb10u1
```

Can anyone else try this and let me know if the Chromium issue has been resolved, before I update this one ,machine?

Thanks
Stuart

Joulinar, 2022-10-07:
> a machine that is forcing me to update it to the very latest version of Debian

At least from DietPi side, we don't force anyone to update. We just notify user about available updates. 

> Can anyone else try this and let me know if the Chromium issue has been resolved, before I update this one ,machine?

You could create a dietpi-backup and check if latest version of Chromium is working. If not, simply perform a restore. If you use SD card, you could create an offline close as well.

MDAR, 2022-10-07:
> based on the Chromium version, this system seems to be still a Buster system.

DAM IT

I'd forgotten about that detail.

I must have picked up my only Buster based box..... I wonder how that missed the "great upgrade"

MichaIng, 2022-10-07:
Hehe, I'm not aware of any info/notification a system would get about this upgrade, but our MOTD 😉.

MichaIng, 2022-11-21:
Finally the issue got solved with latest Kodi package on Bullseye: https://bugs.debian.org/1006457

closed 2022-11-21 by MichaIng

MDAR, 2022-11-21:
@MichaIng 

What an absolute fabulous fella you are.

Thank you so much for all your effort.

When I get back to my office, I'll update my production machine and let you know how it goes.

(I'm sure it'll be fine)

🎉🎊

MDAR, 2022-12-22:
Hello

I can confirm that everything now works really well with a fresh install of DietPi V8.12.1 on Odroid C4 machines