MichaIng/DietPi #5301: dietpi-backup 1 is blocking completion of my script

N7-BADA, 2022-02-20:
i've made a script to run and report some commands/info to my telegram.

below here is a part of the code and it seems as if ```/boot/dietpi/dietpi-backup 1 1> /dev/null 2>&1``` is waiting for user interaction whe it completed as i can see it in ```htop```

htop : ```26052 root       3928   0  0.0  0:00.01 whiptail --title DietPi-Backup --backtitle RPi 4 Model B (armv7l) | IP: 192.168.178.250 --msgbox Backup completed:\n - /mnt/Pi-backups/PiPlex-dietpi-backup/dietpi-backup --ok-button Ok 8 120```

```
if [ $(date +%A) = Monday ] || [ $(date +%A) = Sunday ]; then
    sendMessage "Running Checks.."  1> /dev/null 2>&1
    if [ -f "/mnt/Pi-backups/PiPlex-dietpi-backup/dietpi-backup/.dietpi-backup_stats" ] ; then
        sendMessage "Starting Backup.."  1> /dev/null 2>&1
	/boot/dietpi/dietpi-backup 1 1> /dev/null 2>&1
	sleep 1
    fi
    if [ $(date +%A) = Sunday ]; then
        sendMessage "Setting Permissions.."  1> /dev/null 2>&1
	chmod 775 -R /mnt/Plex >/dev/null 2>&1
	chown -R qbittorrent:root /mnt/Plex/Torrents >/dev/null 2>&1
        sleep 1
    fi
```

is there any way to run ```dietpi-backup 1``` without the promts after thit so my script can continue?

MichaIng, 2022-02-20:
Do you call your script manually (from an interactive shell)? In cron jobs and systemd units, it will skip the whiptail prompts automatically.

Else, you can do:
```sh
G_INTERACTIVE=0 /boot/dietpi/dietpi-backup 1
```

N7-BADA, 2022-02-20:
> Do you call your script manually (from an interactive shell)? In cron jobs and systemd units, it will skip the whiptail prompts automatically.
> 
> Else, you can do:
> 
> ```shell
> G_INTERACTIVE=0 /boot/dietpi/dietpi-backup 1
> ```

im planning to add this in a cronjob, but atm im just manually calling it for testing

MichaIng, 2022-02-20:
Okay. As said, in the cron job you could omit the flag, but it doesn't hurt either.

closed 2022-02-20 by N7-BADA