MichaIng/DietPi #5302: Can not disable MPD in v8.1 Allo GUI

6SN7dual, 2022-02-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | v8.1
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

All services can be disabled except MPD. The Save Changes function never works after selecting to disable that service.
#### Expected behaviour

- 
- ...

#### Extra details

Also, there is an unusual grep reference in the Allo GUI MPD window:

Status :ACTIVE
Freq :Grep: /Etc/Mpd.conf: Permission Denied
HzBit :Grep: /Etc/Mpd.conf: Permission Denied
- ...

MichaIng, 2022-02-20:
Many thanks for your report. Could you check:
```sh
ls -dl /etc /etc/mpd.conf
```

6SN7dual, 2022-02-21:
root@DietPi:~# ls -dl /etc /etc/mpd.conf
drwxr-xr-x 79 root root  4096 Feb 20 17:01 /etc
-rw-r-----  1 mpd  root 13497 Dec 15 20:16 /etc/mpd.conf

MichaIng, 2022-02-21:
```sh
chmod 0644 /etc/mpd.conf
```
But I see the strict modes are the default. Looks like we need to read that file via `sudo`. Will be fixed soon.

6SN7dual, 2022-02-21:
That was speedy, thank you!!!

MichaIng, 2022-03-03:
Fixed with current beta, along with a larger rework if the implementation: #5328

If you find time, would be great if you could test it: #5319

MichaIng, 2022-03-05:
It has now been released. The Allo GUI update will be applied automatically on DietPi update.

closed 2022-03-05 by MichaIng