MichaIng/DietPi #5303: Unmet Requirement for MicroK8S Installation

dmgolembiowski, 2022-02-21:
Follow up issue on behalf of @silverraindog and @MichaIng related to the MicroK8s installation. I'm invested in a boot image for this so I'm opening this issue.

The details from the pull request included:
---
> > Thanks!
> > 
> > > Can you verify it on your system via modinfo vxlan?
> > 
> > Yup, I see similar output to what you've posted.
TL;DR - I believe this was what they were talking about:

<p>
Hmm, reading though the linked solution from your link (https://github.com/k3s-io/k3s/issues/4234) indicates that on Ubuntu it was the missing `vxlan` kernel module, which is indeed shipped with that package on Ubuntu: https://packages.ubuntu.com/linux-modules-extra-raspi

With the official Raspberry Pi kernel we use, it is however shipped OOTB:
```
# modinfo vxlan
filename:       /lib/modules/5.10.92-v7+/kernel/drivers/net/vxlan.ko
alias:          rtnl-link-vxlan
description:    Driver for VXLAN encapsulated traffic
author:         Stephen Hemminger <stephen@networkplumber.org>
version:        0.1
license:        GPL
srcversion:     A772666323B3A673FA6A7F1
depends:        udp_tunnel,ip6_udp_tunnel
intree:         Y
name:           vxlan
vermagic:       5.10.92-v7+ SMP mod_unload modversions ARMv7 p2v8
parm:           udp_port:Destination UDP port (ushort)
parm:           log_ecn_error:Log packets received with corrupted ECN (bool)
2022-02-17 04:39:51 root@micha:/var/log# modinfo vxlan
filename:       /lib/modules/5.10.92-v7+/kernel/drivers/net/vxlan.ko
alias:          rtnl-link-vxlan
description:    Driver for VXLAN encapsulated traffic
author:         Stephen Hemminger <stephen@networkplumber.org>
version:        0.1
license:        GPL
srcversion:     A772666323B3A673FA6A7F1
depends:        udp_tunnel,ip6_udp_tunnel
intree:         Y
name:           vxlan
vermagic:       5.10.92-v7+ SMP mod_unload modversions ARMv7 p2v8
parm:           udp_port:Destination UDP port (ushort)
parm:           log_ecn_error:Log packets received with corrupted ECN (bool)
```
Can you verify it on your system via `modinfo vxlan`?

_Originally posted by @MichaIng in https://github.com/MichaIng/DietPi/issues/5239#issuecomment-1042544209_
</p>

> Thanks!
> 
> > Can you verify it on your system via modinfo vxlan?
> 
> Yup, I see similar output to what you've posted.
> ...
> Merging this now, so we can test on dev branch. We can fine tune via dedicated PRs/commits.
---
Anything I can do to help debug?

MichaIng, 2022-03-26:
@dmgolembiowski 
Is this still an issue without our current MicroK8s implementation?

dmgolembiowski, 2022-03-27:
Sorry, I haven't tested this yet.

MichaIng, 2022-04-03:
I'll mark this as closed for now. In the meantime next Linux LTS is available on most SBCs, so if the issue persists, let's open a new issue with fresh logs and comparison with other systems where CPU load is significantly lower.

closed 2022-04-03 by MichaIng