MichaIng/DietPi #5305: Can't update even after changing mirrors.

Jalh, 2022-02-21:
#### Details:
- Date           | Mon Feb 21 16:08:45 GMT 2022
- Bug report     | 086f9979-a982-424f-b713-46a9af00df7a
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Tried to update the system using the dietpi-update command.
2. ...
#### Expected behaviour:

- Update the IS
#### Actual behaviour:

-  Getting connection error.
#### Extra details:

- System is running pihole, but i had it disable prior to updating the system to avoid conflicts
#### Additional logs:
```
Ign:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:2 http://raspbian.raspberrypi.org/raspbian buster Release
  Cannot initiate the connection to raspbian.raspberrypi.org:80 (93.93.128.193). - connect (101: Network is unreachable)
Ign:3 https://archive.raspberrypi.org/debian buster InRelease
Err:4 https://archive.raspberrypi.org/debian buster Release
  Cannot initiate the connection to archive.raspberrypi.org:443 (93.93.135.117). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (93.93.135.118). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (93.93.135.141). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (176.126.240.84). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (176.126.240.86). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (176.126.240.167). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (46.235.227.39). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (46.235.230.122). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (46.235.231.111). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (46.235.231.145). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (46.235.231.151). - connect (101: Network is unreachable) Cannot initiate the connection to archive.raspberrypi.org:443 (93.93.130.212). - connect (101: Network is unreachable)
Reading package lists...
E: The repository 'http://raspbian.raspberrypi.org/raspbian buster Release' no longer has a Release file.
E: The repository 'https://archive.raspberrypi.org/debian buster Release' no longer has a Release file.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MichaIng, 2022-02-21:
It looks like your network is not configured correctly. "Network is unreachable" usually means that there is no (valid) default route/gateway:
```sh
ip r
```

Jalh, 2022-02-21:
> It looks like your network is not configured correctly. "Network is unreachable" usually means that there is no (valid) default route/gateway:
> 
> ```shell
> ip r
> ```
Returns:
192.168.10.0/24 dev eth0 proto kernel scope link src 192.168.10.53

MichaIng, 2022-02-21:
So indeed no default route. Please verify your network settings. Probably `ifup --force eth0` solves it already, when the interface with gateway is defined correctly in `/etc/network/interfaces`. Did you use `dietpi-config` to configure the network, or did not touch those at all (so that defaults are used)? If so, interesting would be also:
```sh
cat /etc/network/interfaces
```

Jalh, 2022-02-21:
> ifup --force eth0

returns:

RTNETLINK answers: File exists
ifup: failed to bring up eth0

but I was able to fix it by modifying the static gateway, it was incorrectly set up.

Greetings.

closed 2022-02-21 by Jalh