MichaIng/DietPi #5310: Migration to a fresh image using script dietpi-cloud-migration almost succesful

echtfrank, 2022-02-23:
Hello,

I have migrated my Odroid HC4 to a new image. Using the dietpi-cloud-migration script. I was able to do a Nextcloud backup on the old system and on the new system installed Nextcloud again.

_Originally posted by @MichaIng in https://github.com/MichaIng/DietPi/issues/5269#issuecomment-1048131993_

On my lan I can login to Nextcloud by all users and it is working fine! However, I can't reach it from the internet anymore. Letsencrypt seems to be installed correctly and pinging to my domainname gives me the correct outside IP of my router. I haven't changed anything in portforwarding or things like that.

What can I try or verify?

Thanks in advance,
Frank

echtfrank, 2022-02-23:
@Joulinar What a quick response!
- locally: HTTPS (IP) does NOT work
- locally: HTTP (IP) DOES work
- from outside, if I browse to HTTP://<DOMAIN> (without "/nextcloud")  I get to a NGINX-page.

I think but I am not sure that my IP is the same.

Joulinar, 2022-02-23:
Pls run `dietpi-letsencrypt` to check if your certificate was already created. If not, just created one. Just running the installation of `dietpi-letsencrypt` will not create the certificate automatically.

echtfrank, 2022-02-23:
Was it THAT easy??? Blush, blush... That was it... Working good now!

closed 2022-02-23 by echtfrank

Joulinar, 2022-02-23:
Small things could make a huge difference 😉