MichaIng/DietPi #5313: Automatic setup fails because of armbian.lv.auroradev.org being down

jpeg2600, 2022-02-23:
I'm trying to streamline the whole process to deploy on-edge devices running Node-Red. The whole thing works well but it's been failing in the last hour because it can't reach armbian.lv.auroradev.org. When it does, it stops the automation process altogether. 

In my case, it's easier the rewrite the image and start over than going from there. I'll try removing that source before the first boot since it looks unreliable from other tickets.

jpeg2600, 2022-02-23:
Tagging #5190 

Looks like I can't remove that mirror myself.

MichaIng, 2022-02-23:
The Armbian repository is added via mirror director, so by chance that one is picked or a single connection or not. But actually, I can access it just fine here, probably it was done when you tried but is now up again? https://armbian.lv.auroradev.org/

jpeg2600, 2022-02-24:
Yes it's fine now. Maybe the automation script could retry a few times before before giving up the whole process? I think an installation that hangs a bit is better than having to manually finish it or start over if it fails. 

Thanks!

MichaIng, 2022-02-24:
Hmm, for `apt update` it probably makes sense, at least when it is an automated first run setup. However, it is likely that just trying two times after another would fail both, while would require a retry probably one hour later when the issue is solved APT server side 🤔. Also many other failure reasons would just fail on a retry, like when DNS is not working, a certificate expired, or the system time out of date so that certificates are seen as "not yet valid". So currently I guess an automatic retry would help only in very rare cases.

closed 2022-09-17 by MichaIng