MichaIng/DietPi #5314: Can't start sway

97carmine, 2022-02-24:
I have done a minimum installation of DietPi on a Rock Pi 4, I install sway from the Debian repositories but it is impossible for me to start it showing the following error:

```
dietpi@DietPi:~$ sway
XDG_RUNTIME_DIR is not set in the environment. Aborting.
```

#### Information
- DietPi version:
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version: `bulleye`
- Kernel version: `Linux DietPi 5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux`
- SBC model: `ROCK Pi 4 (aarch64)`

MichaIng, 2022-02-24:
This seems to be a Wayland compositor, do you run a desktop on Wayland (instead of X11) on your system? If it asks for `XDG_RUNTIME_DIR`, it needs to be started from within the Wayland session, or as part of the session startup scripts (I have no idea how this is handled with Wayland).

I see it ships with an `/usr/share/wayland-sessions/sway.desktop` desktop icon, so this should work. I guess you can add it to the desktop's/window manager's autostart applications to have it loaded automatically when starting the session.

At least it is expected that it cannot be started like that from raw console or SSH.

97carmine, 2022-02-24:
I've fixed it with [this](https://unix.stackexchange.com/questions/239489/dbus-system-failed-to-activate-service-org-freedesktop-login1-timed-out/573105#573105).

I close it.

closed 2022-02-24 by 97carmine

MichaIng, 2022-02-24:
Ah, okay makes sense 🙂.