MichaIng/DietPi #5315: PLex Media server (optimized software 42) is not working to create library, or does not allow connection

KalixtShawxo, 2022-02-25:
two issues.  
I had previously installed Plex from DietPi-Software on a v7.x and it was fine. 
I am now trying to do it on a v8.1.2 and it runs through all the hoops, and when I go to the web UI, IF it gives me a chance to login, it skips the screen to create the server and add your library, it goes to a screen to down load media server to your computer. 
after several iterations of the same thing happening from clean installs,,,, I tried installing Plex from outside sources, and all that got me was a grey screen saying "not authorized" you do not have access to this server. 
I've uninstalled the standard versions, and the optimized version.... but no luck. 
Which is very odd cause like I said I did have it installed on another SBC in v 7.x which has been upgraded to latest without issues.  
Now, I just can't get it to work.   First it installed, but was missing functionality, now I can't even get into it.  
I'm about to blow out the eMMC on this C2 and put a completely new build on and go right into Plex..... unless someone much smarter than me has some other suggestions?  
This was an SBC that was previously installed with DietPi but was sitting unused for the last couple weeks.
 
thanks in advance for any help.

KalixtShawxo, 2022-02-25:
So here's what I did to further troubleshoot. 
using the Odroid C2 image  for ARM8, I put it on a 16GB eMMC module and booted off that and ran through the setup via SSH. 
upon completion of the automatic updates and upgrades, I did nothing else but run the Plex Media Server (#42) install from Dietpi-software. 
after install of Plex, I remtely went to the web UI at the IP address :32400/web
I was at this time presented with the sign in screen asking me if I wanted to sign in to my plex account, which was already anmed.
I clicked on "sign In".  after a second it took me to the "Your home screen is empty" page. 
from there all I had on the left menu was a "your media" and when I clicked on that, I get the "Add yoour media to plex" screen with the only option to "get Plex Media Server" for the computer I am on.

so what is broken here?  I should have Plex working on the SBC and I should be able to create a library, but I can't. 
All I am trying to use it for is to create a slide show for the collections of pictures I sync'd from my daughter's google photos of my granddaughter for the past two years.  

I hope this is a better explanation.
[image]
[image]

MichaIng, 2022-02-26:
Many thanks for your report. If this was a fresh Plex install, please try to purge and reinstall it:
```sh
apt purge plexmediaserver
rm -R /var/lib/plexmediaserver
apt install plexmediaserver
```

Did you try a different browser (cleared cache), assured that it is possible to use 2 Plex servers with one account? And are there any error messages?
```sh
journalctl -u plexmediaserver
```

I can try to replicate on an `aarch64` SBC, just to assure it generally works, but I suggest, collecting the above info, that you search or ask at the Plex forum: https://forums.plex.tv/
We only install the official DEB package from their official APT repository, so I'd be a bug their end, or a limitation or usage issue where the Plex community will be much better able to help with. You can ping me there @MichaIng as well, when you open a topic.

MichaIng, 2022-04-14:
Marking as closed due to outstanding reply. Feel free to reopen if required.

closed 2022-04-14 by MichaIng

KalixtShawxo, 2022-04-15:
never got it working via the dietpi-software. removed all traces, installed from standard PLEX methods and not from the dietpi-software and was able to get it working.

 
 

Closed #5315.

MichaIng, 2022-04-15:
Did your try the package reinstall and checked for error messages as suggested above?

KalixtShawxo, 2022-04-18:
I had tried purge and reinstall, same problems, did the clean install from PLEX and it worked. never got the error messages. This was on two different SBCs with DietPi. a ROCK64 and Odroid C2. Neither would install and work properly from a clean Dietpi install with the dietpi-software options. so it could still be me, but not looking like it.  I got the PLEX working now, so I'm good. but there's something somewhere. I'll let this close now, I just wanted to make you aware it works not using the dietpi-software version.  for me at least.

MichaIng, 2022-04-20:
> did the clean install from PLEX and it worked

What do you mean by that? `dietpi-software` shouldn't do any different: It installs the official Plex APT repository and installs the `plexmediaserver` package from it.

> I just wanted to make you aware it works not using the dietpi-software version.  for me at least.

Generally this is great, so we can fix bugs, if present. However, I cannot replicate any issues with Plex installed via `dietpi-software`, so to find out what went wrong in your case, it is hence required to see the logs as suggested above:
```sh
journalctl -u plexmediaserver
```

Just to be sure I'll try it again on an ARMv8/aarch64, maching your two SBCs.

jimmyfitz, 2022-06-06:
If it helps, the only things that show up under settings are the "Account" tree and the Plex Web tree.  No server, no Status tree, and no Settings tree.

MichaIng, 2022-06-26:
Does anyone read my repeatedly suggested command to show Plex server logs 🤔? Since I cannot replicate the issue, without logs, I'm naturally completely unable to help or guess in any way 😉. The only thing that is obvious is that the web frontend has issues to detect/connect to the server backend.

jimmyfitz, 2022-06-26:
Hi @MichaIng, I did check the logs and there were no error messages.  I'm happy to report I've worked out that the problem is not DietPi. My use case is a headless server, and I was unaware that the server _only_ signs into Plex when logging into the webserver via 127.0.0.1.  So, I had to install X and RDP/VNC so I could connect remotely, and once signed into 127.0.0.1 everything started working fine.  Weird design, but nothing to do with DietPi.

MichaIng, 2022-06-27:
> I was unaware that the server only signs into Plex when logging into the webserver via 127.0.0.1

Hmm, I can connect from any other system in my LAN via browser perfectly fine 🤔. Was there an "Allow access from outside my home network" checkbox or such when you accessed the web interface the first time? And if so, did you uncheck it?

faxesystem, 2022-10-21:
I'm having a similar problem. Installed Plex a few weeks ago and everything was working perfect. Had to do a fresh install on my Pi3B+ today and now Plex is not letting me claim the server. It is showing me the same screen as in the first post here after I log into my Plex Account. https://www.truenas.com/community/threads/plex-not-authorized-you-do-not-have-access-to-this-server.96858/ I once had this problem with a docker image and I had to enter a claim code into the config.

BUT I installed LXDE on my Pi as a last resort and opened up 127.0.0.1:32400/web and here I'm getting the expected configuration screen to add a new server to my Plex Online account. 

I than tried the option allowedNetworks="10.x.x.x/255.255.255.0" for my home network but this didn't help. 

It is very strange, that the very first installation a few months ago worked perfect and claiming it worked and now the second install on the same hardware and the same network does not (not counting the workaround of installing a GUI and using Firefox to claim under 127.0.0.1.). Also tried another fresh Diet install and it didn't work either. So something must be wrong here.

faxesystem, 2022-10-21:
I think I found the problem. The server has to be removed from the Plex Online Account before doing a new installation. This was never the case on my Unraid Docker but it seems to be for Dietpi. At least everything is working now.

[image]

MichaIng, 2022-10-21:
Probably it is the same issue when you use the same Plex server name again?