MichaIng/DietPi #5320: Implement NYM Node Operators

vanGeek, 2022-02-27:
#### Formal software information
- Node Operators & Validators (when public available)
- Would be nice to run from the PI 
- [https://nymtech.net/get-involved/#node-operators]
- https://github.com/nymtech/nym
- https://nymtech.net/docs/stable/run-nym-nodes/nodes/mixnodes

#### Are there similar/alternative software titles available with DietPi-Software?
Not that I know

#### What makes your requested software better than the above solutions, if available?
/

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
allows installation to less experienced Linux users.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
No. Sorry.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
 Sorry, I am totally inexperienced in any coding

MichaIng, 2022-02-27:
Many thanks for your request.

A short description about what it actually is and does would have been nice, I needed to browse a little back and forth through their website to get away from the "wallet"/"coins" key words, making me think its a crypto currency I didn't know yet, to finally find it being a privacy network.

So it's Tor, but claims to be better then Tor, and the proposed software is hence an alternative to a Tor Relay, for the different Nym network, of course.

vanGeek, 2022-02-27:
Sorry about the missing description. I should have pointed out that it is not just another crypto coin. 

Time will show if this piece of software will make it to the mainstream.