MichaIng/DietPi #5323: dietpi-update fails

INQWT, 2022-03-01:
#### Details:
- Date           | Tue Mar  1 12:51:40 GMT 2022
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ssh as root into DietPi
2. run dietpi-update
#### Expected behaviour:

- should update
#### Actual behaviour:

- fails to update
#### Extra details:

- ...
#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```

Joulinar, 2022-03-01:
Did you try to run `dpkg --configure -a` as requested on the error message?

MichaIng, 2022-03-02:
Account was deleted, marking this as closed.

closed 2022-03-02 by MichaIng