MichaIng/DietPi #5333: Ipv6 disable problem using ssh -X.

cocoflan, 2022-03-04:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

bullseye

Linux EEEPC 5.10.0-11-amd64 #1 SMP Debian 5.10.92-1 (2022-01-18) x86_64 GNU/Linux

When installing X11 to use graphical programs via ssh and at the same time disable ipv6 then the program can't be displayed via ssh. Did some testing and at first i thought it was some installed software problem, but when i enable ipv6 again all works again. I use opensshserver  and not dropbear.

Greetings

MichaIng, 2022-03-06:
Many thanks for your report.

I have no idea how the SSH connection IP family should affect whether X can be tunnelled through it or not. The SSH connection otherwise works fine? Have you tried different clients? Which client OS do you use, in case which local X server on the system (like Xming on Windows)? I can try to replicate by times with Xming on Windows.

cocoflan, 2022-03-06:
I have tested it on openssh server, ipv6 disabled no X11 GUI programs possible, when enabling it back X11 GUI programs work again. tested it 2 times. It was on an intel based system with the recovery install image.

Greetings

MichaIng, 2022-03-06:
What is the client OS, X server implementation and SSH client you use for X forwarding? The server side is clear, but the client side is important if we want to replicate.

cocoflan, 2022-03-06:
Can you give me the commands to see what the logging now is and what it is when i stop ipv6 again to see the difference whe, the problem is back. it was via a terminal via popos, so ubuntu based system.

MichaIng, 2022-03-06:
Logs server side:
```sh
journalctl -u ssh
```
The SSH connection otherwise works fine with IPv6 enabled? Have you tried a very simple X application, like `xterm`? Also check the X server logs on Pop!_OS, if it is invoked at all.

cocoflan, 2022-03-06:
Tested it for the 3th time, it still is so, so there must be a problem! works again after enabling ipv6.

cocoflan, 2022-03-06:
All my other systems are working with -X on dietpi via popos. the program i use to test is gedit.

cocoflan, 2022-03-06:
https://www.angioni.nl/2019/04/14/linux-centos-7-x11-forwarding-broken-when-you-disable-ipv6-solved/

MichaIng, 2022-03-06:
Ah interesting, I totally forgot about that forum thread 😄. An annoying bug indeed, probably we should report this to Debian or upstream directly. It shouldn't be required to explicitly stop OpenSSH from listening on IPv6 for using X11 forwarding, when the network interfaces don't have an IPv6 address and default route..

cocoflan, 2022-03-06:
journalctl -u ssh

With ipv6

-- Journal begins at Sun 2022-03-06 01:47:08 GMT, ends at Sun 2022-03-06 01:53:25 GMT. --
Mar 06 01:47:11 xxxx systemd[1]: Starting OpenBSD Secure Shell server...
Mar 06 01:47:11 xxxx sshd[717]: Server listening on 0.0.0.0 port xxx.
Mar 06 01:47:11 xxxx sshd[717]: Server listening on :: port xxx.
Mar 06 01:47:11 xxxx systemd[1]: Started OpenBSD Secure Shell server.
Mar 06 01:51:10 xxxx sshd[1808]: Accepted password for root from xxxxxxxxxxxxxxxx port xxxx ssh2
Mar 06 01:51:10 xxxx sshd[1808]: pam_unix(sshd:session): session opened for user root(uid=0) by (uid=0)
Mar 06 01:53:06 xxxx dbus-daemon[1942]: [session uid=0 pid=1940] Activating service name='ca.desrt.dconf' requested by ':1.0' (uid=0 pid=1936 comm="gedit ")
Mar 06 01:53:06 xxxx dbus-daemon[1942]: [session uid=0 pid=1940] Successfully activated service 'ca.desrt.dconf'

ipv6 disabled

journalctl -u ssh
-- Journal begins at Sun 2022-03-06 01:54:38 GMT, ends at Sun 2022-03-06 01:55:00 GMT. --
Mar 06 01:54:41 xxxx systemd[1]: Starting OpenBSD Secure Shell server...
Mar 06 01:54:41 xxxx sshd[713]: Server listening on 0.0.0.0 port xxx.
Mar 06 01:54:41 xxxx sshd[713]: Server listening on :: port xxx.
Mar 06 01:54:41 xxxx systemd[1]: Started OpenBSD Secure Shell server.
Mar 06 01:54:52 xxxx sshd[1013]: Accepted password for root from xxxxxxxxxxxxxxxx port xxxx ssh2
Mar 06 01:54:52 xxxx sshd[1013]: pam_unix(sshd:session): session opened for user root(uid=0) by (uid=0)
Mar 06 01:54:52 xxxx sshd[1013]: error: Failed to allocate internet-domain X11 display socket.

cocoflan, 2022-03-06:
I have an other pc installed with the restore image and ipv6 is disabled but x11 works, so i think it is a complicated situation.

MichaIng, 2022-03-06:
> I have an other pc installed with the restore image

Not sure what you mean by this, another DietPi system?

Based on the linked threads it seems to be simply expected that it doesn't work without IPv6 explicitly disabled for the OpenSSH server. It seems to try setting up the X11 stream binding to the IPv6 address space then (`::0`), which doesn't match the one the client is connecting from. This isn't mart and it should be possible to fix it upstream a way that OpenSSH automatically uses the IP family for X11 forwarding which the client uses for the SSH connection itself.