MichaIng/DietPi #5336: Odroid-hc4 - boot.cmd/scr doesn't properly resolve dtb file on petiteboot, causing boot failure

BradWhittington, 2022-03-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='applied'
```
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 5.10.81-meson64 #21.08.6 SMP PREEMPT Mon Nov 22 11:21:51 UTC 2021 aarch64 GNU/Linux`
- SBC model | `Odroid C4/HC4 (aarch64)`
- Power supply used | 15v 4A
- SD card used | Transcend 64GB MICROSDXC CARD - TRA-TS64GUSD300S

#### Steps to reproduce

1. Flash the SD card with `DietPi_OdroidC4-ARMv8-Bullseye.img` and insert into odroid
2. Turn on the odroid
3. Petitboot detects the device, reads config from `boot/boot.scr`
4. Boot fails

#### Expected behaviour

- The odroid should continue to autoboot into dietpi

#### Actual behaviour

- Petitboot is reading the boot.scr file and resolving the dtb file incorrectly, and writes an empty file to /tmp/
- Dropping to a shell and manually copying the correct dtb file to the tmp path resolves the problem

#### Extra details

- It seems as if the problem is with https://github.com/MichaIng/DietPi/blob/master/.build/images/U-Boot/boot.cmd#L42 - `${fdtfile}` isn't populated during petitboot parsing of boot.scr
- I've tried adding fdtfile to dietpiEnv.txt, but it doesn't seem to have any impact
- If I add `setenv fdtfile "amlogic/meson-sm1-odroid-hc4.dtb"` to boot.cmd and rebuild the scr file then it boots fine

MichaIng, 2022-03-06:
Please erase the SPI flash, it is known to conflict with this bootloader config:
```sh
flash_erase /dev/mtd0 0 0
```

BradWhittington, 2022-03-06:
Thanks, just took a read through the issue here - https://forum.odroid.com/viewtopic.php?p=314003#p314003

Is there an appropriate documentation page I could add a note about how to get it working on petitboot based odroids?

BradWhittington, 2022-03-06:
https://www.armbian.com/odroid-hc4/ has nice instructions re: SPI petitboot & u-boot should anyone want more detail :)

closed 2022-03-06 by BradWhittington

MichaIng, 2022-03-06:
Yes, btw `flash_erase /dev/mtd0 0 0` should work from petitboot shell as well.

> Is there an appropriate documentation page I could add a note about how to get it working on petitboot based odroids?

Indeed we should add this info to our download page for Odroid C4 and N2 models. petitboot is depending on legacy U-Boot versions and boot scripts, while Armbian uses mainline kernel and U-Boot, which we follow, also to have some consistency across devices. It simply makes sense to boot via bootloader that ships with the OS instead of hoping that a vendor bootloader is able to correctly boot with the shipped boot scripts and kernel.