MichaIng/DietPi #5337: Problems with sound after update v8.2.2

97carmine, 2022-03-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 5.15.25-rockchip64 #22.02.1 SMP PREEMPT Sun Feb 27 09:05:47 UTC 2022 aarch64 GNU/Linux`
- SBC model | `ROCK Pi 4 v1.72 (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | internal eMMC

#### Actual behaviour
The sound card 'alsa_output.platform-sound.stereo-fallback.monitor' (this ID was obtained from PulseAudio with the previous version of DietPi) linked to the JACK connector has disappeared.
The sound card linked to the HDMI connector appears, but when playing a YouTube video from Firefox only noise sounds.

#### Logs:

<details>
<summary>aplay -l</summary>

```
**** List of PLAYBACK Hardware Devices ****
card 0: hdmisound [hdmi-sound], device 0: ff8a0000.i2s-i2s-hifi i2s-hifi-0 [ff8a0000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
</details>
<details>
<summary>cat /proc/asound/cards</summary>

```
 0 [hdmisound      ]: simple-card - hdmi-sound
                      hdmi-sound
```
</details>
<details>
<summary>dmesg | grep -E 'audio|sound|snd'</summary>

```
[    9.048080] usbcore: registered new interface driver snd-usb-audio
[   20.077851] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[   20.078626] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[   20.080325] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[   20.081076] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[   20.083375] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[   20.084128] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[   20.085799] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[   20.086590] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[   28.714193] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.714206] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.714279] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.714283] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.714356] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.714360] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.714416] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.714420] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.714516] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.714521] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.714951] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.714959] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.715022] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.715026] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.715096] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.715100] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[... 100 lines cut ...]
[   28.716290] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.716295]  ff8a0000.i2s-i2s-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   28.716993] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.717001] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.717011]  ff8a0000.i2s-i2s-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   28.717145] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.717149] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.717155]  ff8a0000.i2s-i2s-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   28.717315] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.717319] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.717324]  ff8a0000.i2s-i2s-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   28.717450] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.717454] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.717459]  ff8a0000.i2s-i2s-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   28.717605] hdmi-audio-codec hdmi-audio-codec.5.auto: Not able to map channels to speakers (-22)
[   28.717609] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_hw_params on i2s-hifi: -22
[   28.717615]  ff8a0000.i2s-i2s-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   28.722911] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[   28.722920] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[   28.722928]  ff8a0000.i2s-i2s-hifi: soc_pcm_open() failed (-22)
[   28.723550] hdmi-audio-codec hdmi-audio-codec.5.auto: Only one simultaneous stream supported!
[   28.723556] hdmi-audio-codec hdmi-audio-codec.5.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -22
[   28.723563]  ff8a0000.i2s-i2s-hifi: soc_pcm_open() failed (-22)
```
</details>

MichaIng, 2022-03-06:
Many thanks for your report.

The kernel upgrade from 5.10 to 5.15 most likely caused this. Hmm, there is no device tree overlay to switch between HDMI and 3.5mm. As of the device name it doesn't sound like it could be configured to use 3.5mm for output, but having a look into the `alsamixer` settings should be worth it to verify that.

Please try to enable the auto-conversion plugin in `dietpi-config` Audio Options and see whether this helps with HDMI playback.

97carmine, 2022-03-06:
When you enable the auto-conversion plugin, there is no longer sound, when you disable it, you run out of sound anyway.

<details>
<summary>alsamixer</summary>

[image]

</details>

MichaIng, 2022-03-06:
You use PulseAudio? Try to purge it and use plain ALSA.

97carmine, 2022-03-06:
Purged PulseAudio and reinstalled ALSA.

<details>
<summary>alsamixer</summary>

[image]
</details>

The resulting sound remains the same, the sound is heard, but with a very robotic and noisy tone at the same time.

<details>
<summary>speaker-test -t wav -c 2</summary>

```
speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 8 to 8192
Period size range from 8 to 2048
Using max buffer size 8192
Periods = 4
was set period_size = 2048
was set buffer_size = 8192
 0 - Front Left
 1 - Front Right
Time per period = 2,860370
 0 - Front Left
 1 - Front Right
```
</details>

MichaIng, 2022-03-06:
Please now try to enable auto-conversion. The `speaker-test` does intentionally only produce a noise, but I think you mean to hear this on regular audio playback.

97carmine, 2022-03-06:
Maybe I have not explained myself well, I add two videos to show both results of the command `speaker-test -t wav -c 2`. Be sure to turn up the player volume.

Good - Arch Linux: 

https://user-images.githubusercontent.com/6619414/156937994-105ac670-9199-469b-bcb6-d37a6ece8325.mp4

Bad - Diet Pi: 

https://user-images.githubusercontent.com/6619414/156938012-090e8ce6-7d76-490b-a5bc-cb26501abacc.mp4

MichaIng, 2022-03-06:
Interesting, `speaker-test` whenever I use it always produces such noise only, never heard the clear voice 😄. But it doesn't say something about actual real audio file playback. Not sure Arch builds it with actual audio files being used by default. Did this ever worked before on any Debian based distro?

You can also pass an audio file:
```sh
speaker-test -c 2 -t wav -w /path/to/file.wav
```

97carmine, 2022-03-06:
`speaker-test -c 2 -t wav -w /path/to/file.wav` has been difficult to make work with a wav file, but for a demonstration of the playback of a wav file I use `aplay /path/to/file.wav`.

https://user-images.githubusercontent.com/6619414/156942479-228ac1b6-e5fb-4f03-862e-ead7d83e6d6e.mp4

MichaIng, 2022-03-06:
Okay, that is horrible indeed. Looks like the device tree or kernel drive for onboard audio received a major regression with Linux v5.15. Quite a problem that there is no maintainer for this SBC on Armbian which would have this in view.

I guess reverting to "legacy" kernel would solve it. This should be Linux 5.10 now, but can you verify:
```sh
apt show linux-image-rockchip64-current linux-dtb-rockchip64-current
```

97carmine, 2022-03-06:
Packages not found

```
dietpi@DietPi:~$ sudo apt show linux-image-rockchip64-current linux-dtb-rockchip64-current
N: No se ha podido localizar el paquete linux-image-rockchip64-current
N: No se ha podido localizar el paquete linux-dtb-rockchip64-current
N: No se ha podido localizar el paquete linux-image-rockchip64-current
N: No se ha podido localizar el paquete linux-dtb-rockchip64-current
E: No se encontró ningún paquete
```

MichaIng, 2022-03-07:
Ah sorry, wrong package names:
```sh
apt show linux-dtb-legacy-rockchip64 linux-image-legacy-rockchip64
```
If the latest ones of these are Linux v5.10, then
```sh
apt install linux-dtb-legacy-rockchip64 linux-image-legacy-rockchip64
reboot
# after reboot check back whether the correct kernel has been loaded
uname -a
# If audio works as well, you can purge the "current" packages
apt autopurge linux-dtb-current-rockchip64 linux-image-current-rockchip64
```

lupa18, 2022-03-17:
Hello ! I have the same problem but no packages is found:
```
apt show linux-image-rockchip64-current linux-dtb-rockchip64-current
N: No se ha podido localizar el paquete linux-image-rockchip64-current
N: No se ha podido localizar el paquete linux-dtb-rockchip64-current
N: No se ha podido localizar el paquete linux-image-rockchip64-current
N: No se ha podido localizar el paquete linux-dtb-rockchip64-current
E: No se encontró ningún paquete

```

MichaIng, 2022-03-18:
@lupa18 
Probably you use an older ROCK64 image with different kernel sources, can you show:
```sh
dpkg -S /lib/modules
```

MichaIng, 2022-03-18:
Ah, you used the wrong package names:
```sh
apt show linux-dtb-current-rockchip64 linux-image-current-rockchip64
```

lupa18, 2022-03-18:
Oh, sorry. 

Here the right way:
```
$ apt show linux-dtb-current-rockchip64 linux-image-current-rockchip64
Package: linux-dtb-current-rockchip64
Version: 22.05.0-trunk.0010
Priority: optional
Section: kernel
Source: linux-5.15.29-rockchip64
Maintainer: Igor Pecovnik <igor.pecovnik@****l.com>
Installed-Size: 8.611 kB
Provides: armbian-current, linux-dtb, linux-dtb-armbian
Homepage: https://www.kernel.org/
Download-Size: 365 kB
APT-Sources: http://apt.armbian.com bullseye/main arm64 Packages
Description: Armbian Linux DTB, version 5.15.29-rockchip64 current
 This package contains device blobs from the Linux kernel, version 5.15.29-rockchip64

Package: linux-image-current-rockchip64
Version: 22.05.0-trunk.0010
Priority: optional
Section: kernel
Source: linux-5.15.29-rockchip64
Maintainer: Igor Pecovnik <igor.pecovnik@****l.com>
Installed-Size: 92,6 MB
Provides: armbian-current, linux-image, linux-image-armbian
Homepage: https://www.kernel.org/
Download-Size: 51,2 MB
APT-Sources: http://apt.armbian.com bullseye/main arm64 Packages
Description: Linux kernel, armbian version 5.15.29-rockchip64 current
 This package contains the Linux kernel, modules and corresponding other
 files, version: 5.15.29-rockchip64.

```

MichaIng, 2022-03-18:
So the same method should solve it for you:
```sh
apt install linux-dtb-current-rockchip64=21.08.2 linux-image-current-rockchip64=21.08.2
apt-mark hold linux-dtb-current-rockchip64 linux-image-current-rockchip64
```
Good to know that it seems to be an RK3399 issue with Armbian's Linux v5.15, not ROCK Pi 4 specifically.

MichaIng, 2022-03-18:
Run `apt update` first. If it still does not work, please show:
```sh
apt show -a linux-image-current-rockchip64
```

MichaIng, 2022-03-21:
It has been btw reported to Armbian: https://forum.armbian.com/topic/20093-rock-pi-4b-rk3399-sound-broken-after-latest-kernel-update-fixed/
It shows "fixed" but it is just the same workaround we use, downgrading the kernel to v5.10, which is a bad workaround as of Dirty Pipe.

MichaIng, 2022-04-02:
Issue got unsolved at the Armbian forum, but no maintainer reaction so far.

97carmine, 2022-06-16:
It's fixed in edge kernel packages:

```
apt install linux-dtb-edge-rockchip64 linux-image-edge-rockchip64
```

<details>
<summary>uname -a</summary>

```
Linux DietPi 5.18.0-rockchip64 #22.05.1 SMP PREEMPT Sat May 28 08:41:43 UTC 2022 aarch64 GNU/Linux
```
</details>

<details>
<summary>aplay -l</summary>

```
**** List of PLAYBACK Hardware Devices ****
card 0: hdmisound [hdmi-sound], device 0: ff8a0000.i2s-i2s-hifi i2s-hifi-0 [ff8a0000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Analog [Analog], device 0: ff880000.i2s-ES8316 HiFi ES8316 HiFi-0 [ff880000.i2s-ES8316 HiFi ES8316 HiFi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
</details>

<details>
<summary>cat /proc/asound/cards</summary>

```
 0 [hdmisound      ]: simple-card - hdmi-sound
                      hdmi-sound
 1 [Analog         ]: Analog - Analog
                      Analog
```
</details>

MichaIng, 2022-06-26:
Okay great, same reported on Armbian forum. Not sure whether it is/will be fixed in "current" kernel (Linux 5.15) as well, and I cannot test on my NanoPi R4S which has no analogue audio, but I'll mark this as closed now since there is not much we can do in any case.

closed 2022-06-26 by MichaIng

kromsam, 2023-08-25:
Just to inform. This seems to haven't been fixed in any newer versions. This problem persists in DietPi bookworm.  Downgrading to bullseye and the 5.10 kernel version is still the only solution for sound on DietPi (RockPro64).

Joulinar, 2023-08-25:
As stated by @MichaIng , nothing we can do on this because we don't do own kernel development for this board.

MichaIng, 2023-08-26:
> Downgrading to bullseye and the 5.10 kernel version

The kernel version is independent of the Debian version. No need to downgrade to Bullseye, only downgrading the kernel makes the difference.

The rockchip64 kernel shipped with Armbian's APT repository is over half a year old (for unknown reasons). Can you test with a recent one?
```sh
cd /tmp
curl -fO https://dietpi.com/downloads/binaries/armbian-firmware.deb
curl -fO https://dietpi.com/downloads/binaries/linux-dtb-current-rockchip64.deb
curl -fO https://dietpi.com/downloads/binaries/linux-image-current-rockchip64.deb
apt install ./armbian-firmware.deb ./linux-*
rm ./armbian-firmware.deb ./linux-*
reboot
```

kromsam, 2023-08-31:
Tried this suggestion in Bookworm

> > Downgrading to bullseye and the 5.10 kernel version
> 
> The kernel version is independent of the Debian version. No need to downgrade to Bullseye, only downgrading the kernel makes the difference.
> 
> The rockchip64 kernel shipped with Armbian's APT repository is over half a year old (for unknown reasons). Can you test with a recent one?
> 
> ```shell
> cd /tmp
> curl -fO https://dietpi.com/downloads/binaries/armbian-firmware.deb
> curl -fO https://dietpi.com/downloads/binaries/linux-dtb-current-rockchip64.deb
> curl -fO https://dietpi.com/downloads/binaries/linux-image-current-rockchip64.deb
> apt install ./armbian-firmware.deb ./linux-*
> rm ./armbian-firmware.deb ./linux-*
> reboot
> ```

But apt overruled me

```
dietpi@dietpi:/tmp$ sudo apt install ./armbian-firmware.deb ./linux-*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'armbian-firmware' instead of './armbian-firmware.deb'
Note, selecting 'linux-dtb-current-rockchip64' instead of './linux-dtb-current-rockchip64.deb'
Note, selecting 'linux-image-current-rockchip64' instead of './linux-image-current-rockchip64.deb'
linux-dtb-current-rockchip64 is already the newest version (23.08.0-trunk).
linux-image-current-rockchip64 is already the newest version (23.08.0-trunk).
The following packages will be DOWNGRADED:
  armbian-firmware
0 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
Need to get 0 B/14.1 MB of archives.
After this operation, 1348 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 /tmp/armbian-firmware.deb armbian-firmware all 23.08.0-trunk [14.1 MB]
debconf: delaying package configuration, since apt-utils is not installed
dpkg: warning: downgrading armbian-firmware from 23.08.0-trunk--1-SA1251-B226d t
o 23.08.0-trunk
(Reading database ... 52946 files and directories currently installed.)
Preparing to unpack /tmp/armbian-firmware.deb ...
Unpacking armbian-firmware (23.08.0-trunk) over (23.08.0-trunk--1-SA1251-B226d) 
...
Setting up armbian-firmware (23.08.0-trunk) ...

```

kromsam, 2023-08-31:
This

> ```shell
> cp /etc/apt/sources.list.d/{dietpi-armbian,armbian-bullseye}.list
> sed -i 's/bookworm/bullseye/g' /etc/apt/sources.list.d/armbian-bullseye.list
> ```

in combination with this

> ```shell
> apt install linux-dtb-current-rockchip64=21.08.2 linux-image-current-rockchip64=21.08.2
> apt-mark hold linux-dtb-current-rockchip64 linux-image-current-rockchip64
> ```

did the trick!

lupa18, 2023-08-31:
hi  ! Yesterday I updated dietpi and no more sounds.
Mi configuration is: 
*  Linux rock64 6.1.46-current-rockchip64
* Debian: 11.7

What do you recommends @MichaIng ?

Where can I ask for applying security patches to that kernel? or other things that we can do on rockchip64 upstream maintainers?

lupa18, 2023-08-31:
> -- samsimons.nl

Sorry, I deleted because it was only a device address change from hw:0,1 to hw:0,0. I fixed it and now it works.

MichaIng, 2023-09-01:
So you mean it does work with Linux 6.1.46 after reselecting the sound card via `dietpi-config`? That would be great!

lupa18, 2023-09-01:
Yes, sound it's working with this config:
* Linux rock64 6.1.46-current-rockchip64
* Debian: 11.7

Sound problem was with mpd, so I only manually modified mpd.conf to get it working again.

MichaIng, 2023-09-01:
Great. If I'm not mistaken, in `mpd.conf` you can use "default" as audio device/PCM, which always matches the one selected in `dietpi-config` respectively defined as "default" in `/etc/around.conf`.