MichaIng/DietPi #5340: Error When Trying to update Dietpi from 8.1.2 to 8.2.2

doopler, 2022-03-06:
When I To update dietpi I get an error:
APT upgrade                                                                                                          │ 
     │  - Command: apt-get -qq upgrade                                                                                      │ 
     │  - Exit code: 100                                                                                                    │ 
     │  - DietPi version: v8.1.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                   │ 
     │  - Image creator: DietPi Core Team                                                                                   │ 
     │  - Pre-image: From scratch                                                                                           │ 
     │  - Error log:                                                                                                        │ 
     │ Setting up sonarr (3.0.7) ...^M                                                                                      │ 
     │ adduser: Only one or two names allowed.^M                                                                            │ 
     │ dpkg: error processing package sonarr (--configure):^M                                                               │ 
     │  installed sonarr package post-installation script subprocess returned error exit status 1^M                         │ 
     │ Errors were encountered while processing:^M                                                                          │ 
     │  sonarr^M                                                                                                            │ 
     │ E: Sub-process /usr/bin/dpkg returned an error code (1)         

I have tried to restart but it didn't help
and also checked sonarr version and it's already updated to v 3.0.7

Thanks in Advance
[image]

MichaIng, 2022-03-06:
Many thanks for your report.

Please run the following command and paste its output here:
```sh
dpkg --configure -a
```

doopler, 2022-03-06:
thanks for quick response
i ran dpkg --configure -a and got the same error

root@DietPi:~# dpkg --configure -a
Setting up sonarr (3.0.7) ...
adduser: Only one or two names allowed.
dpkg: error processing package sonarr (--configure):
 installed sonarr package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 sonarr
root@DietPi:~#

MichaIng, 2022-03-06:
> adduser: Only one or two names allowed.

An invalid `adduser` call in the `postinst` script 🤔. Can you show the following:
```sh
cat /var/lib/dpkg/info/sonarr.postinst
```

doopler, 2022-03-07:
#!/bin/sh
set -e

BUILD_VERSION=3.0.7.1477
UPDATER=BuiltIn

. /usr/share/debconf/confmodule
db_get sonarr/owning_user
USER="$RET"
db_get sonarr/owning_group
GROUP="$RET"
db_get sonarr/owning_umask
UMASK="$RET"
db_get sonarr/config_directory
CONFDIR="$RET"

# Add User and Group
if ! getent group "$GROUP" >/dev/null; then
  groupadd "$GROUP"
fi
if ! getent passwd "$USER" >/dev/null; then
  adduser --system --no-create-home --ingroup "$GROUP" "$USER"
fi

if [ $1 = "configure" ]; then
  # Migrate old Sonarr v3 alpha data dir from /var/opt/sonarr or user home
  if [ -d "/var/opt/sonarr" ] && [ "$CONFDIR" != "/var/opt/sonarr" ] && [ ! -d "$CONFDIR" ]; then
    varoptRoot="/var/opt/sonarr"
    varoptAppData="$varoptRoot/.config/Sonarr"
    sonarrUserHome=`getent passwd $USER | cut -d ':' -f 6`
    sonarrAppData="$sonarrUserHome/.config/Sonarr"
    if [ -f "$varoptRoot/sonarr.db" ]; then
      # Handle /var/opt/sonarr/sonarr.db
      mv "$varoptRoot" "$CONFDIR"
    elif [ -f "$varoptAppData/sonarr.db" ]; then
      # Handle /var/opt/sonarr/.config/Sonarr/sonarr.db
      mv "$varoptAppData" "$CONFDIR"
      rm -rf "$varoptRoot"
    elif [ -f "$sonarrAppData/sonarr.db" ]; then
      # Handle ~/.config/Sonarr/sonarr.db
      mv "$sonarrAppData" "$CONFDIR"
      rm -rf "$sonarrAppData"
    else
      mv "$varoptRoot" "$CONFDIR"
    fi
    chown -R $USER:$GROUP "$CONFDIR"
    chmod -R 775 "$CONFDIR"
  fi

  # Migrate old NzbDrone data dir
  if [ -d "/usr/lib/sonarr/nzbdrone-appdata" ] && [ ! -d "$CONFDIR" ]; then
    NZBDRONE_DATA=`readlink /usr/lib/sonarr/nzbdrone-appdata`
    if [ -f "$NZBDRONE_DATA/config.xml" ] && [ -f "$NZBDRONE_DATA/nzbdrone.db" ]; then
      echo "Found NzbDrone database in $NZBDRONE_DATA, copying to $CONFDIR."
      mkdir -p "$CONFDIR"
      cp $NZBDRONE_DATA/config.xml $NZBDRONE_DATA/nzbdrone.db* "$CONFDIR/"
      chown -R $USER:$GROUP "$CONFDIR"
      chmod -R 775 "$CONFDIR"
    else
      echo "Missing NzbDrone database in $NZBDRONE_DATA, skipping migration."
    fi
    rm /usr/lib/sonarr/nzbdrone-appdata
  fi
fi

# Create data directory
if [ ! -d "$CONFDIR" ]; then
  mkdir -p "$CONFDIR"
fi

# Set permissions on data directory (always do this instead only on creation in case user was changed via dpkg-reconfigure)
chown -R $USER:$GROUP "$CONFDIR"

# Apply patch if present
if [ "$UPDATER" = "BuiltIn" ] && [ -f /usr/lib/sonarr/bin_patch/release_info ]; then
  # It shouldn't be possible to get a wrong bin_patch, but let's check anyway and throw it away if it's wrong
  currentVersion=`cat /usr/lib/sonarr/bin_patch/release_info | grep 'ReleaseVersion=' | cut -d= -f 2`
  currentRelease=`echo "$currentVersion" | cut -d. -f1,2,3`
  currentBuild=`echo "$currentVersion" | cut -d. -f4`
  targetVersion=$BUILD_VERSION
  targetRelease=`echo "$targetVersion" | cut -d. -f1,2,3`
  targetBuild=`echo "$targetVersion" | cut -d. -f4`

  if [ "$currentRelease" = "$targetRelease" ] && [ "$currentBuild" -gt "$targetBuild" ]; then
    echo "Applying $currentVersion from BuiltIn updater instead of downgrading to $targetVersion"
    rm -rf /usr/lib/sonarr/bin
    mv /usr/lib/sonarr/bin_patch /usr/lib/sonarr/bin
  else
    rm -rf /usr/lib/sonarr/bin_patch
  fi
fi

# Set permissions on /usr/lib/sonarr
chown -R $USER:$GROUP /usr/lib/sonarr

# Update sonarr.service file
sed -i "s:User=\w*:User=$USER:g; s:Group=\w*:Group=$GROUP:g; s:UMask=[0-9]*:UMask=$UMASK:g; s:-data=.*$:-data=$CONFDIR:g" /lib/systemd/system/sonarr.service

if [ "$UPDATER" = "BuiltIn" ]; then
  # If we upgraded, signal Sonarr to do an update check on startup instead of scheduled.
  touch $CONFDIR/update_required
  chown $USER:$GROUP $CONFDIR/update_required
fi

# Automatically added by dh_systemd_enable/12.10ubuntu1
if [ "$1" = "configure" ] || [ "$1" = "abort-upgrade" ] || [ "$1" = "abort-deconfigure" ] || [ "$1" = "abort-remove" ] ; then
	# This will only remove masks created by d-s-h on package removal.
	deb-systemd-helper unmask 'sonarr.service' >/dev/null || true

	# was-enabled defaults to true, so new installations run enable.
	if deb-systemd-helper --quiet was-enabled 'sonarr.service'; then
		# Enables the unit on first installation, creates new
		# symlinks on upgrades if the unit file has changed.
		deb-systemd-helper enable 'sonarr.service' >/dev/null || true
	else
		# Update the statefile to add new symlinks (if any), which need to be
		# cleaned up on purge. Also remove old symlinks.
		deb-systemd-helper update-state 'sonarr.service' >/dev/null || true
	fi
fi
# End automatically added section
# Automatically added by dh_systemd_start/12.10ubuntu1
if [ "$1" = "configure" ] || [ "$1" = "abort-upgrade" ] || [ "$1" = "abort-deconfigure" ] || [ "$1" = "abort-remove" ] ; then
	if [ -d /run/systemd/system ]; then
		systemctl --system daemon-reload >/dev/null || true
		if [ -n "$2" ]; then
			_dh_action=restart
		else
			_dh_action=start
		fi
		deb-systemd-invoke $_dh_action 'sonarr.service' >/dev/null || true
	fi
fi
# End automatically added section

exit 0

MichaIng, 2022-03-07:
> adduser --system --no-create-home --ingroup "$GROUP" "$USER"

As if the `$USER` variable was empty 🤔. Can you check:
```sh
sed -n '/^Name: sonarr/,/^$/p' /var/cache/debconf/config.dat
```

doopler, 2022-03-07:
this is what a got..

Name: sonarr/config_directory
Template: sonarr/config_directory
Value: /mnt/dietpi_userdata/sonarr
Owners: sonarr
Flags: seen

Name: sonarr/owning_group
Template: sonarr/owning_group
Value: dietpi
Owners: sonarr
Flags: seen

Name: sonarr/owning_umask
Template: sonarr/owning_umask
Value: 0002
Owners: sonarr

Name: sonarr/owning_user
Template: sonarr/owning_user
Value: 
Owners: sonarr
Flags: seen

doopler, 2022-03-08:
thanks that fixed this issue
but now when i try to run apt upgrade in shows  that 1 package not upgraded

```
root@DietPi:~# apt update && apt upgrade
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://download.mono-project.com/repo/debian buster InRelease                                                             
Hit:3 https://apt.sonarr.tv/debian buster InRelease                                                                              
Hit:4 https://deb.debian.org/debian bullseye InRelease                                                
Hit:5 https://deb.debian.org/debian bullseye-updates InRelease                  
Hit:6 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:7 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:8 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@DietPi:~# apt list upgradble
Listing... Done
root@DietPi:~# 

```

MichaIng, 2022-03-08:
It's `apt list --upgradable` 😉. I think a full upgrade should cover the 1 not upgraded package, along with a dependency change:
```sh
apt full-upgrade
```

doopler, 2022-03-08:
thanks
i tried apt full-upgrade and it's the same

```
root@DietPi:~# apt list --upgradable
Listing... Done
root@DietPi:~# apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@DietPi:~# 
```

MichaIng, 2022-03-08:
Hmm, how to list that "not upgraded" package 🤔. Can you try:
```sh
apt upgrade -o APT::Get::Show-Upgraded=true
```
If this doesn't show the package name, can you please show:
```sh
for i in /etc/apt/preferences{,.d/*}
do
[[ -f $i ]] || continue
echo "$i:"
cat "$i"
done
```

doopler, 2022-03-08:
this is what i got..
```
root@DietPi:~# apt upgrade -o APT::Get::Show-Upgraded=true
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@DietPi:~# for i in /etc/apt/preferences{,.d/*}
do
[[ -f $i ]] || continue
echo "$i:"
cat "$i"
done
/etc/apt/preferences.d/dietpi-lxde:
Package: openbox* obconf* libob* pcmanfm* libfm* gtk-* libgtk* libgail* gir1.2-gtk-3.0 lx*
Pin: origin archive.raspberrypi.org
Pin-Priority: -1
root@DietPi:~# 
```

MichaIng, 2022-03-08:
Nasty that there seems to be no way to show the skipped package, but then I guess it's one related to the RPi desktop, which conflicts with Debian's LXDE, so we block it via APT pinning.

Joulinar, 2022-03-08:
We don't implement any specific version as we install RealVNC directly from available apt package source

doopler, 2022-03-08:
ok
thanks very much for your assistance 🤘

MichaIng, 2022-03-09:
Ah, one thing we forgot is checking whether there is a package set on hold:
```sh
apt-mark showhold
```

This "X not upgraded" only shows up when something is preventing the package upgrade, so whether RealVNC is old or not wouldn't play a role unless there is an upgrade available but prevented 😉.

doopler, 2022-03-09:
got it
I checked 
`apt-mark showhold`
nothing showed

MichaIng, 2022-03-09:
Okay, should be all fine then. I'll mark this as closed. Feel free to reopen if required.

closed 2022-03-09 by MichaIng