MichaIng/DietPi #5341: [New as of DietPi 8.2.2] Keep getting "Unsupported SSH client terminal" popup window upon login

jahway603, 2022-03-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye
- Kernel version | Linux HOSTNAME 4.14.241+ # 1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux
- SBC model | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)
- Power supply used | 12V/2A
- SD card used | SanDisk ultra for /boot and Desktop hard drive for rootfs

#### Additional Information (if applicable)
N/A

#### Steps to reproduce
0. I have two identical hardware DietPi 8.2.2 systems - one I installed "ncurses-term" package and the other does not have it installed.
1. SSH into DietPi 8.2.2 system.
2. Get presented with the "Unsupported SSH client terminal" TUI.
3. Hit OK or Cancel and then brought to the CLI.
4. Does not matter what I do, the "Unsupported SSH client terminal" screen continues to appear. This does not happen in previous versions of DietPi.

#### Expected behaviour
- I'm aware that I am using a unique Terminal Emulator called Kitty and have been for years. In [their FAQ](https://sw.kovidgoyal.net/kitty/faq/#i-get-errors-about-the-terminal-being-unknown-or-opening-the-terminal-failing-when-sshing-into-a-different-computer) I am able to use `kitty +kitten ssh dietpi-system` which places a file on the DietPi system as `~/.terminfo/x/xterm-kitty`. This method worked perfectly fine on DietPi prior to the latest 8.2.2 release. I have other DietPi systems that are working as expected - i.e. the "Unsupported SSH client terminal" popup does not appear on the non-upgraded DietPi systems with version prior to 8.2.2.

#### Actual behaviour
- See "Steps to reproduce"

#### Extra details
- Unsure of what has changed for SSH terminal checking from 8.1.x to 8.2.2, but it was working perfectly fine before this release.

MichaIng, 2022-03-07:
In KiTTY settings Connection > Data, what is shown as "Terminal-type string"?

jahway603, 2022-03-08:
> In KiTTY settings Connection > Data, what is shown as "Terminal-type string"?

I think you mean this as kitty doesn't have menu options as you've described.

[WARNING] Your SSH client passed an unsupported terminal: TERM=xterm-kitty

After following the FAQ on older versions of DietPi, it copies the `xterm-kitty` file to the `~/.terminfo/x/xterm-kitty` directory on the server and then the "nag screen" never appears. I'm not certain how OpenSSH is checking, but it works to get rid of that "nag screen" on all my DietPi systems before the 8.2.2 release.

MichaIng, 2022-03-09:
PuTTY has this setting and KiTTY is a fork of PuTTY with some additional features, so it has this setting as well:

[image]

Change it to `xterm`. It doesn't make much sense to pass a TERM variable which is not understood OOTB by any Linux distribution, or by any OS at all. That variable is used to determine which capabilities, control codes etc the terminal understands, and using one that is unknown breaks this ability and causes issues with various command-line tools. Not sure why KiTTY chose to do so, but IMO this isn't clever, at least not to do this by default, not every user reads the FAQ 😉.

But your notice makes me aware of a regression we introduced with DietPi v8.2, which I already recognised before the release but forgot to solve 🙈. I thought [this](https://github.com/MichaIng/DietPi/commit/d35d1a5#diff-f3542f46b27f733cda6d9c3fa778078e4982cd2d9aea642fca0162a4816b6897) was a nice step to reduce overhead on SSH logins a bid, but there are other locations where such terminal info files are read from, like in the users home where you copied the `xtern-kitty` definition to. I'll compare checking `~/.terminfo`, `/etc/terminfo`, `/lib/terminfo` and `/usr/share/terminfo` with the previous `infocmp` call

jahway603, 2022-03-09:
I see where the confusion is here. I've never heard of [KiTTY, the fork of PuTTY](https://www.9bis.net/kitty/index.html) ever until your response above. I have heard of PuTTY before. I'm not using either of those.

What I am using is [kitty, The fast, feature-rich, GPU based terminal emulator](https://sw.kovidgoyal.net/kitty/), which is completely different and is a terminal emulator much like [any of the many terminal emulators available on Linux](https://wiki.archlinux.org/title/List_of_applications#Terminal_emulators). It has no GUI menus like PuTTY.

I've been successfully using the kitty terminal emulator for 4 years now. Their FAQ tells me exactly what to do, which I explained above in that it copies the `xterm-kitty` file to the server in `~/.terminfo/x/xterm-kitty` and it's been working on all Linux and FreeBSD systems I use (even NixOS), but it now fails on DietPi 8.2.2 -- it does work perfectly fine on DietPi 8.1.x, so perhaps [the change you shared](https://github.com/MichaIng/DietPi/commit/d35d1a5#diff-f3542f46b27f733cda6d9c3fa778078e4982cd2d9aea642fca0162a4816b6897) might be why this is now an issue.

akaij, 2022-03-09:
Just to provide a workaround for now:
`# cp ~/.terminfo/x/xterm-kitty /lib/terminfo/x`

MichaIng, 2022-03-09:
Ah lol, okay very interesting and confusing that there is a terminal emulator with the exact same name as a famous SSH client 😄. Now things start to make sense.

Yes, the workaround of `akaij` works, until our next DietPi release fixes the regression introduced with DietPi v8.2. Note that this prompt has no effect on whether the terminal emulator is working or not, out login script just checks that a wrong way, not considering all valid directories. So the workaround basically mutes the false prompt without changing anything effectively.

jahway603, 2022-03-10:
It is confusing as I had not heard of the PuTTY fork and it's not listed in the [extensive list on the Arch Linux wiki](https://wiki.archlinux.org/title/List_of_applications#Terminal_emulators). Thank you @akaij for the work-around.

@MichaIng I noticed it did not change anything functionally as programs like `htop` wouldn't run if that were the case. I appreciate the help & support as it is now an extra "nag" screen I don't have to worry about.

MichaIng, 2022-03-10:
I guess it really depends on how one uses Linux the most. I use it for servers and development, hence usually use SSH clients, no desktops 🙂.

MichaIng, 2022-03-26:
Solved with: https://github.com/MichaIng/DietPi/commit/6db67e2

closed 2022-03-26 by MichaIng