MichaIng/DietPi #5342: "Page unresponsive" Chromium Kiosk issue in two separate devices, regardless of URL

pastaholic, 2022-03-07:
Hi all,

I am running DietPi 8.1.2 with Chromium in kiosk mode on two separate devices, a Pi Zero W2 and a 3B+.
There's nothing else running in these two machines.

They are displaying a web page (different one on each device). Both of them, on the same day (at some point 2 weeks ago) started showing the exact same error that can be seen [here](https://preview.redd.it/ja4fpnh3tkl81.jpg?width=1600&format=pjpg&auto=webp&s=8a6437e405d7d6022d23c596521a8764f3c95ade). Both devices have been running flawlessly for quite some time, and kept working through minor and major updates which I do regularly. Strangely enough, this happened without me changing anything or even applying an update of any kind.

This error shows no matter what URL I use for the page I want to autoload.
So I reinstalled everything from scratch.
Loaded the microSD card with the ARMv7 32-bit image.
After installation finished, I installed Chromium using:
`sudo dietpi-software install 113`
 No other software is running apart from the SSH server.
 Then I run:
`sudo dietpi-autostart`
and chose the page I wanted to load.
Rebooted, and got what you see in the screenshot. On both machines.
Two different RPI's, different page on each.
Also tried other pages (i.e. google.com). Same result.

I posted this on [Reddit](https://www.reddit.com/r/dietpi/comments/t7ard3/same_chromium_issue_in_two_separate_devices/) and someone else had the same problem. He said it has to do with a recent Chromium update. The kind person who found this said that the problem is caused by hardware acceleration and once he disabled it the problem went away. He accessed Chromium via VNC, but I don't have a VNC server installed

Does anyone know what terminal command disables Chromium's hardware acceleration?
I found [this page](https://peter.sh/experiments/chromium-command-line-switches/#disable-accelerated-video-decode) with many helpful commands, but I don't know how to implement it into DietPi.

Any ideas?

Joulinar, 2022-03-07:
Related forum post https://dietpi.com/phpbb/viewtopic.php?t=10122

pastaholic, 2022-03-08:
Hi @MichaIng 
Yes, it's me from the forum. I just had to ask here because @Joulinar was busy and I didn't want to disturb him any further (I had already asked him once).
The first command didn't make any difference. After rebooting, the same problem appeared exactly as before.

The second command(s) made the difference. Now everything is loading as expected.
Thank you for your kind help.

MichaIng, 2022-03-08:
Let's keep the issue here open, it's easier to track for me with tags etc.

> The second command(s) made the difference. Now everything is loading as expected.

You mean enabling KMS and using EGL?
```sh
G_CONFIG_INJECT 'dtoverlay=vc4-f?kms-v3d' 'dtoverlay=vc4-kms-v3d' /boot/config.txt
echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --use-gl=egl"' > /etc/chromium.d/egl
reboot
```
or was it required to disable GPU acceleration?
```sh
echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --disable-gpu"' > /etc/chromium.d/disable-gpu
```

pastaholic, 2022-03-08:
Hi @MichaIng , yes, I enabled KMS and used EGL and that worked just fine.

pastaholic, 2022-03-20:
Hi @MichaIng 
Today I found some time to sort out the second of the devices that was showing the same problem.
I had it turned off all this time.
This is the Rpi 3B+

The problem I am facing now is that the screen orientation displayed is landscape, but I want to turn it to portrait.
After applying

`G_CONFIG_INJECT 'dtoverlay=vc4-f?kms-v3d' 'dtoverlay=vc4-kms-v3d' /boot/config.txt

echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --use-gl=egl"' > /etc/chromium.d/egl`

The Chromium issue goes away as expected, but the orientation changes to landscape. Using dietpi-config I changed the orientation to "1" (rotate 90 degrees) and rebooted, but the screen still shows as landscape.
Any ideas?

Also, should I do everything from scratch? Have you had the time to get KMS/DRM enabled by default on Chromium install, for Bullseye systems and above?

MichaIng, 2022-03-20:
Indeed, using the `display_hdmi_rotate` setting in `/boot/config.txt` does not support 90°/270° rotation with KMS enabled:
- https://www.raspberrypi.com/documentation/computers/config_txt.html#display_hdmi_rotate
- https://www.raspberrypi.com/documentation/computers/configuration.html#rotating-your-display

Now I'm wondering whether it is possible at all to rotate the plain console with KMS, since `xrandr` is an X11 utility, affecting X/desktop sessions only, AFAIK. But could be tried:
```sh
apt install x11-xserver-utils

pastaholic, 2022-03-20:
When trying to run xrandr I'm getting an error saying "Can't open display"

MichaIng, 2022-03-20:
Same with:
```sh
xrandr --output HDMI-0 --rotate right
xrandr --output HDMI-2 --rotate right
xrandr --rotate right
```
right? Basically I'd take that as indication that it works within an X/desktop session only, or with one running at least, i.e. we need to find out whether it's possible/how to rotate the console with KMS used.

MichaIng, 2022-03-20:
Ah here it is: #3406

Does it work to define the display like that:
```sh
DISPLAY=:0 xrandr --output HDMI-0 --rotate left
```
_`left` should be usually the "right" direction 😉._

pastaholic, 2022-03-20:
`root@MagicMirror:~# xrandr --output HDMI-1 --rotate right
Can't open display
root@MagicMirror:~# xrandr --output HDMI-0 --rotate right
Can't open display
root@MagicMirror:~# xrandr --output HDMI-2 --rotate right
Can't open display
root@MagicMirror:~# xrandr --rotate right
xrandr: --rotate must be used after --output
Try 'xrandr --help' for more information.
root@MagicMirror:~# xrandr --help
usage: xrandr [options]
  where options are:
  --display <display> or -d <display>
  --help
  -o <normal,inverted,left,right,0,1,2,3>
            or --orientation <normal,inverted,left,right,0,1,2,3>
  -q        or --query
  -s <size>/<width>x<height> or --size <size>/<width>x<height>
  -r <rate> or --rate <rate> or --refresh <rate>
  -v        or --version
  -x        (reflect in x)
  -y        (reflect in y)
  --screen <screen>
  --verbose
  --current
  --dryrun
  --nograb
  --prop or --properties
  --fb <width>x<height>
  --fbmm <width>x<height>
  --dpi <dpi>/<output>
  --output <output>
      --auto
      --mode <mode>
      --preferred
      --pos <x>x<y>
      --rate <rate> or --refresh <rate>
      --reflect normal,x,y,xy
      --rotate normal,inverted,left,right
      --left-of <output>
      --right-of <output>
      --above <output>
      --below <output>
      --same-as <output>
      --set <property> <value>
      --scale <x>x<y>
      --scale-from <w>x<h>
      --transform <a>,<b>,<c>,<d>,<e>,<f>,<g>,<h>,<i>
      --off
      --crtc <crtc>
      --panning <w>x<h>[+<x>+<y>[/<track:w>x<h>+<x>+<y>[/<border:l>/<t>/<r>/<b>]]]
      --gamma <r>:<g>:<b>
      --brightness <value>
      --primary
  --noprimary
  --newmode <name> <clock MHz>
            <hdisp> <hsync-start> <hsync-end> <htotal>
            <vdisp> <vsync-start> <vsync-end> <vtotal>
            [flags...]
            Valid flags: +HSync -HSync +VSync -VSync
                         +CSync -CSync CSync Interlace DoubleScan
  --rmmode <name>
  --addmode <output> <name>
  --delmode <output> <name>
  --listproviders
  --setprovideroutputsource <prov-xid> <source-xid>
  --setprovideroffloadsink <prov-xid> <sink-xid>
  --listmonitors
  --listactivemonitors
  --setmonitor <name> {auto|<w>/<mmw>x<h>/<mmh>+<x>+<y>} {none|<output>,<output>,...}
  --delmonitor <name>
root@MagicMirror:~# xrandr --output --rotate right
xrandr: unrecognized option 'right'
Try 'xrandr --help' for more information.
`

pastaholic, 2022-03-20:
Is there a way to revert to a previous version of Chromium and then block it from updating?

MichaIng, 2022-03-20:
Whoops, it should be HDMI-1:
```sh
DISPLAY=:0 xrandr --output HDMI-1 --rotate left
```

pastaholic, 2022-03-20:
Aaaaalmost there...

```
DISPLAY=:0 xrandr --output HDMI-1 --rotate left
```
This worked (with "right" instead of left due to how my screen is), but the setting doesn't persist after a reboot

MichaIng, 2022-03-20:
> but the setting doesn't persist after a reboot

Yes that is expected. You can either set this via `/etc/X11/xorg.conf.d/` like [here](https://github.com/MichaIng/DietPi/issues/3406#issuecomment-594872449) (would be my preferred method) or add it as postboot script:
```sh
echo -e '#!/bin/dash\nDISPLAY=:0 xrandr --output HDMI-1 --rotate left' > /var/lib/dietpi/postboot.d/rotate-screen.sh
```
or add it to the Chromium autostart script so that it applies only for Chromium autostart.

pastaholic, 2022-03-20:
Perfect, this worked! (I used your preferred method)
Thank you very much for all your help @MichaIng

MichaIng, 2022-03-20:
Great. Looks like another thing we need to update in `dietpi-config` for KMS.

MichaIng, 2022-04-02:
Topic at the Raspberry Pi forum: https://forums.raspberrypi.com/viewtopic.php?t=330711

Looks like not only KMS but also `--use-gl=egl` is required now, which isn't default on Raspberry Pi OS either. Verifying this on RPi 2 now.

MichaIng, 2022-04-02:
Interesting, `dietpi.com` opens fine here with legacy framebuffer driver, also other websites. Opening a new tab however throws the error, at first, but browsing a bit back and forth and now also new tabs open fine. Hmm, but indeed when using kiosk mode to open `dietpi.com` directly, the error is thrown again, exiting the prompt and reloading the page works. ... okay same issue indeed, but it does not show up every time and reloading pages helps.

KMS indeed solves the issue. `--use-gl=egl` however is not required here, it seems to be applied automatically, I see the related `GL is egl` in the startup log, regardless whether the option is passed or not.

MichaIng, 2022-04-02:
Solved with: https://github.com/MichaIng/DietPi/commit/0d2791c

closed 2022-04-02 by MichaIng

pastaholic, 2022-04-03:
Aw Snap... again.
[https://i.imgur.com/MQt7OGt.jpg](url)
This happened a few minutes after actually loading the page.

Joulinar, 2022-04-03:
Did you adjusted the chromium start options and added `--use-gl=egl` to `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh` 

@MichaIng 
Workaround confirmed by a user on the forum running an n Acepc T11 x64 machine. https://dietpi.com/phpbb/viewtopic.php?p=44889#p44889

MichaIng, 2022-04-03:
Better add it to `/etc/chromium.d/dietpi`, so it applies to all Chromium starts. This is how it is done now on Chromium installs/reinstalls.

Joulinar, 2022-04-03:
But using `reinstall` would do the same right? So should be fine.

MichaIng, 2022-04-03:
Yes exactly.

pastaholic, 2022-04-03:
@Michalng @Joulinar 
After a full reinstall, I run the following:
```
G_CONFIG_INJECT 'dtoverlay=vc4-f?kms-v3d' 'dtoverlay=vc4-kms-v3d' /boot/config.txt
echo 'export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --use-gl=egl"' > /etc/chromium.d/egl
```
Then I did:
```
sudo nano /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
```
and added at the bottom of the file the line:
`--use-gl=egl`

After rebooting, the same issue appeared.
What did I do wrong?

MichaIng, 2022-04-03:
Adding `--use-gl=egl` to `chromium-autostart.sh` is not required, the second step applies this option for all Chromium starts.

> After rebooting, the same issue appeared.

Please show the files again:
```sh
cat /etc/chromium.d/egl
grep 'dtoverlay' /boot/config.txt
```

pastaholic, 2022-04-03:
```
cat /etc/chromium.d/egl
```
returns
`export CHROMIUM_FLAGS="$CHROMIUM_FLAGS --use-gl=egl"`

```
grep 'dtoverlay' /boot/config.txt
```
returns
`dtoverlay=vc4-kms-v3d`

and
```
ls -l /dev/dri
```
returns
total 0
```
drwxr-xr-x 2 root root         80 Apr  3 17:27 by-path
crw-rw---- 1 root video  226,   0 Apr  3 17:27 card0
crw-rw---- 1 root render 226, 128 Apr  3 17:27 renderD128
```

pastaholic, 2022-04-03:
Same error after rebooting

MichaIng, 2022-04-03:
Is it a Bullseye or Buster system? On Buster, AFAIK, the issue does not exist, but who knows. The Chromium option would then need to be applied differently.

Also, when you start Chromium manually from SSH (so that you can see startup logs), do you see a line stating `GL is egl`?
```sh
sudo xinit /usr/bin/chromium-browser
```

pastaholic, 2022-04-03:
It's the latest DietPi v8.3 ARMv7 32bit (Bullseye), downloaded this morning from https://dietpi.com/
If you can point me to a DietPi Buster image I can download, I can try with that too.

The command

`sudo xinit /usr/bin/chromium-browser`

returns

```
(EE)
Fatal server error:
(EE) Server is already active for display 0
        If this server is no longer running, remove /tmp/.X0-lock
        and start again.
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE)
X connection to :0 broken (explicit kill or server shutdown).
```

This is after Chromium crashes

pastaholic, 2022-04-03:
I then disabled Chromium from dietpi-autostart, rebooted and then run the same command. Here's the result:

`X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Raspbian
Current Operating System: Linux MagicMirror 5.15.30-v7+ #1536 SMP Mon Mar 28 13:43:34                                                                                         BST 2022 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=                                                                                        0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1280x800M@60 vc_mem.mem_base=0x3ec00000 vc                                                                                        _mem.mem_size=0x40000000  root=PARTUUID=b458407c-02 rootfstype=ext4 rootwait fsck.rep                                                                                        air=yes net.ifnames=0 logo.nologo console=tty1
Build Date: 04 January 2022  10:24:42AM
xorg-server 2:1.20.11-1+rpt1+deb11u1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sun Apr  3 19:04:04 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
/usr/bin/chromium-browser: line 140: lsb_release: command not found
/usr/bin/chromium-browser: line 141: lsb_release: command not found
dpkg-query: no packages found matching bluealsa
[986:1023:0403/190420.317752:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[986:1023:0403/190420.366346:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1029:1029:0403/190421.413602:ERROR:gl_surface_egl.cc(808)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[1029:1029:0403/190421.509003:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is
[1029:1029:0403/190422.196740:ERROR:gl_surface_egl.cc(808)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[986:1023:0403/190422.390800:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[986:1023:0403/190422.391218:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[986:1023:0403/190424.459448:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[986:1023:0403/190424.459923:ERROR:bus.cc(397)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[1029:1029:0403/190426.273444:ERROR:gl_surface_egl.cc(808)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[1029:1029:0403/190426.282495:ERROR:gles2_cmd_decoder.cc(2876)] [GroupMarkerNotSet(crbug.com/242999)!:5C37E255]GL ERROR :GL_INVALID_ENUM : BackFramebuffer::Create: <- error from previous GL command
[986:1080:0403/190426.554338:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[986:1080:0403/190426.555871:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[986:1080:0403/190426.557106:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[1029:1029:0403/190428.422739:ERROR:gl_surface_egl.cc(808)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[1029:1029:0403/190428.439990:ERROR:gles2_cmd_decoder.cc(2876)] [GroupMarkerNotSet(crbug.com/242999)!:5C9BE255]GL ERROR :GL_INVALID_ENUM : BackFramebuffer::Create: <- error from previous GL command
[1029:1029:0403/190432.012128:ERROR:gl_surface_egl.cc(808)] EGL Driver message (Error) eglCreateContext: dri2_create_context
[1029:1029:0403/190432.024029:ERROR:gles2_cmd_decoder.cc(2876)] [GroupMarkerNotSet(crbug.com/242999)!:5C777156]GL ERROR :GL_INVALID_ENUM : BackFramebuffer::Create: <- error from previous GL command
[986:1064:0403/190434.606699:ERROR:chrome_browser_main_extra_parts_metrics.cc(227)] START: ReportBluetoothAvailability(). If you don't see the END: message, this is crbug.com/1216328.
[986:1064:0403/190434.607607:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()
`

MichaIng, 2022-04-03:
Okay, so KMS/DRM and EGL are used as expected. And with this you still face the same "Page Unrespinsive" error?

Hmm, without KMS I also do not always face the error, and after one reload it's gone. So probably also with KMS it is not always solved (though I never faced it on my tests). Nasty that we have no reply from RPi Chromium maintainer(s) yet. I personally lack the knowledge to further understand/debug the issue behind it 🤔.

pastaholic, 2022-04-03:
That's ok @MichaIng , you've already done more than enough.
Thank you for your help.
I'll try to follow this guide 
After all, it's only a local page I want to display (local as in served by a different machine in the same LAN).

Thanks again.