MichaIng/DietPi #5345: Probl

maury77, 2022-03-09:
### Creating a bug report/issue

#### Required Information
DietPi version | `8.1.2
Distro version | bullseye
Kernel version | linuxux sdomotica 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
SBC model | RPi 3 Model B+ (armv7l)
Power supply used | (EG: 5V 3A)

I can't upgrade to last version for this problem on the package

                                                      │ APT update                                                                                                           │ 
                                                      │  - Command: apt-get -q update                                                                                        │ 
                                                      │  - Exit code: 100                                                                                                    │ 
                                                      │  - DietPi version: v8.1.2 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 6                                   │ 
                                                      │  - Image creator: DietPi Core Team                                                                                   │ 
                                                      │  - Pre-image: Raspbian Lite                                                                                          │ 
                                                      │  - Error log:                                                                                                        │ 
                                                      │ Ign:1 https://dtcooper.github.io/raspotify jessie InRelease                                                          │ 
                                                      │ Err:2 https://dtcooper.github.io/raspotify jessie Release                                                            │ 
                                                      │   404  Not Found [IP: 185.199.110.153 443]                                                                           │ 
                                                      │ Get:3 https://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]                                            │ 
                                                      │ Hit:4 https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease                                         │ 
                                                      │ Hit:5 https://openhab.jfrog.io/openhab/openhab-linuxpkg stable InRelease                                             │ 
                                                      │ Get:6 https://archive.raspberrypi.org/debian bullseye/main armhf Packages [257 kB]                                   │ 
                                                      │ Get:7 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]                                          │ 
                                                      │ Get:8 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]                                │ 
                                                      │ Reading package lists...                                                                                             │ 
                                                      │ E: The repository 'https://dtcooper.github.io/raspotify jessie Release' no longer has a Release file.                │ 
                                                      │ W: Skipping acquire of configured file 'main’/binary-armhf/Packages' as repository                                   │ 
                                                      │ 'https://openhab.jfrog.io/openhab/openhab-linuxpkg stable InRelease' doesn't have the component 'main’' (compon

Joulinar, 2022-03-09:
You are using quite an old repository for Raspotify, which is failing now

```
Err:2 https://dtcooper.github.io/raspotify jessie Release
  404  Not Found [IP: 185.199.110.153 443]
E: The repository 'https://dtcooper.github.io/raspotify jessie Release' no longer has a Release file.
```

It's still Debian Jessie, while your system is already running Debian Bullseye. You can try following to update the list

```
eval "echo 'deb https://dtcooper.github.io/raspotify/ raspotify main' > /etc/apt/sources.list.d/raspotify.list"
```

neilsmall, 2022-03-10:
Received the same error today when running the "Run now to update DietPi from v8.1.2 to v8.2.2"

[image]

MichaIng, 2022-03-10:
"Structure needs cleaning" usually indicates a filesystem errors. Please try to clean it on reboot:
```sh
> /forcefsck
reboot
# after reboot check logs
journalctl -t systemd-fsck
```

Joulinar, 2022-03-10:
yes, it's not the same issue. Complete different one. Maybe we should move it to an own issue.

neilsmall, 2022-03-10:
> "Structure needs cleaning" usually indicates a filesystem errors. Please try to clean it on reboot:
> 
> ```shell
> > /forcefsck
> reboot
> # after reboot check logs
> journalctl -t systemd-fsck
> ```

Thanks - I'll do this later. I had to re-image the SD Card 'cause my Roon Bridge endpoint got trashed; off to a meeting. I'll get back later. Cheers.

neilsmall, 2022-03-10:
> "Structure needs cleaning" usually indicates a filesystem errors. Please try to clean it on reboot:
> 
> ```shell
> > /forcefsck
> reboot
> # after reboot check logs
> journalctl -t systemd-fsck
> ```

After reboot, the update worked correctly. Thanks.

[image]

MichaIng, 2022-03-11:
Interesting, since there were no errors found or resolved. However, good that it worked well now.

closed 2022-03-11 by MichaIng

neilsmall, 2022-03-15:
An update to Roon Bridge was released a few hours ago. It failed on a newly built Pi 4B. I uninstalled Bridge and tried to reinstall Bridge, which has solved this issue in the past. I ended up at the same screens with the same errors as previously posted (this has =never= happened before). It was easiest to wipe the SD Card clean, and quickly configure DietPi with a fresh install of Bridge (build 918), which worked as expected. FYI, as I said, I've never had these difficulties in the past. 

Thanks for the reply.

MichaIng, 2022-03-15:
Just to avoid mixing topics, which error did you face again? The "Structure needs cleaning" during an APT upgrade? Or an issue with the Roon Bridge update (`dietpi-software reinstall 121`)? If the letter, please open a new issue about this. Logs would be very helpful then.

neilsmall, 2022-03-16:
> Just to avoid mixing topics, which error did you face again? The "Structure needs cleaning" during an APT upgrade? Or an issue with the Roon Bridge update (`dietpi-software reinstall 121`)? If the letter, please open a new issue about this. Logs would be very helpful then.

I didn't realize I still had the terminal window open. Screenshot below. Thanks. 

[image]

[image]

[image]

neilsmall, 2022-03-16:
And (still) getting this error. 

[image]

Joulinar, 2022-03-16:
Try to reboot your system and do `apt update && apt upgrade` afterwards.

Btw: no need to create screen prints. Simply copy the output directly from SSH terminal.

MichaIng, 2022-03-16:
This was a concurrently running or hard killed `apt-get` process. If the reboot suggested by Joulinar does not help, try to clean all cache and list files, which should include the lock files:
```sh
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```