MichaIng/DietPi #5347: /boot/dietpi/func/run_ntpd: line 35: ((: == 1 : syntax error: operand expected (error token is "== 1 ")

adminy, 2022-03-09:
### Creating a bug report/issue

#### Required Information
- DietPi version
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version: `bullseye 0`
- Kernel version: `Linux DietPi 5.10.103-v7l+ #1530 SMP Tue Mar 8 13:05:01 GMT 2022 armv7l GNU/Linux`
- SBC model: `RPi 4 Model B (armv7l)`
- Power supply used: MacOS Usb dongle
- SD card used: Boot over usb stick

#### Steps to reproduce

1. install new distro
2. watch logs after changing ssh server from dropbear to openssh

#### Expected behaviour

- there should be no errors

#### Actual behaviour

```
DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | OpenSSH Client will be installed
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
/boot/dietpi/func/run_ntpd: line 35: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
[  OK  ] DietPi-TimeSync | systemctl start systemd-timesyncd
[ INFO ] DietPi-TimeSync | Waiting for time sync (1/60)
[  OK  ] DietPi-TimeSync | Time sync completed
[  OK  ] DietPi-TimeSync | systemctl stop systemd-timesyncd
[  OK  ] DietPi-TimeSync | mkdir -p /run/systemd/timesync
```

Joulinar, 2022-03-09:
Hi,

does running time sync alone is working?

```
/boot/dietpi/func/run_ntpd 1
```

MichaIng, 2022-03-09:
This is visual only. There is an arithmetic input check which shouldn't be one, but it doesn't affect the result.

MichaIng, 2022-03-09:
Solved with: https://github.com/MichaIng/DietPi/commit/a89e888

Thanks for reporting 🙂. Marking this as closed.

closed 2022-03-09 by MichaIng