MichaIng/DietPi #5353: DietPi-Drive_Manager | fstab entry order incompatible with swap on external SSD?

lukaszsobala, 2022-03-12:
- DietPi version: 8.2.2
- Distro version: bullseye 0
- Kernel version: Linux DietPi 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux
- SBC model: RPi 4 Model B (aarch64)
- Power supply used: official Raspberry Pi 4 PS (15W)
- SD card used: WD Purple 32GB

Software affected: Dietpi-Drive_manager

Steps to reproduce

1. Connect an external ssd
2. Mount it in dietpi-drive_manager 
3. Create a swapfile on an ext4 external SSD in Dietpi-drive_manager
4. Reboot

#### Expected behaviour

- the external SSD should be mounted and the swap enabled

#### Actual behaviour
- the external SSD is not detected and not mounted
- raspberry pi is slow to restart and dietpi-drive_manager slow to start
- the external SSD is not visible in dietpi-drive_manager 

#### Extra details
- my ssd has one partition (sdb2) and the swapfile is 4 GB
- to fix the issue, after rebooting raspberry pi I have to physically disconnect the USB3 SSD and reconnect it. Then the drive mounts automatically. I have to invoke `swapon -a` to have the swap space mounted.

MichaIng, 2022-03-12:
Many thanks for your report.

The order in which drives are defined in fstab do not play a role. systemd creates mount units of them with dependencies on parent mounts automatically applied. So external drives for swap spaces work reliably, are even highly recommended when using an SD card as system drive.

I guess in your case for some reason the USB drive itself fails to mount at boot, so that the dependant swap space fails as well. Could you show your fstab? And also check `journalctl` for boot events, which probably reveals the issue why swap space is not enabled automatically.

lukaszsobala, 2022-03-12:
Hello @MichaIng,

Thank you very much for the quick response! Indeed I was wrong in thinking the order makes a difference.

Sure, here are the fstab entries

> tmpfs /tmp tmpfs size=3823M,noatime,lazytime,nodev,nosuid,mode=1777
> tmpfs /var/log tmpfs size=50M,noatime,lazytime,nodev,nosuid,mode=1777
> /mnt/561da332-1b57-446d-9728-93f186b6657f/.swapfile none swap sw
> PARTUUID=8f4dbd00-02 / ext4 noatime,lazytime,rw 0 1
> PARTUUID=8f4dbd00-01 /boot vfat noatime,lazytime,rw 0 2
> UUID=103d76a6-e90f-48e8-b44c-42092f32389d /mnt/103d76a6-e90f-48e8-b44c-42092f32389d ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount
> UUID=561da332-1b57-446d-9728-93f186b6657f /mnt/561da332-1b57-446d-9728-93f186b6657f ext4 noatime,lazytime,rw,nofail,noauto,x-systemd.automount

The disk in question is `561da332-1b57-446d-9728-93f186b6657f`. When I do `journalctl -g 561da332`, this is the output:

> Mar 12 17:42:55 DietPi systemd[1]: Set up automount mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.automount.
> Mar 12 17:44:24 DietPi systemd[1]: dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device: Job dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device/start timed out.
> Mar 12 17:44:24 DietPi systemd[1]: Timed out waiting for device /dev/disk/by-uuid/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:44:24 DietPi systemd[1]: Dependency failed for /mnt/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:44:24 DietPi systemd[1]: Dependency failed for /mnt/561da332-1b57-446d-9728-93f186b6657f/.swapfile.
> Mar 12 17:44:24 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f-.swapfile.swap: Job mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f-.swapfile.swap/start failed with result 'dependency'.
> Mar 12 17:44:24 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.mount: Job mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.mount/start failed with result 'dependency'.
> Mar 12 17:44:24 DietPi systemd[1]: dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device: Job dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device/start failed with result 'timeout'.
> Mar 12 17:44:36 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.automount: Got automount request for /mnt/561da332-1b57-446d-9728-93f186b6657f, triggered by 624 (rpc.mountd)
> Mar 12 17:46:19 DietPi systemd[1]: dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device: Job dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device/start timed out.
> Mar 12 17:46:19 DietPi systemd[1]: Timed out waiting for device /dev/disk/by-uuid/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:46:19 DietPi systemd[1]: Dependency failed for /mnt/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:46:19 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.mount: Job mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.mount/start failed with result 'dependency'.
> Mar 12 17:46:19 DietPi systemd[1]: dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device: Job dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device/start failed with result 'timeout'.
> Mar 12 17:46:19 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.automount: Got automount request for /mnt/561da332-1b57-446d-9728-93f186b6657f, triggered by 624 (rpc.mountd)
> Mar 12 17:47:49 DietPi systemd[1]: dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device: Job dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device/start timed out.
> Mar 12 17:47:49 DietPi systemd[1]: Timed out waiting for device /dev/disk/by-uuid/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:47:49 DietPi rpc.mountd[624]: Cannot export /mnt/561da332-1b57-446d-9728-93f186b6657f, possibly unsupported filesystem or fsid= required
> Mar 12 17:47:49 DietPi systemd[1]: Dependency failed for /mnt/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:47:49 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.mount: Job mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.mount/start failed with result 'dependency'.
> Mar 12 17:47:49 DietPi systemd[1]: dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device: Job dev-disk-by\x2duuid-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.device/start failed with result 'timeout'.
> Mar 12 17:48:45 DietPi systemd[1]: mnt-561da332\x2d1b57\x2d446d\x2d9728\x2d93f186b6657f.automount: Got automount request for /mnt/561da332-1b57-446d-9728-93f186b6657f, triggered by 2795 (dietpi-drive_ma)
> Mar 12 17:48:45 DietPi systemd[1]: Mounting /mnt/561da332-1b57-446d-9728-93f186b6657f...
> Mar 12 17:48:45 DietPi systemd[1]: Mounted /mnt/561da332-1b57-446d-9728-93f186b6657f.
> Mar 12 17:49:16 DietPi kernel: Adding 4194300k swap on /mnt/561da332-1b57-446d-9728-93f186b6657f/.swapfile.  Priority:-2 extents:8 across:4497404k SSFS

The dmesg entries related to the disk:

> [Sat Mar 12 17:48:21 2022] usb 2-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
> [Sat Mar 12 17:48:22 2022] usb 2-1: New USB device found, idVendor=13fd, idProduct=5910, bcdDevice= 0.57
> [Sat Mar 12 17:48:22 2022] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [Sat Mar 12 17:48:22 2022] usb 2-1: Product: 3639S
> [Sat Mar 12 17:48:22 2022] usb 2-1: Manufacturer: Initio
[... cut ...]
> [Sat Mar 12 17:48:22 2022] sd 1:0:0:0: [sdb] Optimal transfer size 33553920 bytes
> [Sat Mar 12 17:48:22 2022]  sdb: sdb2
> [Sat Mar 12 17:48:22 2022] sd 1:0:0:0: [sdb] Attached SCSI disk
> [Sat Mar 12 17:48:44 2022] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
> [Sat Mar 12 17:49:16 2022] Adding 4194300k swap on /mnt/561da332-1b57-446d-9728-93f186b6657f/.swapfile.  Priority:-2 extents:8 across:4497404k SSFS

There is nothing from the period where it was trying to detect the disk in dmesg, only from when I reconnected the USB.

MichaIng, 2022-03-12:
Hmm, waiting for that drive times out three times before it succeeds, I guess when your detach and reattach? Does it have a dedicated power supply? Not that the RPi's USB ports cannot reliably power any 2.5" drive or larger, including SSDs. You require an external docking station/case with dedicated power supply, else such issues at boot, sudden connection losses during I/O peaks, including filesystem corruption and data loss are expected.

lukaszsobala, 2022-03-12:
Yes, the pi seems to be trying to connect to it all the time but the drive times out. And only when I detach and reattach it succeeds after a reboot. It sometimes seems to work after a full power off - power on cycle, but I have not managed to figure out why. 

The drive does not have a dedicated PS, but it is an mSATA SSD in a USB3.0 enclosure and shouldn't need a lot. It has never failed or disconnected while mounted... And I don't think there was ever any data corruption. The other drive has a power supply. I also don't see anything about insufficient power available to the disk. When I connected a drive that actually needed more power than the pi could supply, all drives stopped working and there were a lot of messages about insufficient power in dmesg, and I can't see any now. So I don't think this is a power issue, but I don't know how to confirm this either.

But it probably also isn't a dietpi issue then? What do you think? Would you recommend a way to get to the bottom of this? Thanks!

MichaIng, 2022-03-12:
There are usually no logs about insufficient power for USB drives when it simply does not fully power up in the first place, and also not necessarily later. You get logs only when the SoC itself is undervolted and of course on sudden disconnects and I/O errors.

It stays true that powering any kind of 2.5" USB drive does not work reliable, not only stated by us but also by RPi foundation, on our end proven by a large number of related issues matching yours and similar. Please try the same with a drive with dedicated PSU, respectively the same drive in a cause/station with dedicated PSU.

lukaszsobala, 2022-03-14:
I have now tried it with 2 powered USB hubs and observed the same (wrong) behaviour on startup. However, when I swapped the device into an mSATA-to-SATA adapter and connected using an unpowered SATA-to-USB3 cable, there is no trouble when restarting.

So the issue is related to the adapter and not strictly to power consumption, however, connecting the drive using a powered hub would still probably be safer. 
I have found out now that Raspberry Pi 4 can be picky about USB adapters (https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/) and this is probably one of the unlucky ones. 

 Definitely not dietpi-related then.

MichaIng, 2022-03-14:
> So the issue is related to the adapter and not strictly to power consumption, however, connecting the drive using a powered hub would still probably be safer.

Good find. But yes, for reliability I'd still recommend to use a powered hub.

Another thing is UAS, which is not supported by all drives very well. Scroll down a bit in the linked article how to apply USB quirks to disable UAS for a specific USB device, probably this help, too.

lukaszsobala, 2022-05-04:
I will close this issue as it is resolved now!

closed 2022-05-04 by lukaszsobala