MichaIng/DietPi #5354: DietPi-DDNS | Prefer IPv4 option

TomBayne, 2022-03-12:
### Creating a feature request

DietPi-DDNS will submit a IPv6 address to the server when IPv6 is enabled. This is not necessarily intended behavior. I would like to submit my IPv4 address rather than an IPv6 address, whilst still keeping IPv6 enabled for other services to use

A simple solution to this would be to implement a switch for IPv4 or IPv6 (A record/AAAA record).
An alternative solution would be to allow submitting of **both** an A record and an AAAA record.

MichaIng, 2022-03-12:
Many thanks for your request. This indeed makes sense. Easy would be to check whether there is an IPv6 default route, and if so, show an IPv4/IPv6 toggle to add the one or the other flag to the `curl` request. By default IPv4 makes sense, for best compatibility.

MichaIng, 2022-03-26:
Implemented with: https://github.com/MichaIng/DietPi/commit/013d6da

closed 2022-03-26 by MichaIng