MichaIng/DietPi #5356: Error updating DietPi

doobersn, 2022-03-13:
I run only pihole on my pi zero w on dietpi. Noticed pihole wasnt blocking everything correctly (though some stuff still seems to be blocked) even though status shows everything is fine. So I updated pihole but it made no difference. Tried updating dietpi and it gave me this error:

RPi Zero W (armv6l) | IP: 192.168.1.8Pi-Update ├───────────────────────────────┐
│ APT update                                                                   │
│  - Command: apt-get -q update                                                │
│  - Exit code: 100                                                            │
│  - DietPi version: v7.6.2 (MichaIng/master) | HW_MODEL: 1 | HW_ARCH: 1 |     │
│ DISTRO: 5                                                                    │
│  - Image creator: DietPi Core Team                                           │
│  - Pre-image: Raspbian Lite                                                  │
│  - Error log:                                                                │
│ Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]    │
│ Reading package lists...                                                     │
│ E: Method https has died unexpectedly!                                       │
│ E: Sub-process https received a segmentation fault.                          │
│                                                                              │
│    Retry          : Re-run the last command that failed                 ↑    │
│    DietPi-Config  : Edit network, APT/NTP mirror settings etc           ▮    │
│    Open subshell  : Open a subshell to investigate or solve the issue   ▒    │
│    Send report    : Uploads bugreport containing system info to DietPi  ▒    │
│                   ●─ Devs only ──────────────────────────────────────●  ↓    │
│                                                                              │
│                                                                              │
│                     <Ok>                         <Exit>                      │
│                                                                              │
└──────────────────────────────────────────────────────────────────────────────┘

Any suggestions on what to do to fix it? Thank you very much!

Joulinar, 2022-03-13:
Can you try following

`apt update && apt upgrade`

MichaIng, 2022-03-13:
We saw this error a few times, but never were able to find out what it causes and how to resolve:
- #4088
- #3564 (not exactly the same)
- #2257

Must be some issue with an underlying library used by APT or dpkg. The problem is that we cannot reinstall packages due to exactly this error, without rewriting `sources.list` to use plain HTTP instead of encrypted+authenticated HTTPS. But even stranger, in your case `http://raspbian.raspberrypi.org/raspbian` was used, as the Raspbian repo is not available via HTTPS, but still the "https" method died. But let's see first the output of Joulinar's commands, to get the full ficture.

doobersn, 2022-03-13:
> Can you try following
> 
> `apt update && apt upgrade`

I usually try that first. I also tried (I think it was) `sudo apt-get update && sudo apt-get dist-upgrade` by accident. Here is the output:

`dietpi@DietPi:~$ apt update && apt upgrade
Reading package lists... Error!
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
E: can not open /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_buster_InRelease - fopen (13: Permission denied)
E: The package lists or status file could not be parsed or opened.`

doobersn, 2022-03-13:
> We saw this error a few times, but never were able to find out what's causing it and how to resolve:
> 
> * [apt and wget segmentation fault #4088](https://github.com/MichaIng/DietPi/issues/4088)
> * [Upgrade 2.28 - 2.30 fails #3564](https://github.com/MichaIng/DietPi/issues/3564) (not exactly the same)
> * [PREP_SYSTEM_FOR_DIETPI.sh fails because https is enabled in sources.list #2257](https://github.com/MichaIng/DietPi/issues/2257)
> 
> Must be some issue with an underlying library used by APT or dpkg. The problem is that we cannot reinstall packages due to exactly this error, without rewriting `sources.list` to use plain HTTP instead of encrypted+authenticated HTTPS. But even stranger, in your case `http://raspbian.raspberrypi.org/raspbian` was used, as the Raspbian repo is not available via HTTPS, but still the "https" method died. But let's see first the output of Joulinar's commands, to get the full ficture.

Thanks a lot! Do you think I messed it up by trying `sudo apt-get update && sudo apt-get dist-upgrade`? I didnt mean to copy and past that, but was googling to see what it did and accidentally did iirc.

Joulinar, 2022-03-13:
The issue has no relation to the usage of `apt-get`

MichaIng, 2022-03-13:
Running an APT list update and package upgrade, including distro upgrades, is a common task, not causing such errors. But indirectly, a filesystem corruption or such may have happened. Can you also check:
```sh
dmesg -l emerg,alert,crit,err
```

MichaIng, 2022-03-13:
If the above does not give further hints. Please try to reinstall the `apt` package manually, which includes the method file `/usr/lib/apt/methods/https`:
```sh
cd /tmp
curl -O 'http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.3_armhf.deb'
dpkg -i apt_1.8.2.3_armhf.deb
rm apt_1.8.2.3_armhf.deb
```

doobersn, 2022-03-13:
> Running an APT list update and package upgrade, including distro upgrades, is a common task, not causing such errors. But indirectly, a filesystem corruption or such may have happened. Can you also check:
> 
> ```shell
> dmesg -l emerg,alert,crit,err
> ```

For this I got:

`dietpi@DietPi:~$ dmesg -l emerg,alert,crit,err
[   17.111755] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.418007] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.418266] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   17.418422] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   17.419743] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   24.261064] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   27.525326] EXT4-fs error (device mmcblk0p2): ext4_mb_generate_buddy:805: group 25, block bitmap and bg descriptor inconsistent: 7994 vs 10042 free clusters
[  101.610978] mmc0: timeout waiting for hardware interrupt.`

Should I go ahead and reinstall the apt package like you mentioned? Thank you so much for helping me!

MichaIng, 2022-03-13:
There is at least one filesystem error in the logs, though seems to be not critical and was shortly after boot. However, please do an fsck scan and repair on reboot first:
```sh
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
```
If you have a screen attached: Depending on the SD card size and speed this may take a while. Leave it running, the greenish LED should give a hint about I/O activity, so when it calms down, it's ready for SSH access.

doobersn, 2022-03-13:
When 

> There is at least one filesystem error in the logs, though seems to be not critical and was shortly after boot. However, please do an fsck scan and repair on reboot first:
> 
> ```shell
> > /forcefsck
> reboot
> # after reboot
> journalctl -t systemd-fsck
> ```
> 
> If you have a screen attached: Depending on the SD card size and speed this may take a while. Leave it running, the greenish LED should give a hint about I/O activity, so when it calms down, it's ready for SSH access.

When I try to run `> /forcefsck` (even with sudo) it says: `Permission denied`. My gosh, im not cut out for this, haha.

Joulinar, 2022-03-13:
Try using `root` user

MichaIng, 2022-03-13:
Ah, do so:
```sh
sudo eval '> /forcefsck'

doobersn, 2022-03-13:
> Ah, do so:
> 
> ```shell
> sudo eval '> /forcefsck'
> ```
> 
> `sudo` does not span across shell redirects, so those need to be wrapped into `eval`, a subshell or similar.

Alrighty, that gives me:

`dietpi@DietPi:~$ sudo eval '> /forcefsck'
sudo: eval: command not found
dietpi@DietPi:~$ sudo eval > /forcefsck
-bash: /forcefsck: Permission denied`

I tried with and without the `! Insanity! Thanks for sticking with me through this. :)

Joulinar, 2022-03-13:
Try using `root` user. This will simplify thinks.

MichaIng, 2022-03-13:
Or:
```
sudo dash -c '> /forcefsck'
```

doobersn, 2022-03-13:
> Try using `root` user. This will simplify thinks.

I do `sudo -I` for that right?

> Or:
> 
> ```
> sudo dash -c '> /forcefsck'
> ```

This worked (I think) but didnt say anything. Then when I did `reboot` it said `Failed to connect to bus: No such file or directory`. So Im thinking maybe its running? Or maybe not, lol.

MichaIng, 2022-03-13:
```sh
sudo reboot
```
How did you reboot before? 😄

doobersn, 2022-03-13:
> ```shell
> ```shell
> sudo reboot
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
> How did you reboot before? 😄
> ```

🤦‍♂️ Whoops, lol.

Okie dokie so, I'm here now:

`dietpi@DietPi:~$ journalctl -t systemd-fsck
Hint: You are currently not seeing messages from other users and the system.
      Users in the 'systemd-journal' group can see all messages. Pass -q to
      turn off this notice.
No journal files were opened due to insufficient permissions.
dietpi@DietPi:~$ sudo journalctl -t systemd-fsck
Illegal instruction`

MichaIng, 2022-03-13:
Use code fences for multi line console pasts, like so:
````
```
line 1
line 2
```
````

`Illegal instruction` hmm, that doesn't sound so good. Can you please login as `root` user and try that again (without `sudo`)? Also it makes things easier for package reinstalls etc. Does `journalctl` (without any arguments) return `Illegal instructions` as well? Also check again `dmesg -l emerg,alert,crit,err` then.

And let's see whether APT can be repaired now:
```
cd /tmp
curl -LO 'http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.3_armhf.deb'
dpkg -i apt_1.8.2.3_armhf.deb
rm apt_1.8.2.3_armhf.deb
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

doobersn, 2022-03-13:
O

> Use code fences for multi line console pasts, like so:
> 
> ```
> ```
> line 1
> line 2
> ```
> ```
> 
> `Illegal instruction` hmm, that doesn't sound so good. Can you please login as `root` user and try that again (without `sudo`)? Also it makes things easier for package reinstalls etc. Does `journalctl` (without any arguments) return `Illegal instructions` as well? Also check again `dmesg -l emerg,alert,crit,err` then.
> 
> And let's see whether APT can be repaired now:
> 
> ```shell
> cd /tmp
> curl -LO 'http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.3_armhf.deb'
> dpkg -i apt_1.8.2.3_armhf.deb
> rm apt_1.8.2.3_armhf.deb
> /boot/dietpi/func/dietpi-set_software apt-cache clean
> apt update
> apt upgrade
> ```

Ooooh, thank you! Okay, I have no idea how to login as a root user. I tried `sudo -I` and that was an illegal command. I'm seeing varying things when I google. Should I try this?

> You will need to do:
> 
> sudo passwd root
> Then go ahead to set password. Use the root user using
> 
> su root

Which honestly the second part makes no sense to me, lol. Man, I hate this! I feel like im taking up too much of your time. I really should just buy a new sd card and reinstall, but it takes so long.

Joulinar, 2022-03-13:
How do you login as user `dietpi`? Simply use user `root` instead.

doobersn, 2022-03-13:
> O
>
>
> Ooooh, thank you! Okay, I have no idea how to login as a root user. I tried `sudo -I` and that was an illegal command. I'm seeing varying things when I google. Should I try this?
>
>
> Which [...]

```
root@DietPi:~# journalctl
-- Logs begin at Sun 2022-03-13 19:44:08 GMT, end at Sun 2022-03-13 19:45:16 GMT
. --
Mar 13 19:44:08 DietPi kernel: Booting Linux on physical CPU 0x0
Mar 13 19:44:08 DietPi kernel: Linux version 5.10.63+ (d
om@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GN
U ld (GNU Binutils for Ubuntu) 2.34) #1457 Tue Sep 28 11:24:51 BST 2021
Mar 13 19:44:08 DietPi kernel: CPU: ARMv6-compatible processor [410fb767] revisi
on 7 (ARMv7), cr=00c5387d
Mar 13 19:44:08 DietPi kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT non
aliasing instruction cache
Mar 13 19:44:08 DietPi kernel: OF: fdt: Machine model: Raspberry Pi Zero W Rev 1
.1
Mar 13 19:44:08 DietPi kernel: random: fast init done
Mar 13 19:44:08 DietPi kernel: Memory policy: Data cache writeback
Mar 13 19:44:08 DietPi kernel: Reserved memory: created CMA memory pool at 0x1ac
00000, size 64 MiB
Mar 13 19:44:08 DietPi kernel: OF: reserved mem: initialized node linux,cma, com
patible id shared-dma-pool
Mar 13 19:44:08 DietPi kernel: Zone ranges:
Mar 13 19:44:08 DietPi kernel:   Normal   [mem 0x0000000000000000-0x000000001eff
ffff]
Mar 13 19:44:08 DietPi kernel: Movable zone start for each node
Mar 13 19:44:08 DietPi kernel: Early memory node ranges
Mar 13 19:44:08 DietPi kernel:   node   0: [mem 0x0000000000000000-0x000000001effffff]
Mar 13 19:44:08 DietPi kernel: Initmem setup node 0 [mem 0x0000000000000000-0x000000001effffff]
Mar 13 19:44:08 DietPi kernel: On node 0 totalpages: 126976
Mar 13 19:44:08 DietPi kernel:   Normal zone: 1116 pages used for memmap
Mar 13 19:44:08 DietPi kernel:   Normal zone: 0 pages reserved
Mar 13 19:44:08 DietPi kernel:   Normal zone: 126976 pages, LIFO batch:31
Mar 13 19:44:08 DietPi kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Mar 13 19:44:08 DietPi kernel: pcpu-alloc: [0] 0 
Mar 13 19:44:08 DietPi kernel: Built 1 zonelists, mobility grouping on.  Total pages: 125860
Mar 13 19:44:08 DietPi kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=720 bcm2708_fb.fb
height=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:4B:7B:21 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=cb7b86f7-02 r
ootfstype=ext4 fsck.repair=yes rootwait
Mar 13 19:44:08 DietPi kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Mar 13 19:44:08 DietPi kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Mar 13 19:44:08 DietPi kernel: mem auto-init: stack:off, heap alloc:off, heap free:off
Mar 13 19:44:08 DietPi kernel: Memory: 423116K/507904K available (8683K kernel code, 1324K rwdata, 2824K rodata, 420K init, 837K bss, 19252K reserved, 65536K cma-reserved)
Mar 13 19:44:08 DietPi kernel: SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Mar 13 19:44:08 DietPi kernel: ftrace: allocating 30718 entries in 60 pages
Mar 13 19:44:08 DietPi kernel: ftrace: allocated 60 pages with 4 groups
Mar 13 19:44:08 DietPi kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Mar 13 19:44:08 DietPi kernel: random: get_random_bytes called from start_kernel+0x324/0x560 with crng_init=1
Mar 13 19:44:08 DietPi kernel: sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Mar 13 19:44:08 DietPi kernel: clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Mar 13 19:44:08 DietPi kernel: bcm2835: system timer (irq = 27)
Mar 13 19:44:08 DietPi kernel: Console: colour dummy device 80x30
Mar 13 19:44:08 DietPi kernel: printk: console [tty1] enabled
Mar 13 19:44:08 DietPi kernel: Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Mar 13 19:44:08 DietPi kernel: pid_max: default: 32768 minimum: 301
Mar 13 19:44:08 DietPi kernel: LSM: Security Framework initializing
Mar 13 19:44:08 DietPi kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mar 13 19:44:08 DietPi kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mar 13 19:44:08 DietPi kernel: cgroup: Disabling memory control group subsystem
Mar 13 19:44:08 DietPi kernel: CPU: Testing write buffer coherency: ok
Mar 13 19:44:08 DietPi kernel: Setting up static identity map for 0x8200 - 0x8238
Mar 13 19:44:08 DietPi kernel: devtmpfs: initialized
Mar 13 19:44:08 DietPi kernel: VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Mar 13 19:44:08 DietPi kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 361 lines cut ...]
                                 Dload  Upload   Total   Spent    Left  Speed
  0   356    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 1317k  100 1317k    0     0  1348k      0 --:--:-- --:--:-- --:--:-- 2693k
root@DietPi:/tmp# dpkg -i apt_1.8.2.3_armhf.deb
(Reading database ... 25253 files and directories currently installed.)
Preparing to unpack apt_1.8.2.3_armhf.deb ...
Unpacking apt (1.8.2.3) over (1.8.2.3) ...
Setting up apt (1.8.2.3) ...
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Processing triggers for libc-bin (2.28-10+rpt2+rpi1) ...
root@DietPi:/tmp# rm apt_1.8.2.3_armhf.deb
root@DietPi:/tmp# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@DietPi:/tmp# apt update
Reading package lists... Done
E: Method https has died unexpectedly!
E: Sub-process https received signal 4.
root@DietPi:/tmp# apt upgrade
```

MichaIng, 2022-03-13:
```
systemd-fsck[99]: Signal (11) SIGSEGV si_code=SEGV_MAPERR fault addr=0x6da4d2a6
```
`fsck` itself fails with a segmentation fault 🤔. Let's try to reinstall it as well:
```sh
cd /tmp
curl -LO 'http://raspbian.raspberrypi.org/raspbian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u3_armhf.deb'
dpkg -i e2fsprogs_1.44.5-1+deb10u3_armhf.deb
rm e2fsprogs_1.44.5-1+deb10u3_armhf.deb
```

Let's also try to switch to plain HTTP with the RPi repo:
```sh
sed -i 's|https://|http://|' /etc/apt/sources.list.d/raspi.list
apt update
```

doobersn, 2022-03-13:
> ```
> systemd-fsck[99]: Signal (11) SIGSEGV si_code=SEGV_MAPERR fault addr=0x6da4d2a6
> ```
> 
> 
>      
>   
> 
> `fsck` itself fails with a segmentation fault 🤔. Let's try to reinstall it as well:
> 
> 
> ```shell
> cd /tmp
> curl -LO 'http://raspbian.raspberrypi.org/raspbian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u3_armhf.deb'
> dpkg -i e2fsprogs_1.44.5-1+deb10u3_armhf.deb
> rm e2fsprogs_1.44.5-1+deb10u3_armhf.deb
> ```
> 
> 
>      
>   
> 
> Let's also try to switch to plain HTTP with the RPi repo:
> 
> 
> ```shell
> sed -i 's|https://|http://|' /etc/apt/sources.list.d/raspi.list
> apt update
> ```

Alrighty, here's another!

```
root@DietPi:~# sed -i 's|https://|http://|' /etc/apt/sources.list.d/raspi.list
root@DietPi:~# apt update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]                                             
Get:6 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]                                             
Get:7 http://raspbian.raspberrypi.org/raspbian buster/rpi armhf Packages [1,360 B]                                                 
Fetched 13.6 MB in 30s (461 kB/s)                                                                                                  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
44 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:~# cd /tmp
root@DietPi:/tmp# curl -LO 'http://raspbian.raspberrypi.org/raspbian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u3_armhf.deb'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   383    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  519k  100  519k    0     0   610k      0 --:--:-- --:--:-- --:--:-- 1794k
root@DietPi:/tmp# dpkg -i e2fsprogs_1.44.5-1+deb10u3_armhf.deb
(Reading database ... 25253 files and directories currently installed.)
Preparing to unpack e2fsprogs_1.44.5-1+deb10u3_armhf.deb ...
Unpacking e2fsprogs (1.44.5-1+deb10u3) over (1.44.5-1+deb10u3) ...
Setting up e2fsprogs (1.44.5-1+deb10u3) ...
root@DietPi:/tmp# rm e2fsprogs_1.44.5-1+deb10u3_armhf.deb
root@DietPi:/tmp# 
```

Btw, when you post these commands, can I literally copy and paste them when they are multiple lines? Thank you so so much!

doobersn, 2022-03-13:
OMFG!!!!

IT WORKED!!! 

Longest install ever but it apparently even fixed the pihole webgui!! THANK YOU @MichaIng! Also thank you to @Joulinar!!!

MichaIng, 2022-03-13:
I try to have the commands in code blocks so that they can be copy&pasted all together. But it may have some advantage to paste them one by one to have more control.

> 44 packages can be upgraded. Run 'apt list --upgradable' to see them.

44 packages were waiting for an upgrade, which is the reason it took a while 🙂.

The time zone can be adjusted via `dietpi-config` or `dpkg-reconfigure tzdata`.

Great that it worked now. Still, I'd retry the `fsck` and after that worked try to fix HTTPS for APT:
```sh
sudo dash -c '> /forcefsck'
sudo reboot
# after reboot
sudo journalctl -t systemd-fsck
```

MichaIng, 2022-04-03:
Marking this as closed now. Feel free to reopen when required.

closed 2022-04-03 by MichaIng