MichaIng/DietPi #5364: Add dietpi to community pi-imager

guysoft, 2022-03-17:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Hey, I got a request to add dietpi to the community rpi-imager I am maintaining called pi-imager:
https://github.com/guysoft/pi-imager
I am also the creator of CustomPiOS :)

#### Describe the solution you'd like:
To serve the latest imager, a json file needs to be created and hosted pointing to the latest image for people to download.
an example would look like this:
https://unofficialpi.org/rpi-imager/rpi-imager-hotspotos.json

#### Describe alternatives you've considered:

- ...

#### Additional context
Another option could be that you could fork your own imager. I added ci/cd in place so every commit builds a windows/mac/linux build.
Or you could submit a request to the official rpi-imgaer. I just know that my pi-imager would welcome this :)

Talks where this was suggested:
https://forums.balena.io/t/i-forked-rpi-imager-and-thought-it-might-be-an-improvment-for-you-over-balena-etcher/354705/2

Joulinar, 2022-03-17:
We offer quite a lot of images for various SBC. Are you looking for images related to RPI platform only?

guysoft, 2022-03-17:
I suggested that on the raspberry rpi-imager issue tracker and my comment got removed.
That followed an email and a talk with rpi trading limited. We are on good terms, and I really don't want to piss of the Rpi foundation. I like what they do when it comes to device quality and promoting open source hardware.

Honestly I am quite conflicted with that since I have quite a lot of my open source work revolving raspberrypi. but I don't like concept of vendor lock-in neither. So at the moment my stand is I will do that only if there is overwhelming request and usage for this that would justify making such a move that could cost me personally (which is not the case at the moment, the imager is not that popular yet being released this month).

You are also welcome to fork the current imager, as mentioned  in the Additional context section.

MichaIng, 2022-03-17:
I tried to implement it into `rpi-imager` (though I'm no C coder) and one issue is that `7z` archives are not well supported. The underlying archive library does support it, but archives are handled a way which does not work with the DietPi images currently. We're already thinking to switch to single xz-compressed image files, having a hash and a signature as dedicated download, but until then it is at least difficult to implement. I'll have a look into `pi-imager`, probably it is easier there.

nafanz, 2025-03-30:
How about we return to the discussion of including the image in the official `rpi-imager`?

This is being discussed again here https://github.com/raspberrypi/rpi-imager/issues/1025

MichaIng, 2025-04-19:
Yeah we were digging into it already: #7450
Seems to be sufficiently easy to take the relevant part of the JSON, host it on our server (update hashes there whenever images are updated), and do a request with the provided online form.

StephanStS, 2025-06-15:
@MichaIng : Can this also be closed?
See #7450