MichaIng/DietPi #5366: installing ncurses-term does not remove warning for unsupported terminal

Dagon, 2022-03-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
  G_DIETPI_VERSION_SUB=2
  G_DIETPI_VERSION_RC=2
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
- Distro version | bullseye
- Kernel version | Linux tinker 5.15.25-rockchip #22.02.1 SMP PREEMPT Sun Feb 27 09:23:03 UTC 2022 armv7l GNU/Linux
- SBC model | ASUS Tinker Board (armv7l)
- Power supply used | (POE adapter)

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?  Fresh DietPi install
- Can this issue be replicated on a fresh installation of DietPi? Yes, this is fresh

- Bug report ID |  Reference code: 3bca477d-8df9-4b7b-a183-3697a86241be

#### Steps to reproduce

1. Using OpenSSH server (if that matters)
2. Use PuTTY from windows to connect

#### Expected behaviour

- First time, warning that it needs to install ncurses-term to have the terminfo for putty-256color
- After installing ncurses-term, warning should no longer appear

#### Actual behaviour

- continues to give warning for unknown terminal putty-256color even after ncurses-term is installed

#### Extra details

- Caused by fix to https://github.com/MichaIng/DietPi/issues/2034 - that supports installing ncurses-term, which is good and correct, but it's looking for files in /lib/terminfo rather than /usr/share/terminfo, where ncurses-term puts it.   
- It should look in both directories for the file (actually, all directories mentioned in /etc/terminfo/README, which is 
  ncurses will search ${HOME}/.terminfo first, then /etc/terminfo (this
  directory), then /lib/terminfo, and last not least /usr/share/terminfo.
- Likely better to use /usr/bin/toe rather than looking for files in terminfo directories.

MichaIng, 2022-03-18:
Many thanks for your report. Indeed a bug our end, see: #5341

MichaIng, 2022-03-26:
Solved with: https://github.com/MichaIng/DietPi/commit/6db67e2

closed 2022-03-26 by MichaIng