MichaIng/DietPi #5367: Dietpi-ddns | Adding cloudflare API support

Tooloco, 2022-03-18:
### feature request

#### Is your feature request related to a problem? Please describe:

- I just want to use cloudflare as my dynamic dns and they dont have an api setup that the custom entry can work with

#### Describe the solution you'd like:

- Cloudflare option for the dietpi-ddns menu

#### Describe alternatives you've considered:

- Editing /var/lib/dietpi/dietpi-ddns/update.sh manually to a custom curl command to update the dns record

#### Additional context

- Okay the thing is its a bit more complicated to do this on cloudflare than any other ddns provider.
- There's no user + password combination to update the record here's how you do it:
- Get the zone ID (identifier) for your domain name
- Create an API token with sufficient permissions to edit the zone the dns record is in
- List DNS records to find record identifier. api request: `GET zones/:zone_identifier/dns_records`
- Find the DNS record identifier you want to update (it already has to exist, so you have to manually create it on the cloudflare dashboard)
- When you gave both zone and dns record identifiers contruct the API request to update the record: `PUT zones/:zone_identifier/dns_records/:identifier`
- Construct the full curl command. In this case the full url would be like this: `https://<base_url>/zone/<zone_id>/dns_records/<identifier>`
- Full example (not mine taken from cloudflare docs):
- `curl -X PUT "https://api.cloudflare.com/client/v4/zones/023e105f4ecef8ad9ca31a8372d0c353/dns_records/372e67954025e0ba6aaa6d586b9e0b59" -H "Content-Type:application/json" -H "Authorization: Bearer YQSn-xWAQiiEh9qM58wZNnyQS7FUdoqGIUAbrh7T" --data '{"type":"A","name":"example.com","content":"127.0.0.1","ttl":3600,"proxied":false}'`
- Authorization: Bearer is the API token that we created earlier.
- Sources and documentation is here: [Base url + API token ](https://api.cloudflare.com/#getting-started-requests), [updating dns record](https://api.cloudflare.com/#dns-records-for-a-zone-update-dns-record)

So im thinking to do a pull request and do it myself but im really busy atm and I would have to lean about shell and understand the current dietpi-ddns script. What im thinking on doing is editing update.sh to the full curl command for now. So I thought I would submit a feature request so somebody that has more experience can work on it, if not I'll do it myself (I dont think its that complicated so it seems feasible for me, although only time will tell).

Joulinar, 2022-03-18:
At the end, do you have a static URL that could be executed to update the DDNS record at cloutflare? If yes, you could try using the custom option.

Tooloco, 2022-03-18:
> At the end, do you have a static URL that could be executed to update the DDNS record at cloutflare? If yes, you could try using the custom option.

Yes, but you have to include other information in the curl command. the data has to have the type, name, content, ttl, and proxied (https://api.cloudflare.com/#dns-records-for-a-zone-update-dns-record)

MichaIng, 2022-03-18:
Cloudflare is not a classic DDNS provider, but a regular DNS provider with consequent web APIs for all their settings. The required `curl` call hence differs quite significantly from what is used with DDNS providers. It can be added, but requires a little rework of the script.

Tooloco, 2022-03-19:
Ok so the easiest solution for me was adding a CNAME DNS record pointing to my DDNS domain

MichaIng, 2022-03-19:
It is safe to adjust the `update.sh`, just assure that its strict modes remain so that only the `dietpi-ddns` user is able to read it, for the sake of keeping the token safe.