MichaIng/DietPi #5369: Exit Code 28 trying to install Lidarr (and other softwares)

johnisprettyneat, 2022-03-19:
#### Details:
- Date           | Sat Mar 19 15:39:11 EDT 2022
- DietPi version | v8.2.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v7+ #1530 SMP Tue Mar 8 13:02:44 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `curl -ILfvm 5 https://github.com/Lidarr/Lidarr/releases/download/v0.8.1.2135/Lidarr.master.0.8.1.2135.linux-core-arm.tar.gz`
- Exit code      | 28
- Software title | DietPi-Software
#### Steps to reproduce:

1. Use "sudo dietpi-software" to install a package, in this example Lidarr
#### Expected behaviour:

- Should install Lidarr to the machine
#### Actual behaviour:

- is timing out with error code 28, even after trying different mirrors
#### Extra details:

- Not sure what is going on should be relatively straight forward installation. This happens for several other softwares as well when attempting to install.
#### Additional logs:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0* Resolving timed out after 5000 milliseconds
  0     0    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
* Closing connection 0
curl: (28) Resolving timed out after 5000 milliseconds
```

MichaIng, 2022-03-19:
Many thanks for your report. Resolving the hostname `github.com` failed within 5 seconds. Usually this should take around 0.2 seconds.

Can you try this from console:
```sh
curl -ILfvm 5 https://github.com/
time getent ahost github.com
```

johnisprettyneat, 2022-03-20:
Hi @MichaIng appreciate the response

_Running the first command I get:_
```
* Resolving timed out after 5000 milliseconds
* Closing connection 0
curl: (28) Resolving timed out after 5000 milliseconds
```

_Running the second command (which I think should be ahosts):_
```
140.82.113.4    STREAM github.com
140.82.113.4    DGRAM  
140.82.113.4    RAW    

real    0m5.072s
user    0m0.012s
sys     0m0.002s
```

Joulinar, 2022-03-20:
and what about increasing time out value to 20 sec?

```
curl -ILfvm 20 https://github.com/
time getent ahost github.com
```

MichaIng, 2022-03-21:
> real    0m5.072s

Slightly longer than 5 seconds 😄. Yes increasing the timeout is a workaround, but having > 5 seconds just for hostname isn't great. Is it the same with other hostnames?
```sh
time getent ahost dietpi.com
time getent ahost google.com
```

johnisprettyneat, 2022-03-21:
polled both dietpi.com and google.com, recieved 5.025s and 5.043s respectively. Which is very strange, when I try from my laptop on my same network I am getting 0.326s.

_When I run for 20s as @Joulinar mentions i get the following:_
`*   Trying 140.82.113.4:443...
* Connected to github.com (140.82.113.4) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=github.com
*  start date: Mar 15 00:00:00 2022 GMT
*  expire date: Mar 15 23:59:59 2023 GMT
*  subjectAltName: host "github.com" matched cert's "github.com"
*  issuer: C=US; O=DigiCert Inc; CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0xa0ad20)
> HEAD / HTTP/2
> Host: github.com
> user-agent: curl/7.74.0
> accept: */*
> 
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* Connection state changed (MAX_CONCURRENT_STREAMS == 100)!
< HTTP/2 200 
HTTP/2 200 
< server: GitHub.com
server: GitHub.com
< date: Mon, 21 Mar 2022 00:21:41 GMT
date: Mon, 21 Mar 2022 00:21:41 GMT
< content-type: text/html; charset=utf-8
content-type: text/html; charset=utf-8
< vary: X-PJAX, X-PJAX-Container, Accept-Language, Accept-Encoding, Accept, X-Requested-With
vary: X-PJAX, X-PJAX-Container, Accept-Language, Accept-Encoding, Accept, X-Requested-With
< permissions-policy: interest-cohort=()
permissions-policy: interest-cohort=()
< content-language: en-US
content-language: en-US
< etag: W/"6e9371dfc8832113f407ebadd9c1f273"
etag: W/"6e9371dfc8832113f407ebadd9c1f273"
< cache-control: max-age=0, private, must-revalidate
cache-control: max-age=0, private, must-revalidate
< strict-transport-security: max-age=31536000; includeSubdomains; preload
strict-transport-security: max-age=31536000; includeSubdomains; preload
< x-frame-options: deny
x-frame-options: deny
< x-content-type-options: nosniff
x-content-type-options: nosniff
< x-xss-protection: 0
x-xss-protection: 0
< referrer-policy: origin-when-cross-origin, strict-origin-when-cross-origin
referrer-policy: origin-when-cross-origin, strict-origin-when-cross-origin
< expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
< content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com collector.github.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com github.githubassets.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com collector.github.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com customer-stories-feed.github.com spotlights-feed.github.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/ github.githubassets.com; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com collector.github.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com github.githubassets.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com notebooks.githubuserconte [... cut ...]
< set-cookie: logged_in=no; Path=/; Domain=github.com; Expires=Tue, 21 Mar 2023 00:21:48 GMT; HttpOnly; Secure; SameSite=Lax
set-cookie: logged_in=no; Path=/; Domain=github.com; Expires=Tue, 21 Mar 2023 00:21:48 GMT; HttpOnly; Secure; SameSite=Lax
< accept-ranges: bytes
accept-ranges: bytes
< x-github-request-id: C038:4528:10EF606:1AE76EC:6237C51C
x-github-request-id: C038:4528:10EF606:1AE76EC:6237C51C

< 
* Connection #0 to host github.com left intact`

will increase timeout as mentioned with G_CONFIG_INJECT and report back afger install attempt

johnisprettyneat, 2022-03-21:
ok still getting a fail here is the error:

```
#### Details:
- Date           | Sun Mar 20 20:33:11 EDT 2022
- DietPi version | v8.2.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v7+ #1530 SMP Tue Mar 8 13:02:44 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `curl -ILfvm 10 https://github.com/Lidarr/Lidarr/releases/download/v0.8.1.2135/Lidarr.master.0.8.1.2135.linux-core-arm.tar.gz`
- Exit code      | 28
- Software title | DietPi-Software
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0*   Trying 140.82.112.4:443...
* Connected to github.com (140.82.112.4) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
* TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [2459 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [78 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=github.com
*  start date: Mar 15 00:00:00 2022 GMT
*  expire date: Mar 15 23:59:59 2023 GMT
*  subjectAltName: host "github.com" matched cert's "github.com"
*  issuer: C=US; O=DigiCert Inc; CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
} [5 bytes data]
* Using Stream ID: 1 (easy handle 0x27ed20)
} [5 bytes data]
[... 33 lines cut ...]
* Connection #0 to host github.com left intact
* Issue another request to this URL: 'https://objects.githubusercontent.com/github-production-release-asset-2e65be/90468352/9ac4c200-a03c-11eb-88c2-541c65a03d61?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220321%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220321T003235Z&X-Amz-Expires=300&X-Amz-Signature=d7b5cc3f2356746754b537dfcfd6a490063a07a4b7d5d215555d81105db4112d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=90468352&response-content-disposition=attachment%3B%20filename%3DLidarr.master.0.8.1.2135.linux-core-arm.tar.gz&response-content-type=application%2Foctet-stream'
  0     0    0     0    0     0      0      0 --:--:--  0:00:14 --:--:--     0* Resolving timed out after 10000 milliseconds
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
* Closing connection 1
curl: (28) Resolving timed out after 10000 milliseconds
HTTP/2 302 
server: GitHub.com
date: Mon, 21 Mar 2022 00:32:35 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/90468352/9ac4c200-a03c-11eb-88c2-541c65a03d61?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220321%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220321T003235Z&X-Amz-Expires=300&X-Amz-Signature=d7b5cc3f2356746754b537dfcfd6a490063a07a4b7d5d215555d81105db4112d&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=90468352&response-content-disposition=attachment%3B%20filename%3DLidarr.master.0.8.1.2135.linux-core-arm.tar.gz&response-content-type=application%2Foctet-stream
cache-control: no-cache
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.githubapp.com collector.github.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locati [... cut ...] ithubassets.com identicons.github.com collector.githubapp.com collector.github.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/
content-length: 683
x-github-request-id: 96DA:62A2:29F0B3:97B48F:6237C7B8

```

Joulinar, 2022-03-21:
still a time out. You need to increase the value again. 

`curl: (28) Resolving timed out after 10000 milliseconds`

johnisprettyneat, 2022-03-21:
okay bumped it to 20s and install appears to be successful. Any reason why hostname resolution is taking so long?

Also - very much appreciate the support!

Joulinar, 2022-03-21:
what is the DNS server configured?

```
cat /etc/resolv.conf
```

johnisprettyneat, 2022-03-21:
I have two entries which are the two piholes on my LAN, both addresses look correct. Both piholes run unbound as well.

MichaIng, 2022-03-21:
Probably the query logs of Pi-hole give a hint?

Joulinar, 2022-03-21:
have a look to PiHole Query log for GitHub entries. Will looks like this

[image]

johnisprettyneat, 2022-03-21:
ok I think I figured it out with your help @Joulinar I had pihole A interface settings only allowing local requests (1 hop), which was getting the initial request, blocking it, then would kick it over to the other pihole B which would answer the query. Switched pihole A to respond on eth0 and checked  `time getent ahost github.com` and it resolves immediately

```
140.82.112.4    STREAM github.com
140.82.112.4    DGRAM  
140.82.112.4    RAW    

real    0m0.063s
user    0m0.001s
sys     0m0.006s
```
for reference it is these settings on pihole A

[image]

I think this is resolved and I will change check url timeout back to 5s!
Thank you so much!!!!!

Joulinar, 2022-03-21:
Just out if interest, why are you passing request from PiHole A to PiHole B and not to a public upstream DNS? :thinking:

MichaIng, 2022-03-21:
I think meant was that the DietPi systems uses Pi-hole B after Pi-hole A denied an answer, since both are configured in `/etc/resolv.conf`.

Great that it works now, I'll mark this as closed. Feel free to reopen if required.

closed 2022-03-21 by MichaIng

johnisprettyneat, 2022-03-21:
It shouldn't have been, I should mention that I have multiple VLANs setup on my network to separate devices. So when the dietpi was making its initial DNS request to Pihole A from a seperate VLAN and I had this setting checked on Pihole A 

[image]

It was then rejecting the request, so I assume that made the dietpi try to resolve using pihole B, which is the other name resolver listed in the resolv.conf which would eventually go through but caused all the delay time.

johnisprettyneat, 2022-03-21:
@MichaIng  yep you got it. Please mark as closed. Thank you to both you and @Joulinar for the support, issue solved! :) :)