MichaIng/DietPi #5372: RPi Zero W/2W Wifi Module "Not available" till Bluetooth is enabled as well

scetu, 2022-03-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 1
`
- Kernel version | `uname -a`
`Linux DietPi 5.10.103+ #1530 Tue Mar 8 13:00:30 GMT 2022 armv6l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi Zero W (armv6l)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V 2A`
- SD card used | (EG: SanDisk ultra)
`Some old 8GB Kingston card, not relevant`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`None, fresh install`
- Can this issue be replicated on a fresh installation of DietPi?
`Yes`

#### Steps to reproduce

1. Flash `DietPi_RPi-ARMv6-Bullseye.7z` (for old RPi Zero) or  `DietPi_RPi-ARMv7-Bullseye.7z` (for new Zero 2W)
2. Turn on device without ethernet dongle
3. Skip first time setup, and go to `dietpi-config`
4. Turn on `on-board WiFi`
5. Reboot
6. Go to `dietpi-config` again and you should see "No WiFi adapter available"
7. In `dietpi-config` navigate to `Advanced options` and enable `Bluetooth`
8. You cannot install required `pi-bluetooth` packages without connection, and Wifi and/or Bluetooth is "Not available"

If you start over and power on device with ethernet dongle you will be able to install `pi-bluetooth`, once this is installed and you reboot RPi, WiFi is available even when you will disable bluetooth afterwards.

#### Expected behaviour

- Wifi should be possible setup without additional installation of packages and/or Bluetooth should be enabled as well in order to get Wifi adapter available
- Also I did not noticed anything related to fact that RPi has to be rebooted in order to enable On-Board WiFi, I found out this from other WiFi related bugs on Github

#### Actual behaviour

- On board Wifi adapter is "Not available" even after turning on

#### Extra details

- already went through bellow listed issues, before I tried to turn on Bluetooth and WiFi started working
- https://github.com/MichaIng/DietPi/issues/818
- https://github.com/MichaIng/DietPi/issues/4770
- https://github.com/MichaIng/DietPi/issues/3490
- https://github.com/MichaIng/DietPi/issues/787
- This issue is applicable to RPi Zero W and as well I've tested this on new Zero 2W

MichaIng, 2022-03-20:
Many thanks for your report.

We know a similar issue on Raspberry Pi 3B+ #4512 where WiFi and Bluetooth physically share the same trigger lane, but this was only on firmware basis the `config.txt` setting, not any userland package or driver, it was the other way round, disabling WiFi broke Bluetooth, not the other way round, and AFAIK it is solved in the meantime.

Your description that the `config.txt` setting doesn't play a role but only the `pi-bluetooth` package, is very unexpected. Also that it affects all your RPi's while we have no single other reports among something with 80,000 RPi systems and with Bluetooth disabled by default on DietPi. But let me quickly try to replicate on RPi Zero W (1).

MichaIng, 2022-03-20:
Okay, I just upgraded my RPi Zero W (1) to latest packages and latest DietPi, and if does not require `pi-bluetooth` or any of it's dependencies and it does not require `dtoverlay=disable-bt` to be removed from `/boot/config.txt` to physically enable Bluetooth. Bluetooth is not and never was enabled on this device. So I think in your case something different happened which is not at all related to Bluetooth.

Here is how it should be:
- Our RPi images whip without `dtoverlay=disable-wifi` in `/boot/config.txt`, hence onboard WiFi is physically enabled on first boot, so that it can be configured OOTB with `/boot/dietpi.txt` settings.
- Onboard WiFi is however disabled during first boot (effective from reboot on) if not enabled via `AUTO_SETUP_NET_WIFI_ENABLED=1` in `/boot/dietpi.txt` or manually via `dietpi-config`.
- `dietpi-config` should how `Onboard WiFi : [On]/[Off]` based on `dtoverlay=disable-wifi`, so when you did not set `AUTO_SETUP_NET_WIFI_ENABLED=1` (instructing the first boot script to leave WiFi + onboard WiFi enabled by not adding `dtoverlay=disable-wifi`), then if should show `[Off]`. If you didn't change this on first boot, on reboot onboard WiFi won't be available anymore.

Also to clarify which message you see where: "No WiFi adapter available" is not shown in any of our menus. When you select the WiFi submenu and no WiFi interface is found, it shows "No supported WiFi hardware was found." (and further lines). Is that what you saw? The WiFi interface is obtained via this command:
```sh
G_GET_NET -q -t wlan iface
```
Then it is checked whether this kernel tunable exists:
```sh
ls -l /sys/class/net/<iface>
```
As long as you do not have any additional WiFi adapter attached, it should be `wlan0` in any case. The only way this is missing is when `dtoverlay=disable-wifi` was added to `/boot/config.txt`, i.e. you did neither set `AUTO_SETUP_NET_WIFI_ENABLED=1` in `/boot/dietpi.txt`, nor set "Onboard WiFi" to `[On]` in `dietpi-config`. Is that correct?

closed 2022-03-21 by scetu

MichaIng, 2022-03-21:
> once onboard Wifi is activated, dietpi-config does not require reboot and it should tell user to reboot to finish activating onboard wifi

On first boot, it is not required, as said, but when enabled it after first boot, it is, you're right. I'll have a look into it: At best we find a way to detect without parsing config files whether onboard WiFi is currently enabled or not, to now show a reboot message unnecessarily. Probably the device tree allows to do that.