MichaIng/DietPi #5373: Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked

krosseyed, 2022-03-24:
### Creating a bug report/issue

#### Required Information
- DietPi version
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version
`bullseye 0`

- Kernel version
`Linux rpi4cm01 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`

- SBC model 
`RPi CM 4 (aarch64)`

- Power supply used
https://www.amazon.com/gp/product/B00VDVCQ84

- SD card used 
SanDisk 32gb SDXC microsd card

#### Additional Information (if applicable)
- Software title
dietpi-update

- Was the software title installed freshly or updated/migrated?
Fresh install

- Can this issue be replicated on a fresh installation of DietPi?
In theory, yes.

- Bug report ID
fae03544-8d55-4a40-b5b6-17aceec7cbf3

#### Steps to reproduce
run dietpi-update

#### Expected behaviour
update dietpi operating system

#### Actual behaviour
  - Command: systemctl start systemd-timesyncd
│  - Exit code: 1
│  - DietPi version: v7.6.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3
│ DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: From scratch
│  - Error log:
│ Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is
│ masked. 

#### Extra details
I have installed proxmox on this dietpi device, and it has removed systemd-timesyncd and installed chrony.

Joulinar, 2022-03-24:
In this case, you would need to disable time sync within `dietpi-config` > `advanced options`

krosseyed, 2022-03-24:
I set it to custom and it now updates just fine. Thank you for that info!

closed 2022-03-24 by krosseyed