MichaIng/DietPi #5376: NO_PUBKEY

zarigata, 2022-03-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | latest
- Distro version | DIY
- Kernel version | `uname -a`
- SBC model |ORANGE PI 4 LTS
- Power supply used | 5V USB-C 5A
- SD card used | 16gbs EEMC
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... try to install
2. ...

#### Expected behaviour

- ... install

#### Actual behaviour

- ...
- │ W: GPG error: https://deb.debian.org/debian bullseye-updates InRelease: The
│ following signatures couldn't be verified because the public key is not
│ available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
│ E: The repository 'https://deb.debian.org/debian bullseye-updates InRelease'
│ is not signed.
│ W: GPG error: https://deb.debian.org/debian-security bullseye-security
│ InRelease: The following signatures couldn't be verified because the public
│ key is not available: NO_PUBKEY 112695A0E562B32A NO_PUBKEY 54404762BBB6E853
│ E: The repository 'https://deb.debian.org/debian-security bullseye-security
│ InRelease' is not signed.
│ W: GPG error: https://deb.debian.org/debian bullseye-backports InRelease:
│ The following signatures couldn't be verified because the public key is not
│ available: NO_PUBKEY 648ACFD622F3D138 NO_PUBKEY 0E98404D386FA1D9
│ E: The repository 'https://deb.debian.org/debian bullseye-backports
│ InRelease' is not signed.

#### Extra details

- ...

Joulinar, 2022-03-25:
Can you share some more information what are you trying to do? Did you run the PREP script?and if yes, what was the base image?

zarigata, 2022-03-25:
im using the 'bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-imager')"' script, and i am tringt to create a dietpi in my orangepi

Joulinar, 2022-03-25:
Are you sure you used the imager tool? This will create an image of an existing system. Means the system would need to exist. 

Maybe you like to share the image you used as starting point to create your DietPi system?

zarigata, 2022-03-25:
`http://www.orangepi.org/downloadresources/Orange%20Pi%204%20LTS/2022-03-09/OrangePi4LTS1610374db1f734d71110fbb2.html` that is the original image,

Joulinar, 2022-03-25:
This is an Ubuntu image. Dietpi is based in Debian. Means you can't use this image as source.

MichaIng, 2022-03-25:
The link doesn't work, but from the html file name it can be derived from here: http://www.orangepi.org/downloadresources/

Ah @Joulinar was too fast, so yeah you need to take the Debian image.

zarigata, 2022-03-25:
> This is an Ubuntu image. Dietpi is based in Debian. Means you can't use this image as source.

i was using an debian, and the same error occurred,  thats why i changed to ubuntu

MichaIng, 2022-03-25:
Please try again with the Debian image and post the output of the script here.

zarigata, 2022-03-25:
same issue, gonna try to make it on SD card

zarigata, 2022-03-25:
found the issue, everytime it is on the EMMC the setup does not work, maybe something on the debian or the get data

closed 2022-03-25 by zarigata