MichaIng/DietPi #5378: RK3328 | Latest kernel breaks USB3

fra87, 2022-03-25:
### Creating a bug report/issue

#### Required Information
- DietPi version |
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `bullseye`
- Kernel version | `Linux ServerCasaCN 5.15.25-rockchip64 #22.02.1 SMP PREEMPT Sun Feb 27 09:05:47 UTC 2022 aarch64 GNU/Linux`
- SBC model | `ROCK64 (aarch64)`
- Power supply used | (5V several Amps (at least 3))
- SD card used | (Samsung Evo Plus 32GB)

#### Steps to reproduce
1. Update the kernel to the latest version (`linux-image-current-rockchip64=22.02.1`)
2. Reboot

#### Expected behaviour
- USB3 port works (in particular I have a USB drive)

#### Actual behaviour

- USB3 port is not available; dmesg shows
```
usb 5-1: USB disconnect, device number 2
scsi host0: uas_eh_device_reset_handler FAILED err -19
sd 0:0:0:0: Device offlined - not ready after error recovery
sd 0:0:0:0: Device offlined - not ready after error recovery
sd 0:0:0:0: rejecting I/O to offline device
[...]
usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
```

#### Extra details
- Reverting to 21.08.2 solved the issue. I did as highlighted in the workaround here: [Issue 5337](https://github.com/MichaIng/DietPi/issues/5337), but left DTB to the current version since the issue was solved just by changing the image:
```
apt install linux-image-current-rockchip64=21.08.2
apt-mark hold linux-image-current-rockchip64
```

Joulinar, 2022-03-25:
Should be similar to this one https://dietpi.com/phpbb/viewtopic.php?t=10195

fra87, 2022-03-25:
@Joulinar yes, it's the same.. Sorry then for the double-posting.

With respect to that thread, my command has actually the bullseye repo (even if the versions look the same):

```
dietpi@rock64:~$ sudo apt list -a linux-image-current-rockchip64
Elencazione... Fatto
linux-image-current-rockchip64/bullseye 22.02.1 arm64 [aggiornabile da: 21.08.2]
linux-image-current-rockchip64/bullseye,now 21.08.2 arm64 [installato, aggiornabile a: 22.02.1]
linux-image-current-rockchip64/bullseye 21.08.1 arm64
linux-image-current-rockchip64/bullseye 21.08 arm64
linux-image-current-rockchip64/bullseye 21.05.9 arm64
linux-image-current-rockchip64/bullseye 21.05.4 arm64
linux-image-current-rockchip64/bullseye 21.05.1 arm64
linux-image-current-rockchip64/bullseye 21.02.3 arm64
linux-image-current-rockchip64/bullseye 21.02.2 arm64
```

Then, however, I don't get all the modinfo errors (sorry for the italian)

```
dietpi@rock64:~$ sudo apt install linux-image-current-rockchip64=22.02.1
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze... Fatto
Lettura informazioni sullo stato... Fatto
I seguenti pacchetti saranno aggiornati:
  linux-image-current-rockchip64
1 aggiornati, 0 installati, 0 da rimuovere e 0 non aggiornati.
È necessario scaricare 51,2 MB di archivi.
Dopo quest'operazione, verranno liberati 123 MB di spazio su disco.
Scaricamento di:1 http://armbian.12z.eu/apt bullseye/main arm64 linux-image-current-rockchip64 arm64 22.02.1 [51,2 MB]
Recuperati 51,2 MB in 12s (4.165 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Lettura del database... 27008 file e directory attualmente installati.)
Preparativi per estrarre .../linux-image-current-rockchip64_22.02.1_arm64.deb...
Removing obsolete file uInitrd-5.10.63-rockchip64
Estrazione di linux-image-current-rockchip64 (22.02.1) su (21.08.2)...
Configurazione di linux-image-current-rockchip64 (22.02.1)...
update-initramfs: Generating /boot/initrd.img-5.15.25-rockchip64
update-initramfs: Converting to U-Boot format
```

Then I rebooted, and the issue was present. Executed the two commands MichaIng showed (`depmod -a` and `update-initramfs -u`) and rebooted again, and the issue was still there. Both commands were successful

Now I reverted to 21.08.2, reverting also DTB; I'm modifying the description of the ticket removing the part where I write I reverted only the image (it's better to revert both)

Joulinar, 2022-03-25:
Basically we don't do any kernel development as we use the original kernel as it is provided by the underlying base image. In this case it's Armbian.

MichaIng, 2022-03-26:
Could you try adding/setting `overlays=usb4` in `/boot/armbianEnv.txt` and reboot? If an `overlays=` line exists already, append `usb4` to it, space-separated.

fra87, 2022-03-26:
Done; the content of that file is now

```
verbosity=4
bootlogo=false
overlay_prefix=rockchip
rootdev=UUID=19e46d43-2ef4-45d3-be06-ceee5e2cb50f
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
extraargs=net.ifnames=0
docker_optimizations=off
overlays=usb4
```

Issue is still present

MichaIng, 2022-03-27:
If someone finds time to test with latest kernel + upgrading U-Boot:
```sh
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$(findmnt -Ufnro SOURCE -M /)")"
```
The user on the forum found an older official Armbian image, upgraded to latest kernel, with broken USB3.0 as well, but latest Armbian image seems to work.

I'll also compare boot configs, which may have changed as well.

MichaIng, 2022-03-27:
Comparing the /boot/armbianEnv.txt, some USB quirks have been removed, this line does not exist anymore:
[code]usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u[/code]
Could be tried to remove it. I mean it should affect individual USB drives only, disabling UAS for them, but who knows.

fra87, 2022-03-27:
Currently I'm not at home (and I prefer not to mess with it remotely without possibility to physically access it).

If nobody else performs tests I will test the two things on Tuesday/Wednesday.

Just one thing: how can I "revert" the u-boot upgrade? Just in case it messes completely the image ;)

Thank you

MichaIng, 2022-03-27:
> Just one thing: how can I "revert" the u-boot upgrade? Just in case it messes completely the image ;)

The U-Boot package can be downgraded the same way and then re-flashed with the same method as above. But of course if boot fails... Hmm, if you have a second ARM system where you can attach the ROCK64's system drive, it can be also installed and flashed there. The second argument of the `write_uboot_platform` function is the drive where U-Boot gets flashed to, so that can be freely adjusted.

But it should be very safe when you flash the new U-Boot with the new kernel, and when you downgrade the kernel, downgrade and flash U-Boot as well.

fra87, 2022-03-27:
I totally forgot that here I have another server, which is not on a rock64 but on a nanopi neo 3. Which has the same RK3328 chip inside.

I upgraded it to 5.15.25 and... It works flawlessly... WTF??? The USB3 drive works perfectly even with the new kernel (I rebooted twice just to be sure, and it still works).

The content of the armbianEnv.txt file for the nanopi neo3 is the following

```
verbosity=4
bootlogo=false
overlay_prefix=rockchip
fdtfile=rockchip/rk3328-nanopi-r2-rev00.dtb
rootdev=UUID=7f26bae3-11cc-4a08-bcc6-df7f80c7d91b
rootfstype=ext4
console=serial
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
docker_optimizations=on
extraargs=net.ifnames=0
```

The only differences are the `docker_optimizations=on`, which I don't think is relevant here (and honestly I don't know what this optimizations are; however I have docker only on this system) and `fdtfile=rockchip/rk3328-nanopi-r2-rev00.dtb`. Do you think there can be anything linked to this parameter?

As for the uboot downgrading, I see... I thought they were separate, but from what I can guess from the script each image contains a u-boot version that can be flashed...

MichaIng, 2022-03-27:
Same USB quirks. Did you test with the same USB 3.0 drive on ROCK64 and NEO3? If the USB quirks are indeed the issue, it affects specific drives, not the USB port itself.

fra87, 2022-03-27:
No, they are two different systems, each with its own USB drive.

Anyway the USB quirks are the same also on the rock64 21.08.2 and 22.02.1, so if it is USB quirk + USB3 drive then it shall also fail there...

MichaIng, 2022-03-27:
> Anyway the USB quirks are the same also on the rock64 21.08.2 and 22.02.1, so if it is USB quirk + USB3 drive then it shall also fail there...

My idea is that the USB quirks cause issues only with the new kernel for some reason. It is at least one difference between an older Armbian image (and our current ROCK64 image) after upgrading the kernel, which has the same USB issues according to a user in our forum, and the latest Armbian image, which does not contain the quirks and does not have the issue after upgrading to the latest kernel. This and probably the bootloader.

fra87, 2022-03-27:
I'll try to test the two modifications on Tuesday when I'm back to the house with the ROCK64 (if b0b33140 or someone else does not check everything before that date).

In any case I'm not so much confident in the quirks solving the issue, since if it was depending only on them then also nanopi neo 3 would be affected. I think u-boot has a better chance to be the core issue, since u-boot from nanopi and from rock64 are different

(just for my understanding, and sorry for the slight OT here, but... do you know if u-boot gets updated "automatically" through `apt upgrade` for instance or if it needs to be explicitly upgraded? That's because I saw [here for instance](https://armbian.tnahosting.net/apt/pool/main/l/linux-u-boot-rock64-current/) that the last uboot versions are 2 months old)

MichaIng, 2022-03-28:
> if it was depending only on them then also nanopi neo 3 would be affected.

As said, those quirks contain and affect specific hardware identifiers, basically two drives which have shown to not work well with UAS in the past. So when the attached drives are different models, the quirks may affect the one attached to ROCK64 but not the one attached to NEO3.

U-Boot is never flashed automatically. The package upgrades just the "offline" files on the filesystem, basically the script sources with the command I provided above and the actual U-Boot binaries the executed function takes and flashes to the drive. This is generally a reasonable behaviour since it is not 100% failsafe to know which drive U-Boot is loaded from since it doesn't necessarily need to be the same where the root filesystem is located. And it is rarely the case that U-Boot really needs to be updated, so it's fine to leave it as a method to try solving issues but not doing it automatically.

wpwentzell, 2022-03-29:
> If someone finds time to test with latest kernel + upgrading U-Boot:
> 
> ```shell
> . /usr/lib/u-boot/platform_install.sh
> write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
> ```
> 
> The user on the forum found an older official Armbian image, upgraded to latest kernel, with broken USB3.0 as well, but latest Armbian image seems to work, which probably has a newer U-Boot.
> 
> I'll also compare boot configs, which may have changed as well.

Is this all that needs tested at this point? I may be able to do this if @fra87 is unable to.

wpwentzell, 2022-03-29:
OK, as an update to this, I tried upgrading U-Boot and installing latest kernel with no success:

```
root@DietPi:~# . /usr/lib/u-boot/platform_install.sh
root@DietPi:~# write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
root@DietPi:~# reboot
```
```
Get:2 http://armbian.chi.auroradev.org/apt buster/main arm64 linux-image-current-rockchip64 arm64 22.02.1 [51.2 MB]
Get:1 http://mirrors.fossho.st/armbian/apt buster/main arm64 linux-dtb-current-rockchip64 arm64 22.02.1 [365 kB]
Fetched 51.6 MB in 4s (14.5 MB/s)
(Reading database ... 27136 files and directories currently installed.)
Preparing to unpack .../linux-dtb-current-rockchip64_22.02.1_arm64.deb ...
Unpacking linux-dtb-current-rockchip64 (22.02.1) over (21.08.2) ...
Preparing to unpack .../linux-image-current-rockchip64_22.02.1_arm64.deb ...
update-initramfs: Deleting /boot/initrd.img-5.10.63-rockchip64
Removing obsolete file uInitrd-5.10.63-rockchip64
Unpacking linux-image-current-rockchip64 (22.02.1) over (21.08.2) ...
Setting up linux-image-current-rockchip64 (22.02.1) ...
update-initramfs: Generating /boot/initrd.img-5.15.25-rockchip64
update-initramfs: Converting to u-boot format
Setting up linux-dtb-current-rockchip64 (22.02.1) ...
```

Here's the output of dmesg after I could no longer reach my USB3 attached drive:

```
[   50.599862] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[   50.600664] usb 5-1: USB disconnect, device number 2
```
I then tried deleting the USB quirks from `/boot/armbianEnv.txt` as such:

```
verbosity=7
overlay_prefix=rockchip
rootdev=UUID=7567a27d-52bf-4abb-aed4-16d9bb93c7c3
rootfstype=ext4
console=none

```
Nothing going, so I added them back, and downgraded back to 21.08.02 via:

```
apt install linux-dtb-current-rockchip64=21.08.2 linux-image-current-rockchip64=21.08.2
apt-mark hold linux-dtb-current-rockchip64 linux-image-current-rockchip64
```

Everything works. Needed to remount my drive, but I'm back in business.

MichaIng, 2022-03-29:
> I then tried deleting the USB quirks from `/boot/armbianEnv.txt` as such:

To be sure, you rebooted afterwards? Also can you show the IDs of the attached USB devices? We can test the opposite, adding an actual UAS blacklist for the ones you use:
```sh
lsusb
```

wpwentzell, 2022-03-29:
Yes, I rebooted after removing the quirks. I also checked armbianEnv.txt to ensure they were gone.

I'll have to give you the ID tonight as I'm away from my device.

MichaIng, 2022-03-29:
I'm dump, I totally forgot that I have a NEO3 here with same SoC where I can debug myself 😄.

wpwentzell, 2022-03-30:
```
root@DietPi:~# lsusb
Bus 005 Device 002: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

fra87, 2022-03-30:
I confirm also on my board the issue is not solved.

This is the `armbianEnv.txt` file, but unlike the user on the forum the quirks did not reappear:

```
root@rock64:/home/dietpi# cat /boot/armbianEnv.txt
verbosity=4
bootlogo=false
overlay_prefix=rockchip
rootdev=UUID=19e46d43-2ef4-45d3-be06-ceee5e2cb50f
rootfstype=ext4
extraargs=net.ifnames=0
docker_optimizations=off
```

So it's not due to the quirks.

As a side note, please consider that the uBoot upgrade was executed in no time and with no output. Is this correct or the flashing shall take some time (>1s)?

An additional thing I noted is that if I boot without the drive attached and attach it later it works:

```
KERNEL 5.10:
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Communications, Driver=cdc_acm, 12M
    |__ Port 1: Dev 2, If 1, Class=CDC Data, Driver=cdc_acm, 12M

KERNEL 5.15, drive attached since  boot (drive not recognized)
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 3, If 0, Class=Communications, Driver=cdc_acm, 12M
    |__ Port 1: Dev 3, If 1, Class=CDC Data, Driver=cdc_acm, 12M

KERNEL 5.15, drive attached after boot
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 3, If 0, Class=Communications, Driver=cdc_acm, 12M
    |__ Port 1: Dev 3, If 1, Class=CDC Data, Driver=cdc_acm, 12M
```

Of course I cannot leave it detached and attach it only when the machine has booted, but maybe this can help finding why it happens

BTW if I understood correctly the USB quirks are used to let the kernel use the usb-storage driver instead of uas. Normally my driver uses the uas driver, so I tried adding my USB drive in the quirks and reboot: the drive used the usb-storage driver, but nothing changed (if attached before boot the drive did not work, attached after boot it worked).

NOTE: if I booted with the drive attached, then detached and reattached it then it did not work. I had to boot without the drive, then attach it later

wpwentzell, 2022-03-30:
Looks like SkiffOS supports the Rock64 board with the mainline kernel. No idea if the board boots with USB3 support using this kernel + U-Boot.

https://github.com/skiffos/SkiffOS#supported-systems

MichaIng, 2022-03-31:
Same as before, after kernel upgrade, the NEO3 does not boot anymore at all 😢. Need to open the case for serial console access.

MichaIng, 2022-03-31:
https://forum.armbian.com/topic/20133-nanopi-neo3-dtb-gone-since-linux-dtb-current-rockchip64-22021/

MichaIng, 2022-03-31:
Okay, the R2 rev00 device tree works. GPIO devices are still there and the USB 3 flash drive can still be used like before 🤔. So seems to be a ROCK64 specific issue then.

MichaIng, 2022-03-31:
@wpwentzell 
Can you try the following:
```
G_CONFIG_INJECT 'usbstoragequirks=' 'usbstoragequirks=0x174c:0x55aa:u' /boot/armbianEnv.txt
reboot
```
This disables UAS for your USB drive.

fra87, 2022-03-31:
@MichaIng what is neo3? As I wrote above, I have a NanoPi Neo3, and my upgrade went smoothly (no problems at all, even with the usb3)

wpwentzell, 2022-03-31:
> @wpwentzell Can you try the following:
> 
> ```shell
> G_CONFIG_INJECT 'usbstoragequirks=' 'usbstoragequirks=0x174c:0x55aa:u' /boot/armbianEnv.txt
> reboot
> ```
> 
> This disables UAS for your USB drive.

Did not work. As @fra87 has already investigated, and I've just now confirmed, the device will not boot with the drive attached. However, the system will recognize the drive perfectly well if I attach it after boot. The quirks make no difference.

MichaIng, 2022-03-31:
@fra87 
What, you are running Linux 5.15 on the NanoPi NEO3 without issues? Can you show:
```sh
uname -a
cat /boot/armbianEnv.txt
```
The latest kernel definitely does not ship the NEO3 device tree, so as long as some older image did not use An NanoPi R2 device tree for NEO3 already, it should break boot.

@wpwentzell 
> the device will not boot with the drive attached.

You mean the drive won't be recognised when attached at boot?

> However, the system will recognize the drive perfectly well if I attach it after boot.

Ah, I actually attached the drive to the NEO3 as well after boot. Will test whether it can boot with the drive attached already or not.

> The quirks make no difference.

😢, but was worth a test.

wpwentzell, 2022-03-31:
> @wpwentzell
> 
> > the device will not boot with the drive attached.
> 
> You mean the drive won't be recognised when attached at boot?

Correct. When trying to boot with the drive attached, USB3 hangs. The `dmesg` output is as follows (from my test above, nothing has changed).

```
[   50.599862] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[   50.600664] usb 5-1: USB disconnect, device number 2
```

Attaching the drive after boot, everything works as normal. 

I should also add that I'm booting from eMMC, which I doubt makes any difference. @fra87, are you using eMMC or microSD for boot? In any case, the eMMC is recognized fine. It's purely USB3 related.

fra87, 2022-04-02:
@MichaIng sure: this is the output:

```
dietpi@nanoPi:~$ uname -a
Linux nanoPi 5.15.25-rockchip64 #22.02.1 SMP PREEMPT Sun Feb 27 09:05:47 UTC 2022 aarch64 GNU/Linux
dietpi@nanoPi:~$ cat /boot/armbianEnv.txt
verbosity=4
bootlogo=false
overlay_prefix=rockchip
fdtfile=rockchip/rk3328-nanopi-r2-rev00.dtb
rootdev=UUID=7f26bae3-11cc-4a08-bcc6-df7f80c7d91b
rootfstype=ext4
console=serial
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
docker_optimizations=on
extraargs=net.ifnames=0
```

If I'm not wrong I reinstalled everything (so taking dietpi image) around mid-october last year; from then I always upgraded, but never reinstalled from scratch the image.

@wpwentzell I'm using a microSD, so we are covering both cases ;)

Just FYI, I have two servers, one with a Rock64 and the other with a NanoPi NEO3. Both have the same architecture: always on server, booting from microSD, with the bulk of the OS on the microSD and some user folders (like dietpi_userdata) and samba share on an external USB3 drive. The NanoPI NEO3 boots without issues even with the USB drive attached, the Rock64 with the new kernel takes forever to login (I assume this is due to the lack of dietpi_userdata, since the drive is not connected). If I boot without the drive attached and then attach it later then it works, but being a headless server I cannot detach/attach manually every time I reboot

wpwentzell, 2022-04-02:
Well, I finally gave up and just installed DietPi on my Dell optiplex 3050 I had sitting around. Ditched the Rock64, couldn't afford anymore downtime. 

Good news is the UEFI PC image works flawlessly, and the performance difference is night and day between the devices. Bad news is I'm no longer going be helping debug this.

Good luck, chaps.

MichaIng, 2022-04-02:
@fra87 
> fdtfile=rockchip/rk3328-nanopi-r2-rev00.dtb

So yours is using the R2 device tree already. I remember that for our very first NEO3 images it was the same issue. Good to know that this exact device tree is an alternative which worked before.

@wpwentzell 
Understandable. Many thanks for helping debugging until now. I will create a new ROCK64 image tonight after DietPi v8.3 release. Probably this helps like it did for the user on the forum, while I'd still no understand why 😄. Otherwise I'll open a [P2P topic](https://forum.armbian.com/forum/25-p2p-help/) at the Armbian forum.

Btw, I just verified that my NEO3 has no issues with the USB3.0 flash drive even when it was attached before boot already. So really just a ROCK64 issue.

fra87, 2022-05-30:
I saw today that kernel 22.05.1 was released (I'm not sure when it was released).

I tried installing it with

```
apt-mark hold linux-dtb-current-rockchip64 linux-image-current-rockchip64
apt upgrade
```

and then rebooting.

The issue is still present; I also tried with

```
. /usr/lib/u-boot/platform_install.sh
write_uboot_platform "$DIR" "$(lsblk -npo PKNAME "$G_ROOTFS_DEV")"
```

(no output from either command).

FYI, the version was upgraded is

```
dietpi@rock64:~$ uname -a
Linux rock64 5.15.35-rockchip64 #22.05.1 SMP PREEMPT Sat May 28 08:37:27 UTC 2022 aarch64 GNU/Linux
```

I had to restore 21.08.1 as done in the first post.

MichaIng, 2022-05-30:
😞, but many thanks for testing!

MichaIng, 2022-07-31:
A fix has been done, should be available already with latest kernel package or available soon.

closed 2022-07-31 by MichaIng

fra87, 2022-09-15:
I feel a bit "stupid", since I waited for 2 months to get a "22.08.X" version, thinking 22.05.X versions were broken, but actually the 22.05.4 was released on 4 July (so hopefully after the fix).

However I tried installing it. It was correctly installed, since `uname -a` returned

```
Linux rock64 5.15.52-rockchip64 #22.05.4 SMP PREEMPT Mon Jul 4 09:40:29 CEST 2022 aarch64 GNU/Linux
```

I also tried reinstalling the bootloader and putting overlay=usb4 at the end of armbianEnv.txt. And...

```
[    2.967613] usb 5-1: new SuperSpeed USB device number 2 using xhci-hcd
[    2.988985] usb 5-1: New USB device found, idVendor=0bc2, idProduct=231a, bcdDevice= 7.12
[    2.989018] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.989034] usb 5-1: Product: Expansion
[    2.989047] usb 5-1: Manufacturer: Seagate
[    2.989060] usb 5-1: SerialNumber: NAAXFMAB
[    3.020644] scsi host0: uas
[    3.022251] usbcore: registered new interface driver uas
[    3.022400] scsi 0:0:0:0: Direct-Access     Seagate  Expansion        0712 PQ: 0 ANSI: 6
[   33.775458] sd 0:0:0:0: tag#20 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD
[   33.775491] sd 0:0:0:0: tag#20 CDB: opcode=0x0 00 00 00 00 00 00
[   33.775521] sd 0:0:0:0: tag#14 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD IN
[   33.775538] sd 0:0:0:0: tag#14 CDB: opcode=0xa0 a0 00 00 00 00 00 00 00 10 00 00 00
[   33.791460] scsi host0: uas_eh_device_reset_handler start
[   37.879496] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[   41.967497] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[   46.055476] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[   50.143475] usb usb5-port1: Cannot enable. Maybe the USB cable is bad?
[   50.144267] usb 5-1: USB disconnect, device number 2
[   50.144815] scsi host0: uas_eh_device_reset_handler FAILED err -19
[   50.144846] sd 0:0:0:0: Device offlined - not ready after error recovery
[   50.144862] sd 0:0:0:0: Device offlined - not ready after error recovery
[   50.144963] sd 0:0:0:0: rejecting I/O to offline device
[   50.145028] sd 0:0:0:0: [sda] Read Capacity(16) failed: Result: hostbyte=0x01 driverbyte=DRIVER_OK
[   50.145462] sd 0:0:0:0: [sda] Sense not available.
[   50.145532] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x01 driverbyte=DRIVER_OK
[   50.145549] sd 0:0:0:0: [sda] Sense not available.
[   50.145609] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[   50.145626] sd 0:0:0:0: [sda] 0-byte physical blocks
[   50.145759] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[   50.145797] sd 0:0:0:0: [sda] Asking for cache data failed
[   50.146312] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   50.149118] sd 0:0:0:0: [sda] Attached SCSI disk
```

😔

Reverted back to good old 21.08.2

MichaIng, 2022-09-18:
I guess the fix is in edge kernel only:
```sh
apt install linux-image-edge-rockchip64 linux-dtb-edge-rockchip64 linux-u-boot-rock64-edge
```

fra87, 2022-09-21:
I did not know the edge kernel, and I was not able to find info about it.

Do you suggest to switch to that kernel? What are the pros/cons? If I'm not wrong this sounds like the "development" version of the kernel, but I can't understand whether this will be actively developed in the future and/or suffer from bugs.

What do you think?

fra87, 2022-09-29:
Aaaaand..... Boom!

`Linux rock64 5.15.69-rockchip64 #22.08.2 SMP PREEMPT Wed Sep 21 19:28:26 UTC 2022 aarch64 GNU/Linux`

About 8 hours after I tested it the new 22.08.2 version was uploaded. I noticed it today, removed the apt-mark and tested. First reboot, and my USB drive is present.

I did not switch to an edge kernel, so it was put in the mainline (or stable, I don't know how it is called)

Thank you for the support :)