MichaIng/DietPi #5381: apt update Eror

born2faq, 2022-03-26:
#### Details:
- Date           | Sat Mar 26 21:25:27 EET 2022
- DietPi version | v8.2.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://download.opensuse.org/repositories/home:/jcorporation/Debian_11  InRelease
Hit:2 https://downloads.plex.tv/repo/deb public InRelease
Hit:3 https://deb.debian.org/debian bullseye InRelease
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Get:5 https://apt.mopidy.com buster InRelease [14.1 kB]
Hit:6 https://deb.debian.org/debian bullseye-updates InRelease
Hit:7 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:8 https://deb.debian.org/debian bullseye-backports InRelease
Err:5 https://apt.mopidy.com buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943
Reading package lists...
W: GPG error: https://apt.mopidy.com buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943
E: The repository 'https://apt.mopidy.com buster InRelease' is not signed.
```

Joulinar, 2022-03-26:
Once you hit by the error, you should see some options on the bottom of the screen. There you can open an additional command shell. Just do this and execute following

```
G_EXEC eval "echo 'deb https://apt.mopidy.com/ ${G_DISTRO_NAME/bullseye/buster} main contrib non-free' > /etc/apt/sources.list.d/mopidy.list"
```

This should overwrite the source file and create the one we use as workaround. Once done, exit the sub shell and retry on the error screen. In general the issue will be fixed on next release.

born2faq, 2022-03-26:
[image]

Joulinar, 2022-03-26:
You can scroll down in the 2 options on the bottom

born2faq, 2022-03-26:
I executed the command as root and it worked.
Thank you very much for your assistance

Joulinar, 2022-03-26:
Ok I'm going to close this. Feel free to reopen if needed

closed 2022-03-26 by Joulinar